openwrt/mt76 #819: Branch openwrt-22.03 compilation error

seer300, 2023-09-20:
I am using version 22.03 of OpenWrt and have encountered compilation errors using the latest submission from September 11, 2023.
Here is the error message:

```
/home/work/openwrt-2203/mt7628/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2023-09-11-bdf8ea71/tx.c: In function 'mt76_tx':
/home/work/openwrt-2203/mt7628/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2023-09-11-bdf8ea71/tx.c:328:44: error: passing argument 1 of 'ieee80211_is_bufferable_mmpdu' makes integer from pointer without a cast [-Werror=int-conversion]
  328 |             !ieee80211_is_bufferable_mmpdu(skb)) {
      |                                            ^~~
      |                                            |
      |                                            struct sk_buff *
In file included from /home/work/openwrt-2203/mt7628/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/net/mac80211.h:20,
                 from /home/work/openwrt-2203/mt7628/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2023-09-11-bdf8ea71/mt76.h:17,
                 from /home/work/openwrt-2203/mt7628/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2023-09-11-bdf8ea71/tx.c:6:
/home/work/openwrt-2203/mt7628/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/linux/ieee80211.h:744:57: note: expected '__le16' {aka 'short unsigned int'} but argument is of type 'struct sk_buff *'
  744 | static inline bool ieee80211_is_bufferable_mmpdu(__le16 fc)
      |                                                  ~~~~~~~^~
cc1: all warnings being treated as errors
```
I have to create a patch file myself to fix this error in the future
```
--- a/tx.c	2023-09-20 15:29:50.477165451 +0800
+++ b/tx.c	2023-09-20 15:30:11.821037090 +0800
@@ -325,7 +325,7 @@
 	if ((dev->drv->drv_flags & MT_DRV_HW_MGMT_TXQ) &&
 	    !(info->flags & IEEE80211_TX_CTL_HW_80211_ENCAP) &&
 	    !ieee80211_is_data(hdr->frame_control) &&
-	    !ieee80211_is_bufferable_mmpdu(skb)) {
+	    !false) {
 		qid = MT_TXQ_PSD;
 	}
```
Is this a bug? Or is it that my compilation environment or kernel version is incorrect?

Chinese original text:
我正在使用openwrt的22.03版本，使用2023.9.11最新的提交编译出错。
下面是报错信息：

--信息见英语原文--

我不得不自己制作一个补丁文件，来临时修复这个错误

--信息见英语原文--

这是BUG吗？还是说是我的编译环境或者内核版本不对？

DragonBluep, 2023-09-20:
The fork you compiled is definitely not the official openwrt. I successfully built it a few days ago.

seer300, 2023-09-21:
> The fork you compiled is definitely not the official openwrt. I successfully built it a few days ago.

Yes, my openwrt branch is a modified version of my company.
But what I'm puzzled about is that the function that reported an error (ieee80211_is_bufferable_mmpdu) should be provided by the Linux kernel. Why do parameter errors occur?

是的，我这个openwrt分支是我的公司修改过的版本。
但是我疑惑的是，报错的函数（ieee80211_is_bufferable_mmpdu）应该是Linux内核提供的，为何会出现参数错误的情况？

DragonBluep, 2023-09-21:
> Yes, my openwrt branch is a modified version of my company. But what I'm puzzled about is that the function that reported an error (ieee80211_is_bufferable_mmpdu) should be provided by the Linux [...]

No. The header files are come from mac80211 backport.

nbd168, 2023-09-21:
@seer300, your tree is missing package/kernel/mac80211/patches/subsys/347-wifi-ieee80211-correctly-mark-FTM-frames-non-buffera.patch

closed 2023-09-21 by nbd168

seer300, 2023-09-21:
> @seer300, your tree is missing package/kernel/mac80211/patches/subsys/347-wifi-ieee80211-correctly-mark-FTM-frames-non-buffera.patch

Thank you. So it's like this