openwrt/mt76 #823: MT7621: crash OpenWRT

prokudin07, 2023-09-27:
**Desription:**
On the phone wifi is on channel 56 (5 GHz). Then I go out onto the balcony and the phone switches to channel 1 (2.4 GHz) and after that (after a few seconds or a couple of minutes) I go back - openwrt crashes. If the phone on the balcony completely switches to cellular communication, then when you return back - everything works.
I simulated this error several times and it is during such actions that the system is crashed

**Firmware:**
23.05-rc2 - standart; 23.05-rc3 -standart ; snapshot r24041-74e7f8ebbd / LuCI Master git-23.236.53281-98e3743 kernel 5.15.132 (with wpad-mbedtls) 
with DAWN and without DAWN

**Model**
Beeline SmartBox PRO - 2 unit

**Architecture**
MediaTek MT7621 ver:1 eco:3

**LOGS:**
phone MAC: 66:9a:f1:f6:43:43
[link1](https://pastebin.com/QMKbiKh2) 
[link2](https://pastebin.com/ULHYPzdS)

Config:
[network](https://pastebin.com/48cErUjM) on main router
[wireless](https://pastebin.com/RsN82QS1) on main router
Second router - stupid AP with the same settings for WiFi (only other channel for 5GHz - 64,  for 2.4GHz - 2)

I would be grateful for your help in solving the problem

prokudin07, 2023-09-30:
The same on 23.05-rc4

csharper2005, 2023-10-01:
Indeed, both issues has similar `skbuff: skb_under_panic` error.

This issue (mt7602 wireless) https://pastebin.com/QMKbiKh2:
```
Thu Sep 28 00:20:55 2023 daemon.debug hostapd: phy0-ap0: STA 66:9a:f1:f6:43:43 MLME: MLME-DELETEKEYS.request(66:9a:f1:f6:43:43)
Thu Sep 28 00:21:02 2023 kern.emerg kernel: [ 8035.553151] skbuff: skb_under_panic: text:82fd1a94 len:122 put:2 head:85c36c00 data:85c36bfe tail:0x85c36c78 end:0x85c36f00 dev:phy1-ap1
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.577636] Kernel bug detected[#1]:
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.584741] CPU: 2 PID: 819 Comm: mt76-tx phy1 Not tainted 5.15.132 #0
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.597722] $ 0   : 00000000 00000001 0000007c 006b2000
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.608134] $ 4   : 81016340 81016340 81017908 82e6fa88
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.618546] $ 8   : ffffefff 00000000 ffffffea 00000000
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.628948] $12   : 82e6fa94 00000182 808b6a08 ffffffff
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.639353] $16   : 830997c0 81572180 0000001a 85e06964
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.649764] $20   : 83561b14 82e6fccc 81572180 00000000
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.660167] $24   : 00000018 80413830
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.670576] $28   : 82e6e000 82e6fc38 830997c0 804ff2b8
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.680992] Hi    : 00000125
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.686716] Lo    : 122f2000
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.692429] epc   : 804ff2b8 skb_panic+0x5c/0x60
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.701633] ra    : 804ff2b8 skb_panic+0x5c/0x60
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.710815] Status: 1100fc03	KERNEL EXL IE
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.719141] Cause : 50800024 (ExcCode 09)
Thu Sep 28 00:21:02 2023 kern.warn kernel: [ 8035.727105] PrId  : 0001992f (MIPS 1004Kc)
```

That issue (mt7921 wireless) https://github.com/openwrt/openwrt/issues/11796:
```
[  672.580038] skbuff: skb_under_panic: text:82f10f04 len:242 put:4 head:82ce3e60 data:82ce3e5e tail:0x82ce3f50 end:0x82ce46c0 dev:wlan2
[  672.592071] Kernel bug detected[#1]:
[  672.592071] Kernel bug detected[#1]:
[  672.595636] CPU: 0 PID: 4380 Comm: mt76-tx phy2 Not tainted 5.10.161 #0
[  672.602218] $ 0   : 00000000 00000001 00000079 00781000
[  672.607442] $ 4   : 81000378 81000378 810058f8 84677a88
[  672.612659] $ 8   : ffffefff 00000000 ffffffea 00000000
[  672.617876] $12   : 84677a94 000001c8 807ce868 ffffffff
[  672.623094] $16   : 80fcf780 000000ee 8328a84c 00000002
[  672.628311] $20   : 82cd2040 83287208 00000002 84596184
[  672.633528] $24   : 00000018 80403fb8
[  672.638744] $28   : 84676000 84677c40 00000000 804e45fc
[  672.643961] Hi    : 00000125
[  672.646826] Lo    : 122f2000
[  672.649694] epc   : 804e45fc 0x804e45fc
[  672.653508] ra    : 804e45fc 0x804e45fc
[  672.657322] Status: 11008403 KERNEL EXL IE
[  672.661498] Cause : 50800024 (ExcCode 09)
[  672.665485] PrId  : 0001992f (MIPS 1004Kc)
```

whitslack, 2023-10-01:
@prokudin07: A stack trace would be very helpful in locating the errant code path. This probably requires you to build your own OpenWrt image with some debugging options enabled. (I don't know how to do that.)

prokudin07, 2023-10-06:
The same crash was on WiFire S1500.NBN / Sercomm S1500 BUC

prokudin07, 2023-10-06:
@nbd168 Hello, I was advised to contact you and you might be interested in this error.
And maybe can you help me with the problem?
I would be grateful for your help in solving the problem

aphorise, 2023-10-29:
Using 23.05 on a AMD64 system I get the following crash:

```bash
uname -arv
  # Linux OpenWrt 5.15.134 #0 SMP Mon Oct 9 21:45:35 2023 x86_64 GNU/Linux

dmesg
```

```
[  864.362215] mt7921u 2-3.3:1.3: Message 00020002 (seq 13) timeout
[  864.592207] mt7921u 2-3.3:1.3: timed out waiting for pending tx
[  864.615670] ------------[ cut here ]------------
[  864.622242] WARNING: CPU: 5 PID: 544 at 0xffffffff810dbf77
[  864.629022] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mt7921u mt7921s mt7921e mt7921_common mt76x0u mt76x0e mt76x0_common iwldvm cdc_mbim sr9700 smsc95xx smsc75xx sierra_net rndis_host qmi_wwan pppox ppp_generic plusb nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt792x_usb mt792x_lib mt7915e mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76_sdio mt76_connac_lib mt76 iwlmvm mcs7830 mac80211 lzo kalmia iwlwifi huawei_cdc_ncm
[  864.629067]  dm9601 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix aqc111 usbnet ts_kmp ts_fsm ts_bm smsc slhc rtl8150 r8169 r8152 pegasus nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c lan78xx kaweth ipheth igc hso forcedeth e1000e crc_ccitt compat cdc_wdm bnx2 ax88796b atlantic asn1_decoder fuse i2c_dev snd_pcsp msdos ixgbe e1000 amd_xgbe mdio oid_registry snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore udf crc_itu_t ntfs lockd sunrpc grace hfsplus hfs autofs4 cachefiles fscache netfs nls_utf8 nls_koi8_r nls_cp1255 nls_iso8859_6 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_cp950 nls_cp936 nls_euc_jp nls_cp932 nls_cp866 nls_cp864 nls_cp862 nls_cp852 nls_cp850 nls_cp775 nls_cp1251 nls_cp1250 macsec ena sha512_ssse3 sha512_generic sha256_ssse3 sha256_generic libsha256 seqiv jitterentropy_rng drbg md5 hmac cmac crypto_acompress
[  864.728206]  nls_iso8859_1 nls_cp437 uas xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl igb vfat fat exfat button_hotplug tg3 realtek microchip mii efivarfs
[  864.846506] CPU: 5 PID: 544 Comm: kworker/u16:5 Not tainted 5.15.134 #0
[  864.854408] Hardware name: Intel BQM5/BQM5, BIOS B009_L0 06/07/2023
[  864.861946] Workqueue: mt76 0xffffffffa088c4a0 [mt7921_common@0000000059370f4c+0xb000]
[  864.871267] RIP: 0010:0xffffffff810dbf77
[  864.876343] Code: 74 31 31 c0 5b 41 5c 5d c3 cc cc cc cc 0f 0b 4c 8b a3 90 05 00 00 a8 04 74 a8 0f 0b b8 da ff ff ff 5b 41 5c 5d c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb d1 0f 0b eb cb 66 66 2e 0f 1f 84 00 00 00
[  864.899079] RSP: 0018:ffffc90000387d78 EFLAGS: 00010202
[  864.905593] RAX: 0000000000000004 RBX: ffff88810fbcc9c0 RCX: ffff88810961f8a8
[  864.914144] RDX: ffffffff824f8248 RSI: 0000000000000297 RDI: ffff88810fbcc9c0
[  864.922687] RBP: ffffc90000387d88 R08: ffffffff824f8248 R09: ffffffff824f8248
[  864.931232] R10: 0000000000000001 R11: 0000000000000001 R12: ffff888106d02680
[  864.939774] R13: 0000000000000000 R14: 0000000000000020 R15: 0000000000000100
[  864.948312] FS:  0000000000000000(0000) GS:ffff88849fb40000(0000) knlGS:0000000000000000
[  864.957893] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  864.964941] CR2: 00007fb999043ff8 CR3: 000000000340a000 CR4: 0000000000350ee0
[  864.973466] Call Trace:
[  864.976993]  <TASK>
[  864.980125]  ? 0xffffffff81a86086
[  864.984529]  ? 0xffffffff81a861b0
[  864.988918]  ? 0xffffffff810bd714
[  864.993281]  ? 0xffffffff810dbf77
[  864.997621]  ? 0xffffffff813f57e6
[  865.001936]  ? 0xffffffff81ad5d28
[  865.006227]  ? 0xffffffff81ad5e38
[  865.010498]  ? 0xffffffff81c00b4b
[  865.014744]  ? 0xffffffff810dbf77
[  865.018967]  0xffffffffa06eb5a1 [mt76_usb@00000000295b177b+0x5000]
[  865.026234]  ? 0xffffffff810fc910
[  865.030433]  0xffffffffa08a1579 [mt7921u@000000004f0950ab+0x3000]
[  865.037573]  0xffffffffa088c525 [mt7921_common@0000000059370f4c+0xb000]
[  865.045254]  0xffffffff810d6008
[  865.049237]  0xffffffff810d61be
[  865.053182]  ? 0xffffffff810d6170
[  865.057292]  0xffffffff810dcfa8
[  865.061190]  ? 0xffffffff810dce80
[  865.065251]  0xffffffff81003c2f
[  865.069116]  </TASK>
[  865.071952] ---[ end trace 3bb606dd140202cb ]---
[  865.217083] mt7921u 2-3.3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
[  865.217083]
[  865.238893] mt7921u 2-3.3:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
[  912.048799] br-lan: port 3(phy1-ap0) entered disabled state
[  912.956574] br-lan: port 3(phy1-ap0) entered disabled state
[  913.019959] device phy1-ap0 left promiscuous mode
[  913.025313] br-lan: port 3(phy1-ap0) entered disabled state
[  942.274480] br-lan: port 3(phy1-ap0) entered blocking state
[  942.280915] br-lan: port 3(phy1-ap0) entered disabled state
[  942.287340] device phy1-ap0 entered promiscuous mode
[  942.343461] br-lan: port 3(phy1-ap0) entered blocking state
[  942.350083] br-lan: port 3(phy1-ap0) entered forwarding state
[... cut ...]
[  942.842160] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[  942.849878] br-lan: port 3(phy1-ap0) entered blocking state
[  942.856026] br-lan: port 3(phy1-ap0) entered forwarding state
[ 1672.733383] mt7921u 2-3.3:1.3: Message 00020002 (seq 9) timeout
[ 1672.973379] mt7921u 2-3.3:1.3: timed out waiting for pending tx
[ 1673.129056] mt7921u 2-3.3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
[ 1673.129056]
[ 1673.153809] mt7921u 2-3.3:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
```

prokudin07, 2023-11-06:
After yesterday updated 23.05 and installed netfid, dawn, umdns and other packages and  the device stopped crashing.

closed 2023-11-06 by prokudin07

aphorise, 2023-11-07:
In my case the crashes continue at random interval the same in my last posting although it does seem to restart ok & I've put some cron jobs to perform `wifi ;` whenever needed.

csharper2005, 2023-11-07:
> In my case the crashes continue at random interval

What device do you use?

aphorise, 2023-11-11:
@csharper2005 - [ALFA - AXE3000 AWUS036AXML](https://www.alfa.com.tw/products/awus036axml?variant=39754360684616)

The drivers seem overall less stable than with Linux desktop based distro (using 6.x+ kernels). On idle with no clients it never crashes versus earlier crashes I shared above that will happen at least twice a day.

esp0xdeadbeef, 2026-11-15:
Possibly the same class as #711 and #550: `skb_under_panic` is the mt76 USB tx path pushing below `skb->head` because the driver did not reserve headroom before `skb_push()`.

In mt7925 the fix is a `skb_cow_head()` before the push, mirroring what mt7921 already does:

```c
err = skb_cow_head(skb, MT_SDIO_TXD_SIZE + MT_SDIO_HDR_SIZE);
if (err)
	return err;
```

https://github.com/esp0xdeadbeef/nixos/blob/cca1932f/patches/mt7925-tx-headroom.patch

esp0xdeadbeef, 2026-11-15:
Correction: the function lives in `mt7925/mac.c`, not `mt7925/usb.c`.

Current mainline, for reference:
- `mt7921/mac.c:781` in `mt7921_usb_sdio_tx_prepare_skb()` has `skb_cow_head(skb, MT_SDIO_TXD_SIZE + MT_SDIO_HDR_SIZE)` before the push
- `mt7925/mac.c:1645` in `mt7925_usb_sdio_tx_prepare_skb()` does not