openwrt/mt76 #824: Tx power issue, it does not exceed 3dbm unable to fix it with firmware upgrade  Xiaomi Mi Router 4A Gigabit Edition v1

jupiterou, 2023-10-04:
I do own two similar Xiaomi Mi Router 4A Gigabit Edition v1 and I do not know  
why one of the routers both of the radios (5ac and 2.4n) works fine and the TX power could reach upto 20dbm.
But the other one regardless of which OpenWrt firmare I flash it the 5GHz radio is always limited to 3db (I have checked it with Uci and Luci) and even manual modification of the Tx power for radio1 (5GHz) on the /etc/config/wireless did not solve the issue..

This problem is so annoying that I could not connect to the router 5GHz from distances further than 2 meters.

Kindly let me know why one router is working fine and the other the issue with the radios eixsts.
Please note that I have changed the Country and channels for the faulty router and it did not fix the issue.

I am such tired of this issue that I wanted to clone the factory pariton of the working router on the faulty router (/dev/mtd3) but I 
do not know how to do it since that partition is read-only  
By the way I have checked the stcok firmware on both of the routers for each of the 2.4 and 5 GHz radios and everything worked fine 
with both of the routers 2.4 and 5 GHz radios. 
So there is no hardware flaw on the faulty router.

Kindly let me know how can I solve this issue, and how can I change the default maximum Tx power of the routers Radio without needing to build an image?
In case if there is no option than rebuilding the image which file and parameters should I change to fix the issues the transmit power of the radios?

Note that both of the routers are flahsed with the following image specs,

MediaTek MT7621 ver:1 eco:3
Target Platform	ramips/mt7621
 (5GHz radio)MediaTek MT76x2E
Kernel Version 5.15.127
OpenWrt 23.05.0-rc3 r23389-5deed175a5 / LuCI openwrt-23.05 branch git-23.236.53405-fc638c8

teuchezh, 2024-07-20:
same problem

HiGarfield, 2024-08-06:
The issue lies in the EEPROM (factory partition). The Tx power will be significantly reduced if an external power amplifier (PA) is specified in the EEPROM. I am not sure whether this is a bug in the mt76 module.

It is conceivable that your two Xiaomi Mi Router 4A routers were manufactured in different batches. The one functioning correctly with OpenWrt likely features an internal PA, whereas the other one experiencing issues may have an external PA. Stock firmware can handle them correctly, but mt76 seems not to.

teuchezh, 2024-08-07:
> The issue lies in the EEPROM (factory partition). The Tx power will be significantly reduced if an external power amplifier (PA) is specified in the EEPROM. I am not sure whether this is a bug in [...]

Yes, they explained to me on another forum that the problem was with the firmware, I restored my router without a dump using a programmer, clearing all the memory and flashing the bootloader there, then using the bootloader (uboot) I flashed OpenWRT