openwrt/mt76 #825: [MT7628] 2.4 GHz packet injection causes kernel crash

ShredRum, 2023-10-05:
Hello. I put the router into monitor mode, then try to deauthenticate via mac, after which the router immediately reboots

I found a similar report here
https://github.com/openwrt/openwrt/issues/5102
https://github.com/openwrt/mt76/issues/504

The crash occurs when trying to send deauthentication packets using aireplay, but also occurs when checking the driver with the command:

aireplay-ng-9 phy0-mon

I only get
05:27:48 Trying broadcast probe requests...
After which the router reboots

logread -f only captures
Thu Oct 5 05:27:48 2023 kern.info kernel: [542.910446] device phy0-mon0 entered promiscuous mode

Xiaomi Mi Router 4A (100M International Edition)
MediaTek MT7628AN ver:1 eco:2
ramips/mt76x8
OpenWrt SNAPSHOT r24072-a181b9f0f9 / LuCI Master git-23.266.27574-7744ad0
Kernel: 5.15.132

lukasz1992, 2023-10-06:
Crashlog please

ShredRum, 2023-10-06:
> Crashlog please

How can I get it?
The router reboots immediately after the crash and does not allow any information to be received.

lukasz1992, 2023-10-06:
pstore is not enabled for your device; you will need to open the device case, connect serial port, crash the kernel and grab crash log.

ShredRum, 2023-10-08:
> pstore is not enabled for your device; you will need to open the device case, connect serial port, crash the kernel and grab crash log.

I don't have the quick technical ability to do this now. Is there an alternative option? Perhaps you can rebuild the firmware or something like that?

damolp, 2024-11-05:
I'm running into the same issue here. [netconsole](https://www.kernel.org/doc/Documentation/networking/netconsole.txt) seemed to be easier to configure than [ramoops](https://docs.kernel.org/admin-guide/ramoops.html) with pstore from U-Boot (as I couldn't get into the serial console to set the pstore environment variable).

```
1,211,611140225,-;CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 828b8d54, ra == 828b8e10
4,212,611161452,-;Oops[#1]:
4,213,611165972,-;CPU: 0 PID: 1818 Comm: aireplay-ng Tainted: G           O       6.6.59 #0
4,214,611181167,-;$ 0   : 00000000 7fb318e6 82d3d2ac 00000000
4,215,611191566,-;$ 4   : 4001009d 00000000 818e39c0 00000001
4,216,611201978,-;$ 8   : 0000000e 00000000 818e39e0 00000004
4,217,611212372,-;$12   : 00000fff 82b9b1b8 00000000 0000001d
4,218,611222765,-;$16   : 818e39c0 82d3d1c8 82d396e0 00000000
4,219,611233159,-;$20   : 82e95c4c 00000000 82d8e168 00000000
4,220,611243552,-;$24   : ffffffe0 00000001                  
4,221,611253944,-;$28   : 82e94000 82e95ba8 807d0000 828b8e10
4,222,611264338,-;Hi    : 00000009
4,223,611270051,-;Lo    : 00000070
4,224,611275770,-;epc   : 828b8d54 mt76_tx+0xa4/0x1ec [mt76]
4,225,611286027,-;ra    : 828b8e10 mt76_tx+0x160/0x1ec [mt76]
4,226,611296433,-;Status: 1100f403\x09KERNEL EXL IE 
4,227,611304754,-;Cause : 0080000c (ExcCode 03)
4,228,611312707,-;BadVA : 00000000
4,229,611318421,-;PrId  : 00019655 (MIPS 24KEc)
4,230,611326374,-;Modules linked in: mt76x2e(O) mt76x2_common(O) mt76x02_lib(O) mt7603e(O) mt76(O) mac80211(O) cfg80211(O) netconsole pppoe ppp_async nft_fib_inet nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat(O) sha512_generic seqiv sha3_generic drbg hmac geniv rng cmac leds_gpio gpio_button_hotplug(O) crc32c_generic [last unloaded: cfg80211(O)]
4,231,611447823,-;Process aireplay-ng (pid: 1818, threadinfo=e5b3e5df, task=afcf51d1, tls=77eb2890)
4,232,611464403,-;Stack : 82d397d4 82b9b1b8 818e39c0 000000ff 00000001 82e95c48 00000000 82e95c4c
4,233,611481030,-;        82d384c0 00000000 00000000 82f3ba24 82e95c48 818e39c0 82d384c0 82b9a4c0
4,234,611497649,-;        00000000 00000000 82d8e168 82e95c4c 82b9a4c0 00000000 82d384c0 82d38220
4,235,611514265,-;        82b9a4c0 82f3dc7c 818e39c0 82b9a4c0 00000000 82d384c0 00000000 82b9a4c0
4,236,611530887,-;        82d8e168 818e39c0 82b9a4c0 82f43be4 00000000 81ba5e0a 81ba5e0c 0000000f
4,237,611547506,-;        ...
4,238,611552373,-;Call Trace:
4,239,611557226,-;[<828b8d54>] mt76_tx+0xa4/0x1ec [mt76]
4,240,611566798,-;[<82f3ba24>] ieee80211_reserve_tid+0x2f0/0x55c [mac80211]
4,241,611579929,c;
4,242,611582904,-;Code: ae020000  ae030004  ac500004 <ac700000> 8c430008  3c04828c  24630001  ac430008  24050200 
4,243,611602295,-;
4,244,611605337,-;---[ end trace 0000000000000000 ]---
0,245,611614536,-;Kernel panic - not syncing: Fatal exception in interrupt
0,246,611627143,-;Rebooting in 3 seconds..
```

mt76 version:
```
PKG_SOURCE_URL:=https://github.com/openwrt/mt76
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2024-10-11.1
PKG_SOURCE_VERSION:=ecca0e77b4bce629ec1f79d83bbd14a68f919188
```

damolp, 2024-11-06:
<details>
<summary>Trace after compiliation w/Debug Symbols</summary>

```
1,216,482782020,-;CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 822b9c78, ra == 822b9d4c
4,217,482803257,-;Oops[#1]:
4,218,482807780,-;CPU: 0 PID: 1954 Comm: aireplay-ng Tainted: G           O       6.6.59 #0
4,219,482822975,-;$ 0   : 00000000 7fde8316 822ad424 00000000
4,220,482833377,-;$ 4   : 4001009d 00000000 82cb3300 00000001
4,221,482843778,-;$ 8   : 0000000e 00000000 82cb3320 00000004
4,222,482854175,-;$12   : 00000fff 82d691b8 00000000 0000001d
4,223,482864589,-;$16   : 82cb3300 822ad340 822a96e0 00000000
4,224,482874990,-;$20   : 82f6dc34 00000000 8295cf68 00000000
4,225,482885388,-;$24   : ffffffe0 00000001                  
4,226,482895784,-;$28   : 82f6c000 82f6db90 807e4134 822b9d4c
4,227,482906183,-;Hi    : 00000009
4,228,482911900,-;Lo    : 00000070
4,229,482917615,-;epc   : 822b9c78 mt76_tx+0x9c/0x24c [mt76]
4,230,482927873,-;ra    : 822b9d4c mt76_tx+0x170/0x24c [mt76]
4,231,482938285,-;Status: 1100f403\x09KERNEL EXL IE 
4,232,482946611,-;Cause : 0080000c (ExcCode 03)
4,233,482954565,-;BadVA : 00000000
4,234,482960282,-;PrId  : 00019655 (MIPS 24KEc)
4,235,482968236,-;Modules linked in: mt76x2e(O) mt76x2_common(O) mt76x02_lib(O) mt7603e(O) mt76(O) mac80211(O) cfg80211(O) netconsole jitterentropy_rng pppoe ppp_async nft_fib_inet nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat(O) sha512_generic seqiv sha3_generic drbg hmac geniv rng cmac leds_gpio gpio_button_hotplug(O) crc32c_generic [last unloaded: cfg80211(O)]
4,236,483092800,-;Process aireplay-ng (pid: 1954, threadinfo=99321860, task=f3dc6bce, tls=77e28890)
4,237,483109381,-;Stack : 822a97d4 82d691b8 82cb3300 000000ff 00000001 82f6dc30 00000000 82f6dc34
4,238,483126007,-;        822a84c0 00000000 00000000 8223eb94 82f6dc30 82cb3300 822a84c0 82d684c0
4,239,483142628,-;        00000000 00000000 8295cf68 82f6dc34 82d684c0 00000000 822a84c0 822a8220
4,240,483159256,-;        82d684c0 82240e2c 82cb3300 82d684c0 00000000 822a84c0 00000000 82d684c0
4,241,483175879,-;        8295cf68 82cb3300 82d684c0 82246d8c 00000000 81bbe48a 81bbe48c 0000000f
4,242,483192501,-;        ...
4,243,483197368,-;Call Trace:
4,244,483202221,-;[<822b9c78>] mt76_tx+0x9c/0x24c [mt76]
4,245,483211788,-;[<8223eb94>] ieee80211_reserve_tid+0x2f0/0x55c [mac80211]
4,246,483225219,c;
4,247,483228195,-;Code: ae020000  ae030004  ac500004 <ac700000> 8c430008  3c04822c  24630001  ac430008  24050200 
4,248,483247597,-;
4,249,483250640,-;---[ end trace 0000000000000000 ]---
```
</details>

```
(gdb) l *(mt76_tx+0x9c)
0x9ca8 is in mt76_tx (./include/linux/skbuff.h:2198).
2193		 * for the opposite READ_ONCE()
2194		 */
2195		WRITE_ONCE(newsk->next, next);
2196		WRITE_ONCE(newsk->prev, prev);
2197		WRITE_ONCE(((struct sk_buff_list *)next)->prev, newsk);
2198		WRITE_ONCE(((struct sk_buff_list *)prev)->next, newsk);
2199		WRITE_ONCE(list->qlen, list->qlen + 1);
2200	}
2201	
2202	static inline void __skb_queue_splice(const struct sk_buff_head *list,
(gdb) l *(mt76_tx+0x170)
0x9d7c is in mt76_tx (/home/ubuntu/openwrt_wav/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2024.10.11.1~ecca0e77/tx.c:350).
345	
346		if (wcid && !(wcid->tx_info & MT_WCID_TX_INFO_SET))
347			ieee80211_get_tx_rates(info->control.vif, sta, skb,
348					       info->control.rates, 1);
349	
350		info->hw_queue |= FIELD_PREP(MT_TX_HW_QUEUE_PHY, phy->band_idx);
351	
352		if ((info->flags & IEEE80211_TX_CTL_TX_OFFCHAN) ||
353		    (info->control.flags & IEEE80211_TX_CTRL_DONT_USE_RATE_MASK))
354			head = &wcid->tx_offchannel;
```

damolp, 2024-11-06:
Patching the driver to use `__mt76_tx_queue_skb` instead of `__skb_queue_tail` fixes the crash.

<details>
<summary>patch</summary>

```
--- a/tx.c	2024-11-06 02:50:39.160702615 +0000
+++ b/tx.c	2024-11-06 02:51:10.053329752 +0000
@@ -333,7 +333,9 @@
 	struct mt76_wcid *wcid, struct sk_buff *skb)
 {
 	struct ieee80211_tx_info *info = IEEE80211_SKB_CB(skb);
-	struct sk_buff_head *head;
+	struct mt76_dev *dev = phy->dev;
+	struct mt76_queue *q;
+	int qid = skb_get_queue_mapping(skb);
 
 	if (mt76_testmode_enabled(phy)) {
 		ieee80211_free_txskb(phy->hw, skb);
@@ -349,22 +351,12 @@
 
 	info->hw_queue |= FIELD_PREP(MT_TX_HW_QUEUE_PHY, phy->band_idx);
 
-	if ((info->flags & IEEE80211_TX_CTL_TX_OFFCHAN) ||
-	    (info->control.flags & IEEE80211_TX_CTRL_DONT_USE_RATE_MASK))
-		head = &wcid->tx_offchannel;
-	else
-		head = &wcid->tx_pending;
-
-	spin_lock_bh(&head->lock);
-	__skb_queue_tail(head, skb);
-	spin_unlock_bh(&head->lock);
-
-	spin_lock_bh(&phy->tx_lock);
-	if (list_empty(&wcid->tx_list))
-		list_add_tail(&wcid->tx_list, &phy->tx_list);
-	spin_unlock_bh(&phy->tx_lock);
+	q = phy->q_tx[qid];
+	spin_lock_bh(&q->lock);
+	__mt76_tx_queue_skb(phy, qid, skb, wcid, sta, NULL);
+	dev->queue_ops->kick(dev, q);
+	spin_unlock_bh(&q->lock);
 
-	mt76_worker_schedule(&phy->dev->tx_worker);
 }
 EXPORT_SYMBOL_GPL(mt76_tx);

```

RuhanSA079, 2026-02-02:
Applying this aforementioned patch seems to resolve the issue, but one can only monitor/read or inject frames, not both.
When I do that, something crashes, so it is not a proper fix.
My crude, experimental WiFiPacketRadio application works okay when I "receive" those packets or transmit them on monitor mode.

Upon some digging, I found that the ```tx_pending``` and ```tx_offchannel``` ```sk_buff_head``` is not initialized properly in ```tx.c```, as the packet injection stuff does not initialize those buffers as seen in ```mac80211.c:void mt76_wcid_init()``` , so probably the ```mt76_wcid``` parameter that is passed through to ```mt76_tx``` is partially created, therefore crashing the kernel on non-initialized stuff.

At this stage, I patched my mt76 driver to lazily fix this, and seems to transmit & receive just fine. (reverted your patch)
I have a seperate problem regarding radiotap headers being qwopped, probably some offset. Sigh. I guess custom Linux kernel and driver development is not easy.