openwrt/mt76 #834: Mediatek  RZ616 Wi-Fi 6E 160MH //  WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2  PowerPC P1010RDB-PB

4920441, 2023-10-19:
Hi,

I recently got a now obsolete Sophos RED15w with a minipci slot, and thought, since it has 128 MB Nand 512 MB Ram and Gigabit combined with an P1010E  CPU it should be fun  to replace the old minipci card with a  Mediatek MT7922 AX Card (and an adapter off course)  to make use as a AX Accesspoint with Openwrt.

But, it ain't so easy as I expected.

First I tried 23.05  for this, selected the mt7921e driver and the firmware for the mt7922, build uploaded it and got an error  like   

`mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   10.245053] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin`

when the driver tries to upload the firmware.

I then found this site on the internet which pulls the latest firmware images directly, but that had no effect either

[(https://mans0n.github.io/2023/01/24/diy-6ghz-ap/#MT7922)]

After that, I checked out the master branch of openwrt, compiled and flashed it again, but there is not much of a change.

The device is correctly shown on the pci bus, as well on the linux side as on the u-boot side as well:

  01:00.0     - 14c3:0616 - Network controller

[2023-10-18-ralinkmt7922_2.txt](https://github.com/openwrt/mt76/files/13038459/2023-10-18-ralinkmt7922_2.txt)

Is this a problem with the CPU/platform  or do I miss something totaly obvious here?

I found a couple of stories with the same wifi card combined with x86 openwrt systems and they didn't seem to have any issues at all.

Thanks for your time and your work! Really appreciated.

Cheers

4920441

cokebar, 2023-10-25:
I have a same issue. I run openwrt VM on a PVE host, with PCI passthrough. I've tried intel AX210 is OK. But MT7922 with same issue.

cokebar, 2023-10-25:
I think this may because the ipk package doesn't include the following firmware files
 (You can check the makefile of mt76:https://github.com/openwrt/openwrt/blob/main/package/kernel/mt76/Makefile):
https://github.com/openwrt/mt76/raw/master/firmware/WIFI_RAM_CODE_MT7922_1.bin
https://github.com/openwrt/mt76/raw/master/firmware/WIFI_MT7922_patch_mcu_1_1_hdr.bin

You can try to upload these two files to see if it can help.

4920441, 2023-10-26:
Hi,

I'll try this again, but I don't see any difference with the firmware files I already manually put there, mentioned also in the link I provided:

> I then found this site on the internet which pulls the latest firmware images directly, but that had no effect either

[https://mans0n.github.io/2023/01/24/diy-6ghz-ap/#MT7922](url)

```
wget https://github.com/openwrt/mt76/raw/master/firmware/WIFI_MT7922_patch_mcu_1_1_hdr.bin 
wget https://github.com/openwrt/mt76/raw/master /firmware/WIFI_RAM_CODE_MT7922_1.bin
```

Seems the same, to me....

DragonBluep, 2023-10-30:
If you are using OpenWrt snapshot version, you can install `kmod-mt7922-firmware` package to solve this issue.

Fixed by https://github.com/openwrt/openwrt/commit/9eecf4905375777e2048177dbe4d83fee5da9ee1

4920441, 2023-10-30:
Erm.... no it does not for me - I tried  to download the firmware files in the beginning and I even tried it with the provided links as well, but there is no change.

It tries to load them but fails

It seems as it isn't able to upload the firmware file, but not because it is not there, though anything else is not working at this point:

`[   10.233901] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   10.245053] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   10.314771] kmodloader: done loading kernel modules from /etc/modules.d/*
[   10.909896] random: jshn: uninitialized urandom read (4 bytes read)
[   10.973401] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   10.984547] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   11.573260] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   11.584405] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   11.689122] urngd: v1.0.2 started.
[   12.213321] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   12.224464] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   12.396835] random: jshn: uninitialized urandom read (4 bytes read)
[   12.457033] random: ubusd: uninitialized urandom read (4 bytes read)
[   12.477711] random: ubus: uninitialized urandom read (4 bytes read)
[   12.936546] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   12.947690] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   13.523120] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   13.534264] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   14.133237] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   14.144381] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   14.294479] random: jshn: uninitialized urandom read (4 bytes read)
[   14.338634] random: ubusd: uninitialized urandom read (4 bytes read)
[   14.863177] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   14.874329] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   15.439384] random: crng init done
[   15.442886] random: 35 urandom warning(s) missed due to ratelimiting
[   15.573911] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   15.585056] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   15.863158] mt7921e a000:01:00.0: Direct firmware load for mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin failed with error -2
[   15.874301] mt7921e a000:01:00.0: Falling back to sysfs fallback for: mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
[   16.161376] mt7921e a000:01:00.0: hardware init failed
[   24.268681] fsl-gianfar soc@ffe00000:ethernet@b0000 eth0: Link is Up - 1Gbps/Full - flow control off`

DragonBluep, 2023-10-30:
Anyway, `error -2` means `No such file or directory`.

exkc, 2023-11-03:
i have same issue and i fixed it in a non practical way kmod-mt7922-firmware isn't available for me bc i am not using lastest commit of mt76 so i copyied the fireware from my main pc that run linux.