openwrt/mt76 #838: mt7921au ap mode random crash on openwrt 23.05

ritech, 2023-11-10:
Fri Nov 10 04:11:31 2023 daemon.info hostapd: phy1-ap0: STA 72:bd:41:7c:12:7d IEEE 802.11: authenticated
Fri Nov 10 04:11:31 2023 daemon.info hostapd: phy1-ap0: STA 72:bd:41:7c:12:7d IEEE 802.11: associated (aid 2)
Fri Nov 10 04:11:31 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 72:bd:41:7c:12:7d auth_alg=open
Fri Nov 10 04:12:11 2023 kern.err kernel: [ 417.365011] mt7921u 2-1:1.0: Message 00020002 (seq 6) timeout
Fri Nov 10 04:12:12 2023 kern.err kernel: [ 417.594940] mt7921u 2-1:1.0: timed out waiting for pending tx
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618122] ------------[ cut here ]------------
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618124] WARNING: CPU: 0 PID: 10152 at kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618161] Modules linked in: xt_FULLCONENAT pppoe ppp_async zstd pppox ppp_mppe ppp_generic mt7921u mt7921_common mt76_usb mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 bnx2x amdgpu zstd_decompress zstd_compress xxhash xt_time xt_tcpudp xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_helper xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_FLOWOFFLOAD xt_CT ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip tcp_bbr slhc r8168 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_syslog nf_flow_table nf_conntrack_netlink nf_conncount mlx5_core mlx4_en mlx4_core macvlan libcrc32c iptable_raw iptable_nat iptable_mangle iptable_filter ip_tables igc iavf i915 i40e gpu_sched forcedeth e1000e crc_ccitt compat bnx2 alx 8139too 8139cp ntfs3 evdev i2c_dev cryptodev xt_set
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618239] ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ixgbe igbvf e1000 amd_xgbe mdio ip6_tunnel tunnel6 raid10 raid1 raid0 md_mod nls_utf8 nls_iso8859_1 nls_cp437 zram zsmalloc ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_ssse3 sha512_generic sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg hmac cmac authencesn authenc arc4 crypto_acompress uas sdhci_pltfm xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl igb vfat fat exfat button_hotplug tg3 realtek broadcom bcm_phy_lib mii
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618409] CPU: 0 PID: 10152 Comm: kworker/u2:0 Not tainted 5.15.127 #0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618414] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 11/12/2020
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618417] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618445] RIP: 0010:kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618449] Code: 74 31 31 c0 5b 41 5c 5d c3 cc cc cc cc 0f 0b 4c 8b a3 90 05 00 00 a8 04 74 a8 0f 0b b8 da ff ff ff 5b 41 5c 5d c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb d1 0f 0b eb cb 66 66 2e 0f 1f 84 00 00 00
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618451] RSP: 0018:ffffc90000fcfd78 EFLAGS: 00010202
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618454] RAX: 0000000000000004 RBX: ffff888004ea1d00 RCX: ffff888009807ba8
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618455] RDX: 0000000000000001 RSI: 0000000000000202 RDI: ffff888004ea1d00
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618457] RBP: ffffc90000fcfd88 R08: 0000000000003c00 R09: ffff888003f7a268
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618458] R10: 0000000000000000 R11: ffffffff822ae508 R12: ffff888009403000
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618460] R13: 0000000000000000 R14: 0000000000000020 R15: 0000000000000100
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618463] FS: 0000000000000000(0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618465] CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618468] CR2: 00007f13341f3b7c CR3: 00000000035e2006 CR4: 00000000003706f0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618503] Call Trace:
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618510]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618518] ? show_regs.part.0+0x1e/0x24
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618549] ? show_regs.cold+0x8/0xd
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618551] ? __warn+0x74/0xf0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618569] ? kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618572] ? report_bug+0x86/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618593] ? handle_bug+0x38/0x90
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618606] ? exc_invalid_op+0x18/0x70
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618608] ? asm_exc_invalid_op+0x1b/0x20
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618615] ? kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618618] mt76u_resume_rx+0x301/0xd60 [mt76_usb]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618626] ? wait_woken+0x60/0x60
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618633] mt7921u_mac_reset+0x69/0x1c0 [mt7921u]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618639] mt7921_mac_reset_work+0x85/0x140 [mt7921_common]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618644] process_one_work+0x1f8/0x360
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618654] worker_thread+0x4e/0x410
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618657] ? process_one_work+0x360/0x360
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618660] kthread+0x128/0x150
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618662] ? set_kthread_struct+0x50/0x50
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618665] ret_from_fork+0x1f/0x30
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618676]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [ 417.618678] ---[ end trace 56a2ce3888ddc6b9 ]---
Fri Nov 10 04:12:12 2023 kern.info kernel: [ 417.759857] mt7921u 2-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
Fri Nov 10 04:12:12 2023 kern.info kernel: [ 417.759857]
Fri Nov 10 04:12:12 2023 kern.info kernel: [ 417.770990] mt7921u 2-1:1.0: WM Firmware Version: ____010000, Build Time: 20230526130958

runtime429, 2024-01-26:
Same here on OpenWrt 23.05.2 r23630-842932a63d:
Fri Jan 26 12:09:06 2024 daemon.info hostapd: phy0-ap0: STA aa:01:05:f8:7c:63 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jan 26 12:09:10 2024 kern.err kernel: [ 2608.456870] mt7921u 3-1:1.0: Message 00020002 (seq 9) timeout
Fri Jan 26 12:09:10 2024 kern.err kernel: [ 2608.686914] mt7921u 3-1:1.0: timed out waiting for pending tx
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.701390] ------------[ cut here ]------------
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.702011] WARNING: CPU: 0 PID: 6079 at 0xffffffff810dbf87
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.702705] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mt7921u mt7921_common pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt792x_usb mt792x_lib mt76_usb mt76_connac_lib mt76 mac80211 lzo cfg80211 slhc r8169 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c igc forcedeth e1000e crc_ccitt compat bnx2 i2c_dev ixgbe e1000 amd_xgbe mdio nls_utf8 nls_iso8859_1 nls_cp437 ena sha512_ssse3 sha512_generic seqiv jitterentropy_rng drbg hmac cmac crypto_acompress igb vfat fat button_hotplug tg3 realtek mii
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.709800] CPU: 0 PID: 6079 Comm: kworker/u2:1 Not tainted 5.15.137 #0
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.710701] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.711792] Workqueue: mt76 0xffffffffa039d4a0 [mt7921_common@000000003e7af789+0xb000]
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.712630] RIP: 0010:0xffffffff810dbf87
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.713154] Code: 74 31 31 c0 5b 41 5c 5d c3 cc cc cc cc 0f 0b 4c 8b a3 90 05 00 00 a8 04 74 a8 0f 0b b8 da ff ff ff 5b 41 5c 5d c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb d1 0f 0b eb cb 66 66 2e 0f 1f 84 00 00 00
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.714846] RSP: 0018:ffffc90002e1bd78 EFLAGS: 00010202
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.715477] RAX: 0000000000000004 RBX: ffff8880046fb500 RCX: ffff888014d07ba8
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.716194] RDX: ffffc90000f36000 RSI: 0000000000000246 RDI: ffff8880046fb500
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.716960] RBP: ffffc90002e1bd88 R08: 0000000000003c00 R09: ffff888003b98260
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.717685] R10: 0000000000000000 R11: ffffffff824ae408 R12: ffff888025221900
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.718397] R13: 0000000000000000 R14: 0000000000000020 R15: 0000000000000100
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.719106] FS:  0000000000000000(0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.719905] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.720794] CR2: 00007fd435825b7c CR3: 000000002525a000 CR4: 00000000000006f0
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.721856] Call Trace:
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.722544]  <TASK>
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.723075]  ? 0xffffffff81a86446
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.723652]  ? 0xffffffff81a86570
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.724222]  ? 0xffffffff810bd714
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.724801]  ? 0xffffffff810dbf87
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.725365]  ? 0xffffffff813f5876
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.725938]  ? 0xffffffff81ad6118
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.726497]  ? 0xffffffff81ad6228
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.727090]  ? 0xffffffff81c00b4b
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.727642]  ? 0xffffffff810dbf87
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.728193]  0xffffffffa02de5a1 [mt76_usb@0000000029702392+0x5000]
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.729001]  ? 0xffffffff810fc910
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.729557]  0xffffffffa03a8579 [mt7921u@000000000b9fd408+0x3000]
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.730316]  0xffffffffa039d528 [mt7921_common@000000003e7af789+0xb000]
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.731114]  0xffffffff810d600b
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.731658]  0xffffffff810d61be
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.732201]  ? 0xffffffff810d6170
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.732763]  0xffffffff810dcfbb
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.733301]  ? 0xffffffff810dce90
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.733872]  0xffffffff81003c32
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.734424]  </TASK>
Fri Jan 26 12:09:10 2024 kern.warn kernel: [ 2608.734892] ---[ end trace 9616d2119f9c062e ]---
Fri Jan 26 12:09:10 2024 kern.info kernel: [ 2608.883845] mt7921u 3-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
Fri Jan 26 12:09:10 2024 kern.info kernel: [ 2608.883845]
Fri Jan 26 12:09:10 2024 kern.info kernel: [ 2608.896985] mt7921u 3-1:1.0: WM Firmware Version: ____010000, Build Time: 20230526130958

sbwml, 2024-02-06:
```
[12202.592215] mt7921u 2-1:1.0: Message 00020002 (seq 1) timeout
[12202.832248] mt7921u 2-1:1.0: timed out waiting for pending tx
[12202.843966] ------------[ cut here ]------------
[12202.843986] WARNING: CPU: 1 PID: 127617 at kernel/kthread.c:659 k"����k+0xa8/0xbc
[12202.844016] Modules linked in: oaf(O) pppoe ppp_async mt7921u(O) mt7921_common(O) l2tp_ppp wireguard snd_usb_audio rockchipdrm rndis_host pppox ppp_generic nft_fib_inet nf_flow_table_inet mt792x_usb(O) mt792x_lib(O) mt76_usb(O) mt76_connac_lib(O) mt76(O) mac80211(O) lz4 libchacha20poly1305 ipt_REJECT dw_hdmi_cec dw_hdmi drm_dma_helper chacha_neon cfg80211(O) cec cdc_ether zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbnet usblp ts_fsm ts_bm tls tcp_bbr snd_usbmidi_lib slhc sch_cake rc_core r8152(O) r8125(O) poly1305_neon phy_rockchip_inno_hdmi panfrost nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq
[12202.844316]  nft_log nft_limit nft_hash nft_fullcone(O) nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo_rle lzo lz4hc_compress lz4_decompress lz4_compress libcurve25519_generic libchacha iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables gpu_sched dw_mipi_dsi drm_shmem_helper drm_kms_helper drm_display_helper crc_ccitt compat(O) br_netfilter asn1_decoder ntfs3 sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_codel em_text em_nbyte em_meta em_cmp act_skbmod act_simple act_pedit act_csum cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32
[12202.844568]  cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs pwm_fan cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb dummy sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 xsk_diag netlink_diag tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore xfrm_user xfrm_ipcomp af_key xfrm_algo nls_utf8 nls_cp950 nls_cp936 nls_cp437 zram zsmalloc vxlan udp_tunnel ip6_udp_tunnel
[12202.844833]  shortcut_fe_ipv6(O) shortcut_fe(O) crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha1_generic seqiv drbg md5 hmac echainiv geniv des_generic libdes deflate cmac cbc authencesn authenc sysimgblt sysfillrect syscopyarea fb_sys_fops fb font drm drm_panel_orientation_quirks fsl_mph_dr_of ehci_fsl gpio_button_hotplug(O) xfs vfat fat exfat btrfs xor zstd_decompress zstd_compress zstd_common xor_neon raid6_pq lzo_decompress lzo_compress
[12202.844981] CPU: 1 PID: 127617 Comm: kworker/u8:5 Tainted: G           O       6.6.16 #0
[12202.844993] Hardware name: FriendlyElec NanoPi R5S (DT)
[12202.845000] Workqueue: mt76 _40 [mt7921_common]
[12202.845027] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[12202.845037] pc : k"����k+0xa8/0xbc
[12202.845049] lr : _62+0x1e8/0x2d8 [mt76_usb]
[12202.845065] sp : ffff800084923c70
[12202.845069] x29: ffff800084923c70 x28: ffff00000ae9b468 x27: ffff00000ae920d8
[12202.845084] x26: 0000000000000000 x25: ffff0000073df580 x24: ffff00000ae920d8
[12202.845097] x23: ffff00000ae920b8 x22: 0000000000000100 x21: ffff00000ae98880
[12202.845110] x20: ffff00000bea2280 x19: ffff00000924aa00 x18: 0000000000000000
[12202.845123] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[12202.845135] x14: 0000000000000000 x13: 00000000000003b8 x12: 0000000000000002
[12202.845147] x11: 0000000000000000 x10: 0000000000000980 x9 : ffff800084923b50
[12202.845159] x8 : ffff0000024197e0 x7 : ffff0000ee7b99c0 x6 : ffff8000817830c8
[12202.845172] x5 : 0000000000000000 x4 : 0000000000000001 x3 : ffff00000b08d614
[12202.845184] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000004
[12202.845196] Call trace:
[12202.845200]  k"����k+0xa8/0xbc
[12202.845211]  _62+0x1e8/0x2d8 [mt76_usb]
[12202.845225]  0xffff80007a6554f8
[12202.845236]  _40+0x98/0x16c [mt7921_common]
[12202.845250]  �c����w�k+0x140/0x3b0
[12202.845258]  w�k��h��+0x318/0x448
[12202.845265]  k"��+0x110/0x114
[12202.845276]  ��f�m��k+0x10/0x20
[12202.845285] ---[ end trace 0000000000000000 ]---
[12202.997671] mt7921u 2-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
[12202.997671] 
[12203.008976] mt7921u 2-1:1.0: WM Firmware Version: ____010000, Build Time: 20231109190959
```

sbwml, 2024-02-13:
The latest commit seems to have solved the crash problem of mt7921au AP mode.

I used 6124ea9135ed512671933f5520c46842906c78bc source version + [100-mt76-mt7921-fix-a-potential-association-failure-upon.patch](https://github.com/sbwml/package_kernel_mt76/blob/20465710b9cfc9c449f2f8c02ec83c086169f674/patches/100-mt76-mt7921-fix-a-potential-association-failure-upon.patch) patch. Currently, I have been running in AP mode for two days, sending 2.1TB of traffic and receiving 301GB of traffic. The kernel did not crash.

sbwml, 2024-02-13:
> The latest commit seems to have solved the crash problem of mt7921au AP mode.
> 
> I used [6124ea9](https://github.com/openwrt/mt76/commit/6124ea9135ed512671933f5520c46842906c78bc) source version + [100-mt76-mt7921-fix-a-potential-association-failure-upon.patch](https://github.com/sbwml/package_kernel_mt76/blob/20465710b9cfc9c449f2f8c02ec83c086169f674/patches/100-mt76-mt7921-fix-a-potential-association-failure-upon.patch) patch. Currently, I have been running in AP mode for two days, sending 2.1TB of traffic and receiving 301GB of traffic. The kernel did not crash.

I have been able to confirm that this problem has been solved on my device (nanopi r5s). I used iperf3 for 3 hours of downloading and 3 hours of uploading. Under the continuous high-bandwidth test, the mt7921u driver did not see any exceptions or kernel crashes. (Currently I'm using Linux 6.6.16)

[image]

**usb wifi adapter:** [FU-AX1800](https://fenvi.com/product_detail_44.html)
**usb interface:** 3.0
**Operating frequency:**
```
Mode: ax 
Band: 6GHz
Channel: auto
Width: 80MHz
```

dmesg:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.16 (admin@cooluc.com) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.2.0 r23630-842932a63d) 13.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Mon Feb 12 15:03:50 2024
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 16 MiB at 0x00000000ee800000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 30 pages/cpu s85288 r8192 d29400 u122880
[    0.000000] pcpu-alloc: s85288 r8192 d29400 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: kernel page table isolation forced OFF by mitigations=off
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=tty1 console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait mitigations=off pcie_aspm=off
[    0.000000] PCIe ASPM is disabled
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000e5200000-0x00000000e9200000] (64MB)
[    0.000000] Memory: 3745816K/3930112K available (10880K kernel code, 3092K rwdata, 7428K rodata, 3008K init, 844K bss, 167912K reserved, 16384K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] Dynamic Preempt: none
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[... 524 lines cut ...]
[   16.588503] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   22.220570] br-lan: port 3(phy0-sta0) entered blocking state
[   22.220588] br-lan: port 3(phy0-sta0) entered disabled state
[   22.220634] mt7921u 8-1:1.0 phy0-sta0: entered allmulticast mode
[   22.220850] mt7921u 8-1:1.0 phy0-sta0: entered promiscuous mode
[   29.929032] init oaf sysctl...ok
[   35.100246] br-lan: port 3(phy0-sta0) entered blocking state
[   35.100272] br-lan: port 3(phy0-sta0) entered forwarding state
[   43.868388] docker0: port 1(veth24951c0) entered blocking state
[   43.868413] docker0: port 1(veth24951c0) entered disabled state
[   43.868456] veth24951c0: entered allmulticast mode
[   43.868683] veth24951c0: entered promiscuous mode
[   45.385425] eth0: renamed from vethe8f0dec
[   45.426291] docker0: port 1(veth24951c0) entered blocking state
[   45.426316] docker0: port 1(veth24951c0) entered forwarding state
[24955.551795] r8125: eth1: link up
[24955.555524] br-lan: port 1(eth1) entered blocking state
[24955.555552] br-lan: port 1(eth1) entered forwarding state
[24984.234112] r8125: eth1: link down
[24984.243439] br-lan: port 1(eth1) entered disabled state
[24986.652147] r8125: eth1: link up
[24986.652205] br-lan: port 1(eth1) entered blocking state
[24986.652216] br-lan: port 1(eth1) entered forwarding state
[25626.965347] systemd[154398]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
```