openwrt/mt76 #839: mt7921u monitor mode and packet injection failed

ZerBea, 2023-11-15:
I got an ALFA AWUS036AXML to run some tests (monitor mode and frame injection) and all tests failed.

The test environment is described here:
https://github.com/ZerBea/hcxdumptool/discussions/361#discussion-5845482

I talked with @morrownr about this problem, too. More information is here:
https://github.com/morrownr/USB-WiFi/discussions/260#discussioncomment-7567855

Result of the test:
ALFA AWUS036AXML
ID 0e8d:7961 MediaTek Inc. Wireless_Device

driver: mt7921u (Linux stock kernel driver)

Important information:
Driver reported that monitor mode (active monitor mode) is supported, but neither hcxdumptool nor tshark nor Wireshark got pakets.
Frame injection is also not working and TX power is not shown correct.

```
$ hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  8  11 00c0cab4a4f7 c022509e2039 * wlp22s0f0u9u3i3  mt7921u (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
```

hcxdumptool have not received packets and frame injection is not working, too:
```
Warning: no packets received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection

1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile

exit on sigterm
bye-bye
```

tshark == 0 packets captured
```
$ tshark -i wlp22s0f0u9u3i3
Capturing on 'wlp22s0f0u9u3i3'
 ** (tshark:24366) 16:47:18.849019 [Main MESSAGE] -- Capture started.
 ** (tshark:24366) 16:47:18.849091 [Main MESSAGE] -- File: "/tmp/wireshark_wlp22s0f0u9u3i3DZ9UE2.pcapng"
^Ctshark: 
0 packets captured
```

iw show activated monitor mode:
```
$ iw dev
phy#8
	Interface wlp22s0f0u9u3i3
		ifindex 11
		wdev 0x800000001
		addr c0:22:50:9e:20:39
		type monitor
		channel 6 (2437 MHz), width: 20 MHz (no HT), center1: 2437 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

Some additional information is here, too:

BTW:
 mt76x0u is not affected
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7553512

mt76x2u is not affected
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7550759

ZerBea, 2023-11-20:
Looks like this driver (https://github.com/openwrt/mt76) doesn't compile running Linux 6.6.1:
```
$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/lib/modules/6.6.1-arch1-1/build'
  CC [M]  /tmp/mt76/mmio.o
In file included from /tmp/mt76/mt76.h:19,
                 from /tmp/mt76/mmio.c:6:
/tmp/mt76/testmode.h:196:32: error: array type has incomplete element type 'struct nla_policy'
  196 | extern const struct nla_policy mt76_tm_policy[NUM_MT76_TM_ATTRS];
      |                                ^~~~~~~~~~~~~~
/tmp/mt76/mt76.h: In function 'mt76_put_page_pool_buf':
/tmp/mt76/mt76.h:1647:9: error: implicit declaration of function 'page_pool_put_full_page' [-Werror=implicit-function-declaration]
 1647 |         page_pool_put_full_page(page->pp, page, allow_direct);
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/tmp/mt76/mt76.h: In function 'mt76_get_page_pool_buf':
/tmp/mt76/mt76.h:1655:16: error: implicit declaration of function 'page_pool_dev_alloc_frag' [-Werror=implicit-function-declaration]
 1655 |         page = page_pool_dev_alloc_frag(q->page_pool, offset, size);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
/tmp/mt76/mt76.h:1655:14: error: assignment to 'struct page *' from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
 1655 |         page = page_pool_dev_alloc_frag(q->page_pool, offset, size);
      |              ^
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /tmp/mt76/mmio.o] Error 1
make[1]: *** [/usr/lib/modules/6.6.1-arch1-1/build/Makefile:1913: /tmp/mt76] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.6.1-arch1-1/build'
```

ZerBea, 2023-11-20:
BTW:
I went back
to kernel 6.5.1 (Debian kernel config) -> neither monitor mode nor packet injection is working
to kernel 6.1.21 (Raspbian kernel config) -> neither monitor mode nor packet injection is working

ZerBea, 2023-11-28:
update
After I got this issue report:
https://github.com/ZerBea/hcxdumptool/issues/376
I did some more tests.
If the interface is on monitor mode:
```
$ sudo hcxdumptool -m wlp22s0f0u9u3i3
$ iw dev
phy#12
	Interface wlp22s0f0u9u3i3
		ifindex 15
		wdev 0xc00000001
		addr 00:c0:ca:b5:74:e6
		type monitor
		channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0	
```
it will receive packets:
```
$ tshark -i wlp22s0f0u9u3i3
Capturing on 'wlp22s0f0u9u3i3'
263 packets captured
```
But once the first frame has been injected, every thing stops:
```
$ tshark -i wlp22s0f0u9u3i3
Capturing on 'wlp22s0f0u9u3i3'
^C
0 packets captured
```
Looks like frame injection killed the driver.

morrownr, 2023-12-04:
> This one does not load: Build Time: 20231109190918

It loads here:

$ ethtool -i wlx00c0cab37abb
driver: mt7921u
version: 6.5.0-0.deb12.1-amd64
firmware-version: ____010000-20231109190959

Adapter: Alfa AXML
Distro: Debian 12

Remember that wifi firmware for the 7921 requires two firmware files:

WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7961_1.bin

There is also a bluetooth file but you won't be using it to so you can delete the file from the system:

BT_RAM_CODE_MT7961_1_2_hdr.bin

ZerBea, 2023-12-04:
I double checked this:

old firmware:
```
[16148.856186] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20230526131214
[16148.879434] mt7921u 1-9.3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
```
new firmware:
```
[41144.190166] usb 1-9.3: new high-speed USB device number 11 using xhci_hcd
[41144.321418] usb 1-9.3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[41144.321422] usb 1-9.3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[41144.321424] usb 1-9.3: Product: Wireless_Device
[41144.321426] usb 1-9.3: Manufacturer: MediaTek Inc.
[41144.321427] usb 1-9.3: SerialNumber: 000000000
[41144.428601] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20231109191416
```
only the BT firmware has been loaded.
```
$ iw dev
$
```

All three bin's have been replaced:
```
$ ls *MT7961*.*
BT_RAM_CODE_MT7961_1_2_hdr.bin.zst
WIFI_RAM_CODE_MT7961_1.bin.zst
WIFI_MT7961_patch_mcu_1_2_hdr.bin.zst
```

I give it another try without compressing the files.

morrownr, 2023-12-04:
$ uname -r
6.5.0-0.deb12.1-amd64

I haven't gone to kernel 6.6 yet on this system, which is my main dev box, but will investigate doing a test with 6.6 on another box.

Keep in mind that loading the bluetooth firmware is basically worthless as you can't run USB3 and bluetooth together on a USB WiFi adapter. I'm pretty sure the communications between Mediatek and the makers was poor in this respect as the bt firmware should not be loaded if the adapter is in USB3 mode.

ZerBea, 2023-12-04:
At the moment, I'm running out of ideas.

morrownr, 2023-12-05:
I got nothing. I've never seen firmware compressed but if you say it works, it probably works. Got any other distros to try?

ZerBea, 2023-12-05:
zstd compression is not something new:
https://www.phoronix.com/news/2021-Linux-Zstd-Firmware
and it is working like a charm.

BTW:
At every test my reference is an ALFA AWUS036ACM (mt76x2u) and an ALFA AWUS036ACHM (mt76x0u).
If both devices/drivers are working as expected, everything seems to be fine.
After that, I'll go hunting for the problems of the new device/driver.

morrownr, 2023-12-05:
I wonder if you had a bad download of one of the wifi firmware files?

ZerBea, 2023-12-05:
Good point - let's compare it:
$ md5sum BT_RAM_CODE_MT7961_1_2_hdr.bin
f8e386541ca02a6311d7c0d9441fbab7  BT_RAM_CODE_MT7961_1_2_hdr.bin

$ md5sum WIFI_MT7961_patch_mcu_1_2_hdr.bin
0a4d833efe94a56c502de8a38405d8fe  WIFI_MT7961_patch_mcu_1_2_hdr.bin

]$ md5sum WIFI_RAM_CODE_MT7961_1.bin
8d0a4f6dc2d01a8b442ae0b8d76d9122  WIFI_RAM_CODE_MT7961_1.bin

morrownr, 2023-12-05:
Here are my results from /lib/firmware/mediatek

$ md5sum WIFI_MT7961_patch_mcu_1_2_hdr.bin
0a4d833efe94a56c502de8a38405d8fe  WIFI_MT7961_patch_mcu_1_2_hdr.bin

$ md5sum WIFI_RAM_CODE_MT7961_1.bin
8d0a4f6dc2d01a8b442ae0b8d76d9122  WIFI_RAM_CODE_MT7961_1.bin

I can't check the BT firmware because it does not exist on my system. I delete it to prevent it from loading and using resources. It should not load given that BT is turned off in our adapters (Alfa AXML) but it does... that is a programming mistake that needs to be corrected.

ZerBea, 2023-12-05:
Thanks.  The md5 hashes matches.
I'll compile kernel 6.5 and give it another try.

ZerBea, 2023-12-05:
Very interesting. I moved to kernel 6.5 and from xhci to ehci.
```
[  268.472348] usb 1-1: new high-speed USB device number 5 using ehci-pci
[  268.630453] usb 1-1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  268.630467] usb 1-1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  268.630472] usb 1-1: Product: Wireless_Device
[  268.630476] usb 1-1: Manufacturer: MediaTek Inc.
[  268.630479] usb 1-1: SerialNumber: 000000000
[  268.900014] alg: No test for fips(ansi_cprng) (fips_ansi_cprng)
[  268.998709] mt7921u 1-1:1.3: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
[  269.136424] usb 1-1: reset high-speed USB device number 5 using ehci-pci
[  269.321037] usbcore: registered new interface driver mt7921u
[  269.322004] mt7921u 1-1:1.3: firmware: direct-loading firmware mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
[  269.322895] mt7921u 1-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
[  269.374788] Bluetooth: Core ver 2.22
[  269.374901] NET: Registered PF_BLUETOOTH protocol family
[  269.374904] Bluetooth: HCI device and connection manager initialized
[  269.374914] Bluetooth: HCI socket layer initialized
[  269.374918] Bluetooth: L2CAP socket layer initialized
[  269.374929] Bluetooth: SCO socket layer initialized
[  269.531093] bluetooth hci0: firmware: direct-loading firmware mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin
[  269.536378] usbcore: registered new interface driver btusb
[  269.587392] mt7921u 1-1:1.3: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
[  269.595039] mt7921u 1-1:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
[  270.293275] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  270.293287] Bluetooth: BNEP filters: protocol multicast
[  270.293297] Bluetooth: BNEP socket layer initialized
[  271.280859] mt7921u 1-1:1.3 wlx00c0cab574e6: renamed from wlan0
[  272.779970] Bluetooth: hci0: Device setup in 3175135 usecs
[  272.779983] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  274.784340] Bluetooth: hci0: Opcode 0x c03 failed: -110
```
 Looks like some of the problems are related to xhci.

morrownr, 2023-12-05:
Alright @ZerBea , you know better than to change 2 variables at the same time.

ZerBea, 2023-12-05:
Conclusion:
the new firmware loads fine on kernel 6.5

the ERROR is back:
`[ 4213.904348] usb 1-2: device not accepting address 21, error -71`

and packet injection is still not working.

ZerBea, 2023-12-05:
The last test:
```
[  340.961002] usb 3-2: new high-speed USB device number 4 using xhci_hcd
[  341.105074] usb 3-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  341.105087] usb 3-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  341.105093] usb 3-2: Product: Wireless_Device
[  341.105098] usb 3-2: Manufacturer: MediaTek Inc.
[  341.105103] usb 3-2: SerialNumber: 000000000
[  341.131221] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[  344.731107] Bluetooth: hci1: Device setup in 3517524 usecs
[  344.731115] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  346.804229] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[  348.934225] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[  351.067575] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[  351.196398] usb 3-2: reset high-speed USB device number 4 using xhci_hcd
```
Looks like a problem between driver and xhci (AMD USB controller).
Packet injection is not working.

xhci & new firmware -> firmware doesn't load -> no packet injection
xhci & old fimware -> firmware load -> no packet injection
ehci & new firmware -> firmware load, but after a while device not accepting address 21  -> no packet injection
ehci & old firmware -> firmware load -> no packet injection

tested on kernel 6.5 and on kernel 6.6
tested on Intel system and AMD system

Taken together, this results in 16 different test.
Regardless which firmware, kernel or hardware was used, packet injection failed.

ZerBea, 2023-12-05:
As a final test I compiled kernel 6.1 and got the same results.
Now I give up and wait for a driver update.

ZerBea, 2023-12-05:
Great, thanks.
My test systems:
2 x Intel (ehci)
2 x AMD (xhci)
5 x Raspberry Pi zero
2 x Raspberry Pi A
2 x Raspberry Pi B

Linux kernel 61, 65 and 6.6

ZerBea, 2023-12-07:
I found the problem.
Unfortunately it is similar to this one:
https://github.com/openwrt/mt76/issues/778

Driver report that active monitor mode is possible:
```
$ iw list | grep active
	Device supports active monitor (which will ACK incoming frames)
```

But if hcxdumptool set active monitor mode, it stops working.

If active monitor mode is disabled, everything's fine
```
0 ERROR(s) during runtime
638 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
83 EPB written to pcapng dumpfile

exit on sigterm
```
I don't think the problem is related to hcxdumptool, because it can be reproduced with iw, ip link and tshark, too:
```
$ sudo ip link set wlp22s0f0u4i3 down
$ sudo iw dev wlp22s0f0u4i3 set type monitor
$ sudo ip link set wlp22s0f0u4i3 up
$ tsahrk -i wlp22s0f0u4i3
22 packets captured

$ sudo ip link set wlp22s0f0u4i3 down
$ sudo iw dev wlp22s0f0u4i3 set monitor active
$ sudo ip link set wlp22s0f0u4i3 up 
$ tshark -i wlp22s0f0u4i3
Capturing on 'wlp22s0f0u4i3'
^C
0 packets captured
```

morrownr, 2023-12-07:
Have you modified the original message to reflect this finding?

How does this finding reflect overall? Is packet injection working with active monitor mode off?

I'm a little fuzzy after being sick for so many days. Why is active monitor mode needed?

ZerBea, 2023-12-07:
The head line has been modified.

Packet injection is working like a charm:
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7567045

Background:
Running active monitor mode, the device ACK incoming frames addressed to the virtual MAC of the device.
This feature is really useful to perform PMKID attacks.
At the moment, active monitor mode is working on:
```
mt76x0u
mt76x2u
```

It is not working on:
```
mt7601u
mt7921u
```
I see two options:
active monitor mode should be fixed, or
active monitor mode capability should not be reported by the driver
[code]
mt7601u
$ iw list | grep active
	Device supports active monitor (which will ACK incoming frames)

mt7921u
$ iw list | grep active
	Device supports active monitor (which will ACK incoming frames)
[/code]

morrownr, 2023-12-07:
> The head line has been modified.

It might help since this post has followed a long path to get where it is, if you use "Edit:" at the top of the original post then add what you have added in your last 2 posts so as to make it easy for a person that might fix it to understand without having to track things down. It might also, as am alternative work if you close this report and start a clean new post. I'm going to try to consolidate the information and add it to my main mt7921u bug list at my site. It seems quite clear at this point that active monitor is broken and is the cause of the problem at this point.

@morrownr

ZerBea, 2023-12-08:
Done. Important information and steps how to reproduce is now mentioned in the first comment.
Thanks for pointing me into this direction.

ZerBea, 2023-12-09:
I hope so, too.
The performance of the interface is enormous (when running latest git head hcxdumptool).
It is now number one:
https://github.com/ZerBea/hcxdumptool/discussions/361

ZerBea, 2023-12-09:
Let me explain the difference between active monitor and passive monitor mode:
If hcxdumptool request e.g. an ASSOCIATION by transmitting an ASSOCIATIONREQUEST frame
active monitor mode:
the AP responds with an ASSOCIATIONRESPONSE frame and the device that hcxdumptool use ACK it
passive monitor mode:
the AP responds with an ASSOCIATIONRESPONSE frame and due to missing ACK, it transmit up to 7 retries (which will spam the entire channel).

ZerBea, 2024-02-27:
We can close this report. As a workaround, hcxdumptool's active monitor mode is not running by default.
Even if the driver reports that it is possible, the user must allow it by a command line option.

morrownr, 2024-02-27:
I haven't seen any action on your report over at OpenWRT. This is something that needs to be fixed. I am neck deep in work right now but when I have a little time, I think I know a guy that can track this down and suggest a patch. Remind me in a couple of weeks if we see no action.

Nick

ZerBea, 2024-02-28:
Major reason for me to close this report:
Active monitor mode is a nice add on. Latest hcxdumptool received a workaround if the driver reports active monitor mode capabilities but fails on it.

Now we come to the real problems of this driver. Since kernel 6.7 it is broken.
First this one should be fixed. Than we can talk about active monitor mode. 

System one (AMD) running kernel 6.7:
```
$ uname -r
6.7.6-arch1-1

$ lspci
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:02.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 51)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 7
10:00.0 VGA compatible controller: NVIDIA Corporation AD103 [GeForce RTX 4080] (rev a1)
10:00.1 Audio device: NVIDIA Corporation Device 22bb (rev a1)
16:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] 500 Series Chipset USB 3.1 XHCI Controller
16:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] 500 Series Chipset SATA Controller
16:00.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] 500 Series Chipset Switch Upstream Port
20:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43ea
20:09.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43ea
21:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller 980 (DRAM-less)
2a:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 15)
2b:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller 980 (DRAM-less)
30:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c8)
30:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Renoir Radeon High Definition Audio Controller
30:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
30:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne USB 3.1
30:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne USB 3.1
30:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h/19h HD Audio Controller

$ dmesg
[ 1042.151987] usb 1-2.3: new high-speed USB device number 8 using xhci_hcd
[ 1042.287434] usb 1-2.3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 1042.287442] usb 1-2.3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 1042.287445] usb 1-2.3: Product: Wireless_Device
[ 1042.287447] usb 1-2.3: Manufacturer: MediaTek Inc.
[ 1042.287449] usb 1-2.3: SerialNumber: 000000000
[ 1042.444109] Bluetooth: Core ver 2.22
[ 1042.444136] NET: Registered PF_BLUETOOTH protocol family
[ 1042.444137] Bluetooth: HCI device and connection manager initialized
[ 1042.444141] Bluetooth: HCI socket layer initialized
[ 1042.444144] Bluetooth: L2CAP socket layer initialized
[ 1042.444147] Bluetooth: SCO socket layer initialized
[ 1042.482669] usbcore: registered new interface driver btusb
[ 1042.485428] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[ 1045.596913] Bluetooth: hci0: Device setup in 3041199 usecs
[ 1045.596920] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 1047.735333] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[ 1049.868468] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[ 1052.001823] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[ 1052.080358] usb 1-2.3: reset high-speed USB device number 8 using xhci_hcd
[ 1057.195568] usb 1-2.3: device descriptor read/64, error -110
[... 178 lines cut ...]
[  577.904346] usb 1-3: new high-speed USB device number 7 using ehci-pci
[  578.062423] usb 1-3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  578.062436] usb 1-3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  578.062441] usb 1-3: Product: Wireless_Device
[  578.062445] usb 1-3: Manufacturer: MediaTek Inc.
[  578.062448] usb 1-3: SerialNumber: 000000000
[... cut ...]
[  583.852386] mt7921u 1-3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
[  583.938256] Bluetooth: hci0: Device setup in 123617 usecs
[  583.938272] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  584.111418] mt7921u 1-3:1.3: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
[  584.116576] mt7921u 1-3:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
[  584.182541] Bluetooth: MGMT ver 1.22
[  585.783726] mt7921u 1-3:1.3 wlx00c0cab574e6: renamed from wlan0
```

morrownr, 2024-02-28:
@ZerBea 

FYI: I saw new firmware flow into linux-wireless last week so it should be posted for download this week or next. We have been making enough noise that the devs may have taken a look and found a fix.

As usual, the firmware guide is menu item 8, look at section 3 for this chipset:

https://github.com/morrownr/USB-WiFi

ZerBea, 2024-02-29:
@morrownr 
Problem is not the firmware, because it is the same on all kernels.
On 6.6 it is loaded, on 6.7 not.

morrownr, 2024-02-29:
> Active monitor mode is a nice add on. Latest hcxdumptool received a workaround if the driver reports active monitor mode capabilities but fails on it.

Are you sure you don't want to keep this open?

ZerBea, 2024-02-29:
If my remaining machine received an update to kernel 6.7 I'm no longer able to do some further going tests regarding active monitor mode. Unfortunately.
But I'll reopen this on next kernel 6.8 (or 6.9 or if the the problem mentioned above has been fixed).

ZerBea, 2024-02-29:
At least I found a firmware version that is working (for me) on all kernel versions:
```
Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20230526131214
mt7921u 1-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
mt7921u 1-2:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
```
Active monitor mode is not working.
Now that I can test active monitor mode again, I reopen this issue report as mentioned above.

morrownr, 2024-02-29:
Wow! I have been seeing a pickup in people reporting problems related to this. It is reasonable to assume if the driver is reporting active monitor mode support, then it will work. They try to use it and end up with a mess.

ZerBea, 2024-02-29:
@morrownr 
Unfortunately the same applies to the firmware.

ZerBea, 2024-08-05:
I  got my a solution:
```
$ hcxdumptool -L
...
* active monitor mode available (reported by driver - do not trust it)
```

From now hcxdumptool does not set active monitor mode by default. This has to be done by the user:
```
$ hcxdumptool -h
-A             : ACK incoming frames
                  INTERFACE must support active monitor mode
```
Closed, because there are serious issues as this one.

closed 2024-08-05 by ZerBea

morrownr, 2024-08-11:
@alexl83 @ZerBea 

This issue needs attention. I have no problem making some real noise if we need to do so to get it fixed.

I may be wrong but I think Active Monitor worked in this driver at one point. Can either of you confirm that?

Where I am going with this is if we can show that it worked at one point, then this is a REGRESSION which would add importance to getting a fix in place. If we were able to bisect it and show exactly what patch broke it, then we can clean up a report that I will send to linux-wireless and the Mediatek devs.

@ZerBea 

I know you know how to do a bisect. I know it takes a lot of time. I have never done one but would like to learn someday but right now I'm trying to get a couple of new drivers out of the door. Maybe @alexl83 can work with you on a bisect while you teach him how to do it. If you guys can sort this out, I will send the message.

@morrownr

ZerBea, 2024-08-11:
Active monitor mode is working on mt76x0u.
```
$ hcxdumptool -l
  2	  5	fc3497321ca3	980ee4f3e212	*	wlp48s0f4u2u4   	mt76x0u	NETLINK

$ sudo hcxdumptool -A --rcascan=active
...
^C
69 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
22 PROBERESPONSE(s) captured
```

It is not working on mt76x2u:
```
$ hcxdumptool -l
  3	  6	000fc913411c	a4a6a98a90ae	*	wlp48s0f4u2u4   	mt76x2u	NETLINK

$ sudo hcxdumptool -A --rcascan=active
...
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 driver is broken (most likely)
 no transmitter in range
 frames are filtered out by BPF
Warning: no PROBERESPONSES received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection
```

and it is not working on mt7921u:
```
$ hcxdumptool -l
  4	  7	00c0cab5742a	00c0cab5742a	*	wlp22s0f0u9u3i3 	mt7921u	NETLINK

$ sudo hcxdumptool -A --rcascan=active
...
^C
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 driver is broken (most likely)
 no transmitter in range
 frames are filtered out by BPF
Warning: no PROBERESPONSES received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection

```
So we shouldn't disable it in general for all drivers of the mt76 series (like the patch does).

Bisecting is useless, because it has never worked on the rest of the mt76 series drivers (except the first one: mt76x0u).
I guess that only mt76x0u include the code tor run real active monitor mode.

morrownr, 2024-08-11:
@ZerBea 

I certainly thought you had reported it was working with mt7612u.

Are you sure it is a lack of code or is it broken code?

As we have talked, I really want to get some adapters in the hands of Dubhater so he can take a look at this and a couple of other things. He is busy right now and has not answered my last email addressing this issue.

ZerBea, 2024-08-12:
@morrownr 

Thanks for pointing me into the right direction.
I have two USB hubs. One of them (USB3 hub) is connected to an USB3 port of my AMD motherboard while the other one (USB2 hub) is connected to an USB2 port of my AMD motherboard.
Mostly I run hcxdumptool tests in automatic mode (usb port is detected by hcxdumptool):
`$ sudo hcxdumptool -A --rcascan=active`
It looks like this is an epical fail.
Please take a look at this. Same test as above, but now the device is connected to an USB2 port:
```
$ lsusb
Bus 001 Device 008: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter

$ hcxdumptool -l
  3	  6	000fc913411c	806d97a8f78d	*	wlp48s0f4u2u4   	mt76x2u	NETLINK
...
^C
115 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
12 PROBERESPONSE(s) captured
```
Conclusion:
Active monitor mode on mt76x0u is working on USB2 and USB3 (ASUS USB AC51 is an USB2 adapter)
Active monitor mode on mt76x2u is only working on USB2 (ALLNET WA1200AC is an USB3 adapter)
Active monitor mode on mt7921u is neither working on USB2 nor on USB3 /ALFA AWUS 036AXML is an USB3 C adaper)

I never thought that the problem could be related to the USB part of the interface.

@alexl83 
Exactly that is the problem. Entire mt76 driver series use the same core. Inside the core ACTIVE MONITOR flag is 1. Unfortunately a change in core applies to all chipsets.

morrownr, 2024-08-12:
@ZerBea 

> I've never considered that the problem could be related to USB.

While helping users over the last few years, I have come upon a lot of usb related problems. It is common for the usb wifi adapters to be blamed for the problems. My rules regarding usb:

- USB2 is far more bug free and stable than USB3 so always test your USB3 adapters in a USB2 port, if able, to see if you get a better result. A better result could indicate USB3 is the problem.
- Some USB3 hub chipsets are shit. Pure shit. 
- A lot of powered external USB3 hubs are shit, some powered external USB2 hubs are shit.
- If you use a powered external hub, remove it and plug directly into ports on the system before blaming the adapter.
- The internal USB3 chipset in the RasPi4B is shit. I can help users get stable USB3 AP mode working on a Pi4B but users bitch a lot.
- The USB3 chipset in the Pi4B is shit...opps, I already said that but it pisses me off as it evidently cannot be fixed in software.
- Many USB3 chips in desktop and laptop systems are problematic. And power stability and standards compliance are messed up on some USB setups.
- In fact, the Pi4B USB system is extremely underpowered in the the entire bus is only supplied with 1200 ma when it should be 2800 mA. This is one reason I recommend Mediatek chips for those setting up AP's as the Mediatek chips use far less power on average than do Realtek chips.
- There are usb stack and driver issues as well.

> Why on earth should an USB3 problem have an impact on active monitor mode.

I can't answer that but it does not surprise me.

> Looks like we have to take a look at the USB part of the driver.

Yes, we really need to get adapters into the hands of Dubhater. 

But keep in mind, different systems use different USB chipsets and it may be those chipsets and their drivers that are the problem. Test in different systems before blaming the adapter and always make sure to test in a USB2 port as well.

Remind me of the command line to test active monitor and I will try it tomorrow with multiple adapters on multiple systems.

Nick

ZerBea, 2024-08-12:
@morrownr 
Just to make sure it is not my AMD notebook and it is not my AMD desktop, I put my hands on an Intel System running latest Linux stable Kernel 6.10.4:
The ALFA AWUS 036AXML is connected to one of the USB3 ports (Intel chipset this time);
```
$ cat /proc/cpuinfo
model name	: Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz

$ lspci 
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
```
same error regarding BT firmware:
```
[   10.018946] wlp3s0: authenticate with 3c:37:12:97:c0:22 (local address=b0:c0:90:46:7c:ab)
[   10.018952] wlp3s0: send auth to 3c:37:12:97:c0:22 (try 1/3)
[   10.120820] wlp3s0: send auth to 3c:37:12:97:c0:22 (try 2/3)
[   10.224250] wlp3s0: send auth to 3c:37:12:97:c0:22 (try 3/3)
[   10.327494] wlp3s0: authentication with 3c:37:12:97:c0:22 timed out
[   10.548833] wlp3s0: authenticate with 3c:37:12:97:c0:21 (local address=b0:c0:90:46:7c:ab)
[   10.548841] wlp3s0: send auth to 3c:37:12:97:c0:21 (try 1/3)
[   10.551315] wlp3s0: authenticated
[   10.553869] wlp3s0: associate with 3c:37:12:97:c0:21 (try 1/3)
[   10.558104] wlp3s0: RX AssocResp from 3c:37:12:97:c0:21 (capab=0x1411 status=0 aid=35)
[   10.558328] wlp3s0: associated
[   10.620667] wlp3s0: Limiting TX power to 20 (20 - 0) dBm as advertised by 3c:37:12:97:c0:21
[   11.030763] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[   13.164071] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[   15.297181] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[   15.306403] mt7921u 1-2:1.3: probe with driver mt7921u failed with error -5
[   15.307338] usbcore: registered new interface driver mt7921u
[   15.580500] usb 1-2: new low-speed USB device number 7 using xhci_hcd
[   15.710514] usb 1-2: device descriptor read/64, error -71
[   15.940554] usb 1-2: device descriptor read/64, error -71
[   16.170720] usb 1-2: new high-speed USB device number 8 using xhci_hcd
[   21.394105] usb 1-2: device descriptor read/64, error -110
```
The same happens if it is connected to one of the USB2 ports.

After the firmware
`/usr/lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin.zst`
has been removed, the WiFi part of the device is working.

Unfortunately the same issue as on the AMD systems:
```
$ sudo hcxdumptool -A --rcascan=active -i wlp0s20f0u2i3
^C
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 driver is broken (most likely)
 no transmitter in range
 frames are filtered out by BPF
Warning: no PROBERESPONSES received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection
```

morrownr, 2024-08-12:
Hi @ZerBea 

> But I'm sure this time it is the usb wifi adapter or its driver.

I'm going to go with it being the axml adapter. I think Alfa screwed something up. I don't know if it is an eeprom (efuse) issue or bad connection but I think they left the adapter in a situation where Linux is told that BT is supported but when it is time to turn BT on, nothing is there are the btusb driver does not handle that is a reasonable way. I am an old guy that learned programming with FORTRAN and am certainly not a wireless expert. It can take me a long time to sort out wireless coding issues... on the other hand, we really really need to get some mt adapters in Dubhater's hands as we have a few issues that he can sort out. He sent the new drivers in to linux-wireless yesterday so now the check it then fix it stuff starts which should allow some time here to figure out to get adapters to him.

I have 3 adapters based on the mt7921au chipset. One is the AXML, then I have an Edup and a Comfast. I'll see about doing some comparative testing later today.

ZerBea, 2024-08-16:
@alexl83 I started to bisect the kernel, but I don't expect to find the bug.
This report contains more than one bug:
https://bugzilla.kernel.org/show_bug.cgi?id=218763

I started with latest RC kernel:
```
$ uname -r
6.11.0-rc3-1-git-00156-gd7a5aa4b3c00
```

And it crashed my entire system:
```
[   14.860416] Bluetooth: hci0: command 0xfd98 tx timeout
[   14.860514] Bluetooth: hci0: failed to reset (-19)
[   14.860731] Bluetooth: hci0: Failed to apply iso setting (-110)
[   14.860756] Bluetooth: hci0: Device setup in 5666265 usecs
[   14.860766] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   16.993692] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[   19.130301] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[   21.189120] block nvme0n1: No UUID available providing old NGUID
[   21.260275] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[   21.260814] Oops: general protection fault, probably for non-canonical address 0xdead000000000122: 0000 [#1] PREEMPT SMP NOPTI
[   21.260830] CPU: 7 UID: 0 PID: 372 Comm: (udev-worker) Not tainted 6.11.0-rc3-1-git-00156-gd7a5aa4b3c00 #1 54019de5e52a8a007318718919bd12961f3a8cc1
[   21.260840] Hardware name: ASUSTeK COMPUTER INC. TUF Gaming FX505DT_FX505DT/FX505DT, BIOS FX505DT.316 01/28/2021
[   21.260846] RIP: 0010:hci_unregister_dev+0x45/0x1f0 [bluetooth]
[   21.260936] Code: 89 ef e8 6e d3 70 f6 f0 80 8b e9 0e 00 00 08 48 89 ef e8 ce c0 70 f6 48 c7 c7 68 c0 7d c1 e8 e2 21 71 f6 48 8b 43 08 48 8b 13 <48> 3b 18 0f 85 35 c8 06 00 48 3b 5a 08 0f 85 2b c8 06 00 48 89 42
[   21.260941] RSP: 0018:ffffa15b40993850 EFLAGS: 00010246
[   21.260949] RAX: dead000000000122 RBX: ffff91ff00db6000 RCX: 0000000000000000
[   21.260955] RDX: dead000000000100 RSI: ffff91ff09016c50 RDI: ffffffffc17dc068
[   21.260959] RBP: ffff91ff00db64d0 R08: ffff91ff1ab079c0 R09: 0000000000150011
[   21.260964] R10: 0000000000150011 R11: 0000000000000001 R12: ffff91ff00db6000
[   21.260968] R13: ffffffffc1750278 R14: ffffffffc1750278 R15: ffff91ff0bcf7c50
[   21.260974] FS:  00007f8fab327880(0000) GS:ffff92002ef80000(0000) knlGS:0000000000000000
[   21.260979] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   21.260984] CR2: 000055a91b099008 CR3: 00000001077fe000 CR4: 00000000003506f0
[   21.260989] Call Trace:
[   21.260998]  <TASK>
[   21.261004]  ? __die_body.cold+0x19/0x27
[   21.261017]  ? die_addr+0x3c/0x60
[   21.261027]  ? exc_general_protection+0x17d/0x400
[   21.261040]  ? asm_exc_general_protection+0x26/0x30
[   21.261058]  ? hci_unregister_dev+0x45/0x1f0 [bluetooth b63727ef82c06d4528a67e7731f14560eea7c053]
[   21.261122]  btusb_disconnect+0x67/0x170 [btusb a7d67ad402eace766a5d2b7d1bee422c1cdfa00f]
[   21.261137]  usb_unbind_interface+0x93/0x290
[   21.261149]  device_release_driver_internal+0x19c/0x200
[   21.261160]  usb_forced_unbind_intf+0x75/0xb0
[   21.261167]  usb_reset_device+0xe8/0x290
[   21.261180]  mt7921u_probe+0xab/0x230 [mt7921u 115e3ac356542846e09c4d3b6b8d77557e29d436]
[   21.261193]  usb_probe_interface+0xe8/0x2b0
[   21.261202]  really_probe+0xde/0x340
[   21.261209]  ? pm_runtime_barrier+0x54/0x90
[   21.261218]  ? __pfx___driver_attach+0x10/0x10
[   21.261224]  __driver_probe_device+0x78/0x110
[   21.261232]  driver_probe_device+0x1f/0xa0
[   21.261239]  __driver_attach+0xba/0x1c0
[   21.261247]  bus_for_each_dev+0x8f/0xe0
[   21.261257]  bus_add_driver+0x112/0x1f0
[   21.261265]  driver_register+0x72/0xd0
[   21.261272]  usb_register_driver+0x8c/0x130
[   21.261280]  ? __pfx_mt7921u_driver_init+0x10/0x10 [mt7921u 115e3ac356542846e09c4d3b6b8d77557e29d436]
[   21.261289]  do_one_initcall+0x5b/0x310
[   21.261303]  do_init_module+0x60/0x230
[   21.261313]  init_module_from_file+0x89/0xe0
[   21.261329]  idempotent_init_module+0x121/0x320
[   21.261341]  __x64_sys_finit_module+0x5e/0xb0
[   21.261350]  do_syscall_64+0x82/0x190
[   21.261361]  ? srso_return_thunk+0x5/0x5f
[   21.261371]  ? syscall_exit_to_user_mode+0x10/0x200
[   21.261379]  ? srso_return_thunk+0x5/0x5f
[   21.261385]  ? do_syscall_64+0x8e/0x190
[... 25 lines cut ...]
[   21.261593] RSP: 002b:00007ffd20ef57c8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   21.261600] RAX: ffffffffffffffda RBX: 0000559204c469c0 RCX: 00007f8fab5261fd
[   21.261606] RDX: 0000000000000004 RSI: 00007f8fab321376 RDI: 0000000000000021
[   21.261610] RBP: 00007f8fab321376 R08: 0000000000000001 R09: 00007ffd20ef5810
[   21.261614] R10: 0000000000000050 R11: 0000000000000246 R12: 0000000000020000
[   21.261618] R13: 0000559204c454d0 R14: 0000000000000000 R15: 0000559204c4fdf0
[   21.261631]  </TASK>
[   21.261634] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device amd_atl intel_rapl_msr intel_rapl_common snd_hda_codec_realtek joydev snd_hda_codec_generic snd_hda_scodec_component snd_hda_codec_hdmi mt7921u(+) mt792x_usb snd_hda_intel crct10dif_pclmul mt7921_common crc32_pclmul rtw88_8821ce snd_intel_dspcfg uvcvideo mt792x_lib polyval_clmulni rtw88_8821c snd_intel_sdw_acpi polyval_generic videobuf2_vmalloc mt76_connac_lib vfat rtw88_pci snd_hda_codec ghash_clmulni_intel mt76_usb uvc videobuf2_memops sha512_ssse3 snd_hda_core rtw88_core fat videobuf2_v4l2 btusb sha256_ssse3 mt76 snd_hwdep btrtl sha1_ssse3 ee1004 hid_multitouch aesni_intel btintel r8169 videodev asus_nb_wmi snd_pcm gf128mul sp5100_tco btbcm mac80211 realtek asus_wmi crypto_simd snd_timer videobuf2_common i2c_piix4 btmtk mdio_devres libarc4 platform_profile cryptd cfg802 [... cut ...]
[   21.262000] R10: 0000000000150011 R11: 0000000000000001 R12: ffff91ff00db6000
[   21.262005] R13: ffffffffc1750278 R14: ffffffffc1750278 R15: ffff91ff0bcf7c50
[   21.262010] FS:  00007f8fab327880(0000) GS:ffff92002ef80000(0000) knlGS:0000000000000000
[   21.262015] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   21.262020] CR2: 000055a91b099008 CR3: 00000001077fe000 CR4: 00000000003506f0
```
So we can mark this kernel as bad!

Now I'm going back to kernel 5.15 to test if this kernel is working flawless.

ZerBea, 2024-08-16:
Thanks, but you can't help here.

ZerBea, 2024-08-16:
BTW:
How to bisect a the kernel is explained here:
https://www.kernel.org/doc/html/latest/admin-guide/bug-bisect.html
On Arch:
https://bbs.archlinux.org/viewtopic.php?id=271926
On Debian:
https://wiki.debian.org/DebianKernel/GitBisect
On Ubuntu:
https://wiki.ubuntu.com/Kernel/KernelBisection

ZerBea, 2024-08-16:
I forgot to mention:
The ALFA AWUS036AXML has never(!) worked out of the box for me. I've found (different) bugs in every kernel version.

ZerBea, 2024-08-16:
Kernel 5.15 and 6.1 - not working out of the box as expected:
```
[   19.385811] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[   19.529178] usb 3-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[   19.529189] usb 3-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[   19.529193] usb 3-2: Product: Wireless_Device
[   19.529196] usb 3-2: Manufacturer: MediaTek Inc.
[   19.529199] usb 3-2: SerialNumber: 000000000
[   23.185949] Bluetooth: hci1: Device setup in 3531095 usecs
[   23.185956] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   25.308865] Bluetooth: hci1: Opcode 0x c03 failed: -110
[   27.442581] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[   29.575753] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[   29.710531] usb 3-2: reset high-speed USB device number 3 using xhci_hcd
[   29.832537] usb 3-2: device descriptor read/64, error -71
[   45.239019] usb 3-2: USB disconnect, device number 3
[   45.239430] mt7921u: probe of 3-2:1.3 failed with error -5
[   45.240098] usbcore: registered new interface driver mt7921u
[   52.492338] usb 3-2: new high-speed USB device number 4 using xhci_hcd
[   52.635829] usb 3-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[   52.635840] usb 3-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[   52.635844] usb 3-2: Product: Wireless_Device
[   52.635847] usb 3-2: Manufacturer: MediaTek Inc.
[   52.635850] usb 3-2: SerialNumber: 000000000
[   55.446130] Bluetooth: hci1: Device setup in 2721146 usecs
[   55.446140] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   57.521725] Bluetooth: hci1: Opcode 0x c03 failed: -110
[   59.655143] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[   61.788684] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[   61.914667] usb 3-2: reset high-speed USB device number 4 using xhci_hcd
```

After the BT firmware has been removed:
`$ sudo rm /usr/lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin.zst`
everything looks fine:
```
[  259.956625] usb 3-2: new high-speed USB device number 6 using xhci_hcd
[  260.101393] usb 3-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  260.101404] usb 3-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  260.101408] usb 3-2: Product: Wireless_Device
[  260.101412] usb 3-2: Manufacturer: MediaTek Inc.
[  260.101415] usb 3-2: SerialNumber: 000000000
[  260.126503] bluetooth hci1: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2
[  260.126516] Bluetooth: hci1: Failed to load firmware file (-2)
[  260.126525] Bluetooth: hci1: Failed to set up firmware (-2)
[  260.126530] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  260.251556] usb 3-2: reset high-speed USB device number 6 using xhci_hcd
[  260.398357] bluetooth hci1: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2
[  260.398368] Bluetooth: hci1: Failed to load firmware file (-2)
[  260.398375] Bluetooth: hci1: Failed to set up firmware (-2)
[  260.398380] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  260.445886] mt7921u 3-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a

[  260.708205] mt7921u 3-2:1.3: WM Firmware Version: ____010000, Build Time: 20240219111038
[  262.320588] mt7921u 3-2:1.3 wlp5s0f4u2i3: renamed from wlan0
[  366.545124] device wlp5s0f4u2i3 entered promiscuous mode
[  390.575563] device wlp5s0f4u2i3 left promiscuous mode
```

Frame injection and VIF is working:
```
$ sudo hcxdumptool -i wlp5s0f4u2i3 --rcascan=active
^C
788 Packet(s) captured by kernel
275 Packet(s) dropped by kernel
168 PROBERESPONSE(s) captured

$ sudo hcxdumptool -l
  3	  4	00c0cab574e6	980ee40b5d2f	*	wlp5s0f4u2i3    	mt7921u	NETLINK
```
Active monitor mode not:
```
$ sudo hcxdumptool -i wlp5s0f4u2i3 --rcascan=active -A
...
125 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: no PROBERESPONSES received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection

exit on sigterm
```

ZerBea, 2024-08-16:
Now kernel 6.6.45.

As long as the BT stuff is removed,
VIF is working
```
$ hcxdumptool -l
  2	  5	00c0cab574e6	9c93e4b45d49	*	wlp5s0f3u3i3    	mt7921u	NETLINK
```

Monitor mode and packet injection is working:
```
$ sudo hcxdumptool -i wlp5s0f4u2i3 --rcascan=active

^C
2214 Packet(s) captured by kernel
208 Packet(s) dropped by kernel
651 PROBERESPONSE(s) captured
```

Active monitor mode is not working:
```
$ sudo hcxdumptool -i wlp5s0f4u2i3 --rcascan=active

^C
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 no transmitter in range
 frames are filtered out by BPF
 driver is broken
Warning: no PROBERESPONSES received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection
```

My conclusion:
The ALFA AWUS036AXML has never worked if BT is installed.
Active monitor mode has never worked.
Since Linux kernel 6.11 the entire system crashes if the AXML has been plugged in.

ZerBea, 2024-08-16:
There is no need to patch the kernel, because monitor mode and frame injection is working as expected.
airgeddon is a script that use iw to set monitor mode:
https://github.com/v1s1t0r1sh3r3/airgeddon/blob/master/airgeddon.sh#L1518
It's enough to clear the flags here (flags none):
`$ iw phy "${phy_interface}" interface add "${iface_monitor_et_deauth}" type monitor flags none 2> /dev/null`

_Real deal would be for drivers to be fixed; do you think it is ever going to happen??_
Maybe if someone report this (BT & active monitor mode bug) to the linux wireless mailing list
Since I got this answer I do not report bugs any longer!
https://bugzilla.kernel.org/show_bug.cgi?id=219040#c11

ZerBea, 2024-08-16:
Rude is not the problem.
But my point of view similar bugs should be collected under the same topic (and not in mailing lists) in bugzilla as described here:
https://bugzilla.kernel.org/
"This bugzilla is for reporting bugs against upstream Linux kernels."
But this is obviously not the case:
https://bugzilla.kernel.org/show_bug.cgi?id=219040#c11

I think everyone can imagine what happens when one of this bugs reaches the major penetration testing distributions.

ZerBea, 2024-08-16:
_...but I'm going to start raising kernel bugs on distribution trackers_
If you're running a distribution kernel that is exactly the recommended way:
"If you did not compile your own kernel from scratch, you are probably in the wrong place.
Please use the following links to report a bug to your distribution instead"
https://bugzilla.kernel.org/

Mostly I'm running three different versions (via systemd-boot):
a custom kernel == Linux mainline: 6.11-rc3
(compiled with Arch kernel config and patched with Arch kernel patches) 

Arch Linux kernel == Linux stable: 6.10.5
https://archlinux.org/packages/core/x86_64/linux/

Arch Linux LTS kernel == Linux longterm: 6.6.46
https://archlinux.org/packages/core/x86_64/linux-lts/

If you run hcxlabtool or hcxdumptool you don't need to set active monitor mode.
Due to ASSOCIATION and REASSOCIATION attacks, you don't need to inject stupid DEAUTHENTICATION frames.
Both tools request PMKIDs from APs and EAPOL M2s from CLIENTs. JtR and hashcat can work on this hashes.

ZerBea, 2024-08-16:
@alexl83 
Good know that a workaround (for the AXML) has an impact on mt7921e devices.
Looks like every adapter needs different flags.
That's why hcxdumptool (without -A) leave the default flags untouched while -A set NL80211_ATTR_MNTR_FLAGS to NL80211_MNTR_FLAG_ACTIVE.

morrownr, 2024-08-17:
@ZerBea @alexl83 

Sorry for the delay. Health problems affecting my wife and I. I try to catch up when I can.

I just ran the following on my Edup adapter  based on the mt7921au: (driver = mt7921u)

$ sudo hcxdumptool -i wlxe84e06aa849b --rcascan=active

```
^C
0 ERROR(s) during runtime
11713 Packet(s) captured by kernel
2635 Packet(s) dropped by kernel
2481 PROBERESPONSE(s) captured

exit on sigterm

```

I then ran the following on my Asus PCIe card  based on the mt7922: (driver = mt7921e)

$ sudo hcxdumptool -i wlp2s0 --rcascan=active

```
^C
0 ERROR(s) during runtime
4147 Packet(s) captured by kernel
381 Packet(s) dropped by kernel
703 PROBERESPONSE(s) captured

exit on sigterm

```

I am not an expert with hcxdumptool so you will to tell me what I did and what it means. I can run additional tests. Certainly I would like to test active monitor but will test anything you want. Testing active monitor is a waste of time on any Realtek but I also have adapters based on mt7612u and mt7610u chips. I can also grab my axml to see what it does.

Mt opinion is that Alfa screwed something up in the axml. 

>  I though Mediatek was the best wireless platform but it seems I got it wrong :(

I would argue that Mediatek is the best wireless platform for monitor mode and ap mode use cases. I've helped people with their wireless problems for several years. With usb adapters we only have the choice of Mediatel and Realtek. Mediatek is much better. I think Alfa just screwed this up with the axml because I am not seeing many of the reported problems with my Edup adapter with the same chip as the axml. There is no doubt there is a problem... exactly what the core issue is and how it can best be fixed is not obvious to me at this point but I am trying to see if we can get to the bottom of it and you guys are helping.

ZerBea, 2024-08-17:
@morrownr 
The scan mode is not an attack mode!
This command
`$ sudo hcxdumptool -i wlp2s0 --rcascan=active`
is similar to this command:
`$ sudo iw dev wlp48s0f4u2u4 scan`

The major difference:
hcxdumptool is doing a scan in monitor mode while iw is doing the scan in managed mode.
By default only the most common channels (1,6,11) are scanned.
Therefore hcxdumptool transmit an undirected PROBEREQUEST and counts the BEACONs and PROBERESPONSEs of APs in range.

On exit, you get some general information:
0 ERROR(s) during runtime -> ERRORs reported by raw socket (e.g. failed to transmit a frame, failed to receive a frame, failed to set channel)
4147 Packet(s) captured by kernel -> total received frames
381 Packet(s) dropped by kernel -> frames dropped by kernel https://unix.stackexchange.com/questions/144794/why-would-the-kernel-drop-packets
703 PROBERESPONSE(s) captured -> total PROBERESONSE frames received.

BTW:
I fully agree: The AXML is not a big deal!
Even on OpenWRT 23.05 (compiled via toolchain and running on a Raspberry Pi Zero) it stops working after awhile (on high work load - and hcxdumptool produce high work load when running in attack mode against APs and CLIENTs).

ZerBea, 2024-08-17:
I forgot to mention:
As of today, I'm testing an ancient ALFA AWUS036NHA (longterm test since several weeks).
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7550862
It is working (Raspberry Pi Zero / OpenWRT 23.05 compiled via toolchain) a thousand times better (flawless!) than the AXML!

morrownr, 2024-08-17:
@ZerBea 

Thanks for the explanation on the test I did. Now, what command line should I use to test active monitor? I want to do a test on both my Edup adapter and the Alfa adapter to see if I see variances in results. I'll also test my mt7922 card as it uses the mt7921 driver as well.

ZerBea, 2024-08-17:
@morrownr 
first get information about the device:
```
$ hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  1   4 74da38f2038e 74da38f2038e * wlp48s0f4u2u4    mt7601u (NETLINK)

* active monitor mode available (reported by driver - do not trust it)
+ monitor mode available (reported by driver)
- no monitor mode available
```

If active monitor mode is available (as reported by the driver), run hcxdumptool in active monitor mode:
`$ sudo hcxdumptool -i wlp48s0f4u2u4 --rcascan=active -A`

If you see something like this, it is working:
```
^C
166 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
33 PROBERESPONSE(s) captured

exit on sigterm
```

ZerBea, 2024-08-17:
Add -F to scan all available (allowed by regulatory domain) ffrequencies.

[10 more replies]