openwrt/mt76 #840: AX6S: mt7622 (mt7615): Low (2.4GHz) tx speed to clients with a single antenna & unstable connect

Wizard160888, 2023-11-16:
Good view through one panel wall. Near the router, it's fine. Regardless of openwrt version and settings.

**HW for test:**
**Notebook:** _Lenovo IdeaPad L340-15API_ with intel ax210 (2 antenna) - OK (147Mbit/s)
**Phones:** _Samsung Galaxy M14_ (1 antenna)

[image]

[image]

<details><summary>rm -f /tmp/router_iwinfo.log && while true; do iwinfo wl0-ap0 assoclist >> /tmp/router_iwinfo.log && sleep 2; done</summary>
'------------------------------------------------------------------------------------'

DE:AC:34:5A:6E:55  -58 dBm / -86 dBm (SNR 28)  56 ms ago
	RX: 58.5 MBit/s, MCS 6, 20MHz                 700895 Pkts.
	TX: 43.3 MBit/s, MCS 4, 20MHz                 414597 Pkts.
	expected throughput: 21.1 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -89 dBm (SNR 28)  0 ms ago
	RX: 39.0 MBit/s, MCS 4, 20MHz                 701199 Pkts.
	TX: 19.5 MBit/s, MCS 2, 20MHz                 415052 Pkts.
	expected throughput: 9.1 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -90 dBm (SNR 29)  0 ms ago
	RX: 65.0 MBit/s, MCS 6, 20MHz                 702081 Pkts.
	TX: 19.5 MBit/s, MCS 2, 20MHz                 416162 Pkts.
	expected throughput: 8.0 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -90 dBm (SNR 29)  4 ms ago
	RX: 58.5 MBit/s, MCS 6, 20MHz                 703014 Pkts.
	TX: 21.7 MBit/s, MCS 2, 20MHz                 417385 Pkts.
	expected throughput: 10.7 MBit/s

DE:AC:34:5A:6E:55  -62 dBm / -89 dBm (SNR 27)  0 ms ago
	RX: 65.0 MBit/s, MCS 6, 20MHz                 703817 Pkts.
	TX: 19.5 MBit/s, MCS 2, 20MHz                 418207 Pkts.
	expected throughput: 8.7 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -89 dBm (SNR 28)  8 ms ago
	RX: 52.0 MBit/s, MCS 5, 20MHz                 704727 Pkts.
	TX: 21.7 MBit/s, MCS 2, 20MHz                 419201 Pkts.
	expected throughput: 10.3 MBit/s

DE:AC:34:5A:6E:55  -62 dBm / -91 dBm (SNR 29)  4 ms ago
	RX: 65.0 MBit/s, MCS 6, 20MHz                 705729 Pkts.
	TX: 19.5 MBit/s, MCS 2, 20MHz                 420288 Pkts.
	expected throughput: 10.2 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -91 dBm (SNR 30)  4 ms ago
	RX: 52.0 MBit/s, MCS 5, 20MHz                 706710 Pkts.
	TX: 21.7 MBit/s, MCS 2, 20MHz                 421346 Pkts.
	expected throughput: 10.4 MBit/s

DE:AC:34:5A:6E:55  -62 dBm / -88 dBm (SNR 26)  12 ms ago
	RX: 65.0 MBit/s, MCS 6, 20MHz                 708070 Pkts.
	TX: 28.9 MBit/s, MCS 3, 20MHz                 422446 Pkts.
	expected throughput: 17.2 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -86 dBm (SNR 25)  0 ms ago
	RX: 52.0 MBit/s, MCS 5, 20MHz                 711556 Pkts.
	TX: 21.7 MBit/s, MCS 2, 20MHz                 423866 Pkts.
	expected throughput: 9.8 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -87 dBm (SNR 26)  4 ms ago
	RX: 39.0 MBit/s, MCS 4, 20MHz                 715611 Pkts.
	TX: 26.0 MBit/s, MCS 3, 20MHz                 425639 Pkts.
	expected throughput: 10.8 MBit/s

DE:AC:34:5A:6E:55  -63 dBm / -87 dBm (SNR 24)  0 ms ago
	RX: 52.0 MBit/s, MCS 5, 20MHz                 719013 Pkts.
	TX: 21.7 MBit/s, MCS 2, 20MHz                 427204 Pkts.
	expected throughput: 9.4 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -88 dBm (SNR 27)  4 ms ago
	RX: 43.3 MBit/s, MCS 4, 20MHz                 723015 Pkts.
	TX: 26.0 MBit/s, MCS 3, 20MHz                 429068 Pkts.
	expected throughput: 18.0 MBit/s

DE:AC:34:5A:6E:55  -63 dBm / -75 dBm (SNR 12)  64 ms ago
	RX: 19.5 MBit/s, MCS 2, 20MHz                 723745 Pkts.
	TX: 39.0 MBit/s, MCS 4, 20MHz                 429481 Pkts.
	expected throughput: 15.1 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -79 dBm (SNR 18)  8 ms ago
	RX: 65.0 MBit/s, MCS 6, 20MHz                 726840 Pkts.
	TX: 28.9 MBit/s, MCS 3, 20MHz                 431061 Pkts.
	expected throughput: 14.2 MBit/s

DE:AC:34:5A:6E:55  -60 dBm / -84 dBm (SNR 24)  12 ms ago
	RX: 52.0 MBit/s, MCS 5, 20MHz                 730600 Pkts.
	TX: 26.0 MBit/s, MCS 3, 20MHz                 432763 Pkts.
	expected throughput: 17.4 MBit/s

DE:AC:34:5A:6E:55  -61 dBm / -87 dBm (SNR 26)  0 ms ago
	RX: 58.5 MBit/s, MCS 6, 20MHz                 730731 Pkts.
	TX: 43.3 MBit/s, MCS 4, 20MHz                 432880 Pkts.
	expected throughput: 7.4 MBit/s
'------------------------------------------------------------------------------------'
</details>

_ZTE Blade A5 2020RU_ (1 antenna) ~ same as samsung.
_Honor 5C_ (2 antenna) - OK (~75Mbit/s)

**OpenWrt version:**
23.05.2, r24364-79ee0d2cee [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=79ee0d2ceead54edae002c3bd0460d3abc645ce0) (with/without mt76 version 4a0f839da0f108837baad908ed12b550f493b912)

**OpenWrt target/subtarget:**
mediatek/mt7622

**Device:**
Xiaomi Redmi Router AX6S

jww-sh, 2023-12-08:
Similar issues with the AX6S on a Samsung A33 5g

Wizard160888, 2023-12-09:
r24600-9291f6025f [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=9291f6025f4dd3f0b54cc7254e869914fbdc5ce0) + linux 6.1.66 - **problem solved**! Very very thanks!!!
[image]

Wizard160888, 2023-12-22:
I won't reopen after noticing [mediatek: disable btif for mt7622 devices](https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=aa7fa4b3815fe183abf065145f8fc272aaf614f2)

Wizard160888, 2024-03-23:
Turning off aql helped me.

DianP, 2024-12-04:
Having similar issues wi OpenWrt 24.10.0-rc1 on my AX6S

Issues with Device:
- Macbook Pro M1
- iPad Pro M2
- Samsung S10

vtv160888, 2026-11-08:
The issue persists in version 25.12.6. I tried using calibration data from other sources, but that didn't help. The counter at "/sys/kernel/debug/ieee80211/wl0/statistics/dot11FCSErrorCount" is increasing rapidly.