openwrt/mt76 #841: MT7603E 2.4 GHz Wi-Fi issue

malekairmaroc7, 2023-11-18:
Hello!

I have a TP-Link Archer C6 (EU V3) router running the latest OpenWRT 23.05.0 release.
I still notice unstable Wi-Fi connections when using it. The connection speed is still slow and I sometimes experience complete outage for some minutes.

mb555444, 2023-11-20:
Same here. 2.4 Ghz Wifi is unusable unless you are 1m away from the router. Wifi Analyzers show a 20dB difference in signal strength compared to the MTK driver. I have been forced to compile my own OpenWRT image with the official MTK driver due to this, but I would love to get back to the mt76 driver some day and use the stock OpenWRT image.

Ser9ei, 2023-12-10:
The same issue, and reproduced on the 21.02.7 and 23.05.2 OpenWrt with Xiaomi Mi WiFi Router 3gv2 and Xiaomi Mi WiFi Router 4a Gigabit v2.

mb555444, 2023-12-11:
> The same issue, and reproduced on the 21.02.7 and 23.05.2 OpenWrt with Xiaomi Mi WiFi Router 3gv2 and Xiaomi Mi WiFi Router 4a Gigabit v2.
> 
> upd. I believe this issue can be closed and the issue can continue to be tracked at #719.

The recommendation on #719 was to create a new issue.

lukasz1992, 2024-02-16:
Check the latest snapshot, it should be fixed