openwrt/mt76 #843: mt7615: tx.c: Kernel crash in mt76_tx

cotequeiroz, 2023-11-24:
I've encountered consistent kernel crashes on both Linksys E8450 and Xiaomi Redmi Router AX6S, both mt7622, running OpenWrt Master ae500e62e2.  It happens on 23.05 as well, although I have not saved any logs for them, or even know if it crashed in the same place.  I haven't been able to establish any means to promptly cause the crash, or any action that I can correlate with crashes, but idle routers don't appear to crash.

```
Unable to handle kernel write to read-only memory at virtual address 0000000000000000
CPU: 1 PID: 396 Comm: napi/mtk_eth-4 Tainted: G S                5.15.137 #0
pc : mt76_tx+0xb8/0x634 [mt76]
```

<details><summary>Xiaomi Redmi AX6S Crashlog</summary>
<p>

```
<1>[385471.808583] Unable to handle kernel write to read-only memory at virtual address 0000000000000000
<1>[385471.817576] Mem abort info:
<1>[385471.820455]   ESR = 0x0000000096000045
<1>[385471.824282]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[385471.829685]   SET = 0, FnV = 0
<1>[385471.832818]   EA = 0, S1PTW = 0
<1>[385471.836037]   FSC = 0x05: level 1 translation fault
<1>[385471.841001] Data abort info:
<1>[385471.843960]   ISV = 0, ISS = 0x00000045
<1>[385471.847879]   CM = 0, WnR = 1
<1>[385471.850925] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000454d8000
<1>[385471.857450] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
<0>[385471.866235] Internal error: Oops: 0000000096000045 [#1] SMP
<7>[385471.871885] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc sch_cake poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_reject_ipv4 nf_log_syslog nf_flow_table nf_dup_netdev nf_conntrack_netlink nf_conntrack_bridge nf_conncount
<7>[385471.872072]  libcurve25519_generic libcrc32c libchacha iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hwmon crc_ccitt compat act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport ledtrig_oneshot xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel nls_utf8 nls_iso8859_1 nls_cp437 seqiv uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ohci_platform ohci_hcd ledtrig_transient
<7>[385471.958789]  gpio_button_hotplug vfat fat exfat usbcore usb_common
<7>[385472.050996] CPU: 1 PID: 394 Comm: napi/mtk_eth-4 Not tainted 5.15.137 #0
<7>[385472.057777] Hardware name: Xiaomi Redmi Router AX6S (DT)
<7>[385472.063166] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[385472.070206] pc : mt76_tx+0xb8/0x634 [mt76]
<7>[385472.074395] lr : mt7915_eeprom_get_power_delta+0x1a08/0x23f0 [mt7915e]
<7>[385472.081008] sp : ffffffc009253400
<7>[385472.084400] x29: ffffffc009253400 x28: ffffff80034c08a0 x27: ffffffc0092535c0
<7>[385472.091618] x26: ffffffc0092535c0 x25: ffffff8003cab328 x24: ffffff80034c0db0
<7>[385472.098835] x23: ffffff8003cab328 x22: ffffffc000f10018 x21: ffffff8003cab300
<7>[385472.106052] x20: ffffff80034c2040 x19: ffffff80034fcce8 x18: 0000000000000000
<7>[385472.113270] x17: 0000000000001e40 x16: ffffff8002962000 x15: 0000000006c00000
<7>[385472.120487] x14: 0000000000000008 x13: 00000003aaaa0000 x12: 0000000000000000
<7>[385472.127705] x11: ffffff80002822b8 x10: 0000000000000014 x9 : 0000000000001f37
<7>[385472.134922] x8 : ffffff80034fcc3a x7 : 0000000055b90d00 x6 : ffffff80093484bc
<7>[385472.142140] x5 : ffffff80034c2040 x4 : 0000000000000000 x3 : 0000000000000001
<7>[385472.149356] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff80034fce28
<7>[385472.156574] Call trace:
<7>[385472.159099]  mt76_tx+0xb8/0x634 [mt76]
<7>[385472.162931]  mt7915_eeprom_get_power_delta+0x1a08/0x23f0 [mt7915e]
<7>[385472.169195]  __ieee80211_schedule_txq+0xf50/0x1830 [mac80211]
<7>[385472.175058]  __ieee80211_xmit_fast+0x7b8/0xa90 [mac80211]
<7>[385472.180558]  __ieee80211_subif_start_xmit+0x1e8/0x354 [mac80211]
<7>[385472.186666]  ieee80211_subif_start_xmit+0x3c/0x3bc [mac80211]
<7>[385472.192514]  ieee80211_subif_start_xmit_8023+0x310/0x420 [mac80211]
<7>[385472.198884]  dev_hard_start_xmit+0xe4/0x180
<7>[385472.203149]  __dev_queue_xmit+0x990/0xc50
<7>[385472.207238]  dev_queue_xmit+0x10/0x20
<7>[385472.210978]  br_dev_queue_push_xmit+0x9c/0x1b4
<7>[385472.215504]  br_forward_finish+0x98/0xb0
<7>[385472.219508]  __br_forward+0x138/0x1a0
<7>[385472.223250]  br_forward+0x124/0x150
<7>[385472.226819]  br_handle_frame_finish+0x310/0x4d4
<7>[385472.231430]  br_handle_frame+0x398/0x41c
<7>[385472.235434]  __netif_receive_skb_core.constprop.0+0x254/0xca4
<7>[385472.241259]  __netif_receive_skb_list_core+0xd4/0x1ec
<7>[385472.246390]  netif_receive_skb_list_internal+0x1b4/0x280
<7>[385472.251781]  napi_complete_done+0x64/0x1a0
<7>[385472.255955]  gro_cell_poll+0x80/0xa0
<7>[385472.259615]  __napi_poll+0x54/0x1b0
<7>[385472.263183]  net_rx_action+0x2a8/0x330
<7>[385472.267010]  _stext+0x10c/0x28c
<7>[385472.270232]  do_softirq+0x8c/0xa0
[... 79 lines cut ...]
``` 

</p>

</details> 

gdb points the address of the crash to ./include/linux/skbuff.h:2021
```
(gdb) l *(mt76_tx+0xb8)
0x7ec8 is in mt76_tx (./include/linux/skbuff.h:2021).
```
``` C
static inline void __skb_insert(struct sk_buff *newsk,
       	       	       	       	struct sk_buff *prev, struct sk_buff *next,
       	       	       	       	struct sk_buff_head *list)
{
2019       	WRITE_ONCE(newsk->next, next);
2020       	WRITE_ONCE(newsk->prev, prev);
2021       	WRITE_ONCE(next->prev, newsk);
2022       	WRITE_ONCE(prev->next, newsk);
2023       	WRITE_ONCE(list->qlen, list->qlen + 1);
}
```
The line that starts a long chain of inlined functions that ends with `__skb_insert` is in tx.c:349
https://github.com/openwrt/mt76/blob/2afc7285f75dca5a0583fd917285bf33f1429cc6/tx.c#L348-L350

cotequeiroz, 2023-12-19:
@rany2 
Are having the same problem?

I'm bisecting this:  Here's my bisect log so far:
```
git bisect start
# status: waiting for both good and bad commits
# bad: [2afc7285f75dca5a0583fd917285bf33f1429cc6] wifi: mt76: mt7925: add Mediatek Wi-Fi7 driver for mt7925 chips
git bisect bad 2afc7285f75dca5a0583fd917285bf33f1429cc6
# status: waiting for good commit(s), bad commit known
# good: [969b7b5ebd129068ca56e4b0d831593a2f92382f] wifi: mt76: mt7915: add support for MT7981
git bisect good 969b7b5ebd129068ca56e4b0d831593a2f92382f
# good: [bdf7ca5db70779d5763145f2fefbfde58f094903] wifi: mt76: mt7921: move common register definition in mt792x_regs.h
git bisect good bdf7ca5db70779d5763145f2fefbfde58f094903
# bad: [b3f739a64e6bd5a10a3cfaacfbe25f7b1f6ee0b3] wifi: mt76: mt7996: add eht rx rate support
git bisect bad b3f739a64e6bd5a10a3cfaacfbe25f7b1f6ee0b3
```
```
Bisecting: 19 revisions left to test after this (roughly 4 steps)
[3d65cd7a8dad351190d1bf6678819267a2313090] wifi: mt76: mt7921: move runtime-pm pci code in mt792x-lib
```

I'm waiting for a week without a crash before I declare a good commit.  This may take a while to complete....

I caught the moment when the last crash occurred.  My Windows notebook was waking up early in the morning, without much active usage of other devices, in an area with poor connection, within the range of at least 4 APs (none of them above -76dB) when it happened.  I could not determine if the notebook was connected to the crashing AP, and I would say it was probably not.  I was running a script to show the APs' uptime, and noticed one of the APs was not connecting.  Then I checked the pstore contents, and confirmed the crash. It may just be coincidence, or not the only cause, since at least one of the crashes occurred while the notebook was not on site.  I tried to reproduce the crash for an hour before continuing the bisect, but could not do it.

rany2, 2023-12-19:
@cotequeiroz As the 2g band doesnt get a lot of use this doesn't occur very frequently. It has only occured a few times, not enough to be a nuisance yet.

I have the same issue as you with identical crashlog.

rany2, 2023-12-19:
@cotequeiroz https://github.com/openwrt/mt76/blob/bebd9cffc2aeb2cecb40aadbb8c6eab3bdf7971b/tx.c#L618 shouldn't there be a check if wcid_to_sta returned NULL here?

cotequeiroz, 2023-12-20:
> @cotequeiroz As the 2g band doesnt get a lot of use this doesn't occur very frequently. It has only occured a few times, not enough to be a nuisance yet.

I'm quite sure the issue is not restricted to 2.4G band, as one of the crashed APs had that interface disabled.  Six of my 9 APs at home have crashed.  I'm also monitoring 4 APs in different locations that have never crashed--2 of the last four are at the office, and handle more clients than the ones at home, and the other 2 are sporadically used.  The 2 in the office are serving mostly phones (iOS and Android) and Windows notebooks.  At home, I can confirm crashes without Android phones.  There are 3 Windows notebooks, iOS devices, echo Dots, Chromecasts, and 2 Nest thermostats.  Another difference is that the crashing APs all use WPA3-SAE, while the others run either WPA2-802.1x with MFP and FT or WPA2-PSK/MFP/FT.  I would guess the problem is more related to vendor interoperability than to the security protocol.

> should there be a check if wcid_to_sta returned NULL here?
>
> Edit: nevermind, probably not

Probably not.  IIUC The NULL pointer error is happening in the skb queue.  I imagine the crash happens when another thread is modifying the queue at the same time without the spin_lock.

rany2, 2023-12-20:
> Probably not. IIUC The NULL pointer error is happening in the skb queue. I imagine the crash happens when another thread is modifying the queue at the same time without the spin_lock.

I considered that but the trouble is that `wcid->tx_pending` doesn't get any usage besides in mac80211.c and tx.c and it all seems to be correct usage.

>  Another difference is that the crashing APs all use WPA3-SAE, while the others run either WPA2-802.1x with MFP and FT or WPA2-PSK/MFP/FT

Something interesting is that I was streaming the `logread -f` output on one of my APs and I was lucky enough to see some hostapd spam that goes like this moments prior to the crash:

```
hostapd: wlan1: STA <redacted> IEEE 802.11: authenticated
hostapd: wlan1: STA <redacted> IEEE 802.11: associated (aid 3)
hostapd: wlan1: STA <redacted> IEEE 802.11: authenticated
hostapd: wlan1: STA <redacted> IEEE 802.11: associated (aid 3)
hostapd: wlan1: STA <redacted> IEEE 802.11: authenticated
hostapd: wlan1: STA <redacted> IEEE 802.11: associated (aid 3)
```

cotequeiroz, 2023-12-20:
I have the logs of every crash, but haven't looked for a pattern there.  I'll take a look.  Meanwhile, this is the syslog from the last crash:

```
Dec 18 08:11:07 hostapd: wlan5g: STA <Windows_NB_MAC> IEEE 802.11: authentication OK (open system)
Dec 18 08:11:07 hostapd: wlan5g: STA <Windows_NB_MAC> WPA: event 0 notification
Dec 18 08:11:07 hostapd: wlan5g: STA <Windows_NB_MAC> MLME: MLME-AUTHENTICATE.indication(<Windows_NB_MAC>, OPEN_SYSTEM)
Dec 18 08:11:07 hostapd: wlan5g: STA <Windows_NB_MAC> IEEE 802.11: authenticated
Dec 18 08:11:07 hostapd: wlan5g: STA <Windows_NB_MAC> WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Dec 18 08:11:07 hostapd: wlan5g: STA <Windows_NB_MAC> IEEE 802.11: association OK (aid 3)
Dec 18 08:11:07 hostapd: wlan5g: STA <Windows_NB_MAC> WPA: event 8 notification
Dec 18 08:11:07 kernel: [319568.827358] Unable to handle kernel write to read-only memory at virtual address 0000000000000000
Dec 18 08:11:07 kernel: [319568.836350] Mem abort info:
Dec 18 08:11:07 kernel: [319568.839221]   ESR = 0x0000000096000045
Dec 18 08:11:07 kernel: [319568.843048]   EC = 0x25: DABT (current EL), IL = 32 bits
Dec 18 08:11:07 kernel: [319568.848451]   SET = 0, FnV = 0
Dec 18 08:11:07 kernel: [319568.851603]   EA = 0, S1PTW = 0
Dec 18 08:11:07 kernel: [319568.854825]   FSC = 0x05: level 1 translation fault
Dec 18 08:11:07 kernel: [319568.859801] Data abort info:
Dec 18 08:11:07 kernel: [319568.862769]   ISV = 0, ISS = 0x00000045
Dec 18 08:11:07 kernel: [319568.866695]   CM = 0, WnR = 1
Dec 18 08:11:07 kernel: [319568.869751] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000043e59000
Dec 18 08:11:07 kernel: [319568.876287] [0000000000000000] pgd=0800000040b7b003, p4d=0800000040b7b003, pud=0800000040b7b003, pmd=0000000000000000
```

cotequeiroz, 2023-12-20:
I am posting the syslog at the time of the crashes I recorded before starting the bisect.  I did not find any conclusive pattern, except maybe that the 2.4G interface does not appear as the last thing the AP does before choking.  It does not get much traffic, so I can't rule it out.  Notice that there is a crash during the 4-way HS, other right after it, but most of them happen right after `WPA: event 8 notification`.

<details><summary>Syslogs of crashes before the bisect:</summary>
<p>

```
Oct 18 21:37:37 ap08 hostapd: wlan5g: EAPOL-4WAY-HS-COMPLETED 0f:0f:0f:0f:0f:0f
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: associated (aid 1)
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association OK (aid 1)
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: authenticated
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: binding station to interface 'wlan5g'
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.1X: authorizing port
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f MLME: MLME-DELETEKEYS.request(0f:0f:0f:0f:0f:0f)
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f MLME: MLME-REASSOCIATE.indication(0f:0f:0f:0f:0f:0f)
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f RADIUS: starting accounting session 7EE807515EA33C69
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: event 1 notification
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: event 4 notification
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: event 8 notification
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: pairwise key handshake completed (RSN)
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: received EAPOL-Key frame (2/4 Pairwise)
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: received EAPOL-Key frame (4/4 Pairwise)
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: sending 1/4 msg of 4-Way Handshake
Oct 18 21:37:37 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: sending 3/4 msg of 4-Way Handshake
Oct 18 21:37:58 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association SA Query attempt 1
Oct 18 21:37:58 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association SA Query attempt 2
Oct 18 21:37:58 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: authenticated
Oct 18 21:37:58 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: authentication OK (open system)
Oct 18 21:37:58 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f MLME: MLME-AUTHENTICATE.indication(0f:0f:0f:0f:0f:0f, OPEN_SYSTEM)
Oct 18 21:37:58 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: event 0 notification
Oct 18 21:37:59 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association SA Query attempt 3
Oct 18 21:37:59 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association SA Query attempt 4
Oct 18 21:37:59 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association SA Query attempt 5
Oct 18 21:37:59 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association SA Query timed out
Oct 18 21:38:00 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: association OK (aid 1)
Oct 18 21:38:00 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: authenticated
Oct 18 21:38:00 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f IEEE 802.11: authentication OK (open system)
Oct 18 21:38:00 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f MLME: MLME-AUTHENTICATE.indication(0f:0f:0f:0f:0f:0f, OPEN_SYSTEM)
Oct 18 21:38:00 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: event 0 notification
Oct 18 21:38:00 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: event 8 notification
Oct 18 21:38:00 ap08 hostapd: wlan5g: STA 0f:0f:0f:0f:0f:0f WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Oct 18 21:38:00 ap08 kernel: [181825.072316] Unable to handle kernel write to read-only memory at virtual address 0000000000000000
Oct 20 20:03:24 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:24 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:25 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:26 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:26 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:27 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:28 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:29 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:29 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:29 ap03 hostapd: wlan5g: STA 08:08:08:08:08:08 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Oct 20 20:03:29 ap03 hostapd: wlan5g: STA 08:08:08:08:08:08 MLME: MLME-DEAUTHENTICATE.indication(08:08:08:08:08:08, 2)
Oct 20 20:03:29 ap03 hostapd: wlan5g: STA 08:08:08:08:08:08 MLME: MLME-DELETEKEYS.request(08:08:08:08:08:08)
Oct 20 20:03:30 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:31 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
Oct 20 20:03:43 ap03 hostapd: Beacon request: 08:08:08:08:08:08 is not connected
[... cut ...]
Nov 30 11:56:11 ap03 hostapd: wlan5g: STA 0a:0a:0a:0a:0a:0a MLME: MLME-AUTHENTICATE.indication(0a:0a:0a:0a:0a:0a, OPEN_SYSTEM)
Nov 30 11:56:11 ap03 hostapd: wlan5g: STA 0a:0a:0a:0a:0a:0a WPA: event 0 notification
Nov 30 11:56:11 ap03 hostapd: wlan5g: STA 0a:0a:0a:0a:0a:0a WPA: event 8 notification
Nov 30 11:56:11 ap03 hostapd: wlan5g: STA 0a:0a:0a:0a:0a:0a WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Nov 30 11:56:11 ap03 kernel: [302134.549482] Unable to handle kernel write to read-only memory at virtual address 0000000000000000
```

</p>
</details>

cotequeiroz, 2023-12-20:
> The above would repeat for literally hundreds of lines. That STA would usually connect fine, but for some reason before the AP crashed it would just keep looping like this endlessly at a really fast rate.

Notice that none of my logs show repeating log entries like you described.

rany2, 2023-12-21:
> > The above would repeat for literally hundreds of lines. That STA would usually connect fine, but for some reason before the AP crashed it would just keep looping like this endlessly at a really fast rate.
> 
> Notice that none of my logs show repeating log entries like you described.

I guess that might be an unrelated issue with my ESP8266 not caused by the AP, I think it does that when the channel is set to 13 sometimes. I just removed it from the chan_list in hostapd to see.

Either way, it does appear that this crash occurs right as the device is connecting (either authenticated or associating); so hopefully that might be a clue of some sort.

rany2, 2023-12-24:
@cotequeiroz Do you think this patch https://git.openwrt.org/?p=openwrt/staging/blocktrron.git;a=blob_plain;f=package/kernel/mac80211/patches/subsys/780-avoid-crashing-missing-band.patch;hb=0ff90cbc06a50817872adfdceacc5b0d4ef07170 might be related to this issue? I'm currently testing to see if I get any crash but it takes time for this to trigger.

cotequeiroz, 2023-12-27:
Bisect update:
```
git bisect start
# status: waiting for both good and bad commits
# bad: [2afc7285f75dca5a0583fd917285bf33f1429cc6] wifi: mt76: mt7925: add Mediatek Wi-Fi7 driver for mt7925 chips
git bisect bad 2afc7285f75dca5a0583fd917285bf33f1429cc6
# status: waiting for good commit(s), bad commit known
# good: [969b7b5ebd129068ca56e4b0d831593a2f92382f] wifi: mt76: mt7915: add support for MT7981
git bisect good 969b7b5ebd129068ca56e4b0d831593a2f92382f
# good: [bdf7ca5db70779d5763145f2fefbfde58f094903] wifi: mt76: mt7921: move common register definition in mt792x_regs.h
git bisect good bdf7ca5db70779d5763145f2fefbfde58f094903
# bad: [b3f739a64e6bd5a10a3cfaacfbe25f7b1f6ee0b3] wifi: mt76: mt7996: add eht rx rate support
git bisect bad b3f739a64e6bd5a10a3cfaacfbe25f7b1f6ee0b3
# good: [3d65cd7a8dad351190d1bf6678819267a2313090] wifi: mt76: mt7921: move runtime-pm pci code in mt792x-lib
git bisect good 3d65cd7a8dad351190d1bf6678819267a2313090
```
```
Bisecting: 9 revisions left to test after this (roughly 3 steps)
[1f0fbbee359c1ec4d2ed0235225ffb676f4c90bf] wifi: mt76: mt7915: fix tlv length of mt7915_mcu_get_chan_mib_info
```

cotequeiroz, 2023-12-27:
Here are the commits left to test:
```
 $ git -P bisect view --oneline
b3f739a6 wifi: mt76: mt7996: add eht rx rate support
f1e1e67d (origin/openwrt-23.05) wifi: mt76: fix race condition related to checking tx queue fill status
bbbac7de wifi: mt76: rename mt76_packet_id_init/flush to mt76_wcid_init/cleanup
b5d13611 mt76: mt7915: fix monitor mode issues
b14c2351 wifi: mt76: mt7603: disable A-MSDU tx support on MT7628
85cc5837 wifi: mt76: mt7603: add missing register initialization for MT7628
c03d84c0 wifi: mt76: mt7603: improve stuck beacon handling
a8d9553d wifi: mt76: mt7603: improve watchdog reset reliablity
80b8bcf0 wifi: mt76: mt7603: rework/fix rx pse hang check
24a54ee3 wifi: mt76: mt7915: fix power-limits while chan_switch
1f0fbbee (HEAD) wifi: mt76: mt7915: fix tlv length of mt7915_mcu_get_chan_mib_info
4904ab2f wifi: mt76: fix return value check in mt76x02_mac_process_rx
c19b62fe wifi: mt76: testmode: add nla_policy for MT76_TM_ATTR_TX_LENGTH
9e4cb87a wifi: mt76: mt7921: move mt7921u_disconnect mt792x-lib
58b2b90a wifi: mt76: mt7921: move mt7921_dma_init in pci.c
700041dc wifi: mt76: mt792x: move MT7921_PM_TIMEOUT and MT7921_HW_SCAN_TIMEOUT in common code
cf752af8 wifi: mt76: mt76_connac3: move lmac queue enumeration in mt76_connac3_mac.h
7067ea0f wifi: mt76: mt792x: move mt7921_load_firmware in mt792x-lib module
bd84b8eb wifi: mt76: mt792x: introduce mt792x-usb module
4551f8c8 wifi: mt76: mt7921: move acpi_sar code in mt792x-lib module
```
Judging by the commit titles only*, I bet f1e1e67d (wifi: mt76: fix race condition related to checking tx queue fill status) is the culprit.
* I did look at its code after looking at the titles.

rany2, 2023-12-27:
> Judging by the commit titles only*, I bet [f1e1e67](https://github.com/openwrt/mt76/commit/f1e1e67d97d1e9a8bb01b59ab20c45ebc985a958) (wifi: mt76: fix race condition related to checking tx queue fill status) is the culprit.
> * I did look at its code after looking at the titles.

FYI MTK does revert that commit, so I presume that they did indeed have an issue with it.

nbd168, 2024-01-03:
I've pushed a mac80211 fix to my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary
Please test if it resolves this crash issue.

rany2, 2024-01-03:
For the record, I'm testing it and will let you know if it crashes with your patch.

cotequeiroz, 2024-01-03:
I am away now, and will test it next week.  Thanks for looking into this.

nbd168, 2024-01-03:
FYI, I believe this issue is triggered by roaming to the AP (via FT)

rany2, 2024-01-04:
@nbd168 So far no crashes on any of my nodes, seems promising.

nbd168, 2024-01-04:
Pushed the fix to master and 23.05

cotequeiroz, 2024-01-16:
My routers have been running the same firmware that started this (r24315-ae500e62e2)--updating only `mac80211.ko`--for over a week without crashes.  I'm closing this as fixed.

closed 2024-01-16 by cotequeiroz