openwrt/mt76 #844: mt7622 unstable on 23.05, especially when using 802.1x dynamic VLANs

albertofustinoni, 2023-11-27:
I have a couple of Linksys E8450s that I use as L2 bridged access points with dynamic VLAN assignment based on Radius attributes: I have configured a single WPA2-Enterprise SSID and users get assigned to a VLAN based on their attributes in the RADIUS server. I also configured roaming with 802.11r and the whole R0KH/R1HK keys.

This worked well enough until I upgraded to 22.03.5, after which I started getting really weird behaviour: some devices would connect to the network but not get assigned an IP, or have working IPv4 but not IPv6, or have both and being able to ping destinations but not get TCP traffic working.

I spent the last few days pulling hair and trying a bunch of things, including one SSIDs with WPA2-PSK per VLAN, to varying degrees of success, but things were never 100% working as intended.

As a last ditch attempt, I reverted to 22.03 and the dynamic VLAN config (restored from backup, so it's the exact same) and things seem to be back to how they used to be.

major-mayer, 2024-02-28:
I have similar instabilities with MT7610E using the 5Ghz radio/ AC mode.
I tried regular WPA2 and WPA3 and can confirm that clients get assigned correct IPv4/6 addresses and can ping internal IPs but not external ones.
Regular TCP traffic is also not working for me.

I have no problem using my normal 2.4 Ghz Wifi (MT7620), except that it's very slow (around 16mbit/s), but I think that's a different issue.
My router is a TP-Link Archer C2 v1 and i am on OpenWRT 23.05.2.

I am also thinking about a downgrade back to OpenWRT 22.03...

rany2, 2024-02-28:
Perhaps you're facing https://github.com/openwrt/openwrt/issues/11650?

corvin84, 2024-03-11:
> I have similar instabilities with MT7610E using the 5Ghz radio/ AC mode. I tried regular WPA2 and WPA3 and can confirm that clients get assigned correct IPv4/6 addresses and can ping internal IPs [...]

Did you find any fix?

major-mayer, 2024-03-11:
> Did you find any fix?

Unfortunately not yet, but I posted my problems in the forums: https://forum.openwrt.org/t/tp-link-archer-c2-v1-on-23-05-no-internet-coming-through-5ghz-wifi-slow-2-4-ghz

enmaskarado, 2024-03-16:
> I have no problem using my normal 2.4 Ghz Wifi (MT7620), except that it's very slow (around 16mbit/s), but I think that's a different issue

Unrelated, but try downgrading WPA3 to pure WPA2, or disabling MFP on WPA2. MT7620 doesn't have HW acceleration for MFP and this caps the max speed to ~18Mbps emulating it on software.

major-mayer, 2024-03-16:
> > I have no problem using my normal 2.4 Ghz Wifi (MT7620), except that it's very slow (around 16mbit/s), but I think that's a different issue
> 
> Unrelated, but try downgrading WPA3 to pure WPA2, or disabling MFP on WPA2. MT7620 doesn't have HW acceleration for MFP and software emulation caps the max speed to ~18Mbps.

Ahh thanks for the explanation. I will try it when I'm back home and see if it makes a difference.