openwrt/mt76 #845: Load correct EEPROM based on PA LNA Type

Rondom, 2023-11-27:
Different EEPROM files are required to achieve best performance depending on the PA LNA type which, as far as I understand, can be ePAeLNA, iPAiLNA, iPAeLNA or ePAiLNA.

In case no EEPROM is available via NVMEM, we need to load the correct EEPROM file from the file system. So, provided, that there is no better way, we would probably need a property in device-tree to specify the PA LNA type so that the fitting EEPROM file can be loaded.

DragonBluep, 2024-04-27:
You can use "mediatek,eeprom-data" property.

Rondom, 2024-04-28:
I had considered that but it will take low precedence over NVMEM if I remember it correctly. For the use case where it is meant as a fallback in case NVMEM is damaged or missing a behaviour that is the other way around is desirable.