openwrt/mt76 #855: catch crash log

ptpt52, 2024-01-09:
since commit: f1e1e67d97d1e9a8bb01b59ab20c45ebc985a958

crash on mt7621 + mt7612e + mt7603
log:
```
<7>[   43.726153] phy1-sta0: Limiting TX power to 30 (30 - 0) dBm as advertised by e2:ef:02:96:cb:0e
<6>[   44.958361] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-sta0: link becomes ready
<6>[ 6528.721727] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[ 6528.729742] br-lan: port 1(lan1) entered blocking state
<6>[ 6528.734954] br-lan: port 1(lan1) entered forwarding state
<6>[ 6531.027140] mtk_soc_eth 1e100000.ethernet wan: Link is Down
<6>[ 6531.037118] br-lan: port 5(wan) entered disabled state
<6>[ 6590.900182] mt7530-mdio mdio-bus:1f lan1: Link is Down
<6>[ 6590.905725] br-lan: port 1(lan1) entered disabled state
<6>[ 6593.192199] mt7530-mdio mdio-bus:1f lan1: Link is Up - 10Mbps/Full - flow control off
<6>[ 6593.200148] br-lan: port 1(lan1) entered blocking state
<6>[ 6593.205363] br-lan: port 1(lan1) entered forwarding state
<6>[14301.372598] mt7530-mdio mdio-bus:1f lan1: Link is Down
<6>[14301.378145] br-lan: port 1(lan1) entered disabled state
<6>[14304.040509] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[14304.048529] br-lan: port 1(lan1) entered blocking state
<6>[14304.053782] br-lan: port 1(lan1) entered forwarding state
<6>[14344.604747] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
<6>[14344.610287] mt76x2e 0000:02:00.0: Build: 1
<6>[14344.614475] mt76x2e 0000:02:00.0: Build Time: 201607111443____
<6>[14344.642679] mt76x2e 0000:02:00.0: Firmware running!
<6>[14344.648879] ieee80211 phy1: Hardware restart was requested
<1>[14344.655921] CPU 2 Unable to handle kernel paging request at virtual address 00000004, epc == 843b4ef0, ra == 843b4ed8
<4>[14344.666617] Oops[#1]:
<4>[14344.668909] CPU: 2 PID: 111 Comm: kworker/2:2 Not tainted 5.15.145 #0
<4>[14344.675350] Workqueue: events cfg80211_stop_iface [cfg80211]
<4>[14344.681157] $ 0   : 00000000 00000001 82fd5a40 00000000
<4>[14344.686409] $ 4   : 00000000 00000000 00000000 00000000
<4>[14344.691649] $ 8   : 00000000 8036e6fc 00000000 1ee1d200
<4>[14344.696887] $12   : 00000002 00000000 00000402 82fd5a68
<4>[14344.702137] $16   : 842b97a0 80e99c10 82fd5964 842b97b4
<4>[14344.707382] $20   : 82fd57cc 00000000 80e8d070 00000000
<4>[14344.712650] $24   : 00000040 00000000                  
<4>[14344.717894] $28   : 80e98000 80e99c00 80860000 843b4ed8
<4>[14344.723142] Hi    : 0004cc86
<4>[14344.726025] Lo    : 8a15a40b
<4>[14344.728892] epc   : 843b4ef0 mt76_wcid_cleanup+0x88/0x1a4 [mt76]
<4>[14344.734953] ra    : 843b4ed8 mt76_wcid_cleanup+0x70/0x1a4 [mt76]
<4>[14344.741027] Status: 1100fc03	KERNEL EXL IE 
<4>[14344.745222] Cause : 4080000c (ExcCode 03)
<4>[14344.749218] BadVA : 00000004
<4>[14344.752084] PrId  : 0001992f (MIPS 1004Kc)
<4>[14344.756164] Modules linked in: ksmbd qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash
<4>[14344.756893]  nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat chacha_mips cdc_wdm br_netfilter natflow natcap nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet
<4>[14344.843782]  ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 cifs_arc4 asn1_decoder nfs msdos ip6_gre ip_gre gre sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 crypto_hw_eip93 hmac ecb des_generic libdes cmac authencesn authenc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd
<4>[14344.931418]  gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[14345.030335] Process kworker/2:2 (pid: 111, threadinfo=091994b0, task=de6f8c23, tls=00000000)
<4>[14345.038761] Stack : 80b04ba0 842b97a0 00000000 82fd5964 80e99c10 80e99c10 00000000 842b9454
<4>[14345.047149]         00000000 00000010 842b97a0 82fd5964 00000010 843b5114 836be800 82fd5034
<4>[14345.055531]         836be800 00000001 82fd57cc 80e8d070 842b9c1c 842b97a0 82fd57cc 80e8c4c0
<4>[14345.063913]         842b89e0 843b5240 00000048 00000001 00000001 82fd5000 82fd5000 00000000
<4>[14345.072296]         842b84a0 80e8c4c0 82fd57cc 8430bfd4 80e99c8c 0000000f 00000000 80fb7580
<4>[14345.080674]         ...
<4>[14345.083128] Call Trace:
<4>[14345.085564] [<843b4ef0>] mt76_wcid_cleanup+0x88/0x1a4 [mt76]
<4>[14345.091248] [<843b5114>] __mt76_sta_remove+0x84/0xe8 [mt76]
<4>[14345.096835] [<843b5240>] mt76_sta_state+0xc8/0x1dc [mt76]
<4>[14345.102252] [<8430bfd4>] sta_set_sinfo+0x1088/0x1190 [mac80211]
<4>[14345.108523] 
<4>[14345.110010] Code: 00000000  8e4300e0  8e4400dc <ac830004> ac640000  ae4200dc  ae4200e0  0c1b7349  264400f0 
<4>[14345.119790] 
<4>[14345.121374] ---[ end trace 84e9f81e2e404c4d ]---

```

ptpt52, 2024-01-09:
may be a fix:
```diff
diff --git a/mac80211.c b/mac80211.c
index 3df8cfc4..baa1e31a 100644
--- a/mac80211.c
+++ b/mac80211.c
@@ -1443,6 +1443,9 @@ void __mt76_sta_remove(struct mt76_dev *dev, struct ieee80211_vif *vif,
        struct mt76_wcid *wcid = (struct mt76_wcid *)sta->drv_priv;
        int i, idx = wcid->idx;
 
+       if (!dev->wcid[wcid->idx])
+               return;
+
        for (i = 0; i < ARRAY_SIZE(wcid->aggr); i++)
                mt76_rx_aggr_stop(dev, wcid, i);
 

```

ptpt52, 2024-01-12:
```
<6>[98328.601115] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
<6>[98328.608086] br-lan: port 2(phy0-ap0) entered blocking state
<6>[98328.613721] br-lan: port 2(phy0-ap0) entered forwarding state
<1>[98331.388597] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 830e8560, ra == 830e855c
<4>[98331.399274] Oops[#1]:
<4>[98331.401558] CPU: 0 PID: 9402 Comm: kworker/u8:0 Not tainted 5.15.146 #0
<4>[98331.408162] Workqueue: phy0 mt7603_mac_work [mt7603e]
<4>[98331.413243] $ 0   : 00000000 00000001 00000000 00000000
<4>[98331.418491] $ 4   : 85aa4a60 ffffff0c 00000000 00000000
<4>[98331.423736] $ 8   : 00000000 00000000 00000000 80808080
<4>[98331.428988] $12   : 00000002 00000000 00000402 85aa4a68
<4>[98331.434312] $16   : ffffff0c 831297a0 00000000 8312d190
<4>[98331.439567] $20   : 00000000 83129c1c 80dcbb35 839f3140
<4>[98331.444835] $24   : 00000040 00000000                  
<4>[98331.450078] $28   : 83bc8000 83bc9df0 80860000 830e855c
<4>[98331.455327] Hi    : 0084f9ef
<4>[98331.458198] Lo    : eb3c9664
<4>[98331.461070] epc   : 830e8560 mt76_tx_status_check+0x68/0xa0 [mt76]
<4>[98331.467290] ra    : 830e855c mt76_tx_status_check+0x64/0xa0 [mt76]
<4>[98331.473484] Status: 11000403	KERNEL EXL IE 
<4>[98331.477696] Cause : 40800008 (ExcCode 02)
<4>[98331.481684] BadVA : 00000000
<4>[98331.484561] PrId  : 0001992f (MIPS 1004Kc)
<4>[98331.488665] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic nft_fib_inet mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_DNETMAP xt_DHCPMAC xt_CLASSIFY ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp
<4>[98331.489132]  nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hwmon crc_ccitt compat_xtables compat chacha_mips br_netfilter asn1_decoder natflow natcap nf_nat nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port
<4>[98331.575791]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool nf_defrag_ipv4 jool_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding tls ip6_gre ip_gre gre nat46 nf_defrag_ipv6 sit ip6_tunnel tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel nls_utf8 nls_iso8859_1 nls_base crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 crypto_hw_eip93 hmac ecb des_generic libdes cmac authencesn authenc arc4 leds_gpio softdog gpio_button_hotplug crc32c_generic
<4>[98331.720469] Process kworker/u8:0 (pid: 9402, threadinfo=63c102db, task=566da9c2, tls=00000000)
<4>[98331.729089] Stack : 8100b460 8100dd80 84eb69c0 8db2c165 83bc9e00 83bc9e00 00000000 8100dd80
<4>[98331.737496]         8100dd80 831299c4 839f3100 831297a0 80dcbb00 830f6ce4 8100dd80 84eb69c0
<4>[98331.745884]         84eb6d30 80c06600 80c06600 8005e79c 80c40bc0 00000000 831299c4 839f3100
<4>[98331.754274]         80c06600 80dcbb00 00000000 00000100 80dcbb35 839f3140 80860000 8004801c
<4>[98331.762666]         830300d8 84eb69c0 80c06600 839f3118 80c06618 80860000 80c06600 839f3100
<4>[98331.771063]         ...
<4>[98331.773522] Call Trace:
<4>[98331.775956] [<830e8560>] mt76_tx_status_check+0x68/0xa0 [mt76]
<4>[98331.781838] [<830f6ce4>] mt7603_mac_work+0x44/0xd70 [mt7603e]
<4>[98331.787602] [<8004801c>] process_one_work+0x278/0x4b8
<4>[98331.792668] [<800483b8>] worker_thread+0x15c/0x5cc
<4>[98331.797460] [<80050c20>] kthread+0x130/0x154
<4>[98331.801735] [<80002eb8>] ret_from_kernel_thread+0x14/0x1c
<4>[98331.807136] 
<4>[98331.808621] Code: 0cc3a0d4  02202025  02002825 <8e1000f4> 24a200f4  1453fff8  2610ff0c  27a50010  0cc39f23 
<4>[98331.818416] 
<4>[98331.820010] ---[ end trace 586baa5a30ae09b9 ]---

```

ptpt52, 2024-01-12:
@nbd168 ping
any idea on these crash log?

ptpt52, 2024-03-20:
@rany2  but still issue for me with lastest code.

ptpt52, 2024-05-08:
crashed on latest code, mt7620
```
<1>[34715.686099] CPU 0 Unable to handle kernel paging request at virtual address 00000004, epc == 8313479c, ra == 83134778
<4>[34715.697024] Oops[#1]:
<4>[34715.699345] CPU: 0 PID: 2191 Comm: hostapd Not tainted 6.1.86 #0
<4>[34715.705482] $ 0   : 00000000 00000001 81b3f9f8 00000000
<4>[34715.710846] $ 4   : 00000000 00000200 00000000 00000000
<4>[34715.716198] $ 8   : 8549aa70 0000000f 00000000 809e1700
<4>[34715.721550] $12   : 00000002 00000000 00000402 81b3fa18
<4>[34715.726903] $16   : 831296c0 8314f9ec 81b3f91c 00000010
<4>[34715.732256] $20   : 81b3f784 00000000 8549b020 00000001
<4>[34715.737608] $24   : 00000040 00000000                  
<4>[34715.742961] $28   : 8314e000 8314f9d8 00000000 83134778
<4>[34715.748315] Hi    : 00000013
<4>[34715.751251] Lo    : 33333338
<4>[34715.754187] epc   : 8313479c mt76_wcid_cleanup+0x74/0x180 [mt76]
<4>[34715.760381] ra    : 83134778 mt76_wcid_cleanup+0x50/0x180 [mt76]
<4>[34715.766547] Status: 1100f403	KERNEL EXL IE 
<4>[34715.770830] Cause : 0080000c (ExcCode 03)
<4>[34715.774921] BadVA : 00000004
<4>[34715.777859] PrId  : 00019650 (MIPS 24KEc)
<4>[34715.781948] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat
<4>[34715.782605]  nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat chacha_mips cdc_wdm br_netfilter natflow natcap nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set
<4>[34715.872283]  ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 cifs_arc4 asn1_decoder nfs msdos ip6_gre ip_gre gre sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ecb des_generic libdes cmac arc4 uas usb_storage leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore
<4>[34715.961214]  nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[34716.056755] Process hostapd (pid: 2191, threadinfo=05d7ee1d, task=6c0f3f23, tls=77ee0df4)
<4>[34716.065114] Stack : 00000000 81b3f91c 00000010 8317038c 83079400 8314f9ec 8314f9ec 00000000
<4>[34716.073686]         00000010 831296c0 81b3f91c 831349b0 855f4000 81b3f030 81b3f040 8324445c
<4>[34716.082254]         81b3f784 8549b020 83129b20 831296c0 81b3f784 8549a4c0 831289b0 83134adc
<4>[34716.090822]         8314fa44 83133d74 00000001 00000000 81b3f000 00000000 831284a0 8549a4c0
<4>[34716.099391]         81b3f784 8320b3b8 83126000 0000000f 80740000 00000028 00000000 830b1940
<4>[34716.107959]         ...
<4>[34716.110468] Call Trace:
<4>[34716.112959] [<8313479c>] mt76_wcid_cleanup+0x74/0x180 [mt76]
<4>[34716.118810] [<831349b0>] __mt76_sta_remove+0x84/0xe8 [mt76]
<4>[34716.124551] [<83134adc>] mt76_sta_state+0xc8/0x350 [mt76]
<4>[34716.130115] [<8320b3b8>] sta_set_sinfo+0xfd4/0x1098 [mac80211]
<4>[34716.136519] 
<4>[34716.138039] Code: 00000000  8e4300e0  8e4400dc <ac830004> ac640000  ae4200dc  ae4200e0  8e4300e4  264200e4 
<4>[34716.148044] 
<4>[34716.149624] ---[ end trace 0000000000000000 ]---
```

dobo90, 2024-08-28:
I've got an old laptop with MT7612E used as a station. I've reproduced the same issue on stock kernel (6.10.6-arch1-1). Wireless link is unstable because of lots of hardware restarts (but that's another issue that I can try to bisect later - the card is working fine on 6.6.47 kernel):
```
[  223.674410] ieee80211 phy0: Hardware restart was requested
[  225.090916] mt76x2e 0000:01:00.0: MCU message 1e (seq 11) timed out
[  225.268710] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  225.273176] mt76x2e 0000:01:00.0: Build: 1
[  225.275989] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[  225.302920] mt76x2e 0000:01:00.0: Firmware running!
[  225.307928] ieee80211 phy0: Hardware restart was requested
...
[  238.850933] mt76x2e 0000:01:00.0: MCU message 1f (seq 2) timed out
...
[  238.959866] ieee80211 phy0: Hardware restart was requested
```
Kernel log with KASAN enabled:
```
[  238.850933] mt76x2e 0000:01:00.0: MCU message 1f (seq 2) timed out
[  238.915460] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  238.920230] mt76x2e 0000:01:00.0: Build: 1
[  238.925074] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[  238.954793] mt76x2e 0000:01:00.0: Firmware running!
[  238.959866] ieee80211 phy0: Hardware restart was requested
[  238.964704] Oops: general protection fault, probably for non-canonical address 0xdffffc0000000000: 0000 [#1] PREEMPT SMP KASAN NOPTI
[  238.968000] KASAN: null-ptr-deref in range [0x0000000000000000-0x0000000000000007]
[  238.971188] CPU: 0 PID: 12 Comm: kworker/u8:1 Tainted: G           OE      6.10.6-arch1-1 #1 da7240322b20ecc66a35d39c108844050587fec6
[  238.974524] Hardware name: LENOVO 062223U/062223U, BIOS 8RET54WW (1.17 ) 10/24/2012
[  238.977883] Workqueue: events_unbound cfg80211_wiphy_work [cfg80211]
[  238.981806] RIP: 0010:mt76_wcid_cleanup+0x1af/0x840 [mt76]
[  238.985291] Code: 03 80 3c 01 00 0f 85 dd 04 00 00 48 8b 83 38 01 00 00 48 8b 8b 30 01 00 00 48 bf 00 00 00 00 00 fc ff df 49 89 c0 49 c1 e8 03 <41> 80 3c 38 00 0f 85 9b 05 00 00 48 3b 10 0f 85 13 f2 00 00 49 b8
[  238.992428] RSP: 0018:ffff8880013cf610 EFLAGS: 00010256
[  238.996146] RAX: 0000000000000000 RBX: ffff8880068cecc8 RCX: 0000000000000000
[  238.999829] RDX: ffff8880068cedf8 RSI: ffff8880068cee00 RDI: dffffc0000000000
[  239.003469] RBP: ffff8880013cf6f0 R08: 0000000000000000 R09: ffffed1000279eb7
[  239.007105] R10: 0000000000000003 R11: ffff8880068cee38 R12: ffff8880070c2000
[  239.010725] R13: ffff8880013cf660 R14: 1ffff11000279ec8 R15: ffff8880070c2024
[  239.014309] FS:  0000000000000000(0000) GS:ffff8880c5200000(0000) knlGS:0000000000000000
[  239.017914] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  239.021479] CR2: 000074bc7d6f4200 CR3: 0000000008198000 CR4: 00000000000006f0
[  239.025083] Call Trace:
[  239.028644]  <TASK>
[  239.032171]  ? __die_body.cold+0x19/0x27
[  239.035658]  ? die_addr+0x46/0x70
[  239.039046]  ? exc_general_protection+0x150/0x240
[  239.042497]  ? asm_exc_general_protection+0x26/0x30
[  239.045929]  ? mt76_wcid_cleanup+0x1af/0x840 [mt76 b982b9e5eb1ae439787ef336d828a09860d79e4e]
[  239.049482]  ? __pfx_mt76_wcid_cleanup+0x10/0x10 [mt76 b982b9e5eb1ae439787ef336d828a09860d79e4e]
[  239.053043]  ? __pfx_mt76x02_mac_wcid_setup+0x10/0x10 [mt76x02_lib 3df29cfcd98578e5beba8f133fc84dd9cff2d3c3]
[  239.056650]  ? mt76x02_mac_wcid_set_drop+0x90/0x160 [mt76x02_lib 3df29cfcd98578e5beba8f133fc84dd9cff2d3c3]
[  239.060348]  __mt76_sta_remove+0xe5/0x1b0 [mt76 b982b9e5eb1ae439787ef336d828a09860d79e4e]
[  239.064116]  ? ieee80211_free_sta_keys+0x1ff/0x2e0 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.068471]  mt76_sta_state+0x5da/0x950 [mt76 b982b9e5eb1ae439787ef336d828a09860d79e4e]
[  239.072353]  ? drv_sta_state+0x22f/0x12f0 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.076741]  drv_sta_state+0x22f/0x12f0 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.081161]  ? ieee80211_sta_tear_down_BA_sessions+0x116/0x160 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.085706]  __sta_info_destroy_part2+0x309/0x3d0 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.090280]  __sta_info_flush+0x410/0x680 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.094921]  ? __pfx___sta_info_flush+0x10/0x10 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.099614]  ? ieee80211_send_deauth_disassoc+0x11c/0x3c0 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.104407]  ieee80211_set_disassoc+0x422/0x1690 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.109263]  ? __pfx_ieee80211_set_disassoc+0x10/0x10 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.114135]  ? prb_read_valid+0x6a/0x90
[  239.118404]  ? __pfx_prb_read_valid+0x10/0x10
[  239.122658]  ieee80211_sta_connection_lost+0xc1/0x1c0 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.127581]  ? __pfx_ieee80211_sta_connection_lost+0x10/0x10 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.132579]  ? desc_read_finalized_seq+0x7b/0x130
[  239.137043]  ? desc_read+0x229/0x330
[  239.141468]  ieee80211_sta_work+0x6f0/0x14d0 [mac80211 992140fdb58da16dea4c462fede699b883be8ea9]
[  239.146581]  ? sched_clock+0x10/0x30
[  239.151107]  ? sched_clock_cpu+0x6d/0x510
[  239.155514]  ? psi_group_change+0x4d5/0xb70
[... 16 lines cut ...]
[  239.202395]  ret_from_fork+0x34/0x70
[  239.202411]  ? __pfx_kthread+0x10/0x10
[  239.202429]  ret_from_fork_asm+0x1a/0x30
[  239.202452]  </TASK>
[... cut ...]
[  239.242324] Kernel Offset: 0x31600000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[  239.326744] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

I've also compiled latest openwrt/mt76 (with some slight changes to make it work on 6.10 kernels). It came out that
```mt76_wcid_cleanup+0x1af``` points to [include/linux/list.h:119](https://elixir.bootlin.com/linux/v6.10.6/source/include/linux/list.h#L119) which is an inlined code from [mac80211.c:1575](https://github.com/openwrt/mt76/blob/a897606f7c26f380c26fdfecf855a96b14c601fb/mac80211.c#L1575).

dobo90, 2024-08-29:
@nbd168 I think I know what's the cause of the oops. I've applied following patch:
```
diff --git a/mac80211.c b/mac80211.c
index f405d540..db1dc88a 100644
--- a/mac80211.c
+++ b/mac80211.c
@@ -1489,6 +1489,7 @@ static void
 mt76_sta_remove(struct mt76_dev *dev, struct ieee80211_vif *vif,
                struct ieee80211_sta *sta)
 {
+       pr_err("DBG: %px accessed\n", sta->drv_priv);
        mutex_lock(&dev->mutex);
        __mt76_sta_remove(dev, vif, sta);
        mutex_unlock(&dev->mutex);
diff --git a/mt76x02_mmio.c b/mt76x02_mmio.c
index 960f6dc0..94b1a6f0 100644
--- a/mt76x02_mmio.c
+++ b/mt76x02_mmio.c
@@ -425,6 +425,7 @@ static void mt76x02_reset_state(struct mt76x02_dev *dev)
 
                __mt76_sta_remove(&dev->mt76, vif, sta);
                memset(msta, 0, sizeof(*msta));
+               pr_err("DBG: %px zeroed out\n", msta);
        }
 
        dev->mt76.vif_mask = 0;
```
And it crashes with a following dmesg:
```
[   37.211716] wlan0: associated
[   47.264465] mt76x2e 0000:01:00.0: MCU message 1f (seq 4) timed out
[   47.267746] DBG: ffff8fd1c8834cc8 zeroed out
[   47.323726] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[   47.326881] mt76x2e 0000:01:00.0: Build: 1
[   47.329858] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[   47.357134] mt76x2e 0000:01:00.0: Firmware running!
[   47.362095] ieee80211 phy0: Hardware restart was requested
[   47.364551] DBG: ffff8fd1c8834cc8 accessed
[   47.366827] BUG: kernel NULL pointer dereference, address: 0000000000000000
[   47.368900] #PF: supervisor read access in kernel mode
[   47.370977] #PF: error_code(0x0000) - not-present page
[   47.373067] PGD 0 P4D 0 
[   47.375154] Oops: Oops: 0000 [#1] PREEMPT SMP NOPTI
[   47.377266] CPU: 0 PID: 11 Comm: kworker/u8:0 Tainted: G           OE      6.10.6-arch1-1 #1 703d152c24f1971e36f16e505405e456fc9e23f8
[   47.379452] Hardware name: LENOVO 062223U/062223U, BIOS 8RET54WW (1.17 ) 10/24/2012
[   47.381644] Workqueue: events_unbound cfg80211_wiphy_work [cfg80211]
[   47.384127] RIP: 0010:mt76_wcid_cleanup+0xcd/0x230 [mt76]
[   47.386378] Code: 01 00 00 e8 a5 9b e8 e9 4c 89 f7 e8 1d 18 f3 e9 49 8b 85 30 01 00 00 49 39 c7 74 36 49 8b 85 38 01 00 00 49 8b 95 30 01 00 00 <4c> 3b 38 0f 85 f2 4c 00 00 4c 3b 7a 08 0f 85 e8 4c 00 00 48 89 42
[   47.391014] RSP: 0018:ffffb8270008bbe0 EFLAGS: 00010282
[   47.393355] RAX: 0000000000000000 RBX: ffff8fd1c96c2000 RCX: 0000000000000019
[   47.395699] RDX: 0000000000000000 RSI: 00000000fffffe00 RDI: ffff8fd1c96c2024
[   47.398007] RBP: ffffb8270008bc30 R08: 0000000000000000 R09: ffff8fd1c8834e48
[   47.400279] R10: 0000000000000000 R11: 0000000000000000 R12: ffffb8270008bbe0
[   47.402539] R13: ffff8fd1c8834cc8 R14: ffff8fd1c96c2024 R15: ffff8fd1c8834df8
[   47.404787] FS:  0000000000000000(0000) GS:ffff8fd1db200000(0000) knlGS:0000000000000000
[   47.407042] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   47.409252] CR2: 0000000000000000 CR3: 0000000109aee000 CR4: 00000000000006f0
[   47.411460] Call Trace:
[   47.413636]  <TASK>
[   47.415798]  ? __die_body.cold+0x19/0x27
[   47.417971]  ? page_fault_oops+0x15a/0x2d0
[   47.420124]  ? search_bpf_extables+0x5f/0x80
[   47.422266]  ? exc_page_fault+0x81/0x190
[   47.424407]  ? asm_exc_page_fault+0x26/0x30
[   47.426486]  ? mt76_wcid_cleanup+0xcd/0x230 [mt76 ebea36c80a0918b2c651d881c07777ba850f959d]
[   47.428575]  ? mt76x02_mac_wcid_setup+0xa4/0xd0 [mt76x02_lib 3fb11ba16d800091fb3ced8134b41f80d36d7ec7]
[   47.430652]  __mt76_sta_remove+0x7c/0xc0 [mt76 ebea36c80a0918b2c651d881c07777ba850f959d]
[   47.432700]  mt76_sta_state.cold+0x38/0x5e [mt76 ebea36c80a0918b2c651d881c07777ba850f959d]
[   47.434711]  drv_sta_state+0xf4/0x5f0 [mac80211 4cfae0e05c4fa4076530a00b4b4d7a48743a315c]
[   47.436904]  __sta_info_destroy_part2+0x14b/0x180 [mac80211 4cfae0e05c4fa4076530a00b4b4d7a48743a315c]
[... 25 lines cut ...]
[   47.509354] R13: ffff8fd1c8834cc8 R14: ffff8fd1c96c2024 R15: ffff8fd1c8834df8
[   47.511627] FS:  0000000000000000(0000) GS:ffff8fd1db200000(0000) knlGS:0000000000000000
[   47.513931] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   47.516241] CR2: 0000000000000000 CR3: 0000000109aee000 CR4: 00000000000006f0
[   47.518577] Kernel panic - not syncing: Fatal exception in interrupt
[   47.520994] Kernel Offset: 0x29e00000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[   47.523398] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```
`ffff8fd1c8834cc8` get's zeroed out in `mt76x02_reset_state` and then it's accessed in `mt76_wcid_cleanup.` `wcid->tx_list` is a chunk of zero bytes and it causes a null pointer dereference.

I've removed the memset and for now I couldn't reproduce the crash:
```
diff --git a/mt76x02_mmio.c b/mt76x02_mmio.c
index 94b1a6f0..e0892106 100644
--- a/mt76x02_mmio.c
+++ b/mt76x02_mmio.c
@@ -424,8 +424,7 @@ static void mt76x02_reset_state(struct mt76x02_dev *dev)
                vif = container_of(priv, struct ieee80211_vif, drv_priv);
 
                __mt76_sta_remove(&dev->mt76, vif, sta);
-               memset(msta, 0, sizeof(*msta));
        }
 
        dev->mt76.vif_mask = 0;
```