openwrt/mt76 #856: MT7921e - can't set  DFS channel

danpawlik, 2024-01-13:
Hi,
When I try to set on MT7921e a channel `> 100`, the interface initialization always fail.
For example:

```
# Channel 104, 80MHZ width:
Sat Jan 13 12:43:27 2024 daemon.notice wpa_supplicant[2007]: Set new config for phy phy0
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: Restart interface for phy phy0
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: Remove interface 'phy0'
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Sat Jan 13 12:43:27 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Sat Jan 13 12:43:27 2024 kern.info kernel: [68779.534670] device phy0-ap0 left promiscuous mode
Sat Jan 13 12:43:27 2024 kern.info kernel: [68779.539454] br-lan: port 5(phy0-ap0) entered disabled state
Sat Jan 13 12:43:27 2024 daemon.notice netifd: Network device 'phy0-ap0' link is down
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=PL ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=104 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40-][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=106 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=1 he_oper_centr_freq_seg0_idx=106 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request=0 he_mu_edca_ac_be_aifsn=8 he_mu_edca_ac_be_aci=0 he_mu_edca_ac_be_ecwmin=9 he_mu_edca_ac_be_ecwmax=10 he_mu_edca_ac
Sat Jan 13 12:43:27 2024 kern.info kernel: [68779.881908] br-lan: port 5(phy0-ap0) entered blocking state
Sat Jan 13 12:43:27 2024 kern.info kernel: [68779.887504] br-lan: port 5(phy0-ap0) entered disabled state
Sat Jan 13 12:43:27 2024 kern.info kernel: [68779.893185] device phy0-ap0 entered promiscuous mode
Sat Jan 13 12:43:27 2024 kern.info kernel: [68779.898217] br-lan: port 5(phy0-ap0) entered blocking state
Sat Jan 13 12:43:27 2024 kern.info kernel: [68779.903781] br-lan: port 5(phy0-ap0) entered listening state
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Sat Jan 13 12:43:27 2024 daemon.notice netifd: Wireless device 'radio0' is now up
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: interface state HT_SCAN->DFS
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: DFS-CAC-START freq=5520 chan=104 sec_chan=-1, width=1, seg0=106, seg1=0, cac_time=60s
Sat Jan 13 12:43:27 2024 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Sat Jan 13 12:43:27 2024 daemon.err hostapd: Interface initialization failed
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: interface state DFS->DISABLED
Sat Jan 13 12:43:27 2024 daemon.notice hostapd: phy0-ap0: AP-DISABLED
```

Similar error I got on different channel and different width.

Djfe, 2024-01-27:
the first thing was just fixed, I don't know about the second one
https://github.com/openwrt/openwrt/commit/c909fdad1870361d5a6f0c0a2437d062e988f559

This is not part of any release, yet.
But if you build the 23.05 release branch and use that or use a snaphot then you will get the fix. Else you'll have to wait

danpawlik, 2024-01-29:
@Djfe I'm using snapshot version, but created by my own (the BPI-R4 device branch is still in progress) and the issue is still there. I will try to dig more in few days (if I will have time).

Djfe, 2024-01-30:
which commit?

danpawlik, 2024-01-30:
I checkout to Daniels fork branch in PR: https://github.com/openwrt/openwrt/pull/14140  . Even rebase to HEAD does not change anything.

lukasz1992, 2024-02-05:
I do not think mt7921 supports DFS right now, I do not see necessary code.

especially NL80211_EXT_FEATURE_RADAR_BACKGROUND is not being set.

morrownr, 2024-03-01:
> I do not think mt7921 supports DFS right now, I do not see necessary code.

The mt7612u and mt7610u drivers do not support DFS channels either. I think AP mode DFS support requires additional certification and the Mediatek management decision has been made not to support it. I think this is a BIG mistake. I have adapters that use various Realtek chips and they do support AP mode DFS. Example: rtl8812au chipset with out-of-kernel driver. In fact, this message is coming to you through an AP that is using an adapter with a rtl8812au chipset and it is running on channel 116.

neheb, 2025-08-14:
@morrownr that is highly unfortunate. At least the AWUS036ACM is better than QCA9880.

morrownr, 2025-08-14:
@neheb 

Since AP mode is very popular with Linux users, it would be nice if AP mode DFS operation was supported in our adapters and cards but it is not. This is not limited to Realtek and Mediatek but rather all wireless chip makers when it comes to chips for USB adapters and cards. The hardware is actually capable and it is possible to hack most drivers to make it so but that would be illegal and I do not support illegal activities here.

neheb, 2025-08-14:
hmm? if DFS works, what's illegal about it? I assume lack of DFS is more of a legal problem.

morrownr, 2025-08-14:
> if DFS works, what's illegal about it? I assume lack of DFS is more of a legal problem.

It is a legal problem. Operating AP  mode on DFS channels requires the capability to search for signals that should not be interfered with and if found, there needs to be the ability to rapidly take the device off of the channel.

My wifi router has this capability as I normally run on channel 100 (5 GHz), channel width 160. There is nothing to interfere in my area so I get a totally clean wide bit of air to run at maximum speeds. When I boot the router, the DFS channel is not available for ~60 seconds so as to allow scanning.

Evidently the industry has decided for us that adapters and cards simply do not need AP mode DFS capability. We get a lot of capability, that is just not one of them.