openwrt/mt76 #859: Mode 2.4 GHz does not deliver appropriate speeds and there are many packet losses (GL-INET MT6000)

Mummify1867, 2024-02-13:
### Describe the bug

When connecting to the 2.4 GHz network, there is a considerable loss in speed in Xiaomi and Samsung brands smartphones. This problem does not happen on the PC with an Intel Ax210 network card

### OpenWrt version

r25142-a42075435a

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Connect any smartphone at a frequency of 2.4 GHz

### Actual behaviour

There is a considerable loss of packages in 2.4 GHz, making the speed limited to 60 Mbps

### Expected behaviour

There should be no problems with loss of packages

### Additional info

Test in smartphone: https://postimg.cc/qzZgG61b
Test in pc with AX210: https://postimg.cc/f3rRcVD7

Developers believe it is a problem with the MT76 driver. Therefore, they launched a version of the proprietary firmware with the mediatek drivers in a closed source that partially solved the problem. However, he uses an old OpenWrt version without support (21.02): https://dl.gl-inet.com/?model=mt6000&type=beta

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2024-02-19:
@Mummify1867 they use Mediatek OEM drivers, so would not help in this case.

mrkiko, 2024-02-20:
Mark to follow.

xize, 2024-03-25:
here is some of my foundings but there is probably alot more (maybe multiple issues):

currently connected: Poco X6 Pro
in OpenWrt it reports as:
```-56/-76 dBm | 6.0 Mbit/s, 20 MHz 286.7 Mbit/s, 20 MHz, HE-MCS 11, HE-NSS 2```

the 2.4ghz channel is on auto, country is set to Netherlands, channel width 40mhz (but luci reports my device as 20mhz might be due to dfs).

WED is enabled, and speedtest shows: 73.42 download and 36.42 upload, but when i disable ``Disassociate On Low Acknowledgement`` the quality improves a little and I see: 110mb/s download and 33.47 mb/s upload, though I think that is another issue on it's own probably.

now the strange thing happens, if i start measuring my distance even further the link connection is going to degrade abnormaly even though it seems the signal strength reports normal, from my observations around 8 meters away it degrades and from more than 15 there is basicly no connectivity, either handshaking starts become problematic / hanging  or packets get dropped.

currently i run two Flint 2, one with a custom build checked out on  5d34c835a179405ff9802839f5e3e80ccc3b8d43 from the main branch at kernel 6.6 and one with the propetairy mediatek sdk firmware 4.5.7 this one is completely different also in driver and might use all sort of hacks, but my speeds are there reported above 200mb/s download and there is not a strange range issue present.

from how it looks it seems something is not properly synced with the link connectivity/tracking from wireless to ethernet or ethernet to wireless.

what is the best way i can help 👍, the only thing im not used at is applying git patches to my build, but i can surely help testing outputs if that can help or create a new build via build tools with git fetch and pulling to test something.

nightcore500, 2024-04-23:
A few hours ago, @degen91 described this bugreport in another bugreport (marked above) as presumably identical. It is about high retries on Banana Pi R3 routers which use the MT7975N chipset for the 2.4 GHz band. It is very noticeable that some clients (mobile devices like HTC U12 or OnePlus 10 Pro) in an iperf3 test between router and client show very high retries in the TX direction (from router to client) when the client are connected to the 2.4 GHz module. Do you also experience this high number of retries? If so, does this change if you deactivate bluetooth on the client while the iperf3 test is running or switch the client to standby?
For me, the retries stop abruptly and the bandwidth becomes stable.

nightcore500, 2024-05-15:
@Mummify1867 @xize 
Would be great if you could check this on your devices

xize, 2024-05-15:
unfortunately my issue was invalid, once i removed the usb stick as interference source the issue was gone for me.

though about the high retries I suspect this to be true, and it may be also on 5ghz but I need to do some more observation.

what I suspect is that sometimes the chip gets too hot and throttles down, and sometimes even result into a crash, currently i was more investigating the multicast crash but today it seems i managed to crash without multicast.

On my ayaneo I have set the txpower to lowest via the devmgmt driver settings in Windows, I still have to check this behaviour on this AX210 powered device, SMP did not change the result, but im not ruling out this issue can still be a part of a temperature issue within the router, i think if a chip works slower than expected it will slow down tx or rx and maybe when stressed even worse it crash, so by turning down the txpower of my ayaneo i hope it stress less the chip 👍, but this is very random so I need to observe this a few days.

Orel-A, 2024-08-02:
Yes, I can confirm that 2.4GHz is unstable for some reason:
[image]
OpenWrt 23.05.4

Anything I can run to provide more info?

Orel-A, 2024-10-01:
Okay, I think I understood the root problem.
It seems like the router is boardcasting a single spatial stream?

How to check? On Samsung Galaxy you can enable Intelligent Wi-Fi. Press multiple times on the version on the bottom then you unlock "Connectivity Labs" from there click "Wi-Fi developer options" and then "Connection information".
There you will see the Link speed.

You can set:
`uci set wireless.radio0.htmode='HT20'`
to check it better.

Anyone follows this issue or has it become stale?