openwrt/mt76 #860: mt7981 - mt798x-wmac 18000000.wifi: Message 00005aed timeout

lukasz1992, 2024-02-15:
Hello, with the most recent version of mt76 in OpenWRT 23.05.2 I have some temporary breaks (every 2-4 weeks) on WiFi - both SSID (2.4GHz and 5GHz disappears). Driver/fw/hw is able to recover itself within a minute.

[559161.263353] mt798x-wmac 18000000.wifi: Message 00005aed (seq 10) timeout
[1563791.313702] mt798x-wmac 18000000.wifi: Message 00005aed (seq 1) timeout
[2326012.477521] mt798x-wmac 18000000.wifi: Message 00005aed (seq 4) timeout
[2326776.176547] mt798x-wmac 18000000.wifi: Message 000026ed (seq 1) timeout
[2326776.183360] mt798x-wmac 18000000.wifi: Message 00005aed (seq 2) timeout

The common thing is message 0x5A, which is GET_MIB_INFO (I'd guess 0x26 request timed out because of the next 0x5A request).
The same problem was reported with mt7915 chip, @rany2 found a workaround to revert d9dd7635b0551839d2dc5544855c1ebb5205c800.

I will try it and report if it helps.

Fail-Safe, 2024-03-21:
@lukasz1992 Any chance your issue and this could be related? https://github.com/openwrt/mt76/issues/866

lukasz1992, 2024-03-23:
Crash log looks exactly the same.

I have my own tree when I reverted commit https://github.com/openwrt/mt76/commit/d9dd7635b0551839d2dc5544855c1ebb5205c800 , now I have 0 crashes

rany2, 2024-03-23:
You could use this patch from my tree if you don't want to revert it yourself: https://raw.githubusercontent.com/rany2/openwrt/4ad7ed1d5d5b77d32999c9314cc1cec3ee2a9724/package/kernel/mt76/patches/9009-wifi-mt76-mt7915-do-not-use-event-format-to-get-.patch

Fail-Safe, 2024-03-23:
So with your patch, @rany2, you're able to run `multicast_to_unicast_all` enabled without any crashes?

rany2, 2024-03-23:
@Fail-Safe I never had to daily drive `multicast_to_unicast_all` so I don't know, I guess you'll have to figure that out yourself

lukasz1992, 2024-04-04:
After ~1.5 months of tests I can tell that reverting https://github.com/openwrt/mt76/commit/d9dd7635b0551839d2dc5544855c1ebb5205c800 is a working workaround.
My WiFi is now rock stable, no more MCU timeout.

xize, 2024-04-15:
@rany2 
could this patch also fix certain multi psk situations aswell?

i noticed when I use ``wifi-station`` and ``wifi-vlan`` multicast is active by default.

here is a relevant part:
```
[  658.714376] br-lan: port 12(phy1-ap0-aya) entered disabled state
[  658.720518] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered allmulticast mode
[  658.727865] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered promiscuous mode
[  658.738574] mt798x-wmac 18000000.wifi phy1-ap0-aya: left allmulticast mode
[  658.745449] mt798x-wmac 18000000.wifi phy1-ap0-aya: left promiscuous mode
[  658.752359] br-lan: port 12(phy1-ap0-aya) entered disabled state
[  658.778774] br-lan: port 12(phy1-ap0-aya) entered blocking state
[  658.784779] br-lan: port 12(phy1-ap0-aya) entered disabled state
[  658.790816] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered allmulticast mode
[  658.798121] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered promiscuous mode
[  658.995820] br-lan: port 6(phy1-ap0) entered blocking state
[  659.001419] br-lan: port 6(phy1-ap0) entered forwarding state
[  659.007405] br-lan: port 12(phy1-ap0-aya) entered blocking state
[  659.013402] br-lan: port 12(phy1-ap0-aya) entered forwarding state
[ 1804.642711] mt798x-wmac 18000000.wifi phy1-ap0: left allmulticast mode
[ 1804.649296] mt798x-wmac 18000000.wifi phy1-ap0: left promiscuous mode
[ 1804.655785] br-lan: port 6(phy1-ap0) entered disabled state
[ 1805.289028] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1805.548024] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1805.587960] mt798x-wmac 18000000.wifi phy1-ap0-aya (unregistering): left allmulticast mode
[ 1805.596217] mt798x-wmac 18000000.wifi phy1-ap0-aya (unregistering): left promiscuous mode
[ 1805.604429] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1806.306349] br-lan: port 6(phy1-ap0) entered blocking state
[ 1806.311964] br-lan: port 6(phy1-ap0) entered disabled state
[ 1806.317576] mt798x-wmac 18000000.wifi phy1-ap0: entered allmulticast mode
[ 1806.324532] mt798x-wmac 18000000.wifi phy1-ap0: entered promiscuous mode
[ 1806.332261] br-lan: port 6(phy1-ap0) entered blocking state
[ 1806.337830] br-lan: port 6(phy1-ap0) entered forwarding state
[ 1806.346761] br-lan: port 12(phy1-ap0-aya) entered blocking state
[ 1806.352791] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1806.358825] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered allmulticast mode
[ 1806.366118] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered promiscuous mode
[ 1806.374153] br-lan: port 12(phy1-ap0-aya) entered blocking state
[ 1806.380157] br-lan: port 12(phy1-ap0-aya) entered forwarding state
[ 1882.836808] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1882.896545] mt798x-wmac 18000000.wifi phy1-ap0-aya (unregistering): left allmulticast mode
[ 1882.904804] mt798x-wmac 18000000.wifi phy1-ap0-aya (unregistering): left promiscuous mode
[ 1882.912980] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1883.016297] mt798x-wmac 18000000.wifi phy1-ap0: left allmulticast mode
[ 1883.022836] mt798x-wmac 18000000.wifi phy1-ap0: left promiscuous mode
[ 1883.029359] br-lan: port 6(phy1-ap0) entered disabled state
[ 1883.872420] br-lan: port 6(phy1-ap0) entered blocking state
[ 1883.878016] br-lan: port 6(phy1-ap0) entered disabled state
[ 1883.883607] mt798x-wmac 18000000.wifi phy1-ap0: entered allmulticast mode
[ 1883.890650] mt798x-wmac 18000000.wifi phy1-ap0: entered promiscuous mode
[ 1883.898428] br-lan: port 6(phy1-ap0) entered blocking state
[ 1883.903999] br-lan: port 6(phy1-ap0) entered forwarding state
[ 1884.085102] br-lan: port 6(phy1-ap0) entered disabled state
[ 1904.462971] br-lan: port 12(phy1-ap0-aya) entered blocking state
[ 1904.468990] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1904.475115] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered allmulticast mode
[ 1904.482463] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered promiscuous mode
[ 1904.492963] mt798x-wmac 18000000.wifi phy1-ap0-aya: left allmulticast mode
[ 1904.499860] mt798x-wmac 18000000.wifi phy1-ap0-aya: left promiscuous mode
[ 1904.506708] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1904.563556] br-lan: port 12(phy1-ap0-aya) entered blocking state
[ 1904.569560] br-lan: port 12(phy1-ap0-aya) entered disabled state
[ 1904.575623] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered allmulticast mode
[ 1904.582912] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered promiscuous mode
[ 1904.787762] br-lan: port 6(phy1-ap0) entered blocking state
[ 1904.793349] br-lan: port 6(phy1-ap0) entered forwarding state
[ 1904.799282] br-lan: port 12(phy1-ap0-aya) entered blocking state
[ 1904.805296] br-lan: port 12(phy1-ap0-aya) entered forwarding state
[... 14 lines cut ...]
[54284.673245] x29: ffffffc08c60bc80 x28: 0000000000000001 x27: ffffff8005c32780
[54284.680361] x26: ffffff800782c3b8 x25: ffffff80050408a0 x24: ffffff80050408a0
[54284.687477] x23: ffffffc0790aed10 x22: ffffff800782e0e8 x21: 0000000000000001
[54284.694592] x20: ffffff8005c32780 x19: ffffff800782c000 x18: 0000000000000028
[54284.701707] x17: 0000000000000001 x16: 0000000000006f68 x15: ffffff80050408a6
[54284.708823] x14: ffffffc08104ac90 x13: 0000000000000000 x12: 0000000000000002
[54284.715939] x11: 0000000000000040 x10: ffffffc080b57470 x9 : ffffffc080b57468
[54284.723056] x8 : 0000000000000002 x7 : 000000000000b8c1 x6 : 0000009f9d0fdd46
[54284.730171] x5 : 0000000001000000 x4 : 0000000000000000 x3 : 0000000000000000
[... cut ...]
only custom change i made on mt76 was this patch by leans OpenWrt: https://github.com/coolsnowwolf/lede/blob/master/package/kernel/mt76/patches/001-allow-vht-on-2g.patch though the crash happened on 5ghz for me and it seems from all my network devices only my Ayaneo Geek 1S is the trouble maker (Intel AX210).

by reading this issue about multicast and especially this line: ``[  658.720518] mt798x-wmac 18000000.wifi phy1-ap0-aya: entered allmulticast mode`` makes me think it crashes by this issue aswell?

I compile my own openwrt based on test kernel 6.6, here is my fork: https://github.com/xize/openwrt-flint2-testing

lukasz1992, 2024-05-26:
https://github.com/blocktrron/mt76/commit/7447213e9e655f7bab6f45e54053747c2f1104e4 ?

graysky2, 2024-07-03:
@nbd168 - what do you think about this?  My `flogic/xiaomi_redmi-router-ax6000-ubootmod` is experiencing this as well.
```
dmesg:
[ 3395.669339] mt798x-wmac 18000000.wifi: Message 00005aed (seq 3) timeout

logread:
Sun Jun 30 00:56:36 2024 kern.err kernel: [ 3395.669339] mt798x-wmac 18000000.wifi: Message 00005aed (seq 3) timeout
```

lukasz1992, 2024-07-03:
@graysky2 https://github.com/lukasz1992/openwrt/blob/bc641e7998f445752a9368146f0e567bcdd1ed00/package/kernel/mt76/patches/103-drop-GET_MIB_INFO.patch

graysky2, 2024-07-03:
Thanks @lukasz1992 - I am building from master, it seems your rebase is for a stable release.  Do you also build from master/have a patch for that branch?

lukasz1992, 2024-07-03:
> Thanks @lukasz1992 - I am building from master, it seems your rebase is for a stable release. Do you also build from master/have a patch for that branch?
> 
> EDIT: my bad, this applies.
> 
> [graysky2/openwrt@83c9baf](https://github.com/graysky2/openwrt/commit/83c9baf588f90ef7912c089107cdc0c1c8b5a081)

I am basing on openwrt stable release, but for mt76 I use master.
PS: there were some reports that my fix does not solve the issue in all cases.

lukasz1992, 2024-07-03:
> ^ I also tried playing around with the Intel Ax210 driver if WoWlan or turning SMPS to off would do a change but with no avail.

https://github.com/openwrt/openwrt/pull/15824 ?

graysky2, 2024-07-03:
@lukasz1992
> I am basing on openwrt stable release, but for mt76 I use master.
PS: there were some reports that my fix does not solve the issue in all cases.

That's OK.  I am not entirely sure what I am seeing is the same issue.  See my description here: https://github.com/openwrt/mt76/issues/860#issuecomment-2206400232

degen91, 2024-07-03:
@graysky2 

> That's OK. I am not entirely sure what I am seeing is the same issue. See my description here: [#860 (comment)](https://github.com/openwrt/mt76/issues/860#issuecomment-2206400232)

If using `multicast_to_unicast_all` or VLAN, potentially it can be https://github.com/openwrt/mt76/issues/866 or https://github.com/openwrt/mt76/issues/881, which may be connected. Some interesting comments -- https://github.com/openwrt/mt76/issues/866#issuecomment-2171271224 and https://github.com/openwrt/mt76/issues/881#issuecomment-2171413901 (potential workaround for VLANs). I bring them up because they can cause disconnects or crashes with similar timeout messages.

graysky2, 2024-07-03:
@degen91 - thanks for the reply.  I am not using `multicast_to_unicast_all` but I am using VLANs with the `xiaomi_redmi-router-ax6000-ubootmod` as a dumb access point.

graysky2, 2024-07-05:
@lukasz1992 - the patch you shared does not seem to fix the issue for me.  This time wifi did not crash but I found the same output in dmesg:
```
[10745.991134] mt798x-wmac 18000000.wifi: Message 00005aed (seq 10) timeout
```
and in logread
```
Fri Jul  5 14:40:00 2024 kern.err kernel: [10745.991134] mt798x-wmac 18000000.wifi: Message 00005aed (seq 10) timeout
```

CAMOBAP, 2024-09-08:
I recently started to observe the same issue on my OpenWRT (23.05.2 r23630-842932a63d)

```
[137054.969466] mt7915e 0000:06:10.0: Message 00005aed (seq 2) timeout
...
[137259.763160] mt7915e 0000:06:10.0: Message 00005aed (seq 15) timeout
[137054.969466] mt7915e 0000:06:10.0: Message 00005aed (seq 1) timeout
...
[137259.763160] mt7915e 0000:06:10.0: Message 00005aed (seq 15) timeout
...
``

So seq id reaches 15 max and start over again

In my case it looks like related to some hardware failure

LuisMitaHL, 2024-09-08:
MediaTek recently released a new wifi firmware fixing various bugs: https://github.com/openwrt/mt76/issues/881

You can try a SNAPSHOT OpenWrt firmware (this includes the latest wifi fw), or you can download the new wifi fw to /lib/firmware/mediatek/ from https://github.com/openwrt/mt76/tree/master/firmware

closed 2024-09-09 by lukasz1992