openwrt/mt76 #862: To introduce support of MT7902

andruhon, 2024-03-01:
According to archive of windows driver, this seems to be the same driver as for 7922 and 7961.

The windows drivers available here https://www.asus.com/motherboards-components/motherboards/tuf-gaming/tuf-gaming-b650m-e-wifi/helpdesk_download?model2Name=TUF-GAMING-B650M-E-WIFI 

I'll be glad to contribute testing.

pablodz, 2024-03-01:
x2

rany2, 2024-03-29:
@hyharry the README specifically says

> -- Driver in development not working yet --

andruhon, 2024-03-30:
I seen it probing successfully on Linux MX 23 with `wl`, for example https://linux-hardware.org/?probe=96859b01b7

I tried with wl on my Arch without any successful outcome. I later tried to boot from memory stick with MX and yes indeed it somewhat worked, very unstably and 2.4g only, but worked.

andruhon, 2024-03-30:
https://linux-hardware.org/?probe=e4944abc1f

wlshdjj, 2024-05-21:
Further testing is required:
`
--- a/mt76_connac.h	2024-04-19 19:25:55.000000000 +0800
+++ b/mt76_connac.h	2024-05-21 20:25:21.688943000 +0800
@@ -177,6 +177,11 @@
 	return mt76_chip(dev) == 0x7925;
 }
 
+static inline bool is_mt7902(struct mt76_dev *dev)
+{
+	return mt76_chip(dev) == 0x7902;
+}
+
 static inline bool is_mt7920(struct mt76_dev *dev)
 {
 	return mt76_chip(dev) == 0x7920;
@@ -264,6 +269,7 @@
 {
 	switch (mt76_chip(dev)) {
 	case 0x7961:
+	case 0x7902:
 	case 0x7920:
 	case 0x7922:
 	case 0x7925:
--- a/mt7921/pci.c	2024-04-19 19:25:55.000000000 +0800
+++ b/mt7921/pci.c	2024-05-21 20:22:36.919307000 +0800
@@ -26,6 +26,8 @@
 		.driver_data = (kernel_ulong_t)MT7922_FIRMWARE_WM },
 	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7920),
 		.driver_data = (kernel_ulong_t)MT7920_FIRMWARE_WM },
+	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7902),
+		.driver_data = (kernel_ulong_t)MT7902_FIRMWARE_WM },
 	{ },
 };
 
@@ -562,6 +564,8 @@
 module_pci_driver(mt7921_pci_driver);
 
 MODULE_DEVICE_TABLE(pci, mt7921_pci_device_table);
+MODULE_FIRMWARE(MT7902_FIRMWARE_WM);
+MODULE_FIRMWARE(MT7902_ROM_PATCH);
 MODULE_FIRMWARE(MT7920_FIRMWARE_WM);
 MODULE_FIRMWARE(MT7920_ROM_PATCH);
 MODULE_FIRMWARE(MT7921_FIRMWARE_WM);
--- a/mt792x.h	2024-04-19 19:25:55.000000000 +0800
+++ b/mt792x.h	2024-05-21 20:29:10.234267000 +0800
@@ -37,11 +37,13 @@
 #define MT792x_MCU_INIT_RETRY_COUNT	10
 #define MT792x_WFSYS_INIT_RETRY_COUNT	2
 
+#define MT7902_FIRMWARE_WM	"mediatek/WIFI_RAM_CODE_MT7902_1.bin"
 #define MT7920_FIRMWARE_WM	"mediatek/WIFI_RAM_CODE_MT7961_1a.bin"
 #define MT7921_FIRMWARE_WM	"mediatek/WIFI_RAM_CODE_MT7961_1.bin"
 #define MT7922_FIRMWARE_WM	"mediatek/WIFI_RAM_CODE_MT7922_1.bin"
 #define MT7925_FIRMWARE_WM	"mediatek/mt7925/WIFI_RAM_CODE_MT7925_1_1.bin"
 
+#define MT7902_ROM_PATCH	"mediatek/WIFI_MT7902_patch_mcu_1_1_hdr.bin"
 #define MT7920_ROM_PATCH	"mediatek/WIFI_MT7961_patch_mcu_1a_2_hdr.bin"
 #define MT7921_ROM_PATCH	"mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin"
 #define MT7922_ROM_PATCH	"mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin"
@@ -329,6 +331,8 @@
 static inline char *mt792x_ram_name(struct mt792x_dev *dev)
 {
 	switch (mt76_chip(&dev->mt76)) {
+	case 0x7902:
+		return MT7902_FIRMWARE_WM;
 	case 0x7920:
 		return MT7920_FIRMWARE_WM;
 	case 0x7922:
@@ -343,6 +347,8 @@
 static inline char *mt792x_patch_name(struct mt792x_dev *dev)
 {
 	switch (mt76_chip(&dev->mt76)) {
+	case 0x7902:
+		return MT7902_ROM_PATCH;
 	case 0x7920:
 		return MT7920_ROM_PATCH;
 	case 0x7922:
`

wlshdjj, 2024-05-21:
Further testing is required:
[0999-mt76-add-mt7902.patch](https://github.com/openwrt/mt76/files/15390020/0999-mt76-add-mt7902.patch)

zyrouge, 2024-08-12:
Seems like something is being done [here](https://github.com/samveen/mt7902-dkms) too. Just hoping that it would be supported as early as possible.