openwrt/mt76 #863: Will there be beamforming support added to the wireless driver for mt7622 in the future?

hpx502766238, 2024-03-05:
As mentioned in the title, I am using the latest version of the mt76 driver. When tested with 256qam+40Mhz+2x2mimo, my phone and the Redmi AX6S router negotiate a link speed of 400Mbps, which is consistent with the vendor-provided firmware. However, there is still a significant gap in the actual measured throughput and stability. The throughput differs by a factor of two, with the vendor's official firmware achieving a downlink speed of 250M, while OpenWrt can only measure around 100M. I suspect that this is caused by the mt76 driver not enabling support for beamforming. As we all know, the mt7622b integrates the mt7615n wireless chip, which supports beamforming technology. I have checked the historical issues of mt76, and there was one raised in 2022 stating that there was no issue with the signal strength of mt7615, and the issue was closed. However, I would like to point out that the actual internet speed is still significantly affected.
https://github.com/openwrt/mt76/issues/657
So, will mt76 be able to address this issue in the future?

brada4, 2024-03-05:
Frankly no idea what are you talking about. Beamforming is enabled if n+ has 2 antennas you want it or not.
250/400 means you have extreme interference or density anyway. Btw no such exact ac or ax rate as you mention.

hpx502766238, 2024-03-05:
> Frankly no idea what are you talking about. Beamforming is enabled if n+ has 2 antennas you want it or not. 250/400 means you have extreme interference or density anyway. Btw no such exact ac or ax rate as you mention.

What I mean is that under identical test conditions with the same level of interference and negotiated link speeds, the actual downlink speed of the OpenWrt firmware is half that of the official router firmware (provided by Xiaomi). This suggests that there may be some features that are not enabled in the OpenWrt firmware. If beamforming is indeed enabled, then what other factors could be causing this discrepancy?

brada4, 2024-03-06:
It is up to you to describe a bug, like in beamforming case you caught router red handed not beamfirming using advanced antenna array.
Probably some noise thresholds disable more sub bands in openwrt.

hpx502766238, 2024-03-07:
> It is up to you to describe a bug, like in beamforming case you caught router red handed not beamfirming using advanced antenna array. Probably some noise thresholds disable more sub bands in openwrt.

root@RedmiAX6S:~# iw wlan0 info
Interface wlan0
        ifindex 11
        wdev 0x2
        addr a4:39:b3:70:16:87
        ssid 666666_2.4G
        type AP
        wiphy 0
        channel 6 (2437 MHz), width: 40 MHz, center1: 2447 MHz
        txpower 27.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packe           ts
                0       0       1102    0       0       0       0       213448          1157
root@RedmiAX6S:~# iwinfo wlan0 info
wlan0     ESSID: "666666_2.4G"
          Access Point: A4:39:B3:70:16:87
          Mode: Master  Channel: 6 (2.437 GHz)  HT Mode: HT40
          Center Channel 1: 8 2: unknown
          Tx-Power: 27 dBm  Link Quality: 54/70
          Signal: -56 dBm  Noise: -84 dBm
          Bit Rate: 360.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [MediaTek MT7622]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
Except for these, I have no idea how to offer more useful information. I don't know how to get information about antennas, especially those related to beamforming.

brada4, 2024-03-16:
You can auto-select better channel where you will get more useful bandwidth. Or manually (like in excel) parse radio survey reports and manually choose best channel.

hpx502766238, 2024-03-17:
> You can auto-select better channel where you will get more useful bandwidth. Or manually (like in excel) parse radio survey reports and manually choose best channel.

"I tested under the exactly same noisy environment, and found that no matter how the channel was selected, the network speed performance varied greatly. I repeatedly compared the source code structure of mt76 driver and mtk closed-source driver, and found two issues: 1. There is no beamforming-related code in mt76's code about mt7622. 2. mt76 also lacks codes related to Power Amplifier (PA) and Low Noise Amplifier (LNA), which are present in the closed-source driver and play a crucial role in hardware noise immunity."

brada4, 2024-03-17:
You cannot deduct absent beamforming while seeing full link speed. You are unlikely to connect to 10dbm weaker ap in absence of amps. Stop posting random forum facts.

hpx502766238, 2024-03-17:
> You cannot deduct absent beamforming while seeing full link speed. You are unlikely to connect to 10dbm weaker ap in absence of amps. Stop posting random forum facts.

"I'm not a professional, just a network engineer. I don't have much knowledge about hardware, and these 'facts' are just my guesses. But it's undeniable that, based on the evidence, the performance of the mt76 driver is significantly worse than the mtk closed-source driver in the same environment. I'm not sure what kind of information would be helpful, but I sincerely hope that mt76 can improve their wireless performance."

brada4, 2024-03-17:
It is called airtime fairness, in busy area transmit less timeslots. You should not disable it as it will cause suffering for everybody in proximity. Likely vendor firmware has it disabled or different parameters, and it breaks nearby networks while you get better rates near your router as long as its signal is >>20dbm above others.

castiel652, 2024-03-19:
mt7622's 2.4g is basically a mt7615 which from what I remembered mt76 didn't implement beamforming for it.

castiel652, 2024-03-19:
The chance is low since devs moved on to Wi-Fi 6/7 chips.