openwrt/mt76 #865: mt7603: unstable traffic (stalls/hangs) under load (MT7603EN, MT7628AN)

rmilecki, 2024-03-06:
I experience stability issues using chipsets supported by the `mt7603` driver. When running `iperf` client on STA I observe hiccups (traffic temporarily slows down and sometimes stops).

I first reported this back in 2021 in e-mail thread [Unstable WiFi with mt76 on MT7628AN](https://lore.kernel.org/linux-wireless/7c96d5ee-86c1-8068-1b58-40db6087a24f@gmail.com/). It doesn't seem to be regression as this issue seems to go back to 2019 at least. It is also present in the latest `mt76` (2024).

For a while there were probably two different issues in `mt7603`: PSE hangs and traffic hangs. The first problem was hopefully fixed in 2023 with commits [baa19b2e4b7b](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=baa19b2e4b7bbb509a7ca7939c8785477dcd40ee) [c677dda16523](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=c677dda165231c3efffb9de4bace249d5d2a51b9) [317620593349](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=3176205933494bd184c6acd70e796c382bc729b5) [19e4f271d62e](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=19e4f271d62e14e92d15bebddc567dab2eb47535) [c2fcc83b41a6](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=c2fcc83b41a64ae3baa12dfb7ee5d8587373493d).

Traffic hangs remain unresolved and were observed by multiple people using various devices. See above e-mail for OpenWrt forum reports and GitHub issues #692 #719 #841.

rmilecki, 2024-03-06:
## Netgear R6220 (MT7621ST SoC + MT7603EN Wi-Fi + MT7612EN Wi-Fi)

Example from OpenWrt 23.05.2 (iperf on STA connected to MT7603EN using channel 1 bandwidth 20 MHz):
```
[  3] 25.0-26.0 sec  5.50 MBytes  46.1 Mbits/sec
[  3] 26.0-27.0 sec  4.62 MBytes  38.8 Mbits/sec
[  3] 27.0-28.0 sec  4.75 MBytes  39.8 Mbits/sec
[  3] 28.0-29.0 sec  6.50 MBytes  54.5 Mbits/sec
[  3] 29.0-30.0 sec  7.50 MBytes  62.9 Mbits/sec
[  3] 30.0-31.0 sec  6.50 MBytes  54.5 Mbits/sec
[  3] 31.0-32.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3] 32.0-33.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 33.0-34.0 sec  6.38 MBytes  53.5 Mbits/sec
[  3] 34.0-35.0 sec  6.50 MBytes  54.5 Mbits/sec
[  3] 35.0-36.0 sec  7.25 MBytes  60.8 Mbits/sec
[  3] 36.0-37.0 sec  6.50 MBytes  54.5 Mbits/sec
[  3] 37.0-38.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 38.0-39.0 sec  4.75 MBytes  39.8 Mbits/sec
[  3] 39.0-40.0 sec  3.25 MBytes  27.3 Mbits/sec
[  3] 40.0-41.0 sec  1.88 MBytes  15.7 Mbits/sec
[  3] 41.0-42.0 sec  1.88 MBytes  15.7 Mbits/sec
[  3] 42.0-43.0 sec  2.38 MBytes  19.9 Mbits/sec
[  3] 43.0-44.0 sec   896 KBytes  7.34 Mbits/sec
[  3] 44.0-45.0 sec  1.00 MBytes  8.39 Mbits/sec
[  3] 45.0-46.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 46.0-47.0 sec  1.25 MBytes  10.5 Mbits/sec
[  3] 47.0-48.0 sec  3.00 MBytes  25.2 Mbits/sec
[  3] 48.0-49.0 sec  1.00 MBytes  8.39 Mbits/sec
[  3] 49.0-50.0 sec  1.12 MBytes  9.44 Mbits/sec
[  3] 50.0-51.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 51.0-52.0 sec  1.88 MBytes  15.7 Mbits/sec
[  3] 52.0-53.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 53.0-54.0 sec  1.25 MBytes  10.5 Mbits/sec
[  3] 54.0-55.0 sec  2.75 MBytes  23.1 Mbits/sec
[  3] 55.0-56.0 sec  3.75 MBytes  31.5 Mbits/sec
[  3] 56.0-57.0 sec  6.38 MBytes  53.5 Mbits/sec
[  3] 57.0-58.0 sec  6.50 MBytes  54.5 Mbits/sec
[  3] 58.0-59.0 sec  5.62 MBytes  47.2 Mbits/sec
[  3] 59.0-60.0 sec  6.38 MBytes  53.5 Mbits/sec
[  3] 60.0-61.0 sec  7.50 MBytes  62.9 Mbits/sec
[  3] 61.0-62.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 62.0-63.0 sec  6.38 MBytes  53.5 Mbits/sec
```

Whenever traffic stops I can see that station's TX bitrate reported by router goes down from 72.2 Mbps to 6.5 Mbps.

Switching from `HT20` to `NOHT` results in rate being limited to 54 Mbps (it varies between 54 Mbps and 48 Mbps). I run `iperf` for 8 hours and experienced only one one-second stall/hang over that time. Average `iperf` speed was 19.5 Mbps and it varied between 15 and 23-24 Mbps most of the time.

Commenting out `ieee80211_hw_set(hw, AMPDU_AGGREGATION);` in `mac80211.c` results in cutting average speed by about a half (down to 29 Mbps) but improves stability too (rate stays at 72.2 Mbps and sometimes drops to 65 Mbps for a second). During the first 1,5 iperf session I had a one single stall/hang. During next one that took 3 hours I had none. Average speed was 30.4 Mbps (I mostly was 31 Mbps ± 4 Mbps).

rmilecki, 2024-03-06:
## Xiaomi Mi Router 4C (MT7628AN Wi-Fi SoC)

Example from OpenWrt 23.05.2 (iperf on STA connected to MT7628AN using channel 1 bandwidth 20 MHz):
```
[  3] 75.0-76.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 76.0-77.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 77.0-78.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3] 78.0-79.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 79.0-80.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 80.0-81.0 sec  6.62 MBytes  55.6 Mbits/sec
[  3] 81.0-82.0 sec  5.88 MBytes  49.3 Mbits/sec
[  3] 82.0-83.0 sec  5.25 MBytes  44.0 Mbits/sec
[  3] 83.0-84.0 sec  1.25 MBytes  10.5 Mbits/sec
[  3] 84.0-85.0 sec  2.50 MBytes  21.0 Mbits/sec
[  3] 85.0-86.0 sec  1.12 MBytes  9.44 Mbits/sec
[  3] 86.0-87.0 sec   896 KBytes  7.34 Mbits/sec
[  3] 87.0-88.0 sec  1.00 MBytes  8.39 Mbits/sec
[  3] 88.0-89.0 sec  2.00 MBytes  16.8 Mbits/sec
[  3] 89.0-90.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 90.0-91.0 sec  1.25 MBytes  10.5 Mbits/sec
[  3] 91.0-92.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 92.0-93.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 93.0-94.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 94.0-95.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 95.0-96.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 96.0-97.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 97.0-98.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 98.0-99.0 sec  2.50 MBytes  21.0 Mbits/sec
[  3] 99.0-100.0 sec  6.88 MBytes  57.7 Mbits/sec
[  3] 100.0-101.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3] 101.0-102.0 sec  6.88 MBytes  57.7 Mbits/sec
[  3] 102.0-103.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3] 103.0-104.0 sec  5.50 MBytes  46.1 Mbits/sec
[  3] 104.0-105.0 sec  6.75 MBytes  56.6 Mbits/sec
[  3] 105.0-106.0 sec  7.00 MBytes  58.7 Mbits/sec
[  3] 106.0-107.0 sec  7.00 MBytes  58.7 Mbits/sec
[  3] 107.0-108.0 sec  6.75 MBytes  56.6 Mbits/sec
```

Whenever traffic stops I can see that station's TX bitrate reported by router goes down from 72.2 Mbps to 6.5 Mbps.

Switching from `HT20` to `NOHT` results in rate being limited to 54 Mbps (it varies between 54 Mbps and 48 Mbps, sometimes 36 Mbps). I run `iperf` for an hour without a single stall/hang. Average `iperf` speed was 19.2 Mbps and it slowed from from 20 Mbps down to 9-10 Mbps a few times but never stalled/hanged completely.

rmilecki, 2024-03-06:
It seems that all those slowdowns/stalls/hangs happen with high traffic only. Slowing Wi-Fi traffic down (by disabling HT or AMPDU) seems to mitigate them.

It's in sync with what I observed back in 2021 when I tried limiting `iperf` traffic by using `-b 20M` and `-b 10M`.

rmilecki, 2024-03-06:
I was wondering if hardware still generates any IRQs during those stalls/hangs/slowdowns. I cooked a very trivial & dirty patch: [dbg-rx-irqs.txt](https://github.com/openwrt/mt76/files/14509751/dbg-rx-irqs.txt). It's terrible quality but maybe it shows something interesting? Following is synced output of client's `iperf` and router's kernel:
```
[  3]  5.0- 6.0 sec  6.38 MBytes  53.5 Mbits/sec	[ 1045.490814] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4681 0
[  3]  6.0- 7.0 sec  6.50 MBytes  54.5 Mbits/sec	[ 1046.530712] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4878 0
[  3]  7.0- 8.0 sec  6.25 MBytes  52.4 Mbits/sec	[ 1047.570853] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4832 0
[  3]  8.0- 9.0 sec  7.50 MBytes  62.9 Mbits/sec	[ 1048.610675] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4854 0
[  3]  9.0-10.0 sec  6.25 MBytes  52.4 Mbits/sec	[ 1049.650727] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4911 0
[  3] 10.0-11.0 sec  6.50 MBytes  54.5 Mbits/sec	[ 1050.690658] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4198 2
[  3] 11.0-12.0 sec  3.88 MBytes  32.5 Mbits/sec	[ 1051.730982] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 2251 19
[  3] 12.0-13.0 sec  2.88 MBytes  24.1 Mbits/sec	[ 1052.770621] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 1606 17
[  3] 13.0-14.0 sec  1.12 MBytes  9.44 Mbits/sec	[ 1053.810652] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 1991 0
[  3] 14.0-15.0 sec  3.50 MBytes  29.4 Mbits/sec	[ 1054.850595] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 1669 0
[  3] 15.0-16.0 sec  1.88 MBytes  15.7 Mbits/sec	[ 1055.890584] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 1676 0
[  3] 16.0-17.0 sec  1.88 MBytes  15.7 Mbits/sec	[ 1056.930789] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 1052 0
[  3] 17.0-18.0 sec   896 KBytes  7.34 Mbits/sec	[ 1057.970606] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 1086 0
[  3] 18.0-19.0 sec  1.75 MBytes  14.7 Mbits/sec	[ 1059.010619] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 1297 0
[  3] 19.0-20.0 sec  2.62 MBytes  22.0 Mbits/sec	[ 1060.050603] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4879 0
[  3] 20.0-21.0 sec  6.50 MBytes  54.5 Mbits/sec	[ 1061.090740] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4821 0
[  3] 21.0-22.0 sec  6.25 MBytes  52.4 Mbits/sec	[ 1062.130816] mt7603e 0000:02:00.0: [mt7603_dbg_watchdog] 4682 0
```
It turns out that whenever slow downs happen there are some `MT_INT_RX_DONE(1)` interrupts (MCU interrupts).

I added some debugging to `*_mcu_*_send.*()` functions and none of them is called after early init phase on MT7603EN. So it seems MCU is generating IRQs and sending those packets on its own (those are not replies to MCU requests).

Some debugging in `mt7603_queue_rx_skb()` revealed that those are type 0 (`PKT_TYPE_TXS`?) IRQs and `tdx[1]` is `0x0080cd00` (which means `idx` 0) in my case. They refer to my only station connected to the router and result in `mt76` calling `ieee80211_sta_set_buffered()`.

Is that any good hint on what may be happening? My STA keeps sending traffic with `iperf` so it clearly doesn't go to sleep. Can that be some queuing issue?

nachalni, 2024-03-07:
Hi @rmilecki, I have checked Mediateks proprietary driver and it seems that in that driver idx 0 is not used for client sta, idx used for sta starts from index 1, maybe there is some reason behind that ?

rmilecki, 2024-03-07:
> I have checked Mediateks proprietary driver and it seems that in that driver idx 0 is not used for client sta, idx used for sta starts from index 1, maybe there is some reason behind that ?

As a very quick test I connected another device (smartphone) to R6220's MT7603EN and **then** my ThinkPad notebook with `iperf` client. I experience the same stability issues only this time `tdx[1]` is `0x0080cd01` (which means `idx` 1). I didn't attempt modifying driver code to avoid idx 0 in general.

nachalni, 2024-03-07:
What might be interesting too, is that MT_HIGH_PRIORITY_1 high priority register value is different from the proprietary driver. MT76 has this value set to `0x55555553` in https://github.com/openwrt/mt76/blob/master/mt7603/init.c#L64, while proprietary driver has this value set to `0x55555555`. Also would be interesting to know the reason behind that

rmilecki, 2024-03-07:
I started wondering if MT7603EN may actually stop/slow down on sending packets. I developed another ugly debugging patch for printing max queues lengths over 1 second: [mt76_max_queued.txt](https://github.com/openwrt/mt76/files/14523029/mt76_max_queued.txt)

Here is my manually (accuracy < 1 s) synced output of station (`iperf` output) and AP (kernel output):
```
[  3]  0.0- 1.0 sec  7.75 MBytes  65.0 Mbits/sec	[  303.736585] [mt7603_dbg_watchdog] [IRQs] tx:2166 rx0:4725 rx1:0 [QUEUES] 0 0 17 0 0 1 0
[  3]  1.0- 2.0 sec  6.38 MBytes  53.5 Mbits/sec	[  304.776652] [mt7603_dbg_watchdog] [IRQs] tx:2125 rx0:4708 rx1:0 [QUEUES] 0 0 6 0 0 1 0
[  3]  2.0- 3.0 sec  7.12 MBytes  59.8 Mbits/sec	[  305.816434] [mt7603_dbg_watchdog] [IRQs] tx:2059 rx0:4472 rx1:0 [QUEUES] 0 0 4 0 0 1 0
[  3]  3.0- 4.0 sec  6.50 MBytes  54.5 Mbits/sec	[  306.856286] [mt7603_dbg_watchdog] [IRQs] tx:2181 rx0:4735 rx1:0 [QUEUES] 1 0 10 0 0 1 0
[  3]  4.0- 5.0 sec  6.50 MBytes  54.5 Mbits/sec	[  307.896319] [mt7603_dbg_watchdog] [IRQs] tx:2180 rx0:4784 rx1:0 [QUEUES] 0 0 4 0 0 1 0
[  3]  5.0- 6.0 sec  6.50 MBytes  54.5 Mbits/sec	[  308.936232] [mt7603_dbg_watchdog] [IRQs] tx:1764 rx0:4029 rx1:3 [QUEUES] 0 0 22 0 0 1 0
[  3]  6.0- 7.0 sec  6.62 MBytes  55.6 Mbits/sec	[  309.976276] [mt7603_dbg_watchdog] [IRQs] tx:1330 rx0:3281 rx1:12 [QUEUES] 1 0 63 0 0 1 0	← q_tx[2] gets longer = traffic slow downs
[  3]  7.0- 8.0 sec  4.00 MBytes  33.6 Mbits/sec	[  311.016331] [mt7603_dbg_watchdog] [IRQs] tx:1371 rx0:3269 rx1:13 [QUEUES] 0 0 57 0 0 1 0
[  3]  8.0- 9.0 sec  4.75 MBytes  39.8 Mbits/sec	[  312.056313] [mt7603_dbg_watchdog] [IRQs] tx:640 rx0:1492 rx1:6 [QUEUES] 0 0 63 0 0 1 0
[  3]  9.0-10.0 sec  1.75 MBytes  14.7 Mbits/sec	[  313.096662] [mt7603_dbg_watchdog] [IRQs] tx:1648 rx0:3656 rx1:0 [QUEUES] 0 0 5 0 0 1 0
[  3] 10.0-11.0 sec  3.75 MBytes  31.5 Mbits/sec	[  314.136306] [mt7603_dbg_watchdog] [IRQs] tx:2151 rx0:4775 rx1:0 [QUEUES] 0 0 8 0 0 1 0
[  3] 11.0-12.0 sec  6.62 MBytes  55.6 Mbits/sec	[  315.176833] [mt7603_dbg_watchdog] [IRQs] tx:2114 rx0:4678 rx1:0 [QUEUES] 0 0 10 0 0 1 0
[  3] 12.0-13.0 sec  6.50 MBytes  54.5 Mbits/sec	[  316.216236] [mt7603_dbg_watchdog] [IRQs] tx:2179 rx0:4758 rx1:0 [QUEUES] 0 0 5 0 0 1 0
[  3] 13.0-14.0 sec  6.50 MBytes  54.5 Mbits/sec	[  317.256330] [mt7603_dbg_watchdog] [IRQs] tx:2132 rx0:4745 rx1:0 [QUEUES] 0 0 7 0 0 1 0
[  3] 14.0-15.0 sec  6.62 MBytes  55.6 Mbits/sec	[  318.296688] [mt7603_dbg_watchdog] [IRQs] tx:2168 rx0:4808 rx1:0 [QUEUES] 0 0 6 0 0 1 0
[  3] 15.0-16.0 sec  6.50 MBytes  54.5 Mbits/sec	[  319.336478] [mt7603_dbg_watchdog] [IRQs] tx:2156 rx0:4788 rx1:0 [QUEUES] 0 0 6 0 0 1 0
[  3] 16.0-17.0 sec  7.38 MBytes  61.9 Mbits/sec	[  320.376263] [mt7603_dbg_watchdog] [IRQs] tx:2236 rx0:4777 rx1:0 [QUEUES] 0 0 7 0 0 1 0
[  3] 17.0-18.0 sec  6.62 MBytes  55.6 Mbits/sec	[  321.416134] [mt7603_dbg_watchdog] [IRQs] tx:1345 rx0:2985 rx1:0 [QUEUES] 0 0 3 0 0 1 0
```
```
[  3]  0.0- 1.0 sec  8.38 MBytes  70.3 Mbits/sec	[  439.977705] [mt7603_dbg_watchdog] [IRQs] tx:1608 rx0:4945 rx1:0 [QUEUES] 0 0 22 0 0 1 0
[  3]  1.0- 2.0 sec  7.00 MBytes  58.7 Mbits/sec	[  441.018929] [mt7603_dbg_watchdog] [IRQs] tx:1541 rx0:4946 rx1:0 [QUEUES] 0 0 35 0 0 1 0
[  3]  2.0- 3.0 sec  6.62 MBytes  55.6 Mbits/sec	[  442.057523] [mt7603_dbg_watchdog] [IRQs] tx:1705 rx0:4996 rx1:0 [QUEUES] 0 0 14 0 0 1 0
[  3]  3.0- 4.0 sec  6.62 MBytes  55.6 Mbits/sec	[  443.097840] [mt7603_dbg_watchdog] [IRQs] tx:1577 rx0:4980 rx1:0 [QUEUES] 0 0 31 0 0 1 0
[  3]  4.0- 5.0 sec  6.75 MBytes  56.6 Mbits/sec	[  444.137589] [mt7603_dbg_watchdog] [IRQs] tx:1651 rx0:5030 rx1:0 [QUEUES] 0 0 33 0 0 1 0
[  3]  5.0- 6.0 sec  6.62 MBytes  55.6 Mbits/sec	[  445.178764] [mt7603_dbg_watchdog] [IRQs] tx:1656 rx0:4940 rx1:0 [QUEUES] 0 0 33 0 0 1 0
[  3]  6.0- 7.0 sec  6.62 MBytes  55.6 Mbits/sec	[  446.219212] [mt7603_dbg_watchdog] [IRQs] tx:1650 rx0:4953 rx1:0 [QUEUES] 0 0 31 0 0 1 0
[  3]  7.0- 8.0 sec  6.62 MBytes  55.6 Mbits/sec	[  447.259172] [mt7603_dbg_watchdog] [IRQs] tx:1698 rx0:5006 rx1:0 [QUEUES] 0 0 37 0 0 1 0
[  3]  8.0- 9.0 sec  5.75 MBytes  48.2 Mbits/sec	[  448.297197] [mt7603_dbg_watchdog] [IRQs] tx:1168 rx0:3675 rx1:13 [QUEUES] 1 0 63 0 0 1 0	← q_tx[2] gets longer = traffic slow downs
[  3]  9.0-10.0 sec  3.00 MBytes  25.2 Mbits/sec	[  449.337174] [mt7603_dbg_watchdog] [IRQs] tx:644 rx0:1982 rx1:16 [QUEUES] 0 0 42 0 0 1 0
[  3] 10.0-11.0 sec  2.12 MBytes  17.8 Mbits/sec	[  450.377171] [mt7603_dbg_watchdog] [IRQs] tx:571 rx0:1793 rx1:6 [QUEUES] 0 0 41 0 0 1 0
[  3] 11.0-12.0 sec  2.12 MBytes  17.8 Mbits/sec	[  451.417169] [mt7603_dbg_watchdog] [IRQs] tx:432 rx0:1613 rx1:9 [QUEUES] 0 0 89 0 0 1 0
[  3] 12.0-13.0 sec  1.00 MBytes  8.39 Mbits/sec	[  452.457172] [mt7603_dbg_watchdog] [IRQs] tx:177 rx0:1102 rx1:5 [QUEUES] 0 0 71 0 0 1 0
[  3] 13.0-14.0 sec  2.12 MBytes  17.8 Mbits/sec	[  453.497707] [mt7603_dbg_watchdog] [IRQs] tx:211 rx0:1008 rx1:7 [QUEUES] 0 0 86 0 0 1 0
[  3] 14.0-15.0 sec  1.00 MBytes  8.39 Mbits/sec	[  454.537117] [mt7603_dbg_watchdog] [IRQs] tx:223 rx0:1191 rx1:6 [QUEUES] 0 0 119 0 0 1 0
[  3] 15.0-16.0 sec  1.88 MBytes  15.7 Mbits/sec	[  455.577135] [mt7603_dbg_watchdog] [IRQs] tx:200 rx0:1194 rx1:3 [QUEUES] 0 0 98 0 0 1 0
[  3] 16.0-17.0 sec  0.00 Bytes  0.00 bits/sec	[  456.617086] [mt7603_dbg_watchdog] [IRQs] tx:116 rx0:233 rx1:4 [QUEUES] 0 0 108 0 0 1 0
[  3] 17.0-18.0 sec  0.00 Bytes  0.00 bits/sec	[  457.657417] [mt7603_dbg_watchdog] [IRQs] tx:40 rx0:294 rx1:1 [QUEUES] 0 0 18 0 0 1 0
[  3] 18.0-19.0 sec  1.25 MBytes  10.5 Mbits/sec	[  458.697068] [mt7603_dbg_watchdog] [IRQs] tx:579 rx0:1806 rx1:9 [QUEUES] 0 0 44 0 0 1 0
[  3] 19.0-20.0 sec  1.88 MBytes  15.7 Mbits/sec	[  459.737067] [mt7603_dbg_watchdog] [IRQs] tx:352 rx0:1874 rx1:6 [QUEUES] 0 0 101 0 0 1 0
[  3] 20.0-21.0 sec  2.00 MBytes  16.8 Mbits/sec	[  460.777078] [mt7603_dbg_watchdog] [IRQs] tx:111 rx0:230 rx1:4 [QUEUES] 0 0 81 0 0 1 0
```
(scroll those above horizontally for my comments)

rmilecki, 2024-03-07:
> proprietary driver has this value set to `0x55555555`.

I changed `mt76` to use `0x55555555` but that doesn't help

nachalni, 2024-03-07:
> I started wondering if MT7603EN may actually stop/slow down on sending packets. I developed another ugly debugging patch for printing max queues lengths over 1 second: [...]

And it seems that interrupts with MT_RXQ_MCU are coming at the time of the slowdown, client device goes to powersave mode and tx packets are loopbacked ?

rmilecki, 2024-03-07:
> And it seems that interrupts with MT_RXQ_MCU are coming at the time of the slowdown, client device goes to powersave mode and tx packets are loopbacked ?

Yeah, that's in sync with what I observed and described earlier in https://github.com/openwrt/mt76/issues/865#issuecomment-1980806588

nightcore500, 2024-03-13:
I am currently observing the same behavior on a banana pi r3 which uses the MT7975N chip for 2.4 GHz. 
Configured as an access point. I only noticed it when I realized that some connected clients such as an HTC U12 Plus or a OnePlus 10 pro in 2.4GHz achieve very poor bandwidths in download (from WAN between 8-15mbit). The upload, on the other hand, was great. 
On the Banana Pi R3 I am currently using a self-compiled OpenWRT 23.05.2. To exclude already fixed driver problems I have already changed to kernel version 5.15.141 and commit ce7ccc5 of the mt76 driver. 
The problem remains. A station dump from the BPI shows that the clients switch down to 6mbit (RX) immediately after no further traffic flows. (power save mode?) 
However, this behavior is also partially visible during a running speed test. However, the TX path remains stable at 130Mbit (MCS15).

I ran some iperf3 tests between BPI and client with a runtime of 30 seconds. The client reaches 40-50mbit in the download direction, but shows considerable retries.
In the upload direction, on the other hand, it remains relatively stable at 90-98mbit with 0 retries.
station dump of one of the affected devices:

```
Station 74:f6:1c:00:07:5a (on phy0-ap0)
        inactive time:  14760 ms
        rx bytes:       2521602353
        rx packets:     2150838
        tx bytes:       3965398152
        tx packets:     3434023
        tx retries:     230475
        tx failed:      232093
        rx drop misc:   127
        signal:         -32 [-35, -34, -40, -44] dBm
        signal avg:     -32 [-35, -34, -40, -44] dBm
        tx bitrate:     130.0 MBit/s MCS 15
        tx duration:    403565220 us
        rx bitrate:     6.0 MBit/s
        rx duration:    239091517 us
        last ack signal:-32 dBm
        avg ack signal: -32 dBm
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 11134 seconds
        associated at [boottime]:       84214.016s
        associated at:  1710328312316 ms
        current time:   1710339447102 ms
```

I also used an older surface pro 3 for testing. The WLAN module shows up as Marvell AVASTAR Wireless AC. **The problem is not observed here at all.** 
A speed test via WAN works great. An iperf3 test also shows no retries in either direction. 
The only difference that can be seen per station dump on the BPI is that the surface does not switch down to 6mbit. Constant TX/RX at 130 MBit MCS15.
So it seems to be a problem with the power save mode?

nightcore500, 2024-03-13:
I have just configured a second BPI as a client and connected both via 2.4GHz. As with the surface, these problems do not occur here either.
I would have liked to test what happens as soon as I activate the power save mode on the BPI client. However, this was not possible because the mt76 driver does not seem to support power save on the client mode?
```
iw dev phy0-sta0 set power_save on
command failed: Not supported (-95)
```

nightcore500, 2024-03-14:
I have just done some more testing with the affected devices.
I noticed a strange behavior:
If I leave an iperf3 running on the Android client for several minutes and switch it to standby, the iperf3 no longer shows a single retry after a few seconds. 
In addition, the transmitted bandwidth then remains stable at around 100 mbit at 20mhz channel width. In addition, the RX path on the BPI then shows no further drop to 6 mbit. 
If I bring the device out of sleep mode, the problem starts all over again.

nightcore500, 2024-03-14:
I have just noticed something else:
As soon as I start an iperf3 test and deactivate the bluetooth on the client while the test is running, the problem disappears. No more retries, bandwidth increases to around 100mbit. There is no longer a drop to 6mbit during the running test.
At first I suspected interference from bluetooth. 
But after I ran another iperf test, the problem reappeared despite deactivated bluetooth.
As soon as the client has no more traffic to process and switches back down to 6 mbit through idle, the problem is there again despite deactivated bluetooth. There is only an improvement when the bluetooth on the client is reactivated and deactivated again during the iperf test. 
This behavior can be seen here with all problematic end devices.

Linaro1985, 2024-04-02:
Hi. @rmilecki thank you for reseaching. Also please take a look for https://github.com/openwrt/mt76/commit/a8d9553d8fc4db9c12022451ca1d2368e796c591#commitcomment-130672442
Watchdog functionality was broken. Rolling back this commit restores its functionality.

rmilecki, 2024-04-02:
I developed a simple workaround that seems to fix stability for me with MT7603 and MT7628:
[[PATCH] wifi: mt76: mt7603: add debugfs attr for disabling frames buffering](https://patchwork.kernel.org/project/linux-wireless/patch/20240325223319.30125-1-zajec5@gmail.com/)

I just pushed that under-review PATCH to OpenWrt, see [commit 7236d4f82b57 ("mt76: add mt7603 possible workaround for MT7603EN / MT7628AN stability")](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=7236d4f82b57680d76a52abc934130cb02cc913c)

My both devices seem really stable with `mt7603` as soon as I do:
```
echo N > /sys/kernel/debug/ieee80211/phy0/mt76/frames_buffering
```

enmaskarado, 2024-04-04:
The patch for disabling frames buffering seems to have been deleted a few hours ago on https://github.com/openwrt/openwrt/commit/a10a6fbac794b30885d65ec817ebdcfe9f94d78a

Besides that, a new version of mt76 arrived and two commits from there are fixes for mt7603. That would be the final solution?

Linaro1985, 2024-04-04:
@enmaskarado I think it would be the final solution because of https://github.com/openwrt/mt76/commit/e4de3592c4e3baa82142eff583cb5a761f790709 (read commit description)

By the way, I'm already testing the fixes and so far everything is fine.

everything411, 2024-04-10:
I have **not** experienced any stability issues after https://github.com/openwrt/mt76/commit/e4de3592c4e3baa82142eff583cb5a761f790709 . For me, mt76 is more stable than proprietary driver now. There are some slow downs for a MI 4C MT7628 router using proprietary drivers but mt76 doesn't suffer from that now.

biboc, 2024-04-19:
@everything411 
On which version of Openwrt are you?
Have you got also these mt76_wmac MCU timed out problem? https://github.com/openwrt/mt76/issues/628
I can't fix it yet
Did you change eth driver as well?

From OpenWRT 23, I patched mt76 driver with two commit changes you mentionned, I still have the problem

everything411, 2024-04-20:
@biboc I'm on OpenWrt master. do you backport b14c235? this commit is not in 23.05

biboc, 2024-04-22:
@everything411 I built OpenWRT and I upgraded Makefile https://github.com/openwrt/openwrt/blob/main/package/kernel/mt76/Makefile to 2024-04-03 that includes https://github.com/openwrt/mt76/commit/b14c2351ddb8601c322576d84029e463d456caef
Doesn't it?

It got multiple MCU HANG like describe here: https://github.com/openwrt/mt76/issues/628
It may be the cause of my problem

biboc, 2024-04-23:
@everything411 
I'm on Openwrt 23.05.3 + mt76 PKG_SOURCE_DATE:=2024-04-03 PKG_SOURCE_VERSION:=1e336a8582dce2ef32ddd440d423e9afef961e71

biboc, 2024-04-23:
@nbd168 Any idea on my problem? Why ping and connection to other nodes are so slow? And what is the cause of Beacon stuck and TX hang ?
Thanks,

biboc, 2024-04-23:
I'll open a new issue

lukasz1992, 2024-07-04:
https://github.com/openwrt/mt76/commit/c3eba20da2c0e77acb171434662b86aedb03cbbf ?