openwrt/mt76 #868: WED support for mt7621?

MoetaYuko, 2024-03-23:
mt7621 uses mt7915 wifi chip, but `CONFIG_NET_MEDIATEK_SOC_WED` is NOT enabled in its kernel defconfig, and thus the `wed_enable` module parameter is a no-op. I believe the option is disabled for a reason, maybe WED support for mt7621 is lacking in mt76?

Right now the wifi speed is heavily bottlenecked by the weak CPU performance of mt7621. With hardware flow offload enabled on OpenWrt 23.05, I get 900Mbps LAN->WAN speed through a wired connection, but only 300Mbps through wifi, and CPU usage reaches 100%. In contrast, wifi speed can reach 800Mbps with the proprietary driver used by 3rd-party OpenWrt forks.

spamsool, 2024-03-23:
Could you please list these 3rd-party OpenWrt forks you mentioned so I could do some tests?

MoetaYuko, 2024-03-23:
> Could you please list these 3rd-party OpenWrt forks you mentioned so I could do some tests?

https://github.com/padavanonly/immortalwrt/tree/mt7915_mtwifi

lukasz1992, 2024-03-23:
@MoetaYuko please try trunk, there were some performance issues for mt7621 devices.
There will be no support for WED for mt7621 devices

MoetaYuko, 2024-03-23:
> @MoetaYuko please try trunk, there were some performance issues for mt7621 devices fixed last year.

Will try, but I doubt the performance could improve much w/o offloading. Could you provide links to the fixes?
> Anyway there will be no support for WED for mt7621 devices

May I know the reason?

lukasz1992, 2024-03-23:
@MoetaYuko 
https://github.com/openwrt/openwrt/commit/cd2b74e01e8d5f5b80b82db1cb204c13ed99dd58
https://github.com/openwrt/openwrt/commit/c49654f4f0c217959542a1093e052851a3909119

Looks like all releases 23.05 have them.
Too old architecture or too hard to do it.

MeIsReallyBa, 2024-03-24:
There will never be wed  for mt7621 since wed needs hardware support.

MoetaYuko, 2024-03-24:
> There will never be wed  for mt7621 since wed needs hardware support.

I managed to get 800Mbps wifi speed with the proprietary driver, and the cpu usage is extremely low, doesn't the result indicate that hardware offloading is supported by the chip?

lukasz1992, 2024-03-24:
@MoetaYuko yes, it indicates. But nobody implemented it in OpenWRT.

MoetaYuko, 2024-03-25:
> @MoetaYuko yes, it indicates. But nobody implemented it in OpenWRT.

Understood, and that's what is being requested.

MeIsReallyBa, 2024-03-25:
> > There will never be wed  for mt7621 since wed needs hardware support.
> 
> I managed to get 800Mbps wifi speed with the proprietary driver, and th>e cpu usage is extremely low, doesn't the result indicate that hardware offloading is supported by the chip?

Wed path: ppe<->wed<->wifi. 
While the hardware offloading implemented in mt7621 is ppe<->cpu<->wifi.
Only mt7622 and mt798x supports wed.

MeIsReallyBa, 2024-03-25:
What proprietary driver has implemented in mt7621 is cpu forward the data from ppe to wifi. This is the only wifi offload feature that mt7621 supports.

MoetaYuko, 2024-03-25:
> What proprietary driver has implemented in mt7621 is cpu forwarding the data from ppe to wifi. This is the only wifi offload feature that mt7621 supports.

Thanks for the clarification. Can I expect the offloading path to be implemented in openwrt?

MeIsReallyBa, 2024-03-26:
> > What proprietary driver has implemented in mt7621 is cpu forwarding the data from ppe to wifi. This is the only wifi offload feature that mt7621 supports.
> 
> Thanks for the clarification. Can I expect the offloading path to be implemented in openwrt?

There is no hope.

nbd168, 2024-04-03:
There is no wifi related hardware offloading in MT7621. Maybe you need to turn on flow offloading and packet steering to improve performance in your setup.

closed 2024-04-03 by nbd168

garyttirn, 2024-04-03:
For the record mt76 can forward +600Mbps to/from LAN over Wifi 6 @80Mhz channel on mt7621+mt7915E.

Dumb AP (no NAT, no routing): 
`ASUS RT-AX53U running 23.05.3 r23809-234f1a2efa`

Client :
`Asus Zenbook with Intel AX210 `
 
Unloaded signal info :
```
Level : -50 dBm
1020.6 Mbit/s, 80 MHz, HE-MCS 10, HE-NSS 2, HE-GI 1
1134.2 Mbit/s, 80 MHz, HE-MCS 11, HE-NSS 2, HE-GI 1
```

```
user@zenbook:~$ iperf3 -R -c server
Connecting to host server, port 5201
Reverse mode, remote host server is sending
[  5] local 192.168.255.123 port 59496 connected to 192.168.255.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  79.8 MBytes   668 Mbits/sec                  
[  5]   1.00-2.00   sec  80.0 MBytes   671 Mbits/sec                  
[  5]   2.00-3.00   sec  79.1 MBytes   663 Mbits/sec                  
[  5]   3.00-4.00   sec  79.4 MBytes   666 Mbits/sec                  
[  5]   4.00-5.00   sec  80.8 MBytes   676 Mbits/sec                  
[  5]   5.00-6.00   sec  72.6 MBytes   610 Mbits/sec                  
[  5]   6.00-7.00   sec  80.0 MBytes   670 Mbits/sec                  
[  5]   7.00-8.00   sec  81.2 MBytes   683 Mbits/sec                  
[  5]   8.00-9.00   sec  81.8 MBytes   684 Mbits/sec                  
[  5]   9.00-10.00  sec  80.8 MBytes   679 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   805 MBytes   675 Mbits/sec  1193             sender
[  5]   0.00-10.00  sec   795 MBytes   667 Mbits/sec                  receiver

iperf Done.

user@zenbook:~$ iperf3 -c server
Connecting to host server, port 5201
[  5] local 192.168.255.123 port 40072 connected to 192.168.255.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  64.4 MBytes   541 Mbits/sec    0    687 KBytes       
[  5]   1.00-2.00   sec  72.5 MBytes   608 Mbits/sec    0   1.06 MBytes       
[  5]   2.00-3.00   sec  75.0 MBytes   629 Mbits/sec    0   1.11 MBytes       
[  5]   3.00-4.00   sec  73.8 MBytes   619 Mbits/sec    0   1.31 MBytes       
[  5]   4.00-5.00   sec  70.0 MBytes   587 Mbits/sec    0    781 KBytes       
[  5]   5.00-6.00   sec  72.5 MBytes   608 Mbits/sec    0    645 KBytes       
[  5]   6.00-7.00   sec  76.2 MBytes   640 Mbits/sec    0   1.07 MBytes       
[  5]   7.00-8.00   sec  76.2 MBytes   640 Mbits/sec    0   1.22 MBytes       
[  5]   8.00-9.00   sec  75.0 MBytes   629 Mbits/sec    0   1.12 MBytes       
[  5]   9.00-10.00  sec  81.2 MBytes   682 Mbits/sec    0   1.02 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   737 MBytes   618 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   735 MBytes   616 Mbits/sec                  receiver

iperf Done.
```

rany2, 2024-04-03:
@garyttirn FYI if you overclock the CPU to 1200Mhz, I've been able to achieve full gigabit Ethernet speeds for the iperf3 UDP test and roughly 900Mbit/s on the TCP test.

stamak, 2024-05-31:
Also noticed limitation: maximum ~ 300Mbps through wifi and high CPU (100%).

stamak, 2024-06-21:
unfortunately, rx page pool xdp is not available.

```
root@RT-AX53U:~# ubus call system board
{
	"kernel": "5.15.137",
	"hostname": "RT-AX53U",
	"system": "MediaTek MT7621 ver:1 eco:3",
	"model": "ASUS RT-AX53U",
	"board_name": "asus,rt-ax53u",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "23.05.2",
		"revision": "r23630-842932a63d",
		"target": "ramips/mt7621",
		"description": "OpenWrt 23.05.2 r23630-842932a63d"
	}
}
root@RT-AX53U:~# ethtool -S eth0 | grep pp_
```