openwrt/mt76 #870: MT7986 OFDMA UL

XDjackieXD, 2024-04-15:
Hi!

I'm trying to get a test-setup for OFDMA up and running and I saw, that https://github.com/openwrt/mt76/commit/75977a85e84459320e90b5e36f8bc559af7085dc disables OFDMA UL for MT7915 (which as far as I can tell also affects the MT7986 used in the Bananapi BPI-R3 I'm using).

Is this still an issue even for the other chipsets that are covered under the MT7915 driver?
For testing it can I just set `ofdma_ul_en` to true in `mcu.c`?

- Jakob

XDjackieXD, 2024-04-17:
Ok I've tested to just set `ofdma_ul_en` to `true` but as far as I can tell, OFDMA is still not used.
I'm testing with three devices (one AP, two Stations), all the same hardware (Bananapi BPI-R3 with a 4x4 Panel Antenna).
I can get 2Gbit/s IPerf throughput (or 1Gbit/s bidirectionally) but the muru_stats sysfs enpoint only counts up `HE SU`, none of the other counters.

castiel652, 2024-04-18:
There are other places that you need to modify.
Check out this commit https://github.com/openwrt/mt76/commit/e52283439094bdc8801bcdae9d95b1907514a2d1

XDjackieXD, 2024-07-15:
Sorry for the late reply.
I think I now also added the required parameters in the init function like you suggested but I could never get it to send any trigger frames for RU allocation (for that I compared the PHY/MAC capabilities a Qualcomm based AP with QSDK based firmware sends and cross-checked with the 802.11 standard).
So I checked with the MTK driver (using the BPI-R3 vendor firmware) and even then when enabling OFDMA it never sends any trigger frame (looking at the on-air traffic using an ax210 in monitor mode; I used a BPI-R3 as the AP and either two other BPI-R3 or two AX210 as clients).
So either the chip does not support OFDMA (even though I expected it to support it as the MTK based firmware has checkboxes to enable it) or (not unlikely) I did something completely wrong during my testing.

Do you by any chance know if this setup should result in trigger frames and RU allocations or if not what a good way to test this would be?

lukasz1992, 2024-07-15:
Could you add patch https://github.com/lukasz1992/openwrt/blob/bc641e7998f445752a9368146f0e567bcdd1ed00/package/kernel/mt76/patches/112-fix-rx-filter-bf.patch and test if it changes anything?

XDjackieXD, 2024-09-13:
Short update from me:
- I tried your patch (thx for helping) but it did not change anything
- I tried the firmware from BPI that comes with the Mediatek mtk driver and even there OFDMA did not work (even though there even is a checkbox in the UI...)
- I since got my hands on a BPI-R4 with the BE14 WiFi card (MT7996e). Using the patch and firmware from https://github.com/openwrt/mt76/pull/907 worked like a charm and I got OFDMA to work immediately.

Shall I keep this issue open or just close it with "probably will never work on the MT7986"?
I will probably not invest any more time trying as I now have a solution for my OFDMA testing (I'm trying to do comparisons between OFDMA vs non-OFDMA 802.11ax for point to multi-point directional links with hidden nodes).