openwrt/mt76 #871: MT7975DN + MT7905DAN with 10 dB less power than propietary drivers

phpaiva, 2024-04-18:
First of all, to put in context, this issue is mainly regarding the OpenWRT portability to an unidentified device, as [it is shown in the post](https://forum.openwrt.org/t/add-support-to-mediatek-mt7621-reference-board-sold-by-ruxtek/191074)

The SoC is MT7621AT and the wifi chips: MT7975DN (2.4 GHz?) + MT7905DAN (5 GHz?), where as I understand, I should install the driver **kmod-mt7915e** to put this to work.

I successfully ported OpenWRT (23.05.2) by adding in :

```
define Device/MediaTek_mt7621-RB-1
$(Device/dsa-migration)
$(Device/uimage-lzma-loader)
IMAGE_SIZE := 15744k
DEVICE_VENDOR := MediaTek
DEVICE_MODEL := RB-1
DEVICE_PACKAGES := kmod-mt7915-firmware -uboot-envtools
endef
TARGET_DEVICES += MediaTek_mt7621-RB-1
```
Then, the .dts file is currently:

```
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
	compatible = "mediatek,rb-1", "mediatek,mt7621-soc", "mediatek,mt7621-rfb-ax-nor", "mt7621-rfb-ax-nor";
	
	model = "MediaTek RB-1";

	aliases {
		led-boot = &led_power;
		led-failsafe = &led_power;
		led-running = &led_power;
		led-upgrade = &led_power;
		label-mac-device = &gmac0;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	keys {
		compatible = "gpio-keys";

		button-wps {
			label = "wps";
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
			debounce-interval = <60>;
			linux,code = <KEY_WPS_BUTTON>;
		};

		button-reset {
			label = "reset";
			gpios = <&gpio 18 GPIO_ACTIVE_LOW>;
			debounce-interval = <60>;
			linux,code = <KEY_RESTART>;
		};

	};

	leds {
		compatible = "gpio-leds";

		led_power: right-green {
			label = "green:right";
			gpios = <&gpio 14 GPIO_ACTIVE_LOW>;
		};
[... 575 lines cut ...]
0000d50 fafb f4f3 0001 fcfd fbfb f5f6 0001 fcfd
0000d60 fbfb f5f6 0000 fcfd fcfc f5f7 0000 fcfd
0000d70 fcfc f5f7 0000 0000 0000 0000 0000 0000
0000d80 e8de 00f4 d90c ede1 01f5 1109 0000 0000
0000d90 0001 f9fc f8f9 f3f2 0001 f9fc f8f9 f3f2
0000da0 0002 fafd f8fa f3f2 0002 fafd f8fa f3f2
0000db0 0000 fcfd fcfc f5f7 0000 fcfd fcfc f5f7
0000dc0 0000 0000 0000 0000 0000 0000 0000 0000
*
0000e00

```
**It is obvious that it is not the same**

What I have tried after that? Since I discovered that if the driver does not find EEPROM data with `mediatek,mtd-eeprom` it searches a file in /lib/firmware/mediatek/mt7915_eeprom_dbdc.bin I commented that line and **tried several EEPROM files. I produced a .bin from the dump obtained with iwpriv from the OEM firmware** which is here: [eeprom_factory_OEM.zip](https://github.com/openwrt/mt76/files/15030324/eeprom_factory_OEM.zip). Also tried an EEPROM obtained of an Archer Ax23 which is running OpenWRT, by the same way, this command did it: `sed 's/ //g' eeprom_in_text.txt | xxd -r -p > eeprom.bin`.

I also verified that **the EEPROM obtained from OpenWRT with the WiFi working taking EEPROM from the "Factory" partition works apparently in the same quality when I use that file on firmware in /lib/firmware/mediatek/**

**With the EEPROM obtained from the OEM firmware, the WiFi admits a maximum configuration of 30 dBm but in the reality it simply does not work.**. With the Ax23's EEPROM, the maximum power allowed by configuration is 26 dBm, the WiFi "works" but really worse than before (even less range than before). I also tried to write the EEPROM in the .dts to check if that makes a difference with the `mediatek,eeprom-data` option (which is commented in the .dts i shared here previously) and that did not work, the EEPROM is not being found.

Since I really do not have precise knowledge of all about this, I don't know what is DBDC (I supose something regarding dual band and sharing the same EEPROM or not for both bands or both radios), **I don't have access to a documentation** of the 7915 EEPROM to know what are the meaning of all of these registers (not only its name, also the explanation of what it is precisely), why in code the registed addresses are indicated with three hex digits and not four, among other things... I'm running out of ideas on how to continue.

Any comments to help me continue exploring possibilities will be deeply appreciated.

If you made it this far, thank you very much for reading me.

phpaiva, 2024-04-19:
Tested on OpenWRT last Snapshot. The behavior is the same.

atmospher3, 2024-05-29:
Have you tried this:
https://github.com/immortalwrt/immortalwrt/blob/master/package/firmware/wireless-regdb/patches/600-custom-change-txpower-and-dfs.patch