openwrt/mt76 #874: MT7603 (MT7628 board) - Many TX hang and Beacon Stuck + very low connection (4-15sec ping between nodes)

biboc, 2024-04-23:
Hilink 7628 board uses MT7603 wifi driver from mt76 repo
I use OpenWrt 23.05.3 and I edit mt76 Makefile with 
PKG_SOURCE_DATE:=2024-04-03 
PKG_SOURCE_VERSION:=1e336a8582dce2ef32ddd440d423e9afef961e71

So I get latests mt76 patches https://github.com/openwrt/mt76/commit/b14c2351ddb8601c322576d84029e463d456caef

I have many TX hang and Beacon Stuck + very low connection (4-15sec ping between nodes)
```
# cat /sys/kernel/debug/ieee80211/phy0/mt76/reset
             TX hang: 88
   TX DMA busy stuck: 0
   RX DMA busy stuck: 0
        Beacon stuck: 9172
   RX PSE busy stuck: 0
            MCU hang: 0
    PSE reset failed: 0
```

And a ping which is very long 4 to 15 seconds!

```
# ping 10.201.21.88
PING 10.201.21.88 (10.201.21.88): 56 data bytes
64 bytes from 10.201.21.88: seq=0 ttl=64 time=12517.019 ms
64 bytes from 10.201.21.88: seq=1 ttl=64 time=11659.209 ms
64 bytes from 10.201.21.88: seq=2 ttl=64 time=14620.829 ms
64 bytes from 10.201.21.88: seq=3 ttl=64 time=14952.260 ms
64 bytes from 10.201.21.88: seq=4 ttl=64 time=15381.312 ms
64 bytes from 10.201.21.88: seq=5 ttl=64 time=15897.280 ms
64 bytes from 10.201.21.88: seq=6 ttl=64 time=14896.970 ms
64 bytes from 10.201.21.88: seq=7 ttl=64 time=13912.369 ms
64 bytes from 10.201.21.88: seq=8 ttl=64 time=13224.867 ms
64 bytes from 10.201.21.88: seq=16 ttl=64 time=8219.529 ms
64 bytes from 10.201.21.88: seq=17 ttl=64 time=7528.824 ms
64 bytes from 10.201.21.88: seq=18 ttl=64 time=8228.625 ms
64 bytes from 10.201.21.88: seq=21 ttl=64 time=6560.212 ms
64 bytes from 10.201.21.88: seq=22 ttl=64 time=6444.579 ms
64 bytes from 10.201.21.88: seq=23 ttl=64 time=7337.334 ms
64 bytes from 10.201.21.88: seq=24 ttl=64 time=6492.642 ms
64 bytes from 10.201.21.88: seq=25 ttl=64 time=6079.157 ms
64 bytes from 10.201.21.88: seq=26 ttl=64 time=5514.735 ms
```

Station is closed to this one with ok metric:
```
Station --:--:--:--:--:-- (on mesh0)
        signal:         -49 [-49, -87] dBm
        signal avg:     -48 [-48, -86] dBm
        mesh plink:     ESTAB
        
DEST ADDR         NEXT HOP          IFACE       SN      METRIC  QLEN    EXPTIME DTIM    DRET    FLAGS   HOP_COUNT       PATH_CHANGE
--:--:--:--:--:-- --:--:--:--:--:-- mesh0       1       1366    0       2900    1600    4       0x15    1       1
```

There are ~40 nodes in the mesh network
It worked fine when I was on Openwrt 19
I upgraded ~20 nodes so I have a network with both versions connected and network is unstable has shown previously

lukasz1992, 2024-04-24:
I think you would need to bisect manually mt76 versions since 2019 year to find a commit that introduces problems

lukasz1992, 2024-04-24:
Also check this https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=7236d4f82b57680d76a52abc934130cb02cc913c

biboc, 2024-04-24:
Yes I saw this patch but @nbd168 removes it on its latest release with a patch that fixed it

dfateyev, 2024-06-01:
I cannot reproduce the issue on `openwrt-master` snapshot [7514d9bb6f](https://github.com/openwrt/openwrt/commit/7514d9bb6f7239fc5c962590b31bd3b6cd0ea7c5) (dated on May 27, with mt76 [2024-05-17](https://github.com/openwrt/openwrt/commit/ed23d4d00dcc26f69da6a2fbb89f21d99f3e25d9)).
Probably, it has been finally fixed there.

closed 2024-07-08 by biboc