openwrt/mt76 #876: MT7976AN - MT6000 - 5GHz WiFi freezes - DHCP responses but no more internet connection

b0nef1re, 2024-04-29:
### Subject of the issue
I use the MT6000 with OpenWRT snapshot r26084 as a public WiFi router with a high volume of clients. After some time the 5GHz WiFi is unusable / freezes.

You are still able to connect to the network, receive an IP via DHCP, but no internet traffic is going through anymore.

The connection to the 2,4 GHz network works without problems. There are no errors in the syslog/kernellog except the usual "Invalid band" error message.

https://github.com/openwrt/openwrt/blob/v23.05.3/package/kernel/mac80211/patches/subsys/780-avoid-crashing-missing-band.patch

### Environment
OpenWRT snapshot r26084 + wireguard 
Public WiFi
2,4 GHz and 5GHz SSID are the same

### Steps to reproduce
OpenWRT snapshot r26084, public 5GHz WiFi with a high client volume. SSID on 2,4 and 5GHz are the same. Wireguard (optional). WED is disabled, only software acceleration is activated.

### Expected behaviour
5GHz running stable

### Actual behaviour
After some time it is not possible to connect to the 5GHz channel anymore. DHCP requests are answered but no internet traffic is going through. Even reconnecting several times to the 5GHz WiFi network does not fix the problem. The only solution is to restart the router or to restart the 5G WiFI interface until the problem will show up again. This problem occured so far on all OpenWRT snapshot versions before.

TheFurryPugcart, 2024-06-09:
I am also experiencing this issue on my MT6000. Once it happens, if I try to reconnect to the 5ghz network, my iPhone never receives a gateway IP.

dashernln, 2024-09-29:
i`m experiencing something similar on my asus tuf ax4200 at home, about 10-15 5ghz clients.
what i`ve noticed (in my home setup i have 3 openwrt AP because walls made of reinforced concrete about 10 inch thick):

1. with DAWN disabled, wifi is stable at least for 5 days
2. with DAWN enabled

- with mt76 firmware from 23.05.5 - it takes several hours for bug to reproduce
- with latest firmware, it takes much longer to hung, from 6 to 18 hours, depending on hom active clients is moving from one AP to another

qwerkle, 2024-09-30:
I’ve found this to only be an issue with DAWN enabled. Removing DAWN and only using 802.11k/v/r seems to be working flawlessly for me. Just make sure your Reassociation Deadline is set to 20000 for iOS devices to roam properly.

dashernln, 2024-09-30:
i have no ios devices:) 
i think, that there exists some another cases similar to DAWN enabled behaviour.

knuxyl, 2025-05-18:
i have issues with 5ghz on release and snapshot on a bpi-r3 mini. i try to use the 5ghz as client to connect to another access point but luci seems to ignore all settings, its stuck at 160MHz AX and RX never goes above 6Mbit. The client can get IP from access point but does nothing at all afterwards. After DHCP it just hangs.