openwrt/mt76 #877: MT7986 - temporarily ping spikes and terrible WiFi performance - 5GHz - 160MHz

b0nef1re, 2024-05-02:
### Subject of the issue
Massive ping spikes and a bad WiFi performance (<50Mbit) are randomly occuring with the MT7986 chipset

### Your environment
* MT6000 & TUF-AX6000
* Latest snapshot (r26128) & 23.05.3
* Public WiFi 
* **WED disabled, no software or hardware offloading, 5GHz, 160MHZ**

### Expected behaviour

Stable and fast WiFI performance:

[image]

### Actual behaviour

I'm having random horrible ping and WiFi performance problems with the MT7986 driver. This happens on the MT6000 and Asus TUF-AX6000. The problems always occur every 10-30min lasting for around 5-10min until everything returns to normal. No errors in the logs.

This problem happened so far with every OpenWRT version which i tried.

I have the feeling that its caused by specific WiFi clients, but it also seems to happen when AP-STA-POLL-OK is being sent.

[image]

[image]

pesa1234, 2024-05-04:
When you get the message `AP-STA-POLL-OK` your device is still present on network?

cliobrando, 2024-05-04:
Redmi AX6000 here, same behavior, only some clients.
5Ghz - 80Mhz
Samsung S21 (Wifi 6): > 200Mbps
Vivo Y35 (Wifi 5): < 25Mbps

b0nef1re, 2024-05-14:
> When you get the message `AP-STA-POLL-OK` your device is still present on network?

AP-STA-POLL-OK is getting sent to clients which seem to be already offline.  But the ping spikes also happen without the AP-STA-POLL_OK message.

ehfd, 2024-05-17:
I reproduce this issue in GL.iNet GL-MT6000 on r26308-4341901f05. The ping spike doesn't occur while pining inside the router or wired LAN.

lukasz1992, 2024-06-28:
Maybe there is a fix here: https://github.com/openwrt/openwrt/pull/15824 ?

burny0202, 2024-10-09:
I think I may be experiencing this as well; I had put it down to other things at the time.

@b0nef1re - How did you record the nice latency graphs? I have been sat trying to watch pings

burny0202, 2024-11-08:
I am definitely having lag spikes as well, though I am not sure pingining is giving the full picture.
Sitting in my VMs - It is very noticeable though
[image]

burny0202, 2024-12-06:
So....Next release fix? 

People on the forum are talking about packet corruption ? Related ?

dtk9289, 2025-01-29:
now i know why redmi ax6000 wifi sucks so hard

gitlaman, 2025-06-03:
Having the same issue on my redmi ax6000, i have 4 dumb wifi ap on my network and not every time when i roam to ax6000 i have lag spikes >600ms and bandwith throttle on my Samsung s22 even the lan connections get high ping. 

When i turn OFF than ON the wifi on s22 it is working as it should. 

I tried tweaking the roaming settings, disabled enabled with help from chatGPT various options in wireless settings on ax6000 but still having problems, i tried 3 diferent stable firmwares but with no  success.

When i get the lag spikes and the bandwidth throttle on ax6000 and when i roam to some other ap the lag is gone but the throttle persist, when i roam back to ax6000 also the lag is gone but the throttle persist.

gitlaman, 2025-06-03:
I think that i have found the source of the problem.

My AX6S and AX6000 redmi have the same power supply, so i changed their place. 

After few hours i got the same lags and throttle speeds on the AX6S because it was placed on the place where AX6000 was. 

I have lots of experience for electronic equipment that malfunction because of faulty power supply but i never thought that it can also mess with wifi in such matter as our problem. Never turned off never hang, only problem with lag and throttle.

Few hours and no lags after i changed the power supply where the AX6000 was. I hope that was the problem, but i think it makes a lot of sense.

martivo, 2026-09-10:
> I think that i have found the source of the problem.
> 
> My AX6S and AX6000 redmi have the same power supply, so i changed their place. 12V 2A
> 
> After few hours i got the same lags and throttle speeds on the AX6S because it was placed on the place where AX6000 was.
> 
> I have lots of experience for electronic equipment that malfunction because of faulty power supply but i never thought that it can also mess with wifi in such matter as our problem. Never turned off never hang, only problem with lag and throttle.
> 
> Few hours and no lags after i changed the power supply where the AX6000 was. I hope that was the problem, but i think it makes a lot of sense.

I was trying to figure this out for ages until I read your comment and gave it a go. I bought new adapter for my Xiaomi Redmi Router AX6000 (mediatek/filogic) and the problem went away. The router had no restarts and no errors - kept constantly dropping my devices from 5Ghz to the 2.4Ghz. The 5Ghz SSID always came back a moment later and i could switch back to it. The old adapter was 12v and 2.0A, the new I got 12v and  3.5A. 
I installed the new adapter in the evening and next day all devices stay on 5Ghz without dropping off. You can see the left side of the graph it does to 0 stations frequently. On the rigjt side already 4 connected and no drops to 0 the whole day. (Mind you during the night there are no devices taht is why it was 0).
[image]