openwrt/mt76 #878: MT7628/MT7603 - Use only one antenna

biboc, 2024-05-03:
Hi,

I've connected only one antenna (main) so I see [-56, -95]
How to disable auxilary antenna and only use main?
The datasheet seems to let the possibility to choose. How to do it from command line? Or dts?
Which is impact the impact to have to antenna defined but only one used?

Thanks,

lukasz1992, 2024-05-03:
- from OpenWRT perspective - you can force rxantenna and txantenna for wifi-device in /etc/config/wireless
- from a driver perspective - correct eeprom of the device or manually override it in dts

biboc, 2024-05-03:
Thanks @lukasz1992 
Can you provide code/cmd line? 
Have you tried it?

lukasz1992, 2024-05-03:
Not tried, it is described here: https://openwrt.org/docs/guide-user/network/wifi/basic