openwrt/mt76 #880: MT7615E Extremely slow speed

BigmenPixel0, 2024-05-19:
Xiaomi Mi Router 3 Pro 5/2.4 GHz 

Speedtest on the router:
``` 
Download: 582.10 Mbit/s                                    
Upload: 491.91 Mbit/s  
```

Speedtest on a client:
```
Download: 30.43 Mbit/s
Upload: 26.16 Mbit/s
```

/etc/config/wireless
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel 'auto'
	option band '2g'
	option htmode 'HT40'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt_Legacy'
	option encryption 'psk2'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
	option key 'xxx'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option channel 'auto'
	option band '5g'
	option htmode 'VHT80'
	option cell_density '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt_New'
	option encryption 'sae-mixed'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
	option key 'xxx'
	option ieee80211w '1'
```

zekica, 2024-05-21:
You didn't provide enough details for anyone to really help you with this.

Since you shared OpenWrt configuration, what exact version are you using?

Are you connecting to a 5GHz or 2.4GHz network when doing the speed test?

What kind of a client device are you using? Does it support 802.11n, 802.11ac?

MT7615 is one of the best supported devices with the mt76 driver, so this shouldn't happen if you are using any remotely modern version.

JohnDowson, 2024-08-21:
I have a very similar issue.
Running OpenWrt 23.05.4 (r24012-d8dd03c46f) on a Xiaomi Redmi AC2100, I get massively reduced speed specifically in download direction. Using `speedtest-netperf` package on the router results in speeds of about 360-390 Mbps in both directions, while on a nearby laptop speeds hover around 8-16 Mbps down and 190-220 Mbps up.

```
config wifi-device 'radio1'
  option type 'mac80211'
  option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
  option channel '36'
  option band '5g'
  option htmode 'VHT80'
  option country 'CN'
  option cell_density '1'

config wifi-iface 'default_radio1'
  option device 'radio1'
  option network 'lan'
  option mode 'ap'
  option ssid 'network'
  option encryption 'psk2'
  option key 'password'
```

BigmenPixel0, 2024-08-22:
I moved to the proprietary drivers. Everything is OK now

JohnDowson, 2024-08-22:
> I moved to the proprietary drivers. Everything is OK now

There is a way to get a proprietary driver for mt7615e working on OpenWRT?
If there is, I'm extremely interested in how to achieve that.

rany2, 2024-08-22:
This patch fixes that issue for me: https://github.com/rany2/openwrt/blob/42894a8f58cda08de1e1f789285f23101c41ba47/package/kernel/mt76/patches/9000-wifi-mt76-mt7615-disable-precal.patch

BigmenPixel0, 2024-08-22:
@JohnDowson it's for r3p, but there are links
https://github.com/Azexios/openwrt-r3p-mtk

mikekuznetsov11, 2024-09-03:
Same issue. After switching from Pandora Box to OpenWrt (v.23.05.4), my r3p gives extremely slow and unstable wifi. I have tried the above recommendation

> @JohnDowson it's for r3p, but there are links
> https://github.com/Azexios/openwrt-r3p-mtk

Replaced the firmware with openwrt-22.03.5-ramips-mt7621-xiaomi_mi-router-3-pro-squashfs-sysupgrade.bin . Now it demonstrates a different behavior, however the result is still unsatisfactory. Wifi speed degrades substantially if the client moves a few meters away from the router.

JohnDowson, 2024-09-04:
It seems like something is messed up about power output. Closing a thin wooden door between me and the router causes my laptop to drop from already abysmal single digit mbps to tens of kbps.

JohnDowson, 2024-09-05:
This forum thread may be relevant: https://forum.openwrt.org/t/mt7615e-is-it-me-or-is-it-the-driver/208281/58

dhewg, 2024-09-19:
Same issue here with a mesh setup on a ramips-mt7621-netgear_wac124:
```
$ iperf3 -c mainrouter
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  38.6 MBytes  32.4 Mbits/sec  519             sender
[  5]   0.00-10.02  sec  38.4 MBytes  32.1 Mbits/sec                  receiver
```

The related wifi net:
```
          Mode: Mesh Point  Channel: 149 (5.745 GHz)  HT Mode: VHT80
          Center Channel 1: 155 2: unknown
          Tx-Power: 13 dBm  Link Quality: 52/70
          Signal: -58 dBm  Noise: -90 dBm
          Bit Rate: 866.7 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

> This patch fixes that issue for me: https://github.com/rany2/openwrt/blob/42894a8f58cda08de1e1f789285f23101c41ba47/package/kernel/mt76/patches/9000-wifi-mt76-mt7615-disable-precal.patch

This patch doesn't seem to have any affect for me.

dhewg, 2024-09-20:
The RX side is not affected, just the TX side. IOW: `iperf -R` doesn't show the regression

BigmenPixel0, 2024-09-25:
> I moved to the proprietary drivers. Everything is OK now

No more xD

chirantan300, 2024-10-04:
> I moved to the proprietary drivers. Everything is OK now

Did you compile the firmware or installed standalone drivers? I am really new to this

BigmenPixel0, 2024-10-04:
> Did you compile the firmware or installed standalone drivers? I am really new to this

Just installed [this](https://github.com/Azexios/openwrt-r3p-mtk/releases/tag/22.03.5,)