openwrt/mt76 #888: MT7986 crash with Disassociate On Low Acknowledgement enabled

Xyz00777, 2024-06-07:
Hi,
i have found out that when i have "Disassociate On Low Acknowledgement" enabled on my Acer Predator W6 on 5Ghz WLAN and an device disconnects from the network (going out of range, shutting down laptop, etc) than openwrt restarts (crashes?).
It looks like it just happen on the MT7986 chipset and its reproducible.
how can i help to fix these bug beside of programming?

b0nef1re, 2024-06-07:
Do you have any crash logs?

Xyz00777, 2024-06-08:
i thought it was these option because after deactivating it, it worked for some while. Sadly its still crashing when a device disconnects :(
here is the serial output of start inclusive 2 forced crashes:
```
Script started, output log file is 'typescript'.
[!p[?3;4l[4l>[?1049h[22;0;0t[4l[?1h=[0m(B[1;64r[H[2J[H[2J
: 1040 0000

FA: 1040 06 0033 [0200 00E0 [0200] to BL

NOTICE:  BL2: v2.6(release-dirty
NOTICE:  BL2: Built : 12:WDT: disabl.. bl_ar_ver:0=0+ OK
NOTICE:  Verifying BL Anti-Rollback Version ... bl_ar_ver:0=0+ OK
NL Anti-Rollver:0=0+ OK
22 - 12:14:58 +0800), B6-AX7800-197

CPU:   986
Model: mt7986-rfb
DRAM:  1 GiB
  Less UP/DOWNTER to selecult)de ATF BL2grade ATF FIngle image
[7A0. U-Bootkey to stop any key to stop autoboot
   Hash(es sha1+Riled
Firmware integritEnvironment to MMC... rying to boot from image slot 1
[9;236H 1 [9;236H 0 
 dev # 0, b, count 6553  Trying 'ke150
     Type:       ge
     Compression:  h64
     OS:          e
     Descuncompressea Start:   0x4039971c
e:   8ac56ab00da3350c036e03737c2b
rc32+ sha1+ OK
   Loading fdt from 0x4039971c to 0x47000000
   Booting using that 0x4700000pressing Kernel Image
end 000000007f7f7987 ... OK
 ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.150 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt ges:
[    DMA      [mem 0x0000000040000000-0x000000007fffffff]
[  y node ranges
[    0.00   node   0:00000430300004fbfffff]
.000000] In 0x000000004sci: probingsci: PSCIv1.0000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33112 r8192 d3242ister CPU i] CPU features: kernel   0.000000]otwait root=rnel_part=PAb19-4a4b-a0c4-8cc34f4c2aam.rootfs_data_part=PARTUUID=5881ba72-d78f-4cbb4b8eef boot_param.boot_image_slot=1 boot_param.upgrade_image_slot=0 boot_param.dual_boot dm-mod.create="dm-verity,,,ro,0 31544 verity 1 PARTUUID=32798804-a1a7-4939498675b P8804-a1a7-4f95-ad5a-48939498675b 4096 4096 3943 3944 sha29e4e20b37746f22633e85b178f5db7c143e11f92733a704299cc933 2dd56e34b15c6c84573cf26c4392028421061d2c808975217b45e9a5b49d2087"
[    0.000000] 072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory1048576K available (8384K kernel code, 906K rwdata, 1472K rodata, 448K init, 300K bss, 36152K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[ 10 jiffies.
00] GICv3: r: cp15 timer(s) runninffffffffff  max_idle_ns: 44079520  0.000000] sched_clock 13MHz, resolution 76nsCalibrating555] CPU fe 0.002643] ax_idle_ns: 19112604462    0.006617L|GFP_DMA32 thermal governor 'bang_bang'
[   hermal_sys: Registered rnor 'user_space'
[    0.007148] ASID allocator initialised with 65536 entries
[    0.007545] pstore: Registered ramoops as persistent store backend
[    0.007549] ramoops: using 0x10000@0x42ff0000, ecc: 0
[    0.016079] cryptd: max_cpu_qlen set to 1000
[    0.017944] SCSI subsystem initialized
[    0.018029] libata version 3.00 loaded.
[    0.01830] IP idents hash table entries: 1: 5, 131072 bytes, linear)
[    0.019983] tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 8192 bytes, linear)
[    0.019997] Table-perturb hash table entries: : 6, 262144 bytes, linear)
[    0.020004] TCP established hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.020042] TCP bind hash table entries: 8192 (order: 5, 131072 bytes, linear)
[    0.020128] TCP: Hash tables configured (establishcol family
=0
[    0. (NAND) (SU) (RTIME) (ric (bsg) dre /soc/pcie@en3 11280000 -517
[    16 ports, IRdisabled
[    0.072452serial: ttySase_baud = 2 ST16650V2
[    0.835635] printk: console [ttyS0] enabled
[    0.840454] mtk_rng 1020f000.rng: registered RNG driver
[    0.840513] hwrng: no data available
[... 1060 lines cut ...]
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'
        option txpower '21'
        option country 'DE'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'NewWorld'
        option encryption 'sae-mixed'
        option key 'XXXXX'
        option disassoc_low_ack '0'

config wifi-iface 'wifinet3'
        option device 'radio0'
        option mode 'ap'
        option ssid 'NewWorldOT'
        option encryption 'sae-mixed'
        option key 'XXXXX'
        option wmm '0'
        option network 'IoT_Network
```
The WLAN who causes the crashes is the "NewWorld" one

Xyz00777, 2024-06-26:
could someone help me please :D i would like to use the 5GHz network :/
since i turned it off it didnt crashed anymore

pcslide, 2024-07-01:
ZyXEL EX5700 seems to have this same issue,  https://github.com/openwrt/openwrt/issues/14548

pcslide, 2024-07-15:
@Xyz00777 
Have you tried latest snapshot?

Xyz00777, 2024-07-15:
> @Xyz00777 
> Have you tried latest snapshot? 

Latest snapshot? 👀 If you let me know how I can test and hope :)

Xyz00777, 2024-07-27:
> @Xyz00777 Have you tried latest snapshot?

at leaste 23.05.4 didnt fixed it :/

rotanid, 2024-07-28:
> > @Xyz00777
> > Have you tried latest snapshot?
> 
> Latest snapshot? 👀 If you let me know how I can test and hope :)

https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/

ziswiler, 2024-08-18:
While I still saw one more unsolicited crash in the short testing I did so far, I can confirm that this issue is indeed fixed. Thanks!

Xyz00777, 2024-12-12:
Sorry for late response.
Looks also fixed in my system.
Thanks, would close it now