openwrt/mt76 #892: MT7621 Cudy WR1300 v3 Wfi 5g not working with channel auto

palbiez, 2024-07-01:
Hi,
I'm new in Openwrt and installed yesterday a Cudy WR1300 v3 with the current 23.05.3 release.
After creating a wireless network the 5g device is not coming up successful

The Channel Analisys says no data and my second SSID is says: "Wireless is not associated"
restart is not successful

I changed the Country code to US nothing happens. Changed back to DE 
Changed the channel from Auto to 36. All is working well. 

```
root@WR1300v3:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel 'auto'
        option band '2g'
        option htmode 'HT20'
        option country 'DE'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Froschi'
        option encryption 'psk2+ccmp'
        option key 'password'
        option dtim_period '1'
        option ieee80211k '1'
        option disassoc_low_ack '0'
        option ieee80211r '1'
        option ft_over_ds '0'
        option reassociation_deadline '20000'
        option bss_transition '1'
        option mobility_domain 'b2d7'
        option wnm_sleep_mode '1'
        option ft_psk_generate_local '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel 'auto'
        option band '5g'
        option htmode 'VHT80'
        option country 'DE'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Froschi'
        option encryption 'psk2+ccmp'
        option key 'password'
        option dtim_period '3'
        option ieee80211k '1'
        option disassoc_low_ack '0'
        option ieee80211r '1'
        option ft_over_ds '0'
        option reassociation_deadline '20000'
        option bss_transition '1'
        option mobility_domain 'b2d7'
        option wnm_sleep_mode '1'
        option ft_psk_generate_local '1'

config wifi-iface 'wifinet2'
        option device 'radio0'
        option mode 'ap'
[... 582 lines cut ...]
Jul  1 07:59:29 WR1300v3 kernel: [  572.433286] br-lan: port 8(phy1-ap1) entered disabled state
Jul  1 07:59:29 WR1300v3 kernel: [  572.439449] device phy1-ap1 entered promiscuous mode
Jul  1 07:59:29 WR1300v3 kernel: [  572.444978] br-lan: port 8(phy1-ap1) entered blocking state
Jul  1 07:59:29 WR1300v3 dnsmasq[1]: read /etc/hosts - 12 names
Jul  1 07:59:29 WR1300v3 dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Jul  1 07:59:29 WR1300v3 dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Jul  1 05:59:30 WR1300v3 usteer: Found nl80211 phy on wdev hostapd.phy1-ap0, ssid=Froschi
Jul  1 05:59:30 WR1300v3 usteer: Connecting to local node hostapd.phy1-ap0
Jul  1 05:59:30 WR1300v3 usteer: Creating local node hostapd.phy1-ap1
Jul  1 05:59:30 WR1300v3 usteer: Found nl80211 phy on wdev hostapd.phy1-ap1, ssid=ALBsmart 
Jul  1 05:59:30 WR1300v3 usteer: Connecting to local node hostapd.phy1-ap1
Jul  1 08:03:31 WR1300v3 dnsmasq[1]: read /etc/hosts - 12 names
Jul  1 08:03:31 WR1300v3 dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Jul  1 08:03:31 WR1300v3 dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Jul  1 08:39:10 WR1300v3 kernel: [ 2953.457466] br-lan: port 8(phy1-ap1) entered disabled state
Jul  1 08:39:11 WR1300v3 kernel: [ 2953.861408] device phy1-ap0 left promiscuous mode
Jul  1 08:39:11 WR1300v3 kernel: [ 2953.866404] br-lan: port 6(phy1-ap0) entered disabled state
Jul  1 08:39:12 WR1300v3 dnsmasq[1]: read /etc/hosts - 12 names
Jul  1 08:39:12 WR1300v3 dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Jul  1 08:39:12 WR1300v3 dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Jul  1 08:39:12 WR1300v3 kernel: [ 2955.061692] br-lan: port 6(phy1-ap0) entered forwarding state
Jul  1 08:39:12 WR1300v3 kernel: [ 2955.164769] br-lan: port 6(phy1-ap0) entered blocking state
Jul  1 08:39:12 WR1300v3 kernel: [ 2955.182260] br-lan: port 6(phy1-ap0) entered blocking state
Jul  1 08:39:12 WR1300v3 kernel: [ 2955.187909] br-lan: port 6(phy1-ap0) entered forwarding state
```

zekica, 2024-07-01:
The 5GHz chip - MT7613BE found in this router is handled in the driver as a special case in the MT7615 driver, internally referred to as 7663.

There is code in `mt7615_init_wiphy` that sets up available channels:

```
	if (is_mt7663(&phy->dev->mt76)) {
		wiphy->iface_combinations = if_comb;
		wiphy->n_iface_combinations = ARRAY_SIZE(if_comb);
	} else {
		wiphy->iface_combinations = if_comb_radar;
		wiphy->n_iface_combinations = ARRAY_SIZE(if_comb_radar);
	}
```

where `if_comb` and `if_comb_radar` have the following values:

```

static const struct ieee80211_iface_combination if_comb_radar[] = {
	{
		.limits = if_limits,
		.n_limits = ARRAY_SIZE(if_limits),
		.max_interfaces = MT7615_MAX_INTERFACES,
		.num_different_channels = 1,
		.beacon_int_infra_match = true,
		.radar_detect_widths = BIT(NL80211_CHAN_WIDTH_20_NOHT) |
				       BIT(NL80211_CHAN_WIDTH_20) |
				       BIT(NL80211_CHAN_WIDTH_40) |
				       BIT(NL80211_CHAN_WIDTH_80) |
				       BIT(NL80211_CHAN_WIDTH_160) |
				       BIT(NL80211_CHAN_WIDTH_80P80),
	}
};

static const struct ieee80211_iface_combination if_comb[] = {
	{
		.limits = if_limits,
		.n_limits = ARRAY_SIZE(if_limits),
		.max_interfaces = MT7615_MAX_INTERFACES,
		.num_different_channels = 1,
		.beacon_int_infra_match = true,
	}
}; 
```

This directly causes DFS channels to not be available on MT7613.

**I'm sure that this is intentional** as there is a possibility that MT7663 firmware doesn't support detecting radar pulses (or doesn't detect them properly).

It would be trivial change the code and enable DFS channels, but you may be breaking local regulations if the hardware doesn't detect the radar pulses properly.

chirantan300, 2024-12-28:
MT7613BE driver doesn't support dfs in openwrt, add this to /etc/config/wireless under "config wifi-device 'radio1' " to continue using auto channel selection on 5ghz without dfs channels and the radio crapping out

>      option channel 'auto'
        option channels '36 40 44 48 149 153 157 161'

>