openwrt/mt76 #896: MT7922 stops working after waking from hibernation

fleaz, 2024-07-09:
Hey,

sometimes after waking my Framework13 from hibernation, I can no longer connect to any wireless network.

Let me know, when you need any more information from me.

Best,
Felix

**OS**: NixOS 24.05
**Kernel**: 6.9.6

**Hardware**:
```
01:00.0 Network controller: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter
        Subsystem: MEDIATEK Corp. Device e616
        Flags: bus master, fast devsel, latency 0, IRQ 113, IOMMU group 12
        Memory at 8010900000 (64-bit, prefetchable) [size=1M]
        Memory at 90b00000 (64-bit, non-prefetchable) [size=32K]
        Capabilities: <access denied>
        Kernel driver in use: mt7921e
        Kernel modules: mt7921e
```

**dmesg output**:
```                                                                                                      
[ 4710.819168] mt7921e 0000:01:00.0: Message 00020007 (seq 7) timeout
[ 4710.820029] mt7921e 0000:01:00.0: PM: dpm_run_callback(): pci_pm_restore+0x0/0xe0 returns -110
[ 4710.820055] mt7921e 0000:01:00.0: PM: failed to restore async: error -110
[ 4710.896420] mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20240409163301a
[ 4711.273528] mt7921e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20240409163347
```

**NetworkManager output**:
```
Jul 09 18:52:17 smithers NetworkManager[1390]: <warn>  [1720543937.2539] platform-linux: do-change-link[3]: failure 110 (Connection timed out)
Jul 09 18:52:46 smithers NetworkManager[1390]: <warn>  [1720543966.4388] device (wlp1s0): Activation: (wifi) association took too long, failing activation
```

bharath637462, 2024-07-23:
can i work on this issue?

Pahheb, 2024-08-30:
+1

Same happens with my mt7921e after hibernation, and I have to run the following command to resolve this issue:
`su -c modprobe -rv mt7921e && sleep 3 && modprobe -v mt7921e`

fleaz, 2024-10-19:
FYI: I did a system upgrade a few weeks ago, which upgraded the kernel to 6.10.6 and apparently the problem was gone since then.
Currently at 6.11.3 and still working fine.

closed 2024-10-19 by nbd168

samuela, 2025-06-17:
I'm still seeing this issue on Linux 6.12.30 with NixOS on a Framework 13:
```
❯ sudo dmesg | grep -i 'mt7921e'
[    5.437461] mt7921e 0000:01:00.0: disabling ASPM  L1
[    5.442560] mt7921e 0000:01:00.0: ASIC revision: 79220010
[    5.518905] mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20241106163228a
[    5.535133] mt7921e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20241106163310
[    6.648584] mt7921e 0000:01:00.0 wlp1s0: renamed from wlan0
```