openwrt/mt76 #897: mt7615E AP/VLAN

regae, 2024-07-25:
the only way i can reproduce this is by viewing remote vnc (same local network)
so i think its UDP related, here's what i'm experiencing
- connection got dropped, wifi still connected
- no communication to/from after it got dropped

tested with snapshot - 21.02 branch
tested devices: ea7500 v2, ea8100 v1, re650 v1

after a week of fiddling around,
reverting this commit fix the issue i have, 0ed6a335ebc28936c68ccba82368d4ab1c8e5fcf
or just remove ieee80211_send_bar() inside case IEEE80211_AMPDU_TX_OPERATIONAL

i understand mt7615 is an old device so it may not be the focus anymore,
but please could you take a look at this particular commit @nbd168 

thank you

lukasz1992, 2024-07-26:
Please test latest snapshot or 23.05.4

regae, 2024-07-26:
> Please test latest snapshot or 23.05.4

already tried with snapshot, 23.05, 22.03, 21.02 
every new version of kernel or mac80211 or mt76 already tested.

the only way so far is to remove ieee80211_send_bar()