openwrt/mt76 #898: mt7921u page_pool_create() gave up with errno -22

ranlyticsBrad, 2024-07-25:
Running nightly openwrt with 6.6 kernel or 23.05 with 6.1 kernel on an Allwinner v3s board (licheepi dock) causes the `page_pool_create() `error to appear in the dmesg log when using a mt7921u usb device.  

`[   14.257809] usb 2-1.3: reset high-speed USB device number 4 using ehci-platform
[   14.427716] page_pool_create() gave up with errno -22
[   14.433091] mt7921u: probe of 2-1.3:1.0 failed with error -22`

current Makefile compile version:
`PKG_SOURCE_URL:=https://github.com/openwrt/mt76
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2024-07-13
PKG_SOURCE_VERSION:=3b47d9df427c4833605a172f2a8f0e0012b04c80`

ranlyticsBrad, 2024-07-26:
Verified it to exist on Allwinner chipsets that are armv7. Same thing on Allwinner H3 and H5.

mihail-jdanov, 2024-11-04:
Same issue with mt7612u (Orange Pi Zero (ARMv7), OpenWrt 23.05.4, kernel v. 5.15.162). It there any chance to fix it?