openwrt/mt76 #902: MT7981: MT7915_TX_RING_SIZE > 300 affects performance

romanovj, 2024-07-31:
Two devices with MT7981 with the same drivers, 
without WED, 
160Mhz AX, 
AP <-> STA, 
iperf3 on devices.

MT7915_TX_RING_SIZE = 2048
STA ---> AP
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-30.00  sec  2.88 GBytes   826 Mbits/sec    0             sender
[  5]   0.00-30.01  sec  2.88 GBytes   825 Mbits/sec                  receiver
```
AP ----> STA
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-30.01  sec  3.99 GBytes  1.14 Gbits/sec    0             sender
[  5]   0.00-30.00  sec  3.99 GBytes  1.14 Gbits/sec                  receiver
```

MT7915_TX_RING_SIZE = 256
STA ---> AP
[ ID] Interval           Transfer     Bitrate         Retr
```
[  5]   0.00-30.01  sec  3.99 GBytes  1.14 Gbits/sec    0             sender
[  5]   0.00-30.00  sec  3.99 GBytes  1.14 Gbits/sec                  receiver
```

AP ----> STA
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-30.01  sec  4.51 GBytes  1.29 Gbits/sec    0             sender
[  5]   0.00-30.00  sec  4.51 GBytes  1.29 Gbits/sec                  receiver
```

rany2, 2024-07-31:
Confirmed on MT7915E as well. Interesting discovery.

zhaojh329, 2024-08-01:
I got a speed check. It really improved.
Mainly reflected in STA -> AP.

lukasz1992, 2024-08-01:
I still wonder why we need so large buffers for TX. Perhaps it affects all wifi5 (and higher) devices that this driver handle.

There is one commit with quite interesting description: https://github.com/openwrt/mt76/commit/dbea5151412b224c430b8277bf3372eb48e6a7fc

On the other hand I wonder how about performance when multiple clients are connected, maybe here we need larger ring sizes

Fail-Safe, 2024-08-02:
FWIW, it does seem that the MT7986 is unaffected by this same issue:

Device: GL.iNet GL-MT6000
WED: Disabled
160Mhz AX

MT7915_TX_RING_SIZE = 2048
STA --> AP
```
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  1.41 GBytes  1.21 Gbits/sec                  sender
[  5]   0.00-10.01  sec  1.41 GBytes  1.21 Gbits/sec                  receiver
```

AP --> STA
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec  1.33 GBytes  1.14 Gbits/sec    1             sender
[  5]   0.00-10.00  sec  1.32 GBytes  1.14 Gbits/sec                  receiver

iperf Done.
```

romanovj, 2024-08-02:
@rany2 I'm also seeing packets drops and lots of retries with [this](https://github.com/rany2/openwrt/blob/main/package/kernel/mt76/patches/1006-wifi-mt76-mt7915-drop-undefined-action-frame.patch) patch 
`1006-wifi-mt76-mt7915-drop-undefined-action-frame.patch`
When client is 1x1 80Mhz AC (android phone );

rany2, 2024-08-02:
@romanovj I'll see if I can confirm

romanovj, 2024-08-04:
@rany2 
aslo compiling kernel for mt7981 with perf events increased single stream speed form 1.1 to 1.3 gpbs. **I saw the same behavior with mt7621 CPU**(lower cpu utilisation when kernel compiled with perf events).

AP -> STA, sinlge stream iperf3 on routers

without perfevents - 1.1
```
iperf3 -c 192.168.15.1 -R
Connecting to host 192.168.15.1, port 5201
Reverse mode, remote host 192.168.15.1 is sending
[  5] local 192.168.15.151 port 35410 connected to 192.168.15.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   124 MBytes  1.04 Gbits/sec
[  5]   1.00-2.00   sec   136 MBytes  1.14 Gbits/sec
[  5]   2.00-3.00   sec   134 MBytes  1.13 Gbits/sec
[  5]   3.00-4.00   sec   136 MBytes  1.14 Gbits/sec
[  5]   4.00-5.00   sec   132 MBytes  1.10 Gbits/sec
[  5]   5.00-6.00   sec   131 MBytes  1.10 Gbits/sec
[  5]   6.00-7.00   sec   131 MBytes  1.10 Gbits/sec
[  5]   7.00-8.00   sec   133 MBytes  1.12 Gbits/sec
[  5]   8.00-9.00   sec   132 MBytes  1.10 Gbits/sec
[  5]   9.00-10.00  sec   132 MBytes  1.11 Gbits/sec

- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec  1.29 GBytes  1.11 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.29 GBytes  1.11 Gbits/sec                  receiver
```
with perf events - 1.3

```
[  5] local 192.168.15.151 port 40898 connected to 192.168.15.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   150 MBytes  1.26 Gbits/sec
[  5]   1.00-2.00   sec   154 MBytes  1.29 Gbits/sec
[  5]   2.00-3.00   sec   158 MBytes  1.32 Gbits/sec
[  5]   3.00-4.00   sec   155 MBytes  1.30 Gbits/sec
[  5]   4.00-5.00   sec   154 MBytes  1.29 Gbits/sec
[  5]   5.00-6.00   sec   156 MBytes  1.31 Gbits/sec
[  5]   6.00-7.00   sec   156 MBytes  1.31 Gbits/sec
[  5]   7.00-8.00   sec   156 MBytes  1.31 Gbits/sec
[  5]   8.00-9.00   sec   158 MBytes  1.32 Gbits/sec
[  5]   9.00-10.00  sec   156 MBytes  1.31 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec  1.52 GBytes  1.30 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.52 GBytes  1.30 Gbits/sec                  receiver
```

rany2, 2024-08-04:
@Fail-Safe I forgot to update the issue but it does indeed reduce the retry count on an 802.11ac Android device (it went from several thousand retries in the first second to 1-2 which is reasonable). I dropped it on my tree but it is worth noting that I didn't have this issue everywhere (for example, with Intel AX201 as client this issue doesn't appear).

romanovj, 2024-08-06:
> @Fail-Safe I forgot to update the issue but it does indeed reduce the retry count on an 802.11ac Android device (it went from several thousand retries in the first second to 1-2 which is [...]

that's shouldn't be like this, right? Should I open another issue?

closed 2024-08-11 by romanovj