openwrt/mt76 #912: serious regression: mt7916 firmware: 5ghz AP not  working

lts-rad, 2024-09-04:
- 2.4ghz interface (wlan)
- 5ghz interface does not appear to be broadcasting beacons to nearby devices 

firmware d6611d015efd725706ee285308dedbff24b9ea03, broken 

checked out 65bbd4c394a9d51f1ca5a0531166c22ff07d4e56, confirmed working

maurerle, 2024-09-06:
Which device is here affected?

Can you trace the bug down to one of the commits in the mentioned code delta, by bisecting further?

https://github.com/openwrt/mt76/compare/65bbd4c394a9d51f1ca5a0531166c22ff07d4e56...d6611d015efd725706ee285308dedbff24b9ea03

lts-rad, 2024-09-06:
To clarify -- this is the firmware itself (not kernel code)

stratixx, 2024-09-17:
Hi, 

I have some observations from my side.
I have Netgear EAX12 with MT7915E working as WiFi Access Point, connected to router over wire. Latest release Openwrt 23.05.4 (r24012-d8dd03c46f) has working 2.4GHz and not starting 5GHz wifi. Snapshot version SNAPSHOT (r27426-232cc239b8) 2024-09-16 12:27:11 has working both wifi's.

By not working wifi in my case I mean that SSID is not broadcasted, no wifi access. But radio is enabled, it is able to perform scan via luci network/wireless scan option.

Kernel logs from 23.05.4 while restarting radio looks like that:
[79324.336625] device phy1-ap0 left promiscuous mode
[79324.341560] br-lan: port 3(phy1-ap0) entered disabled state
[79325.801774] br-lan: port 3(phy1-ap0) entered blocking state
[79325.807601] br-lan: port 3(phy1-ap0) entered disabled state
[79325.813937] device phy1-ap0 entered promiscuous mode
[79325.844345] device phy1-ap0 left promiscuous mode
[79325.849625] br-lan: port 3(phy1-ap0) entered disabled state
[79325.921278] br-lan: port 3(phy1-ap0) entered blocking state
[79325.927061] br-lan: port 3(phy1-ap0) entered disabled state
[79325.933162] device phy1-ap0 entered promiscuous mode

Kernel logs from snapshot:
[ 1904.404544] mt7915e 0000:02:00.0 phy1-ap0: left allmulticast mode
[ 1904.410850] mt7915e 0000:02:00.0 phy1-ap0: left promiscuous mode
[ 1904.417124] br-lan: port 3(phy1-ap0) entered disabled state
[ 1905.036022] br-lan: port 3(phy1-ap0) entered blocking state
[ 1905.041777] br-lan: port 3(phy1-ap0) entered disabled state
[ 1905.047487] mt7915e 0000:02:00.0 phy1-ap0: entered allmulticast mode
[ 1905.054692] mt7915e 0000:02:00.0 phy1-ap0: entered promiscuous mode
[ 1905.061889] br-lan: port 3(phy1-ap0) entered blocking state
[ 1905.067583] br-lan: port 3(phy1-ap0) entered forwarding state
[ 1906.122242] br-lan: left allmulticast mode
[ 1906.308905] br-lan: entered allmulticast mode
[ 2615.024473] mt7915e 0000:02:00.0 phy1-ap0: left allmulticast mode
[ 2615.030747] mt7915e 0000:02:00.0 phy1-ap0: left promiscuous mode
[ 2615.036968] br-lan: port 3(phy1-ap0) entered disabled state
[ 2616.877933] br-lan: port 3(phy1-ap0) entered blocking state
[ 2616.883710] br-lan: port 3(phy1-ap0) entered disabled state
[ 2616.889584] mt7915e 0000:02:00.0 phy1-ap0: entered allmulticast mode
[ 2616.896500] mt7915e 0000:02:00.0 phy1-ap0: entered promiscuous mode
[ 2616.903550] br-lan: port 3(phy1-ap0) entered blocking state
[ 2616.909272] br-lan: port 3(phy1-ap0) entered forwarding state
[ 2616.916154] br-lan: port 3(phy1-ap0) entered disabled state
[ 2617.363271] br-lan: port 3(phy1-ap0) entered blocking state
[ 2617.369028] br-lan: port 3(phy1-ap0) entered forwarding state

There are some differences in logs. Not sure how and why.

lts-rad, 2024-09-17:
@stratixx so is Firmware Version 20240823 working for you then ? you should see it in dmesg when the driver loads

stratixx, 2024-09-18:
I see firmware version 6.6.51.2024.09.05~65cc3daf-r1 in snapshot image. It seems working. Later I can reflash and check what firmware version is used in latest release.

stratixx, 2024-09-18:
version 23.05.4 has mt7915 firmware version 5.15.162+2024-04-03-1e336a85-1

lts-rad, 2024-09-18:
maybe unrelated? the firmware that was committed in https://github.com/openwrt/mt76/commit/d6611d015efd725706ee285308dedbff24b9ea03 is even more recent, 20240823 (vs 04 03)

cristian-ciobanu, 2024-09-30:
I can confirm the most recent firmware for MT7916 (version 20240823) is broken and 5GHz interface does not send any beacons.
Once I have downgrade the firmware files to an older version it started to work fine.

leow149, 2024-10-05:
I can also confirm those issues for the MT7916 (AsiaRF AW7916-NPD on Banana Pi R4). Firmware version 20240823 breaks 5ghz. I downgraded to version 20211230 as that was the first version I could find and now its working again.

Headcrabed, 2024-10-29:
> On the latest commit ([6a11994](https://github.com/openwrt/mt76/commit/6a1199463fc8f7758c32122613288ed95c09e3bd)), 5Ghz works now on the new firmware for me.

Hello, which bandwidth and channel configuration are you using? Have you tried 160mhz?

ZerionSeven, 2024-10-29:
> > On the latest commit ([6a11994](https://github.com/openwrt/mt76/commit/6a1199463fc8f7758c32122613288ed95c09e3bd)), 5Ghz works now on the new firmware for me.
> 
> @ZerionSeven Hello, which bandwidth and channel configuration are you using? Have you tried 160mhz?

I use AC mode on channel 100 at 160 MHz. I get about 1.2 Gbps of bandwith on 2x2.

AX mode doesn't work reliably with my phone, but other devices work fine, so It's probably a problem with the phone.