openwrt/mt76 #915: mt7981: sometimes tx stucks at 600Mbps

keytouch, 2024-09-07:
Device: wr30u
Openwrt version: r27312-62d3773bf1
mt76 version: 65cc3da

It's observed that when doing tx speedtest right after small rest of high throughput tx or rx, sometimes bandwidth is throttled at around 600Mbps.
This problem persists a long time, not introduced by recent changes.
With stock firmware, there is no such behavior.
[image]

This problem persists a long time, not introduced by recent changes.

rany2, 2024-09-07:
It might be thermal throttling. cd into `/sys/class/ieee80211/<phy>/hwmon1` and check whether it exceeds/approaches the critical/max temperature. You can also try to modify `temp1_crit`/`temp1_max`. IIRC if the driver doesn't request it to thermal throttle, the firmware will slow it down to a halt. The values are in millicelsius.

```
/sys/class/ieee80211/wl1/hwmon1# grep '' temp1_*
temp1_crit:110000
temp1_input:55000
temp1_max:120000
```

keytouch, 2024-09-09:
I don't think so. I tested again.
```
# when throughput is stucked at 600Mbps (sometimes it stucks even if there is long pause before testing)
root@OpenWrt:~# cat /sys/class/ieee80211/phy1/hwmon2/temp1_input
50000
# when throughput successfully rise up to 930Mbps
root@OpenWrt:~# cat /sys/class/ieee80211/phy1/hwmon2/temp1_input
53000
```
Then I 
```
echo 120000 > /sys/class/ieee80211/phy1/hwmon2/temp1_crit
```
It doesn't help.

RomanovDanila, 2024-09-10:
Can you try on Linux?

keytouch, 2024-09-15:
Ah. I made a capture locally. Receive window size sometimes stucks at about 262656.
On a bridged Linux VM, Receive window size is always high (4M+) and throughput looks good.

closed 2024-09-15 by keytouch