openwrt/mt76 #916: mt7915: invalid airtime channel utilization for dap-x1860

maurerle, 2024-09-09:
For the target ramips-mt7621, the device dlink-dap-x1860 does not show airtime channel utilization in current main.

After bisecting this to a specific commit, I found that the cause of this is in this commit:

b4a917417c856307fe19fb6a1a2819e3270df22e

Further information on this issue is provided in the gluon-issue-tracker at:
https://github.com/freifunk-gluon/gluon/issues/3338

### Solution

Currently, there is no other solution than to use the mt76 commit prior to the mentioned one.

If there is anything I can do to help fix this, I will - though I did not quite understand what goes wrong on this specific device.

The WSM20 and COVR-X1860 are not affected though and this issue does not influence general wireless clients quality

maurerle, 2024-09-17:
Sometimes after 3h or after 36 hours, this bug fixes itself, magically:

[image]

Eventually the device calibrates itself..?

jluebbe, 2025-03-09:
We've seen this as well on a DAP-X1860 A1 with kernel 5.15.167. Note that the reported airtime values flip between different states on the 2.4GHz radio:
```
root@node:~# iw dev client0 survey dump | grep -A5 -B1 'in use'
Survey data from client0
	frequency:			2412 MHz [in use]
	channel active time:		299348068 ms
	channel busy time:		174532578 ms
	channel receive time:		167033368 ms
	channel BSS receive time:	68370 ms
	channel transmit time:		5811151 ms
root@node:~# iw dev client0 survey dump | grep -A5 -B1 'in use'
Survey data from client0
	frequency:			2412 MHz [in use]
	channel active time:		299348970 ms
	channel busy time:		265270456 ms
	channel receive time:		253015456 ms
	channel BSS receive time:	157701 ms
	channel transmit time:		9659057 ms
root@node:~# iw dev client0 survey dump | grep -A5 -B1 'in use'
Survey data from client0
	frequency:			2412 MHz [in use]
	channel active time:		299349688 ms
	channel busy time:		7309232820476867 ms
	channel receive time:		6071033182431372 ms
	channel BSS receive time:	2471702182883 ms
	channel transmit time:		1153194193501947 ms
```

The 5GHz radio always reports bogus values:
```
root@Baustellenblick-next:~# iw dev client1 survey dump | grep -A5 -B1 'in use'
Survey data from client1
	frequency:			5220 MHz [in use]
	channel active time:		299541859 ms
	channel busy time:		3706609920319303 ms
	channel receive time:		287399344478509 ms
	channel BSS receive time:	11135309163732269 ms
	channel transmit time:		4901612923095391 ms
```

jluebbe, 2025-03-09:
@nbd168 Could this be related to/fixed by https://github.com/openwrt/mt76/commit/02f453d8dfc8f61124a96d6aac32b2573df31469?

lukasz1992, 2025-03-10:
Or maybe this patch would fix it: https://[raw.githubusercontent.com/lukasz1992/openwrt/58a62ed60310d821177fa070b2716ba37f878b3d/package/kernel/mt76/patches/111-connac-fix-airtime-spike.patch](https://raw.githubusercontent.com/lukasz1992/openwrt/58a62ed60310d821177fa070b2716ba37f878b3d/package/kernel/mt76/patches/111-connac-fix-airtime-spike.patch)

jluebbe, 2025-03-26:
That patch seems to try to ignore updates if the difference is too large. I don't think that would help in our case where the value is flipping between to "clusters":
```
while sleep 1; do iw dev client0 survey dump | grep -A5 -B1 'in use' | 
grep busy; done
	channel busy time:		18446744063636752 ms
	channel busy time:		18446744063636765 ms
	channel busy time:		176885473 ms
	channel busy time:		176885733 ms
	channel busy time:		176886143 ms
	channel busy time:		18446744063636870 ms
	channel busy time:		176886753 ms
	channel busy time:		176887014 ms
	channel busy time:		176887327 ms
```