openwrt/mt76 #919: MT7915E Incorrectly Trying to use 6 GHz Band 23.05.4 r24012-d8dd03c46f

bai-yi-bai, 2024-09-19:
Hi,

I have three _Xiaomi Redmi Router AX6S_ units with Mediatek/mt7622+MT7915E radios. These are 802.11b/g/n 2.4 GHz and 802.11ac/ax/n 5 GHz capable devices. Two were set up for fast roaming and to provide better coverage. I had disabled DHCP on both. One is a spare/travel router. 

I had installed `OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a` a very long time ago. About a week ago, I decided it was time to upgrade two of the units in the hopes of improved performance. I upgrade to 
`OpenWrt 23.05.4 r24012-d8dd03c46f / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38`.
Within hours, I began receiving complaints that Wi-Fi was working or working intermittently. 

Today, I decided to investigate and discovered that one of the APs had no clients connecting to it on either band. My clients were still only connecting to the other working AP.
In addition, the wireless network was set as "disabled". Restarting it had no effect.

Looking more closely, I see that the frequencies it is attempting to operate on are incorrect.
[image]

This should be in the 5 GHz band. According to wikipedia, channel 48 should be:
5240 MHz with a frequency range of 5230–5250 MHz with a 20 MHz bandwith.

I upgraded the spare device to the same firmware and was able to reproduce the result. I tried channel 120 which should be 5600 MHz which a frequency range of 5590-5610 MHz with a 20 MHz bandwidth, but it is  6.550 GHz

These are "Wi-Fi 6" devices, not  "Wi-Fi6E" device... they don't have the ability to use the 6 GHz band.

bai-yi-bai, 2024-09-19:
Small update:
I tried going all the way back to 
`openwrt-23.05.0-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-sysupgrade.bin` and I encountered the same issue. I will revert back to 22.03.5

wlshdjj, 2024-09-23:
Please upload the syslog about mt76.

danpawlik, 2024-09-25:
Solved here: https://github.com/openwrt/openwrt/issues/16468