openwrt/mt76 #922: MT7981 5GHz ramdomly bad performance

victor186, 2024-10-21:
I'm testing AX3000T on a restaurant for future network upgrade, but a've noticed poor speeds on 5GHz ramdomly, solved with radio restart, but when it occours, the network goes down due to low speed/high latency.

The AP is running on 80MHz/AX mode.
Openwrt 23.05.5.
[image]
[image]

lukasz1992, 2024-10-21:
Is the second device also connected to the network?
I see really bad signal from it, communication with such device can highly decrease performance.

victor186, 2024-10-22:
> Is the second device also connected to the network? I see really bad signal from it, communication with such device can highly decrease performance.

This devices on list is in 2.4GHz

victor186, 2024-10-27:
> Can you list your wifi clients(device models)?

I can't, due this device is running as AP on a restaurant for administrative and client's Wi-Fi

lukasz1992, 2024-11-07:
Do you have driver 10.0.0.1272 for Windows installed?

victor186, 2024-11-07:
> Looks like Qualcomm QCA9377 + windows 10 driver + 5GHz can cause this. No problems on 2.4 band.

I not understood, Wi-Fi 5GHz adapter with QCA9377 is causing 5GHz network bad performance? I don't have QCA9377 on network and the router is mediatek.

victor186, 2024-11-07:
> @victor186
> 
> > I don't have QCA9377 on network
> 
> How can you be sure?
> 
> > device is running as AP on a restaurant for administrative and client's Wi-Fi

The clients only use smartphones.
The unique PC on Wi-Fi is using a realtek wi-fi adapter

nbd168, 2024-11-14:
One thing you could try is copy the latest MT7981 firmware from https://github.com/openwrt/mt76/tree/master/firmware to your device. If that doesn't help, trying a recent snapshot might also be a good idea.

IrineSistiana, 2024-11-20:
Openwrt 23.05.5. H3C Magic NX30 Pro.

Same issue here. Encountered it several times

Almost zero speed (1kb/s) through 5G wifi. Enough for DHCP but anything else will be broken, even ping.

I noticed that when this happening, there are 2 dead clients (which left at the same time) in luci wifi page. With RX Rate / TX Rate 6.0 Mbit/s, 20 MHz. If I manually client the "Disconnect" button, the wifi works again immediately.

IrineSistiana, 2024-11-20:
More info

Also, when I check the log. The log keeps showing that they were AP-STA-POLL-OK. Until I clicked the luci "Disconnect" button.

P.S. **OFFLINE:MAC:1** **OFFLINE:MAC:2** are clients that went away.

```txt
Wed Nov 20 19:33:33 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **OFFLINE:MAC:1**
Wed Nov 20 19:35:31 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **OFFLINE:MAC:2**
Wed Nov 20 19:38:44 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **OFFLINE:MAC:1**
Wed Nov 20 19:40:51 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **OFFLINE:MAC:2**
Wed Nov 20 19:44:03 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **OFFLINE:MAC:1**
...
Wed Nov 20 20:06:42 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED **OFFLINE:MAC:1**
Wed Nov 20 20:06:44 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED **OFFLINE:MAC:2**
Wed Nov 20 20:06:47 2024 daemon.info hostapd: phy1-ap0: STA **OFFLINE:MAC:1** IEEE 802.11: deauthenticated due to local deauth request
Wed Nov 20 20:06:49 2024 daemon.info hostapd: phy1-ap0: STA **OFFLINE:MAC:2** IEEE 802.11: deauthenticated due to local deauth request
```

When I restart the 5g wifi a few minutes later. Another sus log.

```txt
Wed Nov 20 20:13:06 2024 kern.warn kernel: [2135649.716364] Ignoring NSS change in VHT Operating Mode Notification from **OFFLINE:MAC:1** with invalid nss 2
Wed Nov 20 20:13:06 2024 kern.info kernel: [2143605.339316] device phy1-ap0 left promiscuous mode
Wed Nov 20 20:13:06 2024 kern.info kernel: [2143605.354371] br-lan: port 5(phy1-ap0) entered disabled state
Wed Nov 20 20:13:07 2024 daemon.notice wpa_supplicant[1538]: Set new config for phy phy1
Wed Nov 20 20:13:07 2024 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Wed Nov 20 20:13:07 2024 daemon.notice hostapd: Reload config for bss 'phy1-ap0' on phy 'phy1'
Wed Nov 20 20:13:07 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED **AN:ONLINE:CLIENT:MAC:1**
Wed Nov 20 20:13:08 2024 daemon.notice hostapd: Reloaded settings for phy phy1
Wed Nov 20 20:13:08 2024 daemon.notice netifd: Wireless device 'radio1' is now up
Wed Nov 20 20:13:08 2024 daemon.notice netifd: Network device 'phy1-ap0' link is up
Wed Nov 20 20:13:08 2024 kern.info kernel: [2143607.148600] br-lan: port 5(phy1-ap0) entered blocking state
Wed Nov 20 20:13:08 2024 kern.info kernel: [2143607.154384] br-lan: port 5(phy1-ap0) entered disabled state
Wed Nov 20 20:13:08 2024 kern.info kernel: [2143607.160337] device phy1-ap0 entered promiscuous mode
Wed Nov 20 20:13:08 2024 kern.info kernel: [2143607.165646] br-lan: port 5(phy1-ap0) entered blocking state
Wed Nov 20 20:13:08 2024 kern.info kernel: [2143607.171424] br-lan: port 5(phy1-ap0) entered forwarding state
Wed Nov 20 20:13:09 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Wed Nov 20 20:13:09 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Wed Nov 20 20:13:09 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
...
``

IrineSistiana, 2024-11-21:
I reproduced this bug. If a client leaves the WiFi coverage, there is a certain probability that the above bug will occur.

It is almost the same as this issue https://github.com/openwrt/openwrt/issues/14415 . 

Log keeps showing (p.s. I added `option max_inactivity '60'`.)

```TXT
Thu Nov 21 09:25:38 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:26:46 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:27:56 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:29:04 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:30:24 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:31:33 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:32:39 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:33:44 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
Thu Nov 21 09:34:51 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
...
```

```txt
iw dev phy1-ap0 station dump

Station **WENT:AWAY:CLIENT:MAC** (on phy1-ap0)
        inactive time:  46190 ms
        rx bytes:       7315589
        rx packets:     52352
        tx bytes:       66444699
        tx packets:     69473
        tx retries:     6987
        tx failed:      7033
        rx drop misc:   2
        signal:         -95 [-97, -99] dBm
        signal avg:     -91 [-93, -95] dBm
        tx bitrate:     6.0 MBit/s
        tx duration:    83677141 us
        rx bitrate:     6.0 MBit/s
        rx duration:    4720659 us
        last ack signal:-96 dBm
        avg ack signal: -95 dBm
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 8708 seconds
        associated at [boottime]:       2183028.795s
        associated at:  1732143676976 ms
        current time:   1732152384528 ms
```
p.s. Above device is a smartphone with snapdragon FastConnect 6800 (However, I do believe other clients can do the same.). It left the wifi range hour ago and kilometers away from wifi.

If I manually click the "Disconnect" button in luci, the wifi works again immediately, (no restart).

I'm using the offical unmodified Openwrt 23.05.5 image. https://github.com/openwrt/openwrt/issues/14415 seems using a fork openwrt with a modified driver(?).

victor186, 2024-11-21:
> I reproduced this bug. If a client leaves the WiFi coverage, there is a certain probability that the above bug will occur.
> 
> It is almost the same as this issue [openwrt/openwrt#14415](https://github.com/openwrt/openwrt/issues/14415) .
> 
> Log keeps showing (p.s. I added `option max_inactivity '60'`.)
> 
> ```
> Thu Nov 21 09:25:38 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:26:46 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:27:56 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:29:04 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:30:24 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:31:33 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:32:39 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:33:44 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> Thu Nov 21 09:34:51 2024 daemon.notice hostapd: phy1-ap0: AP-STA-POLL-OK **WENT:AWAY:CLINET:MAC**
> ...
> ```
> 
> ```
> iw dev phy1-ap0 station dump
> 
> Station **WENT:AWAY:CLIENT:MAC** (on phy1-ap0)
>         inactive time:  46190 ms
>         rx bytes:       7315589
>         rx packets:     52352
>         tx bytes:       66444699
>         tx packets:     69473
>         tx retries:     6987
>         tx failed:      7033
>         rx drop misc:   2
>         signal:         -95 [-97, -99] dBm
>         signal avg:     -91 [-93, -95] dBm
>         tx bitrate:     6.0 MBit/s
>         tx duration:    83677141 us
>         rx bitrate:     6.0 MBit/s
>         rx duration:    4720659 us
>         last ack signal:-96 dBm
>         avg ack signal: -95 dBm
>         airtime weight: 256
>         authorized:     yes
>         authenticated:  yes
>         associated:     yes
>         preamble:       short
>         WMM/WME:        yes
>         MFP:            no
>         TDLS peer:      no
>         DTIM period:    2
>         beacon interval:100
>         short preamble: yes
>         short slot time:yes
>         connected time: 8708 seconds
>         associated at [boottime]:       2183028.795s
>         associated at:  1732143676976 ms
>         current time:   1732152384528 ms
> ```
> 
> p.s. Above device is a smartphone with snapdragon FastConnect 6800 (However, I do believe other clients can do the same.). It left the wifi range hour ago and kilometers away from wifi.
> 
> If I manually click the "Disconnect" button in luci, the wifi works again immediately, (no restart).
> 
> I'm using the offical unmodified Openwrt 23.05.5 image. [openwrt/openwrt#14415](https://github.com/openwrt/openwrt/issues/14415) seems using a fork openwrt ~with a modified driver(?)~ (I misunderstund, they enabled /sys/module/mt7915e/parameters/wed_enable.).
> 
> I did not set the wed_enable.
> 
> ```
> cat /sys/module/mt7915e/parameters/wed_enable
> N
> ```

It's make sense, because the router as public Wi-Fi have client's entering and quiting the network at all time.
And i noticed via luci some client's with signal -9x dBm that never disconnect's, like your example, client out of range never disapears.

victor186, 2024-11-22:
> Sorry. My router is a main device, It is hard for me to play with it. But I can provide log if needed.
> 
> @victor186 I feel this is a common bug, for all MT7981, but it happens occasionally, hard to reproduce and notice.
> 
> Maybe we could change the title to make it easier for more users to find?
> 
> "MT7981 5GHz occasionally cannot disconnect clients that have left and causes bad performance."

Done

IrineSistiana, 2024-11-22:
A temp dirty fix. Disconnect clients, that have very low signal, via script. Run it via cron every 1 min, should work. Haven't tested it yet.

```sh
#!/bin/sh
# threshold (dBm)
thr=-94
# add other interface names if any, e.g. "phy1-ap0 phy1-ap1 phy1-ap2"
wlanlist="phy1-ap0" 

disconnect() {
        mac=$1
        wlan=$2
        rssi=$3
        echo "disconnecting client at $wlan $mac with $rssi dBm (thr=$thr)" | logger -t disconnected-client-killer
        ubus call hostapd.$wlan del_client "{'addr':'$mac', 'reason':5, 'deauth':true, 'ban_time':1000}"
}

for wlan in $wlanlist; do
        iwinfo ${wlan} assoclist | grep SNR | while read line; do
                mac=$(echo "${line}" | awk '{ print $1 }')
                rssi=$(echo "${line}" | awk '{ print $2 }')
                # echo "if:${wlan} mac:${mac} rssi:${rssi}"
                if [ $rssi -lt $thr ]; then
                        disconnect $mac $wlan $rssi
                fi
        done
done
```

lukasz1992, 2024-11-25:
Maybe adding patch similar to https://github.com/freifunk-gluon/gluon/blob/main/patches/openwrt/0009-mt76-include-fixes-for-MT7603-MT7612.patch would help?

kloon15, 2024-11-26:
> You can try this [patch](https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/487509697c0ad1b1511da74c0ec5baf2070cf057%5E%21/#F0) from mtk

This patch def does some good thing, before i had intermittent packet loss indication every min or less in games, now thats completely fixed with this patch.

lukasz1992, 2024-11-26:
> This patch def does some good thing, before i had intermittent packet loss indication every min or less in games, now thats completely fixed with this patch.

I tried this patch, and speed dropped 2x times with inactive WED.

kloon15, 2024-11-26:
> I tried this patch, and speed dropped 2x times with inactive WED.

I dont notice a speed difference with WED enabled.

oxavelar, 2024-12-07:
Below client has left the house, but the MT6000 still sees/tracks it with a -92/-92 RSSI, ugh

[image]

Using a pretty recent OpenWrt SNAPSHOT, r28242, with:

```
mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20240823160721
mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20240823160840
```

Stressing roamings with DAWN and or disconnects by walking of bounds seem to trigger that odd condition.

I might try the Cron job. Since this is affecting my mesh network as batctl ends with nodes with 0.3 crawling link-speeds.

LearZhou, 2024-12-08:
Observed similar AP-STA-POLL-OK logs with my Flint 2 on 2.4G WiFi.

oxavelar, 2024-12-08:
> A dirty temp fix. Tested, works for me. Do not know if there is any side effect.
> 
> Run this script every minute via cron. 
> 
> It will "disconnect" all clients that have a very very low signal strength (should be the clients that have already left the wifi coverage but still buggy as "associated".).

I have adapted your solution and started using it to workaround this for my case too.

<script src="https://gist.github.com/oxavelar/5be72dc2b08f7743c0b46c22f4f7a01f.js"></script>

This version can be added under init / rc scripts since it spawns a subshell on boot that keeps checking for the condition every N seconds.

Another slight change is there is no need to set a threshold, it instead considers that if the signal is lower than the noise floor and 4000+ ms or more has passed from the last report.

We understand this is just a temporary workaround while we wait for the real solution, and that MTK ref from losing the ACK on AX chips does ring a bell.

Djfe, 2025-02-20:
the MTK patch was kind of merged a couple of days ago (with some changes)
https://github.com/openwrt/mt76/commit/e5fef138524e63314cb96ff8314048d175294e95

oxavelar, 2025-04-18:
I am on SNAPSHOT and the bug is fixed for me. I have removed the workaround in my init.rc!

Looking stabler and better! Promising :-)

Termin0tor, 2025-04-23:
The good news: There is no more freezing of the 5GHz WiFi. The bad news: Sometimes the transmission speed is reduced, e.g. from 600 Mbit to 200Mbit but it can be restored by reconnecting to the 5 GHz WiFi network.

IrineSistiana, 2025-04-24:
Yeah, I'm trying to double check this for few days. I disabled SMPS in the AX210 driver before. But I saw the log said "STA-OPMODE-SMPS-MODE-CHANGED off". So I thought it was fine and set it back to auto again. 

It's quite annoy because it's rare and seems random. I would blame my laptop first because it's always buggy when sleeping. Until I saw the comment from Termin0tor. So I share the info too.

I've disabled SMPS again. Let's see if it won't happen, like, in next few days.

lukasz1992, 2025-04-25:
@IrineSistiana @Termin0tor Is your laptop Lenovo and do you have Windows 11?

Termin0tor, 2025-04-28:
@lukasz1992 I have a Lenovo T430 laptop with an AX210 with Windows 10. But do not use it often.

lukasz1992, 2025-04-28:
Possible trying different versions of drivers can help with the issue.

In my case Lenovo powersaving solution was the issue (notebook on battery, even on performance mode)

Termin0tor, 2025-04-28:
> Possible trying different versions of drivers can help with the issue.
> 
> In my case Lenovo powersaving solution was the issue (notebook on battery, even on performance mode)
> 
> EDIT: I have Pixel 7A and issue never occurs on it.

Btw, do you notice a range and speed difference between the stock gl-inet and openwrt 24 firmware?

lukasz1992, 2025-04-28:
I can not check it, but as I remember range was the same or very similar - and weaker than Xiaomi Mi R3G or Asus RT-AX53U.

oxavelar, 2025-09-27:
> The good news: There is no more freezing of the 5GHz WiFi. The bad news: Sometimes the transmission speed is reduced, e.g. from 600 Mbit to 200Mbit (at least with my Pixel 7) but it can be restored by reconnecting to the 5 GHz WiFi network.

You are right, I am still seeing this in SNAPSHOT to this day.

I am still using the following workaround because of that: https://gist.github.com/oxavelar/5be72dc2b08f7743c0b46c22f4f7a01f