openwrt/mt76 #925: Enabling DFS support on MT7613BE

andyboeh, 2024-10-25:
I've got a Yuncore CPE200 outdoor WiFi bridge with MT7613BEN chipset. Unfortunately, it's quite useless with OpenWrt since DFS is not supported - but DFS is a requirement for using it outdoor. 

Hunting down the code for DFS, according to the [commit message](https://github.com/openwrt/mt76/commit/f96cffa03e5789dd4ee9a219a65532841e237ccc), DFS "just" hasn't been tested on the MT7663 (MT7613BE) device.

Is this really the only reason or have there been (failed) tests in the meantime?

@dangowrt seems like you added support for this device, do you have any experience with this?

andyboeh, 2024-10-27:
It's trivial to enable DFS on this chip, but how can I test if it's really working? I could do a comparison with another Mediatek-based router, but I'm not sure if I'm near enough any radar to trigger detection.

ThatMishakov, 2025-05-03:
I've been using Comfast CF-EW72 v2 (MT7613BE) for a few weeks now with a custom build where I've modified the driver to enable the frequencies, but while it does seem to scan for radars and overall works great, I also think I don't live near radars to check it since it, nor my other 3 mt76 (MT7615E, MT7915E, MT7986) routers have detected any radars. Perhaps this could be revisited and tested?

reinerotto, 2025-09-01:
> where I've modified the driver to enable the frequencies, but while it does seem to scan for radars and overall works great,

Hi, can you provide your modified driver for me to try ? I have a Cudy TR1200, with MT7613BEN, running 24.10.2, to test. Actually living within line of sight with a commercial airport, so chances for a real test.

Shine-, 2025-09-01:
Here you go:
<details><summary>Enable DFS channels for MT7613/7663</summary>

```patch
diff --git a/package/kernel/mt76/patches/345-mt7613-enable-dfs.patch b/package/kernel/mt76/patches/345-mt7613-enable-dfs.patch
new file mode 100644
index 0000000000..03fa1001dd
--- /dev/null
+++ b/package/kernel/mt76/patches/345-mt7613-enable-dfs.patch
@@ -0,0 +1,31 @@
+--- a/mt7615/init.c
++++ b/mt7615/init.c
+@@ -376,13 +376,9 @@
+ 	hw->sta_data_size = sizeof(struct mt7615_sta);
+ 	hw->vif_data_size = sizeof(struct mt7615_vif);
+ 
+-	if (is_mt7663(&phy->dev->mt76)) {
+-		wiphy->iface_combinations = if_comb;
+-		wiphy->n_iface_combinations = ARRAY_SIZE(if_comb);
+-	} else {
+-		wiphy->iface_combinations = if_comb_radar;
+-		wiphy->n_iface_combinations = ARRAY_SIZE(if_comb_radar);
+-	}
++	wiphy->iface_combinations = if_comb_radar;
++	wiphy->n_iface_combinations = ARRAY_SIZE(if_comb_radar);
++
+ 	wiphy->reg_notifier = mt7615_regd_notifier;
+ 
+ 	wiphy->max_sched_scan_plan_interval =
+--- a/mt7615/mac.c
++++ b/mt7615/mac.c
+@@ -2219,9 +2219,6 @@
+ 	enum mt76_dfs_state dfs_state, prev_state;
+ 	int err;
+ 
+-	if (is_mt7663(&dev->mt76))
+-		return 0;
+-
+ 	prev_state = phy->mt76->dfs_state;
+ 	dfs_state = mt76_phy_dfs_state(phy->mt76);
+ 	if ((chandef->chan->flags & IEEE80211_CHAN_RADAR) &&
```
</details>

For a real test, you should compare against another device that does properly detect radar signals. You can't just assume that living near an airport will make your device see radar.

reinerotto, 2025-09-01:
@Shine- Interim results: 1) Your patch seems to be out-of-date for (my) 24.10.2. I guess, wrong line numbers.
2) I edited your changes into various instances of .../mt7615/init.c and mac.c.
In my case,
build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2025.02.14~e5fef138/mt7615/init.c
and 
build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2025.02.14~e5fef138/mt7615/mac.c
finally seem to be the correct ones.
3) Now I finally can use channel 124:
config wifi-device 'radio1'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option channel '124'
        option htmode 'HT40'
        option disabled '0'
        option country 'DE'

Mon Sep  1 08:32:44 2025 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->DFS
Mon Sep  1 08:32:44 2025 daemon.notice hostapd: phy1-ap0: DFS-CAC-START freq=5620 chan=124 sec_chan=1, width=0, seg0=0, seg1=0, cac_time=60s
Mon Sep  1 08:34:34 2025 daemon.notice hostapd: phy1-ap0: DFS-CAC-COMPLETED success=1 freq=5620 ht_enabled=1 chan_offset=1 chan_width=2 cf1=5630 cf2=0 radar_detected=0
Mon Sep  1 08:34:36 2025 daemon.notice hostapd: phy1-ap0: interface state DFS->ENABLED

Will use it for a while, first to check for neg effects.
May be, you can update your patch for 24.10.2 ?