openwrt/mt76 #927: Support for MT7927

markmann447, 2024-10-31:
Hey I noticed that my motherboard's WiFi 7 antenna isn't working with Linux, so I would like to know who to nag about getting drivers for that device. I'm not sure if I have to come here to github or if I have to nag Mediatek via email. Either way I just wanted to open this up since MT7927 is broken on Linux.

navarro967, 2024-11-22:
I've been trying to cobble something together for at least the Bluetooth on my Asus X870E ProArt using the MT7927. Nothing solid yet but I managed to extract the firmware bins here: https://github.com/navarro967/mt7927

There is also https://github.com/greearb/linux-ct-6.10/commit/1fe0c0394c18fb01d1070d5c092c22627d56bf50#diff-53c42db1505f5a372a01d73aecbd9293ddaadf497d3db059d4b7fc360fc4104dR341

Haven't been able to get it to work with the 6639 firmware but open to working with someone more experienced.

VAllens, 2025-03-22:
Am still waiting this support too.

PartiallyUntyped, 2025-04-08:
It seems that we depend on Mediatek to provide drivers.

Denis11212, 2025-05-14:
Я с вами, друзья! Вроде как MT7927 уже поддерживается? Или нет?

lukasz1992, 2025-05-15:
not supported

isvidler, 2025-05-16:
I also would like this to be supported. Currently stuck with a useless WiFi card.

reg3x, 2025-07-09:
Any idea when will the MT7927 chipset be supported by the Linux kernel?  I'm stucked with the ROG CROSSHAIR X870E HERO motherboard, I'm using Linux kernel 6.14.0-23-generic and still there's no support :-(

anthony-barricelli, 2025-08-27:
found this, putting here for visibility

https://github.com/ehausig/mt7927

marcin-fm, 2025-09-01:
> Maybe someone can make it work: https://github.com/Fede2782/MTK_modules/blob/ff18cd6519b9b5f8a1a013eec667f6ebe90d58a9/connectivity/wlan/core/gen4m/os/linux/hif/pcie/pcie.c#L170

Good find. This would suggest that MT7927 is a variant of MT6639 (which seems to be a mobile chip, so MT76 doesn't support it either).

I did some research yesterday. There is a Windows driver for both MT7925 and MT7927 (downloaded from https://rog-forum.asus.com/t5/downloads-for-motherboards/drivers-amd-mediatek-wifi-bluetooth/td-p/857576). When I extracted it, I found two interesting files: `WLAN/mtkwlan.dat` and `BT/mtkbt.dat`. Thiose files contain a set of firmwares, but it's some proprietary TAR-like format. I reverse-engineered this format and wrote a tool to extract those: https://gist.github.com/marcin-fm/e53840817dea293f4ebe9176578ded9a . After extraction, I found this set of files:

```
.
├── mtkbt
│   ├── BT_RAM_CODE_MT6639_2_1_hdr.bin
│   ├── BT_RAM_CODE_MT7925_1_1_hdr.bin
│   ├── mtkbt_6639.dat
│   ├── mtkbt_7925.dat
│   ├── WIFI_MT6639_PATCH_MCU_2_1_hdr.bin
│   ├── WIFI_MT7925_PATCH_MCU_1_1_hdr.bin
│   ├── WIFI_RAM_CODE_MT6639_2_1.bin
│   └── WIFI_RAM_CODE_MT7925_1_1.bin
├── mtkbt.dat
├── mtkwlan
│   ├── BT_RAM_CODE_MT6639_2_1_hdr.bin
│   ├── BT_RAM_CODE_MT7925_1_1_hdr.bin
│   ├── mtkwl7925_1418_1ss1t.dat
│   ├── mtkwl7925_1418_2_1ss1t.dat
│   ├── mtkwl7925_1418_2.dat
│   ├── mtkwl7925_1418a_1ss1t.dat
│   ├── mtkwl7925_1418a_2_1ss1t.dat
│   ├── mtkwl7925_1418a_2.dat
│   ├── mtkwl7925_1418a.dat
│   ├── mtkwl7925_1418.dat
│   ├── mtkwl7925_1ss1t.dat
│   ├── mtkwl7925_2_1ss1t.dat
│   ├── mtkwl7925_2.dat
│   ├── mtkwl7925.dat
│   ├── mtkwl7927_2_1ss1t.dat
│   ├── mtkwl7927_2.dat
│   ├── mtkwl7927.dat
│   ├── WIFI_MT6639_PATCH_MCU_2_1_hdr.bin
│   ├── WIFI_MT7925_PATCH_MCU_1_1_hdr.bin
│   ├── WIFI_RAM_CODE_MT6639_2_1.bin
│   └── WIFI_RAM_CODE_MT7925_1_1.bin
├── mtkwlan.dat
└── unmtk.rb
```

In this set, there are firmwares (as in, similar to ones present in lib/firmware/mediatek/mt7925) for both MT7925 and MT6639 (which is likely another name for MT7927). Perhaps this would be useful to someone.

adityadas8888, 2025-10-19:
Any other avenues where we can push to get this supported?

JoshuaJakowlew, 2025-10-29:
Very sad that it's impossible to use wifi on a new high end motherboard. USB dongles would work, but it doesn't solve original problem

nvaert1986, 2025-10-31:
> Dear all, I think I have made hardware initialization work according to the Android driver implementation. I'm not so sure since I didn't try to write a driver before, and some code was written by AI (Qwen and Claude).
> 
> Please take a look https://github.com/zouyonghao/mt7927.git.
> 
> It gets stuck at loading firmware
> 
> ```
> [5212379.692271] mt7925e 0000:08:00.0: MT7927 detected, initializing CBInfra remap registers
> [5212379.692283] mt7925e 0000:08:00.0: MT7927: CBTOP_PCIE_REMAP_WF    = 0x74037001
> [5212379.692286] mt7925e 0000:08:00.0: MT7927: CBTOP_PCIE_REMAP_WF_BT = 0x70007000
> [5212379.692287] mt7925e 0000:08:00.0: MT7927: Setting PCIE2AP remap
> [5212379.692289] mt7925e 0000:08:00.0: MT7927: PCIE2AP_REMAP_WF_1_BA = 0x18051803
> [5212379.692313] mt7925e 0000:08:00.0: ASIC revision: 720072
> [5212379.743623] mt7925e 0000:08:00.0: MT7927: Starting MCU initialization sequence
> [5212379.746563] mt7925e 0000:08:00.0: MT7927: CONN_INFRA VERSION = 0x03010002 - OK
> [5212379.746565] mt7925e 0000:08:00.0: MT7927: Performing WF/BT subsystem reset
> [5212379.809570] mt7925e 0000:08:00.0: MT7927: WF/BT subsystem reset completed
> [5212379.809576] mt7925e 0000:08:00.0: MT7927: Reinitializing WPDMA after subsystem reset
> [5212379.811569] mt7925e 0000:08:00.0: MT7927: Waiting for MCU IDLE state
> [5212379.811576] mt7925e 0000:08:00.0: MT7927: MCU IDLE (0x00001d1e)
> [5212379.811578] mt7925e 0000:08:00.0: MT7927: Hardware initialization completed
> [5212379.820575] mt7925e 0000:08:00.0: MT7927: Loading firmware
> [5212382.861819] mt7925e 0000:08:00.0: Message 00000010 (seq 1) timeout
> [5212382.861830] mt7925e 0000:08:00.0: Failed to get patch semaphore
> ```

That's a good start. In the code there's a case with a deviation in timers for firmware loading. Perhaps the 7927 might need that too.

zouyonghao, 2025-11-01:
Does https://github.com/Fede2782/MTK_modules/blob/bsp-rodin-v-oss/connectivity/bt/linux_v2/chip/btmtk_chip_6639.c help?

dubhater, 2025-11-01:
How hard can it be? 😂

nvaert1986, 2025-11-02:
> How hard can it be? 😂

Perhaps it's not that hard for you, but I'm not an experienced C programmer with years of experience it any experience in hardware engineering. I'm not even a programmer by job, but a sysadmin just trying to get stuff to work. Python / Perl / Bash / PHP are the languages whether I'm more experienced in.

I know how to make kernel module changes and recompile modules and test, and must say I got a pretty good start where it downloads the firmware onto the card, but it just refuses to pass the wireless test, as it seems it needs an unique sequence to start not by the Bluetooth standard, but by MediaTek their own extensions not yet defined in the btmtk.h header file, which can be seen in btmon. It was not as simple as adding a vid, pid and some header entire l entries as I had hoped. I've tried using the 79xx loader firmware logic too, but that didn't work either. It seems it does this in several sequences.

zouyonghao, 2025-11-06:
That's great! I haven't made much progress on the WLAN recently because I've been busy with other projects. If anyone could take it on, that'd be great.

nvaert1986, 2025-11-06:
> > EDIT: I've uploaded my progress so far: https://github.com/nvaert1986/btmtk-temp/blob/main/btmtk_usb_mt6639.c
> 
> Is this still in a private repo? I couldn't open it.

I just changed it to public.

ccrusher1, 2025-11-06:
Just poking my head in to follow along. I had the Bluetooth working at one point on my arch install. Rocking a Crosshair 870 Extreme, so been desperately trying to figure out what got it working, it how to get it working again.

nvaert1986, 2025-11-07:
> Spend several hours trying to make wlan work but failed...It seems we didn't know many functions in the Android repo, e.g., `wlanSendInitSetQueryCmd`

I've uploaded my snooping logs to: https://github.com/nvaert1986/btmtk-temp/blob/main/snooplog_20251106124407.cfa using MediaTek's own in-driver snooping tool. It only shows commands after the firmware is loaded. This file can be viewed using WireShark.

nvaert1986, 2025-11-07:
Let me also share the firmware protocol analysis (95% generated by AI): https://github.com/nvaert1986/btmtk-temp/blob/main/MT6639%20Firmware%20Protocol%20Analysis.txt and driver analysis (95% generated by AI): https://github.com/nvaert1986/btmtk-temp/blob/main/MT6639%20Driver%20Analysis.txt

hmtheboy154, 2025-11-11:
> Let me also share the firmware protocol analysis (95% generated by AI): https://github.com/nvaert1986/btmtk-temp/blob/main/MT6639%20Firmware%20Protocol%20Analysis.txt and driver analysis (95% generated by AI): https://github.com/nvaert1986/btmtk-temp/blob/main/MT6639%20Driver%20Analysis.txt
> 
> The Microsoft Windows initialization sequence (if understood correctly): https://github.com/nvaert1986/btmtk-temp/blob/main/MTK_BT_INIT_SEQUENCE.md

May I ask, the driver you made here is AI generated or edit from btmtk ?

marcin-fm, 2025-12-07:
There is one more thing, which may be of interest to you, but may as well be a fluke. I have a BPI-R4 with BPI-R4-NIC-BE14. This card contains an MT7995 chip (which is mostly MT7996, but has a 2-3-3 antenna setup instead of 4-4-4).

When executing lspci -vnn, I get the following output:

```
root@Bpir4:~# lspci -vnn | grep -B1 Subsys
0000:01:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7990]
        Subsystem: MEDIATEK Corp. Device [14c3:6639]
--
0001:01:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7991]
        Subsystem: MEDIATEK Corp. Device [14c3:6639]
```

This card has two miniPCIe connections, so there are two separate PCIe devices acting together as a single card. Note the PCI Subsystem ID of those, it's `6639`. Which kind of baffles me, because mt6639 is mt7927. Can this perhaps mean that mt7927 is not a variant of mt7925 but actually of mt7995?

oklxd45, 2025-12-18:
> 还有一件事，可能你会感兴趣，但也可能只是偶然。我有一台BPI-R4，配BPI-R4-NIC-BE14。该卡内置MT7995芯片（主要为MT7996，但采用2-3-3天线而非4-4-4）。
> 
> 执行lspci -vnn时，我得到以下输出：
> 
> ```
> root@Bpir4:~# lspci -vnn | grep -B1 Subsys
> 0000:01:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7990]
>         Subsystem: MEDIATEK Corp. Device [14c3:6639]
> --
> 0001:01:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7991]
>         Subsystem: MEDIATEK Corp. Device [14c3:6639]
> ```
> 
> 这张卡有两个miniPCIe接口，所以有两个独立的PCIe设备作为一张卡一起工作。注意这些 PCI 子系统 ID，是 。这让我有点困惑，因为MT6639其实是MT7927。这是否意味着MT7927并非MT7925的一个变体，而是MT7995的变体？`6639`

bpi-R4+NIC-BE14
root@ImmortalWrt:~# lspci -vnn
0000:00:00.0 PCI bridge [0604]: MEDIATEK Corp. Device [14c3:7988] (rev 01) (prog-if 00 [Normal decode])
	Device tree node: /sys/firmware/devicetree/base/pcie@11300000/pcie@0,0
	Flags: bus master, fast devsel, latency 0
	Memory at 30500000 (64-bit, non-prefetchable) [size=32K]
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: [disabled] [32-bit]
	Memory behind bridge: 30400000-304fffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: 30200000-303fffff [size=2M] [32-bit]
	Capabilities: [80] Express Root Port (Slot-), MSI 00
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting
	Capabilities: [300] Secondary PCI Express
lspci: Unable to load libkmod resources: error -12

0000:01:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7990]
	Subsystem: MEDIATEK Corp. Device [14c3:6639]
	Device tree node: /sys/firmware/devicetree/base/pcie@11300000/pcie@0,0/mt7996@0,0
	Flags: bus master, fast devsel, latency 0, IRQ 124
	Memory at 30200000 (64-bit, prefetchable) [size=2M]
	Memory at 30400000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: [80] Express Endpoint, MSI 00
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108] Latency Tolerance Reporting
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting
	Capabilities: [300] Secondary PCI Express
	Kernel driver in use: mt7990

0001:00:00.0 PCI bridge [0604]: MEDIATEK Corp. Device [14c3:7988] (rev 01) (prog-if 00 [Normal decode])
	Device tree node: /sys/firmware/devicetree/base/pcie@11310000/pcie@0,0
	Flags: bus master, fast devsel, latency 0
	Memory at 38500000 (64-bit, non-prefetchable) [size=32K]
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: [disabled] [32-bit]
	Memory behind bridge: 38400000-384fffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: 38200000-383fffff [size=2M] [32-bit]
	Capabilities: [80] Express Root Port (Slot-), MSI 00
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting
	Capabilities: [300] Secondary PCI Express

0001:01:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7991]
	Subsystem: MEDIATEK Corp. Device [14c3:6639]
	Flags: bus master, fast devsel, latency 0, IRQ 125
	Memory at 38200000 (64-bit, prefetchable) [size=2M]
	Memory at 38400000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: [80] Express Endpoint, MSI 00
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108] Latency Tolerance Reporting
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting
	Capabilities: [300] Secondary PCI Express
	Kernel driver in use: mt7991

0004:00:00.0 Unclassified device [0002]: MEDIATEK Corp. Device [14c3:7981] (prog-if 80)
	Subsystem: Device [7622:14c3]
	Flags: bus master, fast devsel, latency 0, IRQ 121
	Memory at 18000000 (64-bit, non-prefetchable) [size=16M]
	Memory at <unassigned> (32-bit, non-prefetchable) [size=16]
	Memory at <unassigned> (32-bit, non-prefetchable) [size=16]
	Memory at <unassigned> (32-bit, non-prefetchable) [size=16]
	Memory at <unassigned> (32-bit, non-prefetchable) [size=16]
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [78] Power Management version 3
	Capabilities: [80] Express Endpoint, MSI 00

dubhater, 2026-02-12:
Why don't you just ask it to write a driver?

Loong0x00, 2026-02-12:
Yes, that's exactly what I did, and then I kept getting stuck on DMA issues. The community won't be able to create a viable alternative until this problem is solved. I'm planning to buy a Claude and try again.

Loong0x00, 2026-02-12:
$20 per month, more Opus credit requires $200.

Loong0x00, 2026-02-12:
I considered KVM, but GPT told me that breakpoint capture is better for bare-metal testing, but that's quite cumbersome.

dubhater, 2026-02-12:
Hmm, you should pay me instead. 🙂 I can probably do this.

pshirshov, 2026-02-13:
> > Yes, that's exactly what I did, and then I kept getting stuck on DMA issues. The community won't be able to create a viable alternative until this problem is solved. I'm planning to buy a Claude and try again.
> 
> If you actually manage to vibe-code this driver, I will be extraordinarily impressed!

Considering my experiments - that's completely doable but expensive/will take a lot of time.

Codex is more capable at the moment.

saboooor, 2026-02-13:
we need a savior at this point the official drivers arent gonna be released for another year or two 💔😭 i lowkey gave up on this issue and just bought a qcncm865

dubhater, 2026-02-13:
@markmann447 At least enough to buy the card, because I don't have it. But I wouldn't be opposed to 20 euros from everyone who left a thumbs up on this issue.

clemenscodes, 2026-02-14:
> > https://gitlab.com/jfmarliere/linux-driver-mediatek-mt7927-bluetooth
> > Might be of relevance
> 
> Does this work?

Yes, it works! Based on this, I wrote a [NixOS module](https://github.com/clemenscodes/linux-mediatek-mt6639-bluetooth-kernel-module) that patches the current kernels bluetooth driver to support this chip. I had to adjust the Python firmware extraction script though. Using this, I am now finally able to use the Bluetooth chip of my motherboard. The single patch thats needed to get it working can be found [here](https://github.com/clemenscodes/linux-mediatek-mt6639-bluetooth-kernel-module/blob/main/patches/mt6639-bt-6.19.patch).

[image]

syabro, 2026-02-14:
> > https://gitlab.com/jfmarliere/linux-driver-mediatek-mt7927-bluetooth
> > Might be of relevance
> 
> Does this work?

Yup, writing using it from my bt keyboard :)

Loong0x00, 2026-02-15:
我觉得我可能取得了一些新的进展。我的研究完全是用 Claude Code 完成的，以下是所有分析日志。希望不会有人因为我反编译 Windows 版代码而起诉我。这些文件很多可能都没用。我对驱动程序一窍不通，所以所有内容都是从 Claude Code 生成的。如果你有什么想法，可以参考一下。我基于一个项目进行了进一步的研究，但详情请阅读项目描述：[https://github.com/Loong0x00/mt7927]
这个项目比较混乱；我为了开发塞进去了一堆其他文件，不过我让 Claude 解释了它们。感兴趣的人可以看看，包括我已经发现的一些寄存器设置。

seebeen, 2026-02-15:
If anyone can assist with setting this up on fedora 43 - shoot me an email i.am at sib dot ing.
I'm going f*cking nuts.

nvaert1986, 2026-02-16:
I've added a request to add this to mainline: https://bugzilla.kernel.org/show_bug.cgi?id=221096

jetm, 2026-02-21:
@kerberos272 — Thanks for the detailed testing on the Lenovo Legion Pro 7. I've pushed updates based on your findings.

**Important:** You'll need to switch to the new package. The old `btusb-mt7927-dkms` has been replaced by [`mediatek-mt7927-dkms`](https://aur.archlinux.org/packages/mediatek-mt7927-dkms), which includes both BT and WiFi modules. To switch:
```
pacman -R btusb-mt7927-dkms
# then install mediatek-mt7927-dkms from AUR
```
The packages conflict, so the old one must be removed first.

**What changed:**

1. **Lenovo BT USB ID** — Added `0489:e0fa` to the btusb patch alongside the ASUS `0489:e13a`. No manual patch editing needed anymore.

2. **Additional WiFi PCI IDs** — Added `14c3:6639` and `14c3:0738` (AMD RZ738) from the INF device list you extracted. WiFi still won't work (same register space mismatch), but the driver will bind and show clear error messages for users with those device IDs.

3. **Flexible firmware sourcing** — The package no longer requires the ASUS ZIP specifically. Three options now:
   - Place `BT_RAM_CODE_MT6639_2_1_hdr.bin` directly (e.g. from the Station-Drivers package you found)
   - Place `mtkwlan.dat` from any MediaTek driver package
   - Let it auto-download the ASUS ZIP (existing behavior, still the fallback)

   This means Lenovo users can use the newer December 2025 firmware from Station-Drivers without needing the ASUS driver.

4. **GCC/Clang auto-detection** — Already handled, as you confirmed. No change needed.

Your WiFi analysis matches mine exactly — `ASIC revision: 0000` because the mt7925 driver's HIF_REMAP_L1/L2 registers don't exist on MT6639's bus architecture. The subsystem-level MODE2/MODE3/AMD_DEFAULT distinctions from the INF are interesting but as you noted, those are high-level driver behavior settings, not the low-level init that's blocking us.

The complete device ID list from the INF is useful for future reference. I've added the PCI IDs that represent distinct chips (`0x6639`, `0x0738`); the `0x7927` subsystem variants (Foxconn `105B:E0FD`, etc.) all use the same PCI device ID so they'll bind automatically.

If you get a chance, could you test the new `mediatek-mt7927-dkms` package on the Lenovo and let me know how it goes? I don't have Lenovo hardware to test on, so any issues you hit — build errors, BT not binding, firmware path problems — I can fix from my side if you share the output.

Loong0x00, 2026-02-24:
@marcin-fm Oh, you can just copy my code, as long as you follow the open-source license. Claude's code is terrible, but I know absolutely nothing about programming. I also have some reverse engineering results in one of my repositories that you can use for reference.

Loong0x00, 2026-02-24:
@marcin-fm The real challenge comes after the scanning is complete. Implementing the scan only requires porting the firmware and rewriting the identification code. You can read some debug documentation in my file; the chip's operating logic is very strange. It needs to control rings 0, 2, 4, 6, and 15 simultaneously. I don't understand this; this is just what Claude told me. Specifically, ring 0 handles information, ring 2 handles control, ring 4 handles receiving messages, and I'm not sure about the other two. You can look at my code. The operating logic of the MT7927 is extremely cumbersome; it requires configuring many RX and TX pins to function. So, have Codex read my code and documentation directly, or have him test if mine works. I tested it on my computer and it works. Maybe this will help you. Also, my router is only Wi-Fi 5. So, 6GHz and 320MHz might require separate reverse engineering of Windows or porting. And don't try to port the MT76 and MT6639; I've proven it's a huge pitfall, especially in the communication implementation stage. It's best to focus entirely on reverse engineering Windows. You can directly refer to my project documentation and code (it's very poorly written, all generated by AI).

marcin-fm, 2026-02-24:
Thanks @Loong0x00 . I asked it to look at the whole issue in the past. The aim of my project is a patch against mt76. I will give it your and your agent's work later on (I want for now - short term - to keep both attempts separate - I want to benchmark Codex compared to Claude, even if that comparison is not fully scientific - as we are also differ in our approaches - and also I want it to generate as much independent opinions as possible, because both our agents could have made some mistakes which we could reconcile later on).

Loong0x00, 2026-02-24:
Oh, my agent also made quite a few mistakes. I strictly instructed it in CLAUDE.md not to look at MT76 and MT6639 (which I used as references in the early stages), but the model always went astray when it was working on its own. Codex's analysis and reverse engineering are particularly good. My method was to create multiple sub-agents using Claude code to call Codex. This is a good approach, but I find Codex's CLI difficult to use. Using this method for evaluation might not be very valuable. I don't know programming, so I'm just correcting Claude code according to my own ideas. I also need to rely on the web versions of Gemini and GROK to find possible resources.

Loong0x00, 2026-02-24:
Good luck, I hope you succeed. I don't understand this at all, so my code is probably very bad.

marcin-fm, 2026-02-24:
Thanks. Me neither, I'm a programmer but I have little to none experience in kernel things. I could give you (and your agent) access to said laptop later on, but we'd have to synchronize. You are much further ahead with your approach.

Loong0x00, 2026-02-24:
No need. I have an X870E HERO motherboard. I hate that ASUS used a network card with insufficient support. I'll just use my own computer for testing. Good luck!

nvaert1986, 2026-02-24:
> Why don't you just experiment on a computer with a wired network and this network card? My X870E HERO is like that; even laptops should have an RJ45 port, right?

Ask Claude to compare your current kernel code to your WiFi version of the driver. Ask how close the code has resemblances and ask if it can rewrite your driver into a patch on top of the current driver reusing as much as existing kernel code where possible and ask him to tell you what the differences are prior to writing the patch and inspect them. This will make the changes smaller. Ask to make the driver work and stable but with the minimal amount of changes required. This should make the driver more stable and reuse already existing and tested kernel code.

Loong0x00, 2026-02-24:
The reply was generated by claude.
Sure, I've added a detailed architecture document to the repo that covers exactly this:

  👉 docs/ARCHITECTURE.md — https://github.com/Loong0x00/mt7927

  It explains what standard kernel APIs are reused, what had to be implemented from scratch, and specifically why mt76 can't simply be patched (register space, DMA layout, firmware protocol — all fundamentally different). Should answer your questions in
  detail.

jetm, 2026-02-24:
**Update: MT7927 WiFi working on both 2.4GHz and 5GHz - via mt7925e patches**

I've gotten MT7927 WiFi fully working by patching the existing in-kernel mt7925e driver rather than building a standalone module. Both bands are operational:

**2.4GHz:**
```
$ iw dev wlan0 link
Connected to xx:xx:xx:xx:xx:xx (on wlan0)
        SSID: MyNetwork
        freq: 2422 (2.4 GHz)
        signal: -38 dBm
        tx bitrate: 258.1 MBit/s HE-MCS 10 HE-NSS 2 HE-GI 0.8
```

**5GHz:**
```
$ iw dev wlan0 link
Connected to xx:xx:xx:xx:xx:xx (on wlan0)
        SSID: MyNetwork-5G
        freq: 5200 (5 GHz)
        signal: -22 dBm
        tx bitrate: 1200.9 MBit/s HE-MCS 11 HE-NSS 2 HE-GI 0.8 80MHz
```

Hardware: ASUS ROG Crosshair X870E Hero, kernel 6.19.

### Approach

The MT6639 WiFi silicon is architecturally close enough to MT7925 that the existing mt7925e driver can be extended to support it. The total change is ~355 lines across two patches - small enough that it could realistically be proposed for upstream mt76 once stability is proven across more hardware.

### Testing

Available on AUR for Arch Linux:

```bash
pikaur -S mediatek-mt7927-dkms   # or yay/paru
```

After install, reload the driver:

```bash
sudo modprobe -r mt7925e mt7925_common mt792x_lib mt76_connac_lib mt76
sudo modprobe mt7925e

# Verify
dmesg | grep -E 'mt7925e|MT6639'
iw dev
```

The package handles Bluetooth (MT6639 via USB) as well - that side has been working for a while.

For stability testing, there's a non-disruptive monitor script that can run overnight without disrupting your connection: https://gist.github.com/jetm/b680a887778618f98632c8b5d5503c0a

### Acknowledgments

Thank you to @Loong0x00 - your reverse engineering work on the [mt7927 repo](https://github.com/Loong0x00/mt7927) was invaluable. Your documentation of the MT6639 DMA ring layout and initialization sequence saved a huge amount of time and made it possible to understand how the hardware differs from MT7925. The standalone driver proved that this chip can work on Linux, which gave confidence to pursue the mt76-patch approach.

Also thanks to @marcin-fm for the firmware extraction tooling earlier in this thread - that was an important piece of the puzzle.

@kerberos272 - I saw your report about shutdown hangs on the Lenovo Legion Pro 7. That was the BT-only version of the package. I'd appreciate it if you could test the WiFi-enabled version when it's published and report whether the shutdown issue persists. If you can grab `dmesg` output during the hang (via serial console or SSH), that would help narrow it down.

I'll share a detailed technical writeup separately. For now, the focus is on getting more testers - especially on hardware other than the X870E Hero.

Hi-Angel, 2026-02-24:
Thank you very much! I [shared the link on another relevant thread](https://github.com/morrownr/USB-WiFi/issues/517#issuecomment-3954189726), so hopefully the driver will get more testing.

Loong0x00, 2026-02-24:
Scale of impact: MT7927 is not a niche chip

  To put this in perspective, here's a conservative estimate of MT7927 units in consumer hands as of early 2026.

  Desktop motherboards (largest share):
  AMD sold ~18M desktop Ryzen CPUs in 2025 (based on ~59M desktop PC shipments × ~32% AMD share). Of those, ~60-70% are AM5 platform → ~11-13M AM5 CPUs. New AM5 motherboard sales are estimated at 5-8M units, with mid-to-high-end (X870/X870E/B850
  flagships) at roughly 15-25% → 750K-2M boards. MT7927 appears on confirmed flagship models from all three major vendors:

  - ASUS: ROG Crosshair X870E Hero, Strix X870E-E/F/I, ProArt X870E-Creator
  - Gigabyte: X870E AORUS Master and select AORUS models
  - MSI: MPG X870I Edge Ti Evo and select high-end boards

  Conservative 20-30% MT7927 adoption among these → 150K-600K motherboards.

  Laptops: Lenovo Legion Pro 7/9 (AMD configs) ship with MT7927 as a confirmed option. Conservative estimate: 10K-50K units.

  Retail/aftermarket cards: Cross-platform sales data (Taobao, JD.com, Amazon, eBay) suggest 10K-20K units.

  Conservative total: 200K-400K MT7927 devices in the wild. This number will grow as B850 mid-range boards proliferate through 2026.

  Why this matters for Linux:

  High-end AMD desktop and gaming laptop users — exactly the demographic most likely to run Linux — are disproportionately affected. Many are migrating from Intel platforms (post-13th/14th gen reliability issues) where WiFi worked out of the box with
  iwlwifi. They arrive on AM5, and their WiFi simply doesn't work. This is one of the largest Linux hardware support gaps on current-generation mainstream hardware.

  What we need:

  A kernel developer with mt76 subsystem experience to write a proper mt7927/ sub-driver rather than patching mt7925. The full Windows RE documentation (52 files covering hardware init, register maps, DMA layout, MCU commands, firmware download) is
  publicly available at Loong0x00/mt7927/docs/re. The foundation is there — it just needs someone who can turn it into upstream-quality code.

Loong0x00, 2026-02-24:
I've summarized all of this briefly at the end of the documentation in my project repository. Please read it if needed. The increasing use of the MT7927 network card on AMD platforms is an issue that must be addressed.
https://github.com/Loong0x00/mt7927

syabro, 2026-02-25:
**MT7927 working on Arch Linux with DKMS package**

Got MT7927 WiFi working using [@jetm's DKMS package](https://github.com/jetm/mediatek-mt7927-dkms) v2.1-2.

**Hardware:**
- `14c3:7927` — MediaTek MT7927 802.11be 320MHz 2x2 PCIe (Filogic 380)
- Arch Linux, kernel 6.18.7

**dmesg:**
```
mt7925e 0000:6c:00.0: ASIC revision: 0000
mt7925e 0000:6c:00.0: MT6639 raw CHIPID=0x0000, forcing chip=0x7927
mt7925e 0000:6c:00.0: HW/SW Version: 0x8a108a10, Build Time: 20250606200911a
mt7925e 0000:6c:00.0: WM Firmware Version: ____000000, Build Time: 20250606201037
```

**WiFi status (iwd):**
- Connected on 5GHz, channel 48, 802.11ax
- TX/RX: ~1201 Mbps, MCS 11
- RSSI: -30 dBm

**Speed test (rsync over SSH, 500MB file to local server):**
```
524.29M 100%   89.77MB/s    0:00:05
sent 43 bytes  received 524.42M bytes  80.68M bytes/sec
```
~720 Mbps sustained throughput over WiFi. fast.com shows ~750 Mbps.

Bluetooth also detected (`MT7927: skipping section 5/6` — seems cosmetic).

For anyone stuck — the DKMS package works out of the box on Arch. Just `makepkg -si` and reboot.

clemenscodes, 2026-02-25:
I have also published a [NixOS module](https://github.com/clemenscodes/linux-mt7927/blob/main/flake.nix) which installs both the Bluetooth as well as the WiFi patches and their firmware. I'll try to maintain the patch for new developments in the future. Huge thanks to everyone here!

cmspam, 2026-02-25:
> I have also published a [NixOS module](https://github.com/clemenscodes/linux-mt7927/blob/main/flake.nix) which installs both the Bluetooth as well as the WiFi patches and their firmware. I'll try to maintain the patch for new developments in the future. Huge thanks to everyone here!

https://github.com/cmspam/mt7927-nixos/

You beat me to it by 13 minutes, haha.

Mine tracks jetm's github and is set to update in as automated a fashion as possible, but that also means things can go wrong, and I may not be maintaining it very actively, so I think if clemenscodes is going to be maintaining the patch, it'll end up best to use his.

jetm, 2026-02-25:
@lubnin Thanks for the report. The USB ID `13d3:3588` is an IMC Networks/Azurewave module variant of the MT6639 BT chip - different vendor than the Foxconn `0489:e13a` on most ASUS boards, but same silicon.

I've added `13d3:3588` to the BT patch in v2.1-6. If you want to try the full DKMS package (which handles both WiFi and BT), it's at https://github.com/jetm/mediatek-mt7927-dkms. For Fedora, you'll need to adapt the patches manually or check if @giosal's Ubuntu fork at https://github.com/giosal/mediatek-mt7927-dkms works for you.

Regarding Secure Boot: the DKMS-built modules won't be signed by default. You'll need to either enroll a MOK key for DKMS signing or disable Secure Boot for the out-of-tree modules to load.

Thex-Thex, 2026-02-25:
Hello, the installation didn't worked for me i got error while installing the AUR package with yay
```
==> Création du paquet mediatek-mt7927-dkms 2.1-6 (mer 25 fév 2026 22:32:19)
==> Récupération des sources…
  -> mt6639-bt-6.19.patch trouvé
  -> mt7902-wifi-6.19.patch trouvé
  -> mt6639-wifi-init.patch trouvé
  -> mt6639-wifi-dma.patch trouvé
  -> extract_firmware.py trouvé
  -> dkms.conf trouvé
  -> dkms-patchmodule.sh trouvé
==> ATTENTION : Ignore la vérification des signatures PGP des fichiers sources.
==> Validation des fichiers source avec sha256sums…
    mt6639-bt-6.19.patch ... Réussite
    mt7902-wifi-6.19.patch ... Réussite
    mt6639-wifi-init.patch ... Réussite
    mt6639-wifi-dma.patch ... Réussite
    extract_firmware.py ... Réussite
    dkms.conf ... Réussite
    dkms-patchmodule.sh ... Réussite
:: (1/1) Parsing SRCINFO: mediatek-mt7927-dkms
==> Création du paquet mediatek-mt7927-dkms 2.1-6 (mer 25 fév 2026 22:32:20)
==> Vérification des dépendances pour l’exécution…
==> Vérification des dépendances pour la compilation…
==> Récupération des sources…
  -> mt6639-bt-6.19.patch trouvé
  -> mt7902-wifi-6.19.patch trouvé
  -> mt6639-wifi-init.patch trouvé
  -> mt6639-wifi-dma.patch trouvé
  -> extract_firmware.py trouvé
  -> dkms.conf trouvé
  -> dkms-patchmodule.sh trouvé
==> Validation des fichiers source avec sha256sums…
    mt6639-bt-6.19.patch ... Réussite
    mt7902-wifi-6.19.patch ... Réussite
    mt6639-wifi-init.patch ... Réussite
    mt6639-wifi-dma.patch ... Réussite
    extract_firmware.py ... Réussite
    dkms.conf ... Réussite
    dkms-patchmodule.sh ... Réussite
==> Suppression du répertoire $srcdir/ existant…
==> Extraction des sources…
==> Lancement de prepare()…
Verifying DRV_WiFi_MTK_MT7925_MT7927_TP_W11_64_V5603998_20250709R.zip...
/home/thex/.cache/yay/mediatek-mt7927-dkms/DRV_WiFi_MTK_MT7925_MT7927_TP_W11_64_V5603998_20250709R.zip: Réussi
Using driver ZIP: DRV_WiFi_MTK_MT7925_MT7927_TP_W11_64_V5603998_20250709R.zip
==> Les sources sont prêtes.
==> Création du paquet mediatek-mt7927-dkms 2.1-6 (mer 25 fév 2026 22:32:21)
==> Vérification des dépendances pour l’exécution…
==> Vérification des dépendances pour la compilation…
==> ATTENTION : Utilisation de l’arbre $srcdir/ existant
==> Lancement de build()…
Extracted BT_RAM_CODE_MT6639_2_1_hdr.bin: 688341 bytes -> /home/thex/.cache/yay/mediatek-mt7927-dkms/src/firmware/BT_RAM_CODE_MT6639_2_1_hdr.bin
Extracted WIFI_MT6639_PATCH_MCU_2_1_hdr.bin: 299296 bytes -> /home/thex/.cache/yay/mediatek-mt7927-dkms/src/firmware/WIFI_MT6639_PATCH_MCU_2_1_hdr.bin
Extracted WIFI_RAM_CODE_MT6639_2_1.bin: 1596848 bytes -> /home/thex/.cache/yay/mediatek-mt7927-dkms/src/firmware/WIFI_RAM_CODE_MT6639_2_1.bin
Downloading mt76 source for kernel v6.19.3...
  mt76.h
  mt76_connac.h
  mt76_connac2_mac.h
  mt76_connac3_mac.h
  mt76_connac_mcu.h
  mt76_connac_mcu.c
  mt76_connac_mac.c
  mt76_connac3_mac.c
  mmio.c
  util.c
  util.h
  trace.c
  trace.h
  dma.c
[... 210 lines cut ...]
==> WARNING: Possibly missing firmware for module: 'xhci_pci_renesas'
  -> Running build hook: [keymap]
  -> Running build hook: [sd-vconsole]
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: 'qed'
==> WARNING: Possibly missing firmware for module: 'qla2xxx'
==> WARNING: Possibly missing firmware for module: 'aic94xx'
==> WARNING: Possibly missing firmware for module: 'bfa'
==> WARNING: Possibly missing firmware for module: 'wd719x'
==> WARNING: Possibly missing firmware for module: 'qla1280'
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-linux-fallback.img'
  -> Early uncompressed CPIO image generation successful
==> Initcpio image generation successful

```
i also checked `/var/lib/dkms/mediatek-mt7927/2.1/build/make.log`

[make.log](https://github.com/user-attachments/files/25558941/make.log)

idk if this is a me problem or if it could come from another place 

i'm using a **Z790 AORUS MASTER X 1.1**

jetm, 2026-02-25:
@Loong0x00 This was already fixed in v2.1-6 - see https://github.com/openwrt/mt76/issues/927#issuecomment-3961991532. The `13d3:3588` ID was added after lubnin's report on the ASUS X870E-E.

marcin-fm, 2026-02-26:
Since my chip is kind of exotic, it also has a different ID on USB from ones in the patch:

Bus 002 Device 003: ID 0489:e116 Foxconn / Hon Hai Wireless_Device

marcin-fm, 2026-02-26:
@jetm It does work on Windows, yes.

From my agent:

```patch
  diff --git a/drivers/net/wireless/mediatek/mt76/mt7925/mac.c b/drivers/net/wireless/mediatek/mt76/mt7925/mac.c
  index 871b67101976..a2070ef50d36 100644
  --- a/drivers/net/wireless/mediatek/mt76/mt7925/mac.c
  +++ b/drivers/net/wireless/mediatek/mt76/mt7925/mac.c
  @@ -339,6 +339,11 @@ mt7925_mac_fill_rx_rate(struct mt792x_dev *dev,
        case IEEE80211_STA_RX_BW_160:
                status->bw = RATE_INFO_BW_160;
                break;
  +     /* RXV can report 320 in two positions. */
  +     case IEEE80211_STA_RX_BW_320:
  +     case IEEE80211_STA_RX_BW_320 + 1:
  +             status->bw = RATE_INFO_BW_320;
  +             break;
        default:
                return -EINVAL;
        }
  @@ -991,6 +996,10 @@ mt7925_mac_add_txs_skb(struct mt792x_dev *dev, struct mt76_wcid *wcid,
        stats->tx_mode[mode]++;

        switch (FIELD_GET(MT_TXS0_BW, txs)) {
  +     case IEEE80211_STA_RX_BW_320:
  +             rate.bw = RATE_INFO_BW_320;
  +             stats->tx_bw[4]++;
  +             break;
        case IEEE80211_STA_RX_BW_160:
                rate.bw = RATE_INFO_BW_160;
                stats->tx_bw[3]++;
```

  ### MT7927 320 MHz retest (iteration-1 only)

  I retested with only the iteration-1 change set active (no iteration-2/3 logic), on latest Linus-based kernel with current firmware in a Fedora VM using PCIe passthrough. The test AP profile was 6 GHz, WPA3-SAE, 320 MHz channel width, and all results below are from `wlp7s0`.

  Methodology was strict and route-isolated: connect to the 6 GHz 320 profile, remove competing default routes, pin datapath checks to the Wi-Fi interface, then run repeated `iw dev wlp7s0 link` sampling alongside short `ping` and `iperf3` runs. I also performed a reconnect cycle and checked kernel logs for auth/
  assoc/4-way behavior.

  `iw dev wlp7s0 link` shows sustained EHT 320 operation after association. Typical samples report RX/TX rates in EHT MCS10/MCS11 with 2 spatial streams at 320 MHz, with signal in the expected range for this setup. Across a 40-sample capture window, rates remained mostly stable at 320-MHz EHT values. I did observe
  rare one-sample drops to `6.0 MBit/s` on one direction, followed by immediate recovery to high EHT rates.

  Datapath is working in this setup. In short runs, ping had 0% loss, and throughput reached roughly ~1.09 Gbit/s uplink (0 retransmits in that run) and ~0.98 Gbit/s downlink (non-zero retransmits). A quick post-reconnect retest still carried traffic (uplink remained high; downlink showed higher retransmit pressure
  in that sample window).

  Residual issue remains: intermittent WPA3 `4WAY_HANDSHAKE_TIMEOUT` still appears in reconnect cycles, even though the link can recover and pass traffic afterward. So the current conclusion is: iteration 1 is high-impact and appears necessary for usable 320-MHz datapath behavior, but it is not sufficient to fully
  solve handshake/reconnect stability.

-----

Connected to xx:xx:xx:xx:xx:xx (on wlp7s0)
	SSID: xxxxxxx
	freq: 5975.0
	RX: 6167 bytes (51 packets)
	TX: 6464 bytes (60 packets)
	signal: -43 dBm
	rx bitrate: 4803.8 MBit/s 320MHz EHT-MCS 11 EHT-NSS 2 EHT-GI 0
	tx bitrate: 4803.8 MBit/s 320MHz EHT-MCS 11 EHT-NSS 2 EHT-GI 0
	bss flags: short-slot-time
	dtim period: 2
	beacon int: 100

------

The connection speeds are not great, but they likely stem from a fact it's an old laptop with 2.5Gbps PCIe link:

```
		LnkSta:	Speed 2.5GT/s (downgraded), Width x1
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
```

Loong0x00, 2026-02-26:
@jetm I am from China and lack 6GHz support; please excuse the fact that this document is written in my native language, Chinese. This is a possible analysis of the problem.

[320mhz_handshake_timeout_fix.md](https://github.com/user-attachments/files/25562386/320mhz_handshake_timeout_fix.md)

jetm, 2026-02-26:
@Loong0x00 Thanks for the analysis. The 8023 fast path / EAPOL theory is a reasonable hypothesis - `__ieee80211_rx_handle_8023()` does require `fast_rx != NULL`, which is only set after AUTHORIZED state.

A few notes:

1. marcin-fm is actively isolating whether the timeout is bound to SAE or 6GHz - he confirmed it doesn't happen on mt7925, only mt7927. @lmcarneiro is also seeing it reliably (~4s after initial connection on reconnect). So we have two independent data points now.

2. The proposed `cfg80211_rx_control_port()` interception would need to go in mt76's RX path, not mac80211 itself. The signature and available parameters vary significantly across kernel versions (6.1 vs 6.12 vs 6.19), so any implementation would need version guards.

3. The good news is that marcin-fm's RXV/TXS decode fix (now in v2.1-9) resolved the main 320MHz datapath issue - the link carries traffic at full speed. The handshake timeout is a secondary issue that doesn't prevent usage (link recovers after the timeout).

If you have ideas on how to test the EAPOL theory without 6GHz hardware (maybe reproducible on 5GHz with specific AP configurations?), that would help narrow it down.

Package updated to v2.1-9 with the 320MHz fix - please test and report back any new findings. I'm heading to sleep - will continue tomorrow.

[47 more replies]