openwrt/mt76 #930: Linksys e1700

589userr589, 2024-11-07:
Wifi doesn't work, it works as a jammer

bmaehr, 2025-01-26:
@589userr589 Have you been able to solve the issue? I see openwrt in windows and on my galaxy but are not able to connect to the network.

Has anyone working WIFI with e1700? If not then the the #7959 should been updated and said that wireless is NOT working and not instable.

589userr589, 2025-01-28:
No, unfortunately I was unable to connect to the wifi network.

As an additional possible problem, I suspect that there may be a problem with the current 24.10.0-rc6 version with WAN operation or connection. The log shows that it takes a while to connect and repeats the routine several times. I'm not sure, I don't think I encountered this problem in previous versions. Could the problem be in the 64/8Mb 580Mhz capacity?