openwrt/mt76 #933: [Snapshot] AX3000T no CPU clock management

erayrafet, 2024-12-01:
I have the Xiaomi AX3000T and there is no CPU clock management. The CPU directories in /dev do not exist as btop doesn't display clock speeds either. Because of that, the CPU is always hovering at about 60C which is quite hot on the outside as well. Seems like cpufreq isn't enabled.

https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/1782?u=erayrafet

https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/1783?u=erayrafet

[image]
[image]

Bartvz, 2024-12-01:
This is the driver for the wireless chips so this does not look like the right place to me.

erayrafet, 2024-12-01:
You're right. I should've checked. I assumed it was geared towards newer Mediatek SoCs.

closed 2024-12-01 by erayrafet