openwrt/mt76 #934: mt7615 crash log

quarkysg, 2024-12-02:
Router: Linksys E8450

Running own custom master branch build (at commit 92ca322dd1f48158b8829fec59319a12e4ae4295)

Router crashed with the following captured in ramoops pstore:

```
<1>[235082.785489] Unable to handle kernel read from unreadable memory at virtual address 0000000000000001
<1>[235082.794656] Mem abort info:
<1>[235082.797527]   ESR = 0x0000000096000005
<1>[235082.801368]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[235082.806759]   SET = 0, FnV = 0
<1>[235082.809890]   EA = 0, S1PTW = 0
<1>[235082.813113]   FSC = 0x05: level 1 translation fault
<1>[235082.818068] Data abort info:
<1>[235082.821029]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
<1>[235082.826592]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
<1>[235082.831725]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
<1>[235082.837115] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000046658000
<1>[235082.843638] [0000000000000001] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
<0>[235082.852424] Internal error: Oops: 0000000096000005 [#1] SMP
<7>[235082.858076] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_inet nf_flow_table nf_conntrack mt7915e(O) mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT ebtable_nat ebtable_filter ebtable_broute chacha_neon cfg80211(O) xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet slhc poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_compat nf_tables nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ipt_ECN ipheth ip_tables ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype
<7>[235082.858295]  ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 compat(O) cls_flower act_vlan cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_gre ip_gre gre sit ipip ip6_tunnel tunnel6 tunnel4 ip_tunnel tun ovpn_dco_v2(O) udp_tunnel ip6_udp_tunnel sha512_arm64 seqiv geniv usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug(O) usbcore usb_common mii
<7>[235083.026279] CPU: 0 PID: 1089 Comm: mt76-tx phy0 Tainted: G S         O       6.6.56 #0
<7>[235083.034277] Hardware name: Linksys E8450 (UBI) (DT)
<7>[235083.039233] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[235083.046274] pc : mt7615_mac_set_rates+0x44/0x3cc [mt7615_common]
<7>[235083.052381] lr : mt7615_tx_prepare_skb+0x290/0x300 [mt7615e]
<7>[235083.058129] sp : ffffffc0810438e0
<7>[235083.061520] x29: ffffffc0810438e0 x28: fffffffe00000000 x27: 000000000000007f
<7>[235083.068740] x26: 0000000000000000 x25: ffffffc0792ba13c x24: ffffff8003dc2338
<7>[235083.075959] x23: ffffff800267ae30 x22: ffffff8003dc1fe8 x21: ffffff8003ded068
<7>[235083.083177] x20: ffffff8003dc1fe8 x19: ffffff8003de1fe0 x18: 0000000000000000
<7>[235083.090395] x17: ffffffbf9f30d000 x16: ffffffc080000000 x15: ffffff801fea2b80
<7>[235083.097614] x14: ffffffffffffffff x13: 0000000000000008 x12: 0101010101010101
<7>[235083.104833] x11: 00000000000002ad x10: 00000000000008a0 x9 : 0000000000000000
<7>[235083.112051] x8 : ffffffc081043cb0 x7 : 0000000000000000 x6 : ffffffc079139a78
<7>[235083.119268] x5 : ffffffc081043aa0 x4 : 0000000000000000 x3 : ffffff8003dc2338
<7>[235083.126486] x2 : ffffff800267ae30 x1 : 0000000000000000 x0 : 0000000000000000
<7>[235083.133705] Call trace:
<7>[235083.136228]  mt7615_mac_set_rates+0x44/0x3cc [mt7615_common]
<7>[235083.141978]  mt7615_tx_prepare_skb+0x290/0x300 [mt7615e]
<7>[235083.147379]  mt76_dma_cleanup+0x900/0xa44 [mt76]
<7>[235083.152091]  mt76_tx_check_agg_ssn+0xa0/0x114 [mt76]
<7>[235083.157147]  __mt76_tx_complete_skb+0x3e4/0x738 [mt76]
<7>[235083.162377]  __mt76_tx_complete_skb+0x6f8/0x738 [mt76]
<7>[235083.167606]  mt76_tx_worker_run+0x2c/0x60 [mt76]
<7>[235083.172315]  mt7615_tx_worker+0xa0/0x190 [mt7615_common]
<7>[235083.177718]  __mt76_worker_fn+0x8c/0xdc [mt76]
<7>[235083.182253]  kthread+0xd8/0xdc
<7>[235083.185392]  ret_from_fork+0x10/0x20
<0>[235083.189052] Code: 394e043a f9438a60 f9419421 b9403800 (39400439) 
<4>[235083.195223] ---[ end trace 0000000000000000 ]---
```

quarkysg, 2024-12-20:
So I used gdb (with this command `l *mt7615_mac_set_rates+0x44`) and I get the following from gdb:

```
(gdb) l *mt7615_mac_set_rates+0x44
0x90c8 is in mt7615_mac_set_rates (/home/quarkysg/projects/firmware/openwrt-mt7622/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/mt76-2024.10.28~c246fa54/mt7615/mac.c:1109).
1104	{
1105		int wcid = sta->wcid.idx, n_rates = sta->n_rates;
1106		struct mt7615_dev *dev = phy->dev;
1107		struct mt7615_rate_desc rd;
1108		u32 w5, w27, addr;
1109		u16 idx = sta->vif->mt76.omac_idx;
1110	
1111		if (!mt76_is_mmio(&dev->mt76)) {
1112			mt7615_mac_queue_rate_update(phy, sta, probe_rate, rates);
1113			return;
```

So it looks like the `sta` variable passed in by the calling function contains a NULL value (either for `vif` or `mt76`) in it's data structured that was used.

Anyone knows how this could have happened and how to check to prevent kernel crash?

lynxthecat, 2025-02-02:
Seems this may also be the reason for the crashes that I've seen too: 

```

root@OpenWrt:~# cat /sys/fs/pstore/dmesg-ramoops-*
Oops#1 Part1
<6>[400158.817830] br-lan: port 1(lan1) entered blocking state
<6>[400158.831153] br-lan: port 1(lan1) entered forwarding state
<6>[400165.660441] br-lan: port 1(lan1) entered disabled state
<6>[400165.660542] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[400168.582429] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[400168.582485] br-lan: port 1(lan1) entered blocking state
<6>[400168.595792] br-lan: port 1(lan1) entered forwarding state
<6>[400169.426259] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[400169.431625] br-lan: port 1(lan1) entered disabled state
<6>[400172.410793] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[400172.410822] br-lan: port 1(lan1) entered blocking state
<6>[400172.424131] br-lan: port 1(lan1) entered forwarding state
<6>[411604.437978] br-lan: port 1(lan1) entered disabled state
<6>[411604.438082] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[411608.540791] mt7530-mdio mdio-bus:00 lan1: Link is Up - 10Mbps/Full - flow control off
<6>[411608.540817] br-lan: port 1(lan1) entered blocking state
<6>[411608.554043] br-lan: port 1(lan1) entered forwarding state
<6>[411610.428878] br-lan: port 1(lan1) entered disabled state
<6>[411610.428981] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[411614.805162] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control off
<6>[411614.805192] br-lan: port 1(lan1) entered blocking state
<6>[411614.818325] br-lan: port 1(lan1) entered forwarding state
<6>[411616.653019] br-lan: port 1(lan1) entered disabled state
<6>[411616.653124] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[411619.717705] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[411619.717758] br-lan: port 1(lan1) entered blocking state
<6>[411619.731076] br-lan: port 1(lan1) entered forwarding state
<6>[411631.724981] br-lan: port 1(lan1) entered disabled state
<6>[411631.725092] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[411634.982444] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[411634.982473] br-lan: port 1(lan1) entered blocking state
<6>[411634.995788] br-lan: port 1(lan1) entered forwarding state
<6>[427738.040625] br-lan: port 1(lan1) entered disabled state
<6>[427738.040679] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[427739.647664] mt7530-mdio mdio-bus:00 lan1: Link is Up - 10Mbps/Full - flow control rx/tx
<6>[427739.651544] br-lan: port 1(lan1) entered blocking state
<6>[427739.661123] br-lan: port 1(lan1) entered forwarding state
<6>[427740.795176] br-lan: port 1(lan1) entered disabled state
<6>[427740.795279] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[484817.346306] mt7530-mdio mdio-bus:00 lan1: Link is Up - 10Mbps/Full - flow control off
<6>[484817.346334] br-lan: port 1(lan1) entered blocking state
<6>[484817.359599] br-lan: port 1(lan1) entered forwarding state
<6>[484818.680330] br-lan: port 1(lan1) entered disabled state
<6>[484818.680432] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[484821.636763] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control off
<6>[484821.636791] br-lan: port 1(lan1) entered blocking state
<6>[484821.649944] br-lan: port 1(lan1) entered forwarding state
<6>[484825.146803] br-lan: port 1(lan1) entered disabled state
<6>[484825.146904] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[484828.144451] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[484828.144494] br-lan: port 1(lan1) entered blocking state
<6>[484828.157837] br-lan: port 1(lan1) entered forwarding state
<6>[484831.822973] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[484831.828360] br-lan: port 1(lan1) entered disabled state
<6>[484834.799543] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[484834.807411] br-lan: port 1(lan1) entered blocking state
<6>[484834.812891] br-lan: port 1(lan1) entered forwarding state
<6>[484835.611384] br-lan: port 1(lan1) entered disabled state
<6>[484835.611486] mt7530-mdio mdio-bus:00 lan1: Link is Down
<6>[484838.585493] mt7530-mdio mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[484838.585524] br-lan: port 1(lan1) entered blocking state
<6>[484838.598869] br-lan: port 1(lan1) entered forwarding state
<6>[516001.058626] br-lan: port 1(lan1) entered disabled state
<6>[516001.058729] mt7530-mdio mdio-bus:00 lan1: Link is Down
[... 142 lines cut ...]
<7>[593594.498921] x14: ffffffc008cff2d0 x13: 0000000000000002 x12: 000000000000002e
<7>[593594.506138] x11: ffffffc008b2d4e8 x10: 0000000000000001 x9 : 0000000000000000
<7>[593594.513356] x8 : ffffffc008e0bd00 x7 : 0000000049ec2134 x6 : ffffffc000bdfbd0
<7>[593594.520573] x5 : ffffffc008e0baf0 x4 : 0000000000000000 x3 : ffffff80021960d0
<7>[593594.527789] x2 : ffffff800234de30 x1 : 0000000000000000 x0 : 0000000000000000
<7>[593594.535008] Call trace:
<7>[593594.537531]  0xffffffc000bd22c8 [mt7615_common@0000000099475aa1+0x14000]
<7>[593594.544313]  0xffffffc000bdfe7c [mt7615e@00000000433d51f3+0x5000]
<7>[593594.550487]  0xffffffc000bad350 [mt76@00000000c431e310+0x11000]
<7>[593594.556487]  0xffffffc000bb1fd8 [mt76@00000000c431e310+0x11000]
<7>[593594.562488]  0xffffffc000bb3a34 [mt76@00000000c431e310+0x11000]
<7>[593594.568488]  0xffffffc000bb3ba4 [mt76@00000000c431e310+0x11000]
[... cut ...]
<7>[593594.581270]  0xffffffc000bab648 [mt76@00000000c431e310+0x11000]
<7>[593594.587270]  0xffffffc008058e8c
<7>[593594.590488]  0xffffffc00801581c
<0>[593594.593711] Code: 394d643a f9437e60 f9418021 b9403800 (39400439)
<4>[593594.599882] ---[ end trace 7b76ca0d3d36e44c ]---
<0>[593594.615583] Kernel panic - not syncing: Oops: Fatal exception in interrupt
<2>[593594.622540] SMP: stopping secondary CPUs
<0>[593594.626544] Kernel Offset: disabled
<0>[593594.630109] CPU features: 0x0,00001000,00000802
<0>[593594.634719] Memory Limit: none
root@OpenWrt:~#
```

hnyman, 2025-02-02:
> at virtual address 0000000000000001

To me it looks like a function is passed some enumeration index 1 as a pointer argument, instead of the actual pointer to the struct about the item 1.

Looking at the source code location identified by quarkysg, the code is 4-5 years old, so likely the changes are in the calling function.

quarkysg, 2025-02-03:
> To me it looks like a function is passed some enumeration index 1 as a pointer argument, instead of the actual pointer to the struct about the item 1.

From gdb's output, the exception occurs here:

`1109		u16 idx = sta->vif->mt76.omac_idx;`

We know that the `sta` variable is OK as it was used a couple of lines above this line.  So issue must be with either the `vif` or `mt76` pointer in the data structure.  It looks to me that `mt76` or `hostapd` likely has a bug that it is not initialising client stations' data structure properly under some obscure scenario, resulting in this kernel panic.