openwrt/mt76 #935: mt7921 / mt7921e can't change tx power and channel

doublep21, 2024-12-06:
i can't change tx power and it's pretty useless that way , the signal drop too fast . The channel is broken too on mt7921 . Please help me.
[image]

Easton97-Jens, 2024-12-07:
English

Hi everyone,

I’m currently using OpenWrt 24.10-rc2 with a MediaTek MT7922 and the latest mt76 driver. I’ve encountered an issue with the TX power being limited to 3 dBm. This issue occurs with both the MediaTek MT7921e and the MT7922.

This problem has persisted since version 23.05, and I haven’t been able to find a solution so far. Additionally, I’ve downloaded the latest firmware directly from the repository:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek?id=862cd7427f8899104c224084d07416cddff1ea1f

Here are the details of my setup:

System: x64 platform

Wi-Fi Chip: MediaTek MT7922

Driver: Latest version of mt76

Firmware: Latest version from the repository linked above

Country Code: correctly set

Despite this configuration, the TX power remains limited to 3 dBm, leading to very poor range and performance. I’ve tried several adjustments, but so far, there’s been no improvement.

Are there any known solutions or workarounds for this issue? I’m happy to assist with further testing and provide debug logs if needed.

Thank you for your support!

doublep21, 2024-12-23:
> maybe this [greearb/linux-ct-6.13@8e84f02](https://github.com/greearb/linux-ct-6.13/commit/8e84f028d61684818aedabc8a8b0379939af94a9)
> 
> ```
> --- a/drivers/net/wireless/mediatek/mt76/mt7921/init.c
> +++ b/drivers/net/wireless/mediatek/mt76/mt7921/init.c
> @@ -74,6 +74,9 @@ mt7921_regd_channel_update(struct wiphy *wiphy, struct mt792x_dev *dev)
>  	struct ieee80211_channel *ch;
>  	int i, cfreq;
>  
> +	if (!(dev->phy.chip_cap & MT792x_CHIP_CAP_CLC_EVT_EN))
> +		dev->phy.clc_chan_conf = 0xff;
> +
>  	np = mt76_find_power_limits_node(mdev);
>  
>  	sband = wiphy->bands[NL80211_BAND_5GHZ];
> ```

only for the channel or even for the tx power?

morrownr, 2025-01-12:
This problem is actually a cosmetic problem. I sure as hell wish it were fixed because I get asked about it a lot.

doublep21, 2025-01-27:
> This problem is actually a cosmetic problem. I sure as hell wish it were fixed because I get asked about it a lot.

so why the performance and signal are so bad? is the chip itself bad?

morrownr, 2025-01-27:
>  is the chip itself bad?

I have devices with mt7921, mt7922 and mt7925 chips. I rate them as excellent.

> so why the performance and signal are so bad?

I don't see that here. There can be a lot of reasons for poor performance and weak signal.

doublep21, 2025-01-27:
> > is the chip itself bad?
> 
> I have devices with mt7921, mt7922 and mt7925 chips. I rate them as excellent.
> 
> > so why the performance and signal are so bad?
> 
> I don't see that here. There can be a lot of reasons for poor performance and weak signal.

ok , thankyou

closed 2025-01-27 by doublep21

morrownr, 2025-01-27:
Sure. Also, I have a site that is primarily about USB WiFi adapters but we also talk about WiFi cards. We are often able to help users with problems such as yours so please feel free to stop by and read or post:

https://github.com/morrownr/USB-WiFi

zearp, 2025-02-27:
If it was merely cosmetic then I would get the same issues in Windows but I'm not. There are too many reports of low speeds and tx power seemingly stuck at 3dbm. Maybe the tx power part is just cosmetic but it doesn't change the issues people are having.

If I can max out my line speed in Windows but only get ~50mbit/s in any Linux distro I've tried then surely it's more than cosmetic. My affected chipset is ```mt7921e``` and I've tried a lot. Different firmwares, disabling power management, compiling drivers from source and so on.

Seems to me a lot of mediatek devices have some kind of issue and many users including myself connect that to low tx power and also many many retries for tx. Please keep in mind most people reporting things are just end-users. We may not get everything right technically but the problems reported are still very real. 

From what I seen on this issue tracker alone it is mainly ```mt79xx``` and ```mt791x``` that are affected and I would assume there are people who have this issue but are simply not aware of it and might blame their low speed on something else. I sure did for way too long. Only found out it was a driver issue not a user issue after installing Windows 11 as a sanity check.

doublep21, 2025-02-27:
> If it was merely cosmetic then I would get the same issues in Windows but I'm not. There are too many reports of low speeds and tx power seemingly stuck at 3dbm. Maybe the tx power part is just cosmetic but it doesn't change the issues people are having.
> 
> If I can max out my line speed in Windows but only get ~50mbit/s in any Linux distro I've tried then surely it's more than cosmetic. My affected chipset is `mt7921e` and I've tried a lot. Different firmwares, disabling power management, compiling drivers from source and so on.
> 
> Seems to me a lot of mediatek devices have some kind of issue and many users including myself connect that to low tx power and also many many retries for tx. Please keep in mind most people reporting things are just end-users. We may not get everything right technically but the problems reported are still very real.
> 
> From what I seen on this issue tracker alone it is mainly `mt792x` and `mt791x` that are affected and I would assume there are people who have this issue but are simply not aware of it and might blame their low speed on something else. I sure did for way too long. Only found out it was a driver issue not a user issue after installing Windows 11 as a sanity check.

My speed is not capped at 50mbit/s but it decay veri fast per distance , so 1 meter from the source even in line of sight is even less than 50mbit/s . I don't know if maybe is the case to reopen the issue because my wifi card is useless at this moment of time and i'm still using my old fritzbox router as an ap

morrownr, 2025-02-27:
@zearp @doublep21 

> If it was merely cosmetic then I would get the same issues in Windows but I'm not.

If all else besides txpower was equal, this logic would hold, but all else is not equal between Windows and Linux.

> If I can max out my line speed in Windows but only get ~50mbit/s in any Linux distro I've tried then surely it's more than cosmetic.

This is a true statement if you include things besides txpower. Like I said above, Windows and Linux have differences that can explain situations like this. We could discuss this and see if there are ways to correct the issues that you are seeing. Please allow me to invite both of you to my site so we can discuss this:

https://github.com/morrownr/USB-WiFi

Use `Issues` to post some details of what you are seeing. Let's see if we can explore the issues and find ways to make things better.

I know it may look like it is just your adapter that is not performing like you want but the internet is full of wireless issues that have to do with Windows and Linux. Wireless is complicated. I just ran a test to see how the performance of my mt7922 based PCIe card. It uses the mt7921e driver:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.128 port 54758 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   152 MBytes  1.27 Gbits/sec    0   2.49 MBytes       
[  5]   1.00-2.00   sec   145 MBytes  1.22 Gbits/sec    0   2.49 MBytes       
[  5]   2.00-3.00   sec   146 MBytes  1.23 Gbits/sec    0   2.49 MBytes       
[  5]   3.00-4.00   sec   146 MBytes  1.23 Gbits/sec    0   2.49 MBytes       
[  5]   4.00-5.00   sec   145 MBytes  1.22 Gbits/sec    0   2.49 MBytes       
[  5]   5.00-6.00   sec   150 MBytes  1.26 Gbits/sec    0   2.49 MBytes       
[  5]   6.00-7.00   sec   150 MBytes  1.26 Gbits/sec    0   2.49 MBytes       
[  5]   7.00-8.00   sec   146 MBytes  1.23 Gbits/sec    0   2.49 MBytes       
[  5]   8.00-9.00   sec   149 MBytes  1.25 Gbits/sec    0   2.49 MBytes       
[  5]   9.00-10.00  sec   150 MBytes  1.26 Gbits/sec    0   2.49 MBytes       

[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.44 GBytes  1.24 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.44 GBytes  1.24 Gbits/sec                  receiver

```

Note that speed is shown in Gbps. Yes, Gbps. 1 Gbps connection with ethernet cable cannot do this. Am I doing anything special? No. Do I have a good handle on how wireless works? Yes. This test was done through 2 walls with about 14 feet of distance. 

Here are the interface details:

```
	Interface wlp3s0
		ifindex 3
		wdev 0x1
		addr e8:fb:1c:7e:4d:cb
		ssid Bass
		type managed
		channel 100 (5500 MHz), width: 160 MHz, center1: 5570 MHz
		txpower 3.00 dBm

```

Darn! That is 3.00 dBm is messing things up. Not! Actually a patch to correct this cosmetic txpower issue very recently was submitted. I can provide a link if you want to see it.

I also have adapters and cards based on the mt7921au, mt7921 and mt7925. I can do a lot of testing. We can get to the bottom of this. See you at my site.

zearp, 2025-02-27:
>  Am I doing anything special? No. Do I have a good handle on how wireless works? Yes.

> Darn! That is 3.00 dBm is messing things up. Not! 

Wow, I only said that even if it's just cosmetic it doesn't negate the facts people are having issues. but thanks for this condescending response to someone who is only trying to help by investing time to test and report things. How silly of me. It is much easier to just move on and replace Mediatek with Intel m2 cards. Thankfully I have that option but not everybody can do this, too bad for them I guess.