openwrt/mt76 #945: [MT7613BE] -STA does not connect to access points with a bandwidth of 160 MHz.

rah66, 2025-01-17:
Hi,

I have a device _Xiaomi MiWiFi Range Extender AC1200_ (**RC04**), this is an updated 2023 version of the device _Xiaomi MiWiFi Range Extender AC1200_ (**RA75**) - [OpenWRT RA75 link](https://openwrt.org/inbox/toh/xiaomi/mi_wifi_range_extender_ac1200_model_ra75). MT7612EN (in RA75) updated to MT7613BE (in RC04).

I have [compiled a firmware based on RA75](https://firmware-selector.openwrt.org/?version=23.05.5&target=ramips%2Fmt76x8&id=xiaomi_mi-ra75), excluding kmod-mt76x2 module for MT7612EN and adding three modules: kmod-mt7615-firmware (includes: kmod-mt7615-common, kmod-mt7615e), kmod-mt7663-firmware-ap, kmod-mt7663-firmware-sta for MT7613BE.

[bootlogRC04_23.05.5.txt](https://github.com/user-attachments/files/18454525/bootlogRC04_23.05.5.txt)

Wifi module MT7613BE in STA client mode connects to the Access Point (Xiaomi AX3000T RD03 OpenWRT) at a width of 80 MHz and works stably.

[image]

[bootlogRC04_23.05.5_cont.txt](https://github.com/user-attachments/files/18455038/bootlogRC04_23.05.5_cont.txt)

But if switch the access point from 80 to 160 width, the connection immediately drops.

```
[ 5263.722605] phy1-sta0: deauthenticated from 46:f7:70:fc:4b:bb (Reason: 3=DEAUTH_LEAVING)
[ 5267.448731] phy1-sta0: authenticate with 46:f7:70:fc:4b:bb
[ 5267.459617] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 1/3)
[ 5267.469750] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 2/3)
[ 5267.479798] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 3/3)
[ 5267.489841] phy1-sta0: authentication with 46:f7:70:fc:4b:bb timed out
[ 5269.392603] phy1-sta0: authenticate with 46:f7:70:fc:4b:bb
[ 5269.403590] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 1/3)
[ 5269.414075] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 2/3)
[ 5269.424282] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 3/3)
[ 5269.434709] phy1-sta0: authentication with 46:f7:70:fc:4b:bb timed out
[ 5271.839253] phy1-sta0: authenticate with 46:f7:70:fc:4b:bb
[ 5271.850424] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 1/3)
[ 5271.860658] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 2/3)
[ 5271.870686] phy1-sta0: send auth to 46:f7:70:fc:4b:bb (try 3/3)
[ 5271.880946] phy1-sta0: authentication with 46:f7:70:fc:4b:bb timed out
```

This error is not corrected by switching mode AX/AC, channels 36, 40, 44, 48, (64-128), 149, 153, 157, etc., changing WEP encryption and disabling encryption.
I also read that the [region and DFS](https://openwrt.org/unsupported/dfs) may have an impact. Conducted testing on US, PA and RU regions.

```
root@MiWiFi_RC04:/# iw reg get
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

root@MiWiFi_RC04:/# iw reg get
global
country PA: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 36), (N/A)
        (5150 - 5250 @ 80), (N/A, 36), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5470 - 5725 @ 160), (N/A, 30), (N/A)
        (5725 - 5850 @ 80), (N/A, 36), (N/A)
        (57000 - 64000 @ 2160), (N/A, 43), (N/A)

root@MiWiFi_RC04:/# iw reg get
global
country RU: DFS-UNSET
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5350 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (5650 - 5850 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A), NO-OUTDOOR
```
RU region [no DFS and have 160 on 36-64 channels](https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/n/ac/ax/be)).

The problem of connecting MT7613 in client mode to 160 width is confirmed on Xiaomi AX3000T RD03 OpenWRT and Asus RT-AX88U MerlinWRT access point routers.

Linaro1985, 2025-01-24:
@rah66 mt7613be is not support 160 Mhz as I know.

rah66, 2025-01-24:
@Linaro1985 It's very sad, I don't want to install the original Xiaomi firmware....

rah66, 2025-02-17:
@Linaro1985  Hi, could these adjustments/corrections be the reason?
1. [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blobdiff;f=package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh;h=6ff627b03813bb2de0cd1f465014cc74031106a3;hp=b5378e556eb232fdf30f42192d9a00dfb075c456;hb=80e4e2285fdf4a7b19c84532deafe2d1e690ed30;hpb=d864f68232e910f2c8ab06a66347fc08c257dfcc)
2. [Commit bb3937d](https://github.com/openwrt/mt76/commit/bb3937d5c3e0b13c0d08747ec0fc9726fb4fd870)

From this discussion, what are the [solutions](https://forum.openwrt.org/t/solved-802-11ax-worse-than-802-11ac-with-mt76-driver/126466/533) to the problem with Apple devices?

ptpt52, 2025-06-15:
mark