openwrt/mt76 #951: MT7915e - WiFi Hotspot Disappearing when client connected

Sarvan124, 2025-01-28:
**HW Device Details:**
Device with DBDC Mode supporting 2.4 &5GHz using AsiaRF MT7915e(M.2 PCIe Module)

**Driver Details**
dmesg | grep mt7915e
[    5.095784] mt7915e 0000:05:00.0: enabling device (0000 -> 0002)
[    5.217806] mt7915e 0000:05:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220929104113a
[    5.255979] mt7915e 0000:05:00.0: WM Firmware Version: ____000000, Build Time: 20220929104145
[    5.276718] mt7915e 0000:05:00.0: WA Firmware Version: DEV_000000, Build Time: 20220929104205

**OS Issue seen:**
OS   Version | Kernel Version
-- | --
Oracle Linux 9.5 | 5.15.302
Oracle Linux 9.5 | 5.15.304
Ubuntu 24.04.1 LTS (Noble Numbat) | 6.8.0-51-generic

**Issue Details:**
The Device SSID's disappear with reporting timeout issue in Kernel, Logs below

[  194.057968] mt7915e 0000:05:00.0: Message 00005aed (seq 9) timeout

_**********************************************************************************************************

**Logs Attached:**

[dmesg_OL_console.txt](https://github.com/user-attachments/files/18575067/dmesg_OL_console.txt)

lukasz1992, 2025-01-29:
Driver is quite old, but anyway it should still work correctly - you would need to update firmware.

Please take 3 firmware files from https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild/autobuild_5.4_mac80211_release/package/kernel/mt76/src/firmware/mt7915_{wa,wm,rom_patch}.bin and put into /lib/firmware/mediatek/ , reboot and it should fix the issue

Sarvan124, 2025-01-29:
HI Lukasz,
Thanks for the reply, put into /lib/firmware/mediatek/ , reboot is failing with below error- we are trying in Oracle Linux 15.5.302 Kernel & Ubuntu 24.10 version.

Error as below:
[root@surekha-lt mt7915]# insmod mt7915e.ko

insmod: ERROR: could not insert module mt7915e.ko: Invalid module format

 
[root@localhost new-fw]# cp mt7915_* /lib/firmware/mediatek/

cp: overwrite '/lib/firmware/mediatek/mt7915_rom_patch.bin'? y

cp: overwrite '/lib/firmware/mediatek/mt7915_wa.bin'? y

cp: overwrite '/lib/firmware/mediatek/mt7915_wm.bin'? y

[root@localhost new-fw]# sudo chmod 644 /lib/firmware/mediatek/mt7915_*.bin

sudo chown root:root /lib/firmware/mediatek/mt7915_*.bin

[root@localhost new-fw]# chmod 644 /lib/firmware/mediatek/mt7915_*.bin

[root@localhost new-fw]# modprobe -r mt7915e

[root@localhost new-fw]# modprobe mt7915e

[root@localhost new-fw]# 

[root@localhost new-fw]# 

[root@localhost new-fw]# 

[root@localhost new-fw]# modinfo mt7915e

filename:       /lib/modules/5.15.0-302.167.6.el9uek.x86_64/kernel/drivers/net/wireless/mediatek/mt76/mt7915/mt7915e.ko.xz

license:        Dual BSD/GPL

firmware:       mediatek/mt7915_rom_patch.bin

firmware:       mediatek/mt7915_wm.bin

firmware:       mediatek/mt7915_wa.bin

srcversion:     624D0468F76409EFFD28EC2

alias:          pci:v000014C3d00007915sv*sd*bc*sc*i*

alias:          pci:v000014C3d00007916sv*sd*bc*sc*i*

depends:        mac80211,mt76,cfg80211

retpoline:      Y

intree:         Y

name:           mt7915e

vermagic:       5.15.0-302.167.6.el9uek.x86_64 SMP mod_unload modversions 

sig_id:         PKCS#7

signer:         Oracle CA Server

sig_key:        1E:5E:01:71:4F:64:58:8E:57:F8:8F:CB:C5:B9:B5:4A:31:3B:9C:D4

sig_hashalgo:   sha512

signature:      8D:8D:89:14:D3:47:3B:9E:B3:F0:CC:FD:8F:92:FF:94:E0:EB:32:86:

		9F:C8:85:7B:68:7E:1A:08:94:8A:99:DD:98:E9:32:F7:D1:97:6F:1B:

		29:B8:A1:45:A2:DC:86:17:CF:C3:03:E7:54:53:13:F4:4D:FB:F3:D8:

		58:ED:E4:4F:26:B6:3D:78:24:90:E5:A6:FC:57:B6:12:D1:AE:77:FD:

		E2:66:E2:00:C5:A7:6B:32:D5:3F:1F:9C:AB:7F:B5:DF:DD:4B:44:46:

		BC:2B:3E:62:34:95:CA:BB:64:30:3E:65:74:F6:10:73:B8:04:75:39:

		23:F4:A2:C0:AC:95:45:CE:15:B5:DD:29:41:2F:C0:76:BB:0E:52:78:

		5D:0E:4B:57:CE:15:1A:68:2C:FC:8B:D5:FB:C8:B8:D5:F2:4F:B9:1A:

		9D:3A:09:BD:D5:FA:73:6D:05:05:41:CF:27:8F:21:D3:14:90:2F:3F:

		3A:89:4F:72:DF:E7:42:94:D5:23:89:70:7D:A4:68:0E:9F:08:EA:78:

		B8:2D:09:31:B3:A4:90:B8:C8:BE:A6:FB:7A:36:C3:1C:79:DD:C9:7B:

		C0:16:4F:DB:5A:45:37:A8:25:9A:64:85:5A:02:BA:88:A4:AC:C1:E5:

		8A:9D:02:26:87:1C:58:09:CC:23:B6:1D:34:B3:6C:B8:4E:F5:C7:CE:

		D7:7D:CE:F6:E3:1F:88:3E:42:AD:B9:AB:DD:0C:57:CD:D0:DC:C2:F3:

		7C:C6:35:62:B7:26:0B:49:0B:69:02:CD:BE:C3:B8:92:08:AF:2D:F6:

		83:59:5E:72:2B:75:DC:AD:C4:AF:83:B3:58:46:D9:2F:ED:BF:B9:9E:

		73:19:86:2F:FE:D1:C6:B7:E1:6A:55:DC:78:D4:AB:66:30:8E:7A:93:

		E7:C8:18:AA:6E:D3:3D:61:CF:38:03:4F:89:5D:09:2B:A1:6E:77:CC:

		D0:73:1A:57:F6:A2:77:CC:A7:D8:0D:80:60:BD:7F:05:64:A4:30:87:

		30:0B:38:0C:70:7D:D4:39:38:FF:62:27:AC:35:B1:C4:8E:77:96:4A:

		4E:88:BE:1D:77:BF:34:C5:23:7D:6D:22:C5:16:BD:4F:45:25:7F:08:

		02:BC:BF:E3:D1:29:B7:43:AA:1C:33:F2:49:55:10:92:0A:9E:2D:D6:

		CB:D1:98:A5:7E:11:E2:37:F8:55:32:87:1C:F0:D8:4C:64:C7:97:3E:

		D5:BC:02:0E:E7:E6:C0:F7:E2:F3:31:B6:CA:2E:43:08:B2:AF:17:6C:

		24:02:57:AE:55:A3:B2:20:13:DB:47:D1:36:9F:03:30:F3:7C:8B:04:

		56:C4:18:BA:0E:A1:BD:22:02:F3:0E:92

[root@localhost new-fw]# dmesg | grep mt7915

[    5.067859] mt7915e 0000:05:00.0: enabling device (0000 -> 0002)

[    5.118878] mt7915e 0000:05:00.0: HW/SW Version: 0x4320222d, Build Time: <!DOCTYPE html P

[   25.455622] mt7915e 0000:05:00.0: Message 00000010 (seq 3) timeout

[   25.455656] mt7915e 0000:05:00.0: Failed to get patch semaphore

[   25.455780] mt7915e: probe of 0000:05:00.0 failed with error -11

[ 3045.487164] mt7915e 0000:05:00.0: HW/SW Version: 0x4320222d, Build Time: <!DOCTYPE html P

[ 3065.726885] mt7915e 0000:05:00.0: Message 00000010 (seq 3) timeout

[ 3065.726915] mt7915e 0000:05:00.0: Failed to get patch semaphore

[ 3065.727033] mt7915e: probe of 0000:05:00.0 failed with error -11

lukasz1992, 2025-01-29:
I can not help :( HW/SW version starts to be invalid with new FW.

Maybe using dkms and compiling modules from source would help.

Sarvan124, 2025-01-29:
Please suggest any alternate, we are stuck with this timeout issue now.

lukasz1992, 2025-01-30:
I am not a developer. And this is something for:
- Linux firmware maintainers
- Maintainers of OS to use the most recent kernel
- compile the driver by yourself, using dkms (you would need to write Makefile)

rany2, 2025-01-30:
> mt7915e 0000:05:00.0: HW/SW Version: 0x4320222d, Build Time: <!DOCTYPE html P

You downloaded an HTML page instead of the binaries themselves. You need to download the binaries by selecting "text" and then using `base64 -d` to convert to binaries (or by git cloning the repo and just copying the binaries this way, if you want to avoid using that git web UI).

Sarvan124, 2025-02-13:
Thanks Rany, Lukaz,
However, our issue is not fixed yet, still seeing the timeout error with New Firmware 202409 as well.
Below is our further Observation, but with both 5.15 & 6.13 Kernel our issue is not fixed, please help if there is any other WA, fixes around this.
with 5.15 kernel,
failure is occurring from the below function
o	⁠static int
mt7915_mcu_parse_response(struct mt76_dev *mdev, int cmd,
        struct sk_buff *skb, int seq)
{
  struct mt7915_mcu_rxd *rxd;
  int ret = 0;
 
  if (!skb) {
    dev_err(mdev->dev, "Message %08x (seq %d) timeout\n",
      cmd, seq);
    return -ETIMEDOUT;
  } 
  rxd = (struct mt7915_mcu_rxd *)skb->data;
  if (seq != rxd->seq)
    return -EAGAIN;
 
  if (cmd == MCU_CMD(PATCH_SEM_CONTROL)) {
    skb_pull(skb, sizeof(*rxd) - 4);
    ret = *skb->data;
  } else if (cmd == MCU_EXT_CMD(THERMAL_CTRL)) {
    skb_pull(skb, sizeof(*rxd) + 4);
    ret = le32_to_cpu(*(__le32 *)skb->data);
  } else {
    skb_pull(skb, sizeof(struct mt7915_mcu_rxd));
  }
 
  return ret;
}
o	the above if condition is causing the timeout issue.
_________________________________________________________________________________________________
o	with 6.13 kernel,
o	the timeout issue has been handled by MCU RESET.
o	if (!skb) {
o	         [dev_err](https://elixir.bootlin.com/linux/v6.13.2/C/ident/dev_err)([mdev](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mdev)->dev, "Message %08x (seq %d) timeout\n",
o	                       cmd, [seq](https://elixir.bootlin.com/linux/v6.13.2/C/ident/seq));
o	
o	                if ([image]([MT76_MCU_RESET](https://elixir.bootlin.com/linux/v6.13.2/C/ident/MT76_MCU_RESET), &dev->[mphy](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mphy).state)) {
o	                     dev->[recovery](https://elixir.bootlin.com/linux/v6.13.2/C/ident/recovery).[restart](https://elixir.bootlin.com/linux/v6.13.2/C/ident/restart) = [true](https://elixir.bootlin.com/linux/v6.13.2/C/ident/true);
o	                     [wake_up](https://elixir.bootlin.com/linux/v6.13.2/C/ident/wake_up)(&dev->[mt76](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mt76).[mcu](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mcu).[wait](https://elixir.bootlin.com/linux/v6.13.2/C/ident/wait));
o	                 [queue_work](https://elixir.bootlin.com/linux/v6.13.2/C/ident/queue_work)(dev->[mt76](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mt76).[wq](https://elixir.bootlin.com/linux/v6.13.2/C/ident/wq), &dev->[reset_work](https://elixir.bootlin.com/linux/v6.13.2/C/ident/reset_work));
o	                 [wake_up](https://elixir.bootlin.com/linux/v6.13.2/C/ident/wake_up)(&dev->[reset_wait](https://elixir.bootlin.com/linux/v6.13.2/C/ident/reset_wait));
o	                }
o	
o	                return -[ETIMEDOUT](https://elixir.bootlin.com/linux/v6.13.2/C/ident/ETIMEDOUT);
o	 }
o	but now the issue is occurring in other function
o	static int [mt7915_driver_own](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mt7915_driver_own)(struct [mt7915_dev](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mt7915_dev) *dev, [u8](https://elixir.bootlin.com/linux/v6.13.2/C/ident/u8) [band](https://elixir.bootlin.com/linux/v6.13.2/C/ident/band))
o	{
o	      [mt76_wr](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mt76_wr)(dev, [MT_TOP_LPCR_HOST_BAND](https://elixir.bootlin.com/linux/v6.13.2/C/ident/MT_TOP_LPCR_HOST_BAND)([band](https://elixir.bootlin.com/linux/v6.13.2/C/ident/band)), [MT_TOP_LPCR_HOST_DRV_OWN](https://elixir.bootlin.com/linux/v6.13.2/C/ident/MT_TOP_LPCR_HOST_DRV_OWN));
o	       if ([image](dev, [MT_TOP_LPCR_HOST_BAND](https://elixir.bootlin.com/linux/v6.13.2/C/ident/MT_TOP_LPCR_HOST_BAND)([band](https://elixir.bootlin.com/linux/v6.13.2/C/ident/band)),
o	                           [MT_TOP_LPCR_HOST_FW_OWN_STAT](https://elixir.bootlin.com/linux/v6.13.2/C/ident/MT_TOP_LPCR_HOST_FW_OWN_STAT), 0, 500)) {
o	              [dev_err](https://elixir.bootlin.com/linux/v6.13.2/C/ident/dev_err)(dev->[mt76](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mt76).dev, "Timeout for driver own\n");
o	            return -[EIO](https://elixir.bootlin.com/linux/v6.13.2/C/ident/EIO);
o	   }
o	
o	   /* clear irq when the driver own success */
o	 [mt76_wr](https://elixir.bootlin.com/linux/v6.13.2/C/ident/mt76_wr)(dev, [MT_TOP_LPCR_HOST_BAND_IRQ_STAT](https://elixir.bootlin.com/linux/v6.13.2/C/ident/MT_TOP_LPCR_HOST_BAND_IRQ_STAT)([band](https://elixir.bootlin.com/linux/v6.13.2/C/ident/band)),
o	          [MT_TOP_LPCR_HOST_BAND_STAT](https://elixir.bootlin.com/linux/v6.13.2/C/ident/MT_TOP_LPCR_HOST_BAND_STAT));
o	
o	       return 0;
o	}

Sarvan124, 2025-02-20:
Hi All,
Any update on this?

maurerle, 2025-11-07:
Hi @Sarvan124, can you please report back if you still see this issue on latest openwrt 24.10.4?
The `Message 00005aed (seq 9) timeout` and `Failed to get patch semaphore` should be fixed in latest openwrt/mt76.