openwrt/mt76 #953: MT7986: wifi randomly disconnects

moetayuko, 2025-01-31:
My device is JDCloud RE-CP-03, running v24.10.0-rc6, and the configuration is as follows:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/18000000.wifi'
        option band '2g'
        option channel '6'
        option htmode 'HE40'
        option disabled '0'
        option country 'CN'
        option cell_density '0'
        option mu_beamformer '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key 'xxx'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone 'CST-8'
        option wnm_sleep_mode '1'
        option bss_transition '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/18000000.wifi+1'
        option band '5g'
        option channel '48'
        option htmode 'HE160'
        option disabled '0'
        option country 'CN'
        option cell_density '0'
        option mu_beamformer '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key 'xxx'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone 'CST-8'
        option wnm_sleep_mode '1'
        option bss_transition '1'
```
Besides, I have enabled HW flow offload and WED. usteer is employed for band steering, but there are no other APs.

A very annoying issue is that wifi randomly disconnected, which happens once every 2 or 3 days. When it happens, all clients are still connected, but can't send or receive data. ping the router fails. The approach to resolve it is to log in to the router through Ethernet and press the _Restart_ button in _Network -> Wireless_, or run `/sbin/wifi` in the shell. I have added a corn job to run `/sbin/wifi` every 24 hours, but it doesn't help.

There are no warnings or errors in both syslog and dmesg when experiencing the issue.

filippz, 2025-01-31:
I'm struggling with possibly same issue for quite some time on a couple of Asus TUF-AX4200 (MT7986) devices. I was using them for mesh connection and it would fail randomly with the same symptoms but I was able to trigger the same behavior by doing `iw scan`. Eventually the `iw scan` issue was resolved (https://github.com/openwrt/mt76/commit/e3e6d490ab90d28503debdc6b9d5084710f84dad) yet sadly the problem you describe still remains and is seemingly occurs at random.

This issue you're mentioning could be the same one as this: https://github.com/openwrt/mt76/issues/867 and I would say that it will be difficult to fix it until we figure out an exact way to trigger it.

moetayuko, 2025-02-04:
> I'm struggling with possibly same issue for quite some time on a couple of Asus TUF-AX4200 (MT7986) devices. I was using them for mesh connection and it would fail randomly with the same symptoms but I was able to trigger the same behavior by doing `iw scan`. Eventually the `iw scan` issue was resolved ([e3e6d49](https://github.com/openwrt/mt76/commit/e3e6d490ab90d28503debdc6b9d5084710f84dad)) yet sadly the problem you describe still remains and is seemingly occurs at random.
> 
> This issue you're mentioning could be the same one as this: [#867](https://github.com/openwrt/mt76/issues/867) and I would say that it will be difficult to fix it until we figure out an exact way to trigger it.

Yes, they appear to be the same. The title of https://github.com/openwrt/mt76/issues/867 is misleading, the author experienced the problem with Banana Pi R3, an MT7986 device, but he wrote MT7981 in the title.

MT7981 appears to be fine, my rax3000m runs stably with the official 23.05.

moetayuko, 2025-02-04:
> > I have switched to a downstream openwrt fork w/ proprietary driver for my MT7986, which does not suffer from the bug. I hope it will be resolved soon in the mt76 driver.
> 
> Which fork are you using?

https://github.com/hanwckf/immortalwrt-mt798x

> While looking for a solution I noticed that MediaTek firmwares (https://github.com/openwrt/mt76/tree/master/firmware) from this repo are not the ones used by OpenWrt as it seems to be using files from linux-firmware instead. I simply replaced those bin files on my routers with the ones from this repo and it seems to be fine for now - but it has been just a couple of days of testing so it might not be the solution.

It appears that MT7986 offload firmware is from linux-firmware and the rest are from this repo:
https://github.com/openwrt/openwrt/blob/f628467dfd6b32ead172d33a2593f04901395343/package/kernel/mt76/Makefile#L623-L632
https://github.com/openwrt/openwrt/blob/f628467dfd6b32ead172d33a2593f04901395343/package/firmware/linux-firmware/mediatek.mk#L90-L97

This repo hosts the offload firmware but is not packaged for use. Interestingly, linux-firmware has not been upgraded since 2022 https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/mediatek/mt7986_wo_0.bin

filippz, 2025-02-04:
AFAIK, primary source for MT firmwares is https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild/autobuild_5.4_mac80211_release/package/kernel/mt76/src/firmware?autodive=0%2F%2F%2F

I remember even building OpenWRT form there but since I had issues with `iw scan` I reverted back to "vanilla". Eventually `iw scan` turned out to be a different bug altogether so it makes sense to try with bin files from there/here as those are more recent than the ones in linux-firmware.

If it's not too much trouble please try with OpernWRT and replace those bin files with the newer ones and see if it helps.

moetayuko, 2025-02-04:
> AFAIK, primary source for MT firmwares is [...]

I just compared and found that mt7986 firmware in mtk git and this repo are identical. Hence, outdated offload firmware provided by linux-firmware may be the cause.

filippz, 2025-02-10:
> Did you try this patch? https://github.com/cmonroe/feed-wifi-master/raw/refs/heads/smartrg-master/mt76/patches/0925-wifi-mt76-do-not-report-ACK-when-TXS-is-lost.patch

I've built an image with the mentioned patch applied and I have been running trouble free on both of my AX4200 for about 3,5 days. I can't say that it's a definite fix for my problems but it does look promising.

moetayuko, 2025-02-16:
> Did you try this patch? https://github.com/cmonroe/feed-wifi-master/raw/refs/heads/smartrg-master/mt76/patches/0925-wifi-mt76-do-not-report-ACK-when-TXS-is-lost.patch

I guess this patch is superseded by https://github.com/openwrt/mt76/commit/e5fef138524e63314cb96ff8314048d175294e95

filippz, 2025-03-08:
> Did you encounter the issue again, or do you consider it fixed with that patch?

I'd say it's fixed - I had just one intermittent hicckup on one client but I'm thinking that it was caused by the client itself. I can't say if the OP has the same experience as me tho.

In any case, a big thanks to everyone for helping and @nbd168 for adding the patch to the repo.

filippz, 2025-03-08:
> Do I need to wait for the next stable? Without compiling it myself :)

I suggest simply using https://firmware-selector.openwrt.org/ to download sysupgrade snapshot for your device - there's no need to wait for release or build it yourself.

moetayuko, 2025-03-08:
> > > Do I need to wait for the next stable? Without compiling it myself :)
> > 
> > 
> > I suggest simply using https://firmware-selector.openwrt.org/ to download sysupgrade snapshot for your device - there's no need to wait for release or build it yourself.
> 
> I don't think this has been merged yet, right? There is no pull request on the OpenWRT project with the latest changes on the mt76 driver (the changes up to January 22 are included [openwrt/openwrt@17dc079](https://github.com/openwrt/openwrt/commit/17dc0797d03d2f2dd57b38ab39962dafc69d1d34)).

Yes, the patch is not included in any tagged releases yet. You can use the snapshot build before 24.10.1 is available

filippz, 2025-03-09:
@moetayuko 
Today I experienced another "connected but no ping" failure on two clients at the same time which I resolved by restarting 5Ghz WiFi on the AP. Do you still have issues after https://github.com/openwrt/mt76/commit/e5fef138524e63314cb96ff8314048d175294e95 but less frequently than before like I do, or you have no problems at all anymore, or maybe the problems are roughly equally frequent as before?
I'm now at 11 days of runtime on my APs and first experienced an issue with one client a couple of days ago but then it seemed to be caused by the client itself.

moetayuko, 2025-03-11:
> [@moetayuko](https://github.com/moetayuko) Today I experienced another "connected but no ping" failure on two clients at the same time which I resolved by restarting 5Ghz WiFi on the AP. Do you still have issues after [e5fef13](https://github.com/openwrt/mt76/commit/e5fef138524e63314cb96ff8314048d175294e95) but less frequently than before like I do, or you have no problems at all anymore, or maybe the problems are roughly equally frequent as before? I'm now at 11 days of runtime on my APs and first experienced an issue with one client a couple of days ago but then it seemed to be caused by the client itself.

I'm on the proprietary driver so can't tell.