openwrt/mt76 #958: MT7981: zero-wait DFS not working as expected

LukasMikalauskas1, 2025-02-18:
In **OpenWRT**, enabling the **background_radar** option in the network **UCI** configuration is expected to enable **zero-wait DFS**. However, on devices using the **MT7981 SoC** with the **MT76/MT7915** driver, enabling this option appears to break the AP. When **background_radar** is enabled, the AP appears to function normally according to the logs.

Current Behavior:

- The AP starts on a non-DFS channel.
- It begins CAC (Channel Availability Check) in the background.
- Once background CAC completes, the AP switches to the DFS channel and starts CAC on another channel in the background.
- Simulating a radar signal using debugfs also triggers the expected behavior—switching to a completed DFS channel and immediately starting CAC on another.

Issue:

- Client devices are unable to connect to the AP.
- Wireshark analysis shows that the AP does not respond to authentication frames.
- This issue occurs regardless of the channel the AP is currently operating on.
- From the client side logs indicate authentication timeouts.

LukasMikalauskas1, 2025-02-19:
Did some mt76/mt7915 testing and found out that changing this line:
https://github.com/openwrt/mt76/blob/e5fef138524e63314cb96ff8314048d175294e95/mt7915/mcu.c#L2730

To:
`return mt76_connac_mcu_rdd_cmd(&dev->mt76, RDD_START, phy->mt76->band_idx, MT_RX_SEL2, region);`

appears to fix the issue. With this change, client devices can successfully connect to the AP, and everything functions correctly even when a AP jumps to different DFS channel. 

Could this be the actual fix for the issue, or is it just a workaround ?

lukasz1992, 2025-02-19:
Had the same issue. This is great finding!

It looks like 3rd param should be index and 4th param should be rx_sel - and the swap was made only in mt7915_mcu_rdd_background_enable. I am going to check the fix also on my device.

It looks like background radar never had a chance to work correctly.

lukasz1992, 2025-02-19:
```
diff --git a/mt7915/mcu.c b/mt7915/mcu.c
index 9d790f23..a2bad361 100644
--- a/mt7915/mcu.c
+++ b/mt7915/mcu.c
@@ -2701,6 +2701,6 @@ int mt7915_mcu_rdd_background_enable(struct mt7915_phy *phy,
 
        if (!chandef) { /* disable offchain */
-               err = mt76_connac_mcu_rdd_cmd(&dev->mt76, RDD_STOP, MT_RX_SEL2,
-                                             0, 0);
+               err = mt76_connac_mcu_rdd_cmd(&dev->mt76, RDD_STOP, 0,
+                                             MT_RX_SEL2, 0);
                if (err)
                        return err;
@@ -2728,6 +2728,6 @@ int mt7915_mcu_rdd_background_enable(struct mt7915_phy *phy,
        }
 
-       return mt76_connac_mcu_rdd_cmd(&dev->mt76, RDD_START, MT_RX_SEL2,
-                                      0, region);
+       return mt76_connac_mcu_rdd_cmd(&dev->mt76, RDD_START, 0,
+                                      MT_RX_SEL2, region);
 }
 
```

Confirm, this patch solves the issue also for me

nbd168, 2025-02-20:
@lukasz1992, looks good to me. Could you please send me a full patch with description and your signed-off-by, so that I can upstream it? Thanks

closed 2025-04-02 by LukasMikalauskas1

Djfe, 2025-04-13:
Have you seen these? :)
atleast something good came out of this (not sure if it was put into motion due to you reporting this, but I assume so) 
https://github.com/openwrt/mt76/commit/3cec186474c4017e57fc148de5f0302ba647ec8b

https://github.com/openwrt/mt76/commit/9bacdb2bf2c3afa9c7ee1bde900d920b5d145eb9

https://github.com/openwrt/mt76/commit/a19543c7e05e288ed0c5dde589e3228b0a8f78b4

lukasz1992, 2025-04-14:
Yes those patches, are the result of this reported issue here.