openwrt/mt76 #961: mt7996e: 6GHz AP not working correctly

K900, 2025-02-24:
BPi-R4, mt7996e, running kernel 6.14.0-rc3 and hostapd 2.11.

`iw list`: https://bpa.st/25NQ
`iw phy`: https://bpa.st/H3TA
`hostapd.conf`: https://bpa.st/STDA
`hostapd -dd` log: https://bpa.st/64BQ

Everything looks normal, but the AP simply isn't detected on any of the clients I've tried.

abajk, 2025-03-09:
Links don't work.

K900, 2025-03-10:
<details>
<summary>iw list</summary>

```
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 2131 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0xff RX 0xff
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce920d01f30e4e3f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
[... 2949 lines cut ...]
nl80211: Add ifindex 11 for bridge br-lan
nl80211: Add own interface ifindex 11 (ifidx_reason 14)
nl80211: ifindex 11 already in the list
RTM_NEWLINK: ifi_index=14 ifname=wl6g operstate=6 linkmode=0 master=11 ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: RTM_NEWLINK: ifi_index=14 ifname=wl6g ifi_family=7 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: vlan_newlink(wl6g)
nl80211: Event message available
nl80211: Drv Event 60 (NL80211_CMD_FRAME_TX_STATUS) received for wl6g
nl80211: MLME event 60 (NL80211_CMD_FRAME_TX_STATUS) on wl6g(00:0c:43:26:60:12) A1=ff:ff:ff:ff:ff:ff A2=00:0c:43:26:60:12 on link_id=-1
nl80211: MLME event frame - hexdump(len=26): c0 00 00 00 ff ff ff ff ff ff 00 0c 43 26 60 12 00 0c 43 26 60 12 00 00 02 00
nl80211: Frame TX status event A1=ff:ff:ff:ff:ff:ff stype=12 cookie=0x4142 ack=0
wl6g: Event TX_STATUS (16) received
mgmt::deauth cb
VLAN: RTM_NEWLINK: ifi_index=14 ifname=wl6g ifi_family=7 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: vlan_newlink(wl6g)
VLAN: RTM_NEWLINK: ifi_index=14 ifname=wl6g ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: vlan_newlink(wl6g)
VLAN: RTM_NEWLINK: ifi_index=14 ifname=wl6g ifi_family=7 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: vlan_newlink(wl6g)
RTM_NEWLINK: ifi_index=14 ifname=wl6g master=11 operstate=6 ifi_family=7 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
nl80211: Add ifindex 11 for bridge br-lan
nl80211: Add own interface ifindex 11 (ifidx_reason 14)
nl80211: ifindex 11 already in the list
```
</details>

dsbaha, 2025-03-15:
I think its related to the old version of the mt7996e firmware that openwrt uses.  I updated the binary blobs to MTK SDK 4.0 on my branch https://github.com/openwrt/mt76/compare/master...dsbaha:mt76:master and 6ghz works great, along with speed boosts for the other bands.  I haven't tried the MTK SDK 4.1 version of the blobs yet.

dsbaha, 2025-03-24:
I updated this a few hours ago with the MTK MP 4.1 binary blobs.  Seems to work a little better.  Update the Makefile with this;
```
PKG_SOURCE_URL:=https://github.com/dsbaha/mt76
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2025-03-24
PKG_SOURCE_VERSION:=56bef7cc2c1394a4e6cf2d9d1028952500e69ec7
```

leow149, 2025-03-25:
> I updated this a few hours ago with the MTK MP 4.1 binary blobs. Seems to work a little better. Update the Makefile with this;
> 
> ```
> PKG_SOURCE_URL:=https://github.com/dsbaha/mt76
> PKG_SOURCE_PROTO:=git
> PKG_SOURCE_DATE:=2025-03-24
> PKG_SOURCE_VERSION:=56bef7cc2c1394a4e6cf2d9d1028952500e69ec7
> ```

Is there a way to just compile this package? I usually update via owut because i have a pretty complicated package selection that I cannot easily recreate. Would be perfect if I could compile this into an ipk and just install

leow149, 2025-03-25:
I figured it out. When using the SDK, you just can add the above config to the Makefile in this path: `feeds/base/package/kernel/mt76/Makefile`

But sadly that doesnt fix the low txpower issue

dsbaha, 2025-03-29:
> I figured it out. When using the SDK, you just can add the above config to the Makefile in this path: `feeds/base/package/kernel/mt76/Makefile`
> 
> But sadly that doesnt fix the low txpower issue

Ohh, are you just trying to fix the low txpower issue?  Sorry, that is another patch.  iirc, some modules come out with bad eeprom values, and there is a patch to identify these and force set it to the defaults.  Let me see if I can find it.

leow149, 2025-03-29:
> > I figured it out. When using the SDK, you just can add the above config to the Makefile in this path: `feeds/base/package/kernel/mt76/Makefile`
> > 
> > But sadly that doesnt fix the low txpower issue
> 
> Ohh, are you just trying to fix the low txpower issue?  Sorry, that is another patch.  iirc, some modules come out with bad eeprom values, and there is a patch to identify these and force set it to the defaults.  Let me see if I can find it.
> 
> edit; nevermind, seems you found it already https://github.com/openwrt/openwrt/issues/17489#issuecomment-2757567418

Yeah, I already found it
Anyways, thanks for your work :)

ptpt52, 2025-04-01:
how to build latest mt76 base on latest openwrt code? it seems a new mac80211 backport is needed?
```
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/mt7915/main.c:1783:10: error: 'const struct ieee80211_ops' has no member named 'link_sta_rc_update'; did you mean 'sta_rc_update'?
 1783 |         .link_sta_rc_update = mt7915_sta_rc_update,
      |          ^~~~~~~~~~~~~~~~~~
      |          sta_rc_update
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/mt7915/main.c:1783:31: error: initialization of 'void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_sta *, u32)' {aka 'void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_sta *, unsigned int)'} from incompatible pointer type 'void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_link_sta *, u32)' {aka 'void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_link_sta *, unsigned int)'} [-Werror=incompatible-pointer-types]
 1783 |         .link_sta_rc_update = mt7915_sta_rc_update,
      |                               ^~~~~~~~~~~~~~~~~~~~
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/mt7915/main.c:1783:31: note: (near initialization for 'mt7915_ops.sta_rc_update')
cc1: all warnings being treated as errors
scripts/Makefile.build:243: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/mt7915/main.o' failed
make[6]: *** [/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/mt7915/main.o] Error 1
scripts/Makefile.build:480: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/mt7915' failed
make[5]: *** [/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/mt7915] Error 2
/mnt/build/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.6.85/Makefile:1921: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325' failed
make[4]: *** [/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325] Error 2
Makefile:234: recipe for target '__sub-make' failed
make[3]: *** [__sub-make] Error 2
make[3]: Leaving directory '/mnt/build/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.6.85'
Makefile:722: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/.built' failed
make[2]: *** [/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-20250325/.built] Error 2
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/kernel/mt76'
time: package/kernel/mt76/compile#1.05#0.28#1.35
    ERROR: package/kernel/mt76 failed to build.
package/Makefile:187: recipe for target 'package/kernel/mt76/compile' failed
make[1]: *** [package/kernel/mt76/compile] Error 1

```

K900, 2025-05-21:
OK, something in the recent kernel updates seems to have fixed it, and I don't know what. I'll just close the issue.

closed 2025-05-21 by K900